johanmeijer/grott #1: Add authentication to MQTT

Jeltel, 2020-03-30:
I got another sniffer running to locally read out my ShineWifi stick. But we ran into problems with the energy message. And you seems to have solved it :) So I would like to test it.

But I have authentication on my MQTT server. I can probably do a test and disable the authentication. But I'm not comfortable with leaving it that way. 
Can you add MQTT-authentication to your Python script?

johanmeijer, 2020-03-31:
At this moment do not have authentication in MQTT enabled because I only use it in my own (home) network. 

I do think it is an added value to make it possible to enable authentication (via a parameter) and like to experiment with this. Do you use userid/password authentication or another security mechanism?

Jeltel, 2020-03-31:
I use the mosquito home assistant add on: https://github.com/home-assistant/hassio-addons/blob/master/mosquitto/README.md

With the optional logins configuration set to only one account.
I suspect mosquito will have a lot of documentation around Mqtt authentication. Don't think it's to complicated.

johanmeijer, 2020-04-01:
I added an option to specify MQTT autentication. 
You can now specify in the MQTT section of the INI file: 

auth = True
user = youruserid 
password = yourpassword 

Default: 
auth = False
user = grott
password = growatt2020

I hope you are able to test it. At this moment I do not have a secure MQTT available myself (will set this up later).

Jeltel, 2020-04-04:
Works great btw :)

commit 4dffde27 (2023-09-21): Close #1 with the problem in the battery template.
Corrected the sensor unit type for pdischarge1
closed 2020-04-04 by Jeltel