johanmeijer/grott #2: ShineLan not supported?

wilcodl, 2020-03-31:
Hi,

When I'm running grott on a raspberry pi with an Growatt Shinelan stick I get the message:

- No Growatt data processed or SN not found:

Is the shinelan stick not supported? The data it is trying to parse is:

\x00\x01\x00\x02\x00\xd9\x01\x04\x59\x55\x5a\x30\x38\x33\x30\x30\x34\x34\x
51\x4b\x42\x32\x39\x30\x39\x32\x39\x45\x00\x00\x00\x00\x00\x00\x02\x00\x00
\x00\x2c\x00\x01\x00\x00\x05\x8e\x0f\x45\x00\x03\x00\x00\x05\x8e\x00\x00\x
00\x00\x00\x00\x00\x00\x00\x00\x05\x00\x13\x85\x08\xe5\x00\x06\x00\x00\x05
\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x
00\x00\x01\x00\x00\x5b\xa8\x01\x80\xd3\xb1\x00\xf1\x00\x00\x00\x00\x00\x00
\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x42\x00\x00\x00\x00\x
00\x2d\x00\x59\x4e\x20\x00\x00\x00\x05\x00\x00\x00\x01\x00\x00\x61\x60\x00
\x00\x00\x00\x00\x00\x00\x00\x00\x00\x61\x60\x00\x00\x00\x00\x00\x00\x00\x
00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x05
\x00\x08\x00\x00\x0f\xd0\x0f\x57\x0f\x4d\x00\x01\x10\x34\x75\x30\x00\x00\x
0e\x10\x00\xeb\x0e\x10\x00\x03\x00\x02\x00\x2d\x00\x00\x00\x00\x00\x00\x00
\x00

My generic config section looks like:

inverterid = QKB290929E
decrypt = True
minrecl = 100
compat = True
valueoffset = 26

I already tried changing the decrypt and compat settings

johanmeijer, 2020-03-31:
Hi, 

I have not be able to test with a shinelan but at first sight the record is the same and should not give a problem. 

The record is not encrypted and the offset of the data is 6 (default). 

The program should be able to detect your values with the following config: 

inverterid = QKB290929E
decrypt = False
minrecl = 100
compat = False
valueoffset = 6

(last valueoffset is not important while if compat=False the offset is set to 6 by default). 

Can you try it with this values and if that is not working send me the output (from the system journal if you run it as a service). 

I expect at least that your inverter is recognized because it is in the data. If that is the case and the values are not as expected then the offset has to be changed. 

To be honest I have not been able tot test the code with decrypt = False (I only have one growatt and he sends encrypted data) so there might be a code issue. 
If this is the case we can try to fix it :)

wilcodl, 2020-04-01:
I ran the python script with decrypt and compat disabled but now the script crashes, see attached log:

Apr 01 11:07:16 rpi1b grott[1866]:   File "/home/pi/grott/grott.py", line 238, in main
Apr 01 11:07:16 rpi1b grott[1866]:     if(result_string.find(SN) > -1):
Apr 01 11:07:16 rpi1b grott[1866]: AttributeError: 'list' object has no attribute 'find'

[grott-log.txt](https://github.com/johanmeijer/grott/files/4414383/grott-log.txt)

It would be very nice if we can fix this and add ShineLan support :)

johanmeijer, 2020-04-01:
Hi Wilco,

Thank you for the information. I think I found the problem and will upload a new version of the program. 

This is the output I see now from your information: 
- Growatt unscrambled data processed for:  QKB290929E
                 - Growatt unscrambled data:
                 - Growatt processing values for:  QKB290929E
                         - pvserial:       QKB290929E
                         - pvstatus:       1
                         - pvpowerout:     128.0
                         - pvenergytoday:  0.1
                         - pvenergytotal:  2346.4
                         - pv1watt:        142.2
                         - pv2watt:        0.0
                         - pvfrequentie:   49.97
                         - pvgridvoltage:  227.7

Is this the informaton you expected?

wilcodl, 2020-04-01:
Hi Johan,

The information looks valid! Thanks, can't wait to to use grott:)

johanmeijer, 2020-04-01:
I uploaded the changed grott.py (version 1.0.7) 

Please let me know if this works. I am very curious.

wilcodl, 2020-04-02:
Thanks for the fast new release. Unfortunately I get another error with the new version:

Traceback (most recent call last):
  File "grott.py", line 425, in <module>
    main()
  File "grott.py", line 254, in main
    result_string = message.hex();
AttributeError: 'list' object has no attribute 'hex'

johanmeijer, 2020-04-02:
Oeps sorry,  used a wrong data format with my tests. Hope I solved it now. 
I wil add temporary grott108.py to the repository. Can you please test this one? 

Thanky you for testing!

wilcodl, 2020-04-02:
It worked! But the script exited after:  Growatt unscrambled data processed for:  QKB290929E

The 'break' on line 276 has to be removed, after I do that the script keeps running and sends the data to MQTT :)

johanmeijer, 2020-04-02:
Oke thanks for testing and debugging :)

I will make the change to the code and commit it.

johanmeijer, 2020-04-02:
Which growatt inveter do you use? 

If it all works fine I can add it as tested to the documentation.

wilcodl, 2020-04-02:
A Growatt 3000-S with ShineLan stick.

Maybe you can also add enabling IPv4 forwarding in the kernel to the documentation? I had to set net.ipv4.ip_forward to 1 in /etc/sysctl.conf on Raspbian

johanmeijer, 2020-04-02:
Thanks! I will at this to the Wiki,

johanmeijer, 2020-04-04:
Do you agree this issue can be closed?

wilcodl, 2020-04-04:
yes it works great, thanks!

closed 2020-04-04 by wilcodl