johanmeijer/grott #3: How to remove the Iptables rules

Jeltel, 2020-04-02:
I'm trying to get it to work on my Ubuntu 18.04 server. But I'm not getting any data.

So now I want to return to my old proxy, so that the server data is at least updated. But now running that doesn't even receives data. I suspect the iptables rules.

How can I delete them?

Tried deleting the rule bij linenumber and the whole thing. But it says it cannot find the rule number...
https://www.digitalocean.com/community/tutorials/how-to-list-and-delete-iptables-firewall-rules

After this i want to check what's going on and why it's not running properly.

johanmeijer, 2020-04-02:
Did you change the destination address in the ShineWlan to your ubuntu server (via initial setup dialog or growatt website)? 

I run this on a raspberry pi. I will see if I can test it on a ubuntu server. 

My experience is, if you have not made the isptable rules persistent there are gone after a reboot of the server (again on a rpi).

Jeltel, 2020-04-02:
Yes I did. I had a perl proxy running, to divert the data from the Shine-Wifi to the Growatt servers. Worked great, until the datamessage changed.

I did nothing on purpose to make the rules persistant, just copied your rule.

Can you share what you need to see in the iptables after adding the rules?

johanmeijer, 2020-04-02:
I think if you stop the Perl proxy probably no data is sent to the Growatt server(a proxy receives, processes and sends)  that is a main difference with the sniffer. If ipforwarding is working correct the inverter data is always sent to the growatt server. Grott only "sniffers" the network for Growatt records but does not do anything with it. If grott not runs the data is still being sent to Growatt Server.

For the output you post above I think you also used the trace option (-t) and that shows all network trafic (can be a lot). And yes it will also trafic for port 5279 if it is there. 

Using only -v (--verbose) gives only information if Growatt records are processed you will see something like this. 

[image]
[image]

There might be more settings needed for IP (port)  forwading to work. A good description is given here: 
https://www.revsys.com/writings/quicktips/nat.html 

You have to enable ipforwarding in the kernel (not sure if that is necessary for Ubuntu):
echo 1 > /proc/sys/net/ipv4/ip_forward

And to make sure that IP forwarding works after reboot You will need to edit /etc/sysctl.conf and change the line that says net.ipv4.ip_forward = 0 to net.ipv4.ip_forward = 1

To cleanup IP  forwarding I used: 
sudo iptables -t nat -X
iptables -t nat -F
sudo iptables -X
sudo iptables -F

Other ways of cleaning and setting up IP forwarding for Ubuntu are described here: 
https://upcloud.com/community/tutorials/configure-iptables-ubuntu/

The output of the display commands if IP forwading is active:

sudo iptables -t nat -L
[image]

sudo iptables -nvxL
[image]

I hope you can at least go back to the old settings before activating IP forwarding but I almost know for sure that grott should be able to work with Ubuntu as well. And thinks this an elegant solution for getting Growatt information in MQTT (and Nodered to do more processing with it like sent it to domoticz, pvoutput.org, influxdb/grafana),

Jeltel, 2020-04-02:
They are gone now after a second reboot. Don't know what happened.

Thanks for all the info. Cleared up a lot. Will test later, when the sun is shining ;) and report back.

Hopefully it works on a Growatt 2500-MTL-S and Ubunut 18.04 :)

Jeltel, 2020-04-02:
One more question. For the perl solution it was mandatory to put the IP of my server in the shine WiFi stick. Can I just reroute server.growatt.com in my local DNS to the server?
Would save me some time during testing, because I won't have to edit the IP on the stick itself.

johanmeijer, 2020-04-02:
I do not think that this will work. It is a 2 way communication. The traffic back from Growatt to your inverter is also needed. I do not know if it is possible to find the way back if you use dns. 

I am not sure. It might be worthwhile to test.

closed 2020-04-04 by Jeltel