johanmeijer/grott #4: Docker support

Jeltel, 2020-04-02:
Is it an option to set this up in a docker image? This would improve installation drastically.

Not sure about the Iptables section though.

johanmeijer, 2020-04-02:
Nice thought! 

I put it on the todo list. I have to improve  my Dockers skills before I can start (but that was on the todo list as well). Do not expect a solution next week ;)

Network sniffering might be an issue indeed. We will see.

Jeltel, 2020-04-02:
No need for it next week. I know learning new things take time. See it as a challenge ;)

Jeltel, 2020-05-24:
With the proxy/sniffer option it's easier to install and more robust. A docker image would be even more robust and easier to maintain. But the need is a bit lower now :)

You're work is great. Thanks again.

johanmeijer, 2020-09-25:
@Jeltel 

I published a docker container on docker hub under: ledidobe/grott:2.2.0. 

The minimal command to run it is: 
docker run -d --network=host --restart on-failure -e gmqttip="yourip" ledidobe/grott:2.2.0

So only IP of your mqtt server needs to be specified. 

It build with grott version 2.2.0 this version has a lot of nw functionality (like docker support, direct output to PVOUTPUT.org  and command blocking/filtering so inverter can not be hacked) 

Most of the .ini settings can be used with the docker to. For a ini setting there is a envrionmental defined that can be set with -e at startup. e.g. if you want to use mqtt security specify also: 
-e gmqttauth = "True" -e gmqttuser = "grott" -e gmqttpassword = "growatt2020" 

I will publish version 2.2.0 and update the documentation in the comming days. 

I tested it on a Ubuntu Linux on x86 machine. I also published an ARMv32 version for the older Raspberry PI's (ledidobe/grottrpi:2.2.0).

Jeltel, 2020-09-25:
I will test it. 

Can you specify all the environment variables (the -e's). I'm using docker-compose and the mqtt variables are passed, but not the inverterid. It stays at the default id. And also the grott ip is set at 0.0.0.0?

johanmeijer, 2020-09-25:
The new version automatically detects most of the things, like inverterid,  the moment records are processed. 
Grott ip 0.0.0.0 is oke (= local). I do understand the default inverter id looks a little bit strange but is oke.

You only have to specify the mqtt settings (ip, security if needed) as described above.

johanmeijer, 2020-09-25:
You can also specify -e gblockcmd="True"  this will blocks the commands from outside  (e.g. growatt server in china) to the inverter  so nobody can control/hack your inverter.

johanmeijer, 2020-09-25:
With local (0.0.0.0) ip address I mean the ip address of the container or if you specify --network=host (as described)  the ip address of the docker server. 

I will look if it is possible to show the real ip address during start up.

johanmeijer, 2020-09-28:
@jeltel have you been able to run it? 

I uploaded a new version (2.2.1) which shows deviceid=automatic, ip address = default and after start the message which IP address and Port is used. 

You can now run it without specifing any taggs (tags latest is added to the most current version). 

So in your case: 

docker run -d --network=host --restart on-failure -e gmqttip="yourmqttip" -e gmqttauth = "True" -e gmqttuser = "yourmqttuser" -e gmqttpassword = "yourmqttpwd" ledidobe/grott

Jeltel, 2020-09-29:
@johanmeijer I ran it at the same system as the service. I stopped that service and started the docker, but got this message:
```
grott    | Grott proxy mode started
grott    | Traceback (most recent call last):
grott    |   File "grott.py", line 33, in <module>
grott    |     proxy = Proxy(conf)
grott    |   File "/app/grottproxy.py", line 48, in __init__
grott    |     self.server.bind((conf.grottip, conf.grottport))
grott    | OSError: [Errno 98] Address already in use
grott exited with code 1
```

Is stopping the service not enough? Do I need to remove the service and restart the complete server maybe?
I don't have much time during the day to play with it, and I'm not going to restart the server now ;)

johanmeijer, 2020-09-29:
Stopping the service should be enough. 

Do you have maybe multiple grott containers active? 

My expierence is that the docker logging lag a bit behind and sometimes older messages are displayed between the actual messages.

Jeltel, 2020-09-29:
I saw the older 2.2.0 was still running. Stopped and deleted everything I could find and tried again:

```
grott    | Grott proxy mode started
grott    | Traceback (most recent call last):
grott    |   File "grott.py", line 33, in <module>
grott    |     proxy = Proxy(conf)
grott    |   File "/app/grottproxy.py", line 50, in __init__
grott    |     print("IP : ", socket.gethostbyname(socket.gethostname()), ", port : ", conf.grottport, "\n")
grott    | socket.gaierror: [Errno -2] Name or service not known
grott exited with code 1
```

johanmeijer, 2020-09-29:
The docker container will restart automatically.

The docker ps command shows the active containers and also shows if they are running and restarting etc.

Probably the container is restarting. And claims the address.

johanmeijer, 2020-09-29:
Oke. I will look into the error (I do not have on mij rpi and ubuntu docker test environments).

I build this display massage so you can see what the actual ip address that is beeing used. Maybe I delete it again......

johanmeijer, 2020-09-29:
Is this error generated with the docker container or are you running 2.2.1 as service?

Jeltel, 2020-09-29:
That was the docker container. I reverted back to 2.0.1 running as a service. Haven't updated that part yet :)

johanmeijer, 2020-09-29:
Strange. Can you please share your docker run command (without your detailed information of course). So  I can try to recreate the problem?

Jeltel, 2020-09-29:
I'm actually using docker-compose for all my containers, so that I have saved the command in the same place as the data. Just a matter of preference. Should work both ways.

I've converted your run command to a docker-compose with this site: https://www.composerize.com/
Result:

```
version: '3.3'

services:
  grott:
    image: ledidobe/grott
    container_name: grott
    restart: always
    environment:
      - gmqttip=192.168.1.xx
      - gmqttauth=true
      - gmqttuser=user
      - gmqttpassword=password
    network_mode: host
```

johanmeijer, 2020-09-29:
Thanks. I will test it. 

Your docker server is running Ubuntu (or other Linux build) or Windows Server ?

Jeltel, 2020-09-29:
Ubuntu server 18.04 LTS

johanmeijer, 2020-09-29:
I tested it on Windows, Ubuntu 18.04 en 20.04 en RaspberryPi. All go well (Docker and Service). 

After that I came to the conclussion it most have something to do with your network configuration (especially domain name resolution). There seems something different here in your configuration.

Components / definitions involved: 

/etc/hosts
/etc/resolve.conf
DNS server (or maybe missing DNS server). 

I can almost recreate the problem (errno -3) by deleting the /etc/resolv.conf and delete the 127.0.1.1 hostname from /etc/hosts. 

I can not solve your netwerk issue's (if it are issue's if you noticed it untill now). Wat I can do is bypass the error if it occurs. You will miss the IP-address, port information at startup,  but I think we can live whith that don't you think? 

I will change the code and create a new docker image.

johanmeijer, 2020-09-29:
Learning on the fly. 

I discussed this with my son and together we tried to recreate and solve the problem (and test it on his ubuntu 18.04). The moment I changed the /etc/host and used sudo I  get a warning about hostname resolution. 
My son said: we had that always with NUC's at work. 

It seems that NUC by default do not set a hostname. Maybe that can be the issue. The resolution might be adding the hostname to the etc/hosts file. Can you see if there is a line: 127.0.1.1 yourhostname in this file? 
     
I do understand if you don't want to go down this route. Probably tomorrow I will publish the new build that will skip this error. You can also wait on that.

johanmeijer, 2020-09-29:
Last comment for today: I published ledidobe/grott:2.2.1b on docker hub. 

Hopefully your problem is solved with this build.

Jeltel, 2020-10-01:
@johanmeijer That does start and gets the data. It unfortunately doesn't connect correctly to my MQTT server. With the same information as was in the grott.ini. With that information i can login fine with a MQTT explore application.

So one step at a time 👍

```
grott    |       - Grott data record date/time processing started
grott    |       -  2020-10-01T16:55:48
grott    |      - Grott values retrieved:
grott    |               - pvserial:          xxxxxxxxxxx
grott    |               - pvstatus:          1
grott    |               - pvpowerin:         72.6
grott    |               - pvpowerout:        0.0
grott    |               - pvenergytoday:     0.0
grott    |               - pvenergytotal:     52428.8
grott    |               - pv1watt:           33.0
grott    |               - pv2watt:           39.6
grott    |               - pvfrequentie:      0.0
grott    |               - pvgridvoltage:     0.0
grott    |               - pv1voltage:        118.3
grott    |               - pv1current:        0.2
grott    |               - pv2voltage:        115.7
grott    |               - pv2current:        0.3
grott    |               - pvtemperature:     13.4
grott    |               - pvipmtemperature:  2.5
grott    |       - MQTT jsonmsg:
grott    |                       {"device": "ccxzcxzcxzcx", "time": "2020-10-01T16:55:48", "buffered": "yes",
grott    |                       "values": {"pvstatus": 1, "pv1watt": 330, "pv2watt": 396, "pvpowerin": 726,
grott    |                       "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvenergytoday": 0,
grott    |                       "pvenergytotal": 524288, "pv1voltage": 1183, "pv2voltage": 1157,
grott    |                       "pv1current": 2, "pv2current": 3, "pvtemperature": 134, "pvipmtemperature":
grott    |                       25}}
grott    |       - MQTT send failed: Connection Refused: not authorised.
grott    |       - Grott Send data to PVOutput disabled
```

johanmeijer, 2020-10-01:
I sent you also a message via tweakers.

johanmeijer, 2020-10-01:
@Jeltel I found the error I think for the wrong values. Probably this only for the "bufferered" records. (the shine will buffer records if there is no connection with the growatt server en send them later). The real time records will have the right values. 
I will fix that.   

The mqtt problem I only have if there are two grott instances  active (at  2 different servers). But you use also MQTT security so that can be cause an error as well (but then there should be an error in passing userid/password). I will have to look deeper into that.

johanmeijer, 2020-10-01:
Sorry I was to enhousiastic. It was not the error I thought. I need more record information as described above. 
I espcially like to know the layout part (in my example T060104).

johanmeijer, 2020-10-04:
Hopefully (fixed) in grott 2.2.1d. 

Docker container ledidobe/grott:2.2.1d and ledido/grott:latest gepublished. Als docker 2.2.1 branche updated.

Jeltel, 2020-10-05:
@johanmeijer Yes! This works.
Data is processed correctly, and sent to the MQTT server (and thus to Home Assistant in my case).

Great work. The service version has been disabled.

For people with docker, I think this is the easiest way to run the grott-server.

johanmeijer, 2020-10-05:
@Jeltel gblockcmd=True environmental is already available.
 
In principal are all settings from .ini available as environmentals. I will document this in the comming days.

johanmeijer, 2020-10-19:
@Jeltel  can we close this issue?

closed 2020-10-19 by Jeltel