johanmeijer/grott #6: Feature Request add more values to MQTT

Jeltel, 2020-04-04:
The json now is:
```
{
    "device": "xxxxxxxxxxx",
    "time": "2020-04-04T12:38:06",
    "values": {
        "pvstatus": 1,
        "pv1watt": 6,
        "pv2watt:": 18056,
        "pvpowerout": 17493,
        "pvfrequentie": 5000,
        "pvgridvoltage": 2458,
        "pvenergytoday": 74,
        "pvenergytotal": 1601
    }
}
```
But there are more values sent over in the message like:
* PV Amperage
* PV Volts
* Inverter Temperature

This is the output of the script in verbose mode:
```
                 - Growatt plain data:
                           00437269763e75704a50433439333034313100000000000000000000000000000000000000
                           004353453139333130364100000000000000000000000000000000000000001404040e2104
                           030000002c0001000046e502680000000000060917004d000046df000044e7138909a70047
                           000044f600000000000000000000000000000000000000480000063f001ac99d015e000000
                           0000000000000000000000000001580ebd00000000002d00594e2000000000000000000000
                           012b0000004a00000578000006a30000000000000000000000000000000000460000000000
                           00000000000000000000000000000000000000000000000000000000000000000000000000
                           00000000005a0086001914031907180b000000191403170624310000000011010100000000
                           00000011010100000000000000110101000000000000000000000000000000000000000000
                           0000000000000000000000000000000000000000000000007b0b4c
                 - Growatt processing values for:  xxxxxxxx
                         - pvserial:       xxxxxxxx
                         - pvstatus:       1
                         - pvpowerout:     1763.9
                         - pvenergytoday:  7.2
                         - pvenergytotal:  159.9
                         - pv1watt:        0.6
                         - pv2watt:        1814.3
                         - pvfrequentie:   50.01
                         - pvgridvoltage:  247.1

```

I can help with getting the other values by giving the values which they should roughly be:
* PV ampere 1: 0
* PV ampere 2: 7.7
* PV Voltage 1 : 61.5
* PV Voltage 2: Roughly: 226 - 232
* Inverter Temperature: Roughly 35-36 °C

Can you also get these out of the message and sent them over MQTT?

johanmeijer, 2020-04-04:
No problem to add more values. Below the known values in the Growatt record. 

[image]
[image]
 
The values I find in your record:
pv1voltage	61.6	
pv1current	0	
pv2voltatge	232,7 	
pv2current	7,7	
pvtemp 	        35

Jeltel, 2020-04-04:
Great!

What does the High and Low mean in the different attributes?

johanmeijer, 2020-04-04:
It is a 2 or 4 bytes value. (High and low together makes 4 bytes)

johanmeijer, 2020-04-04:
I have made the changes (correct the pv2watt error) and added new values to the json message: 

pvinput 
pv1voltage
pv2voltage
pv1current 
pv2current
pvtemp

I have added the new version to the repository! Can you please test it?

Jeltel, 2020-04-04:
I've got them in :) Thanks.

I do think the pvtemp is a bit too low. Have never seen it that low with the old implementation. More around 35 °C. Just like you said. Can you check?

Latest:
```
                 - Growatt plain data:
                           00c97269763e75704a50433439333034313100000000000000000000000000000000000000
                           0043534531393331303641000000000000000000000000000000000000000014040413112c
                           030000002c000100000134025b00000000002a076e00010000010a0000012e1388094f0001
                           0000013d000000000000000000000000000000000000006d00000664001b4bd400d4000000
                           00000000000000000000000000010e0e4000000000002d00594e2000000000000000000000
                           012b000000700000059e000006c90000000000000000000000000000000000010000000000
                           00000000000000000000000000000000000000000000000000000000000000000000000000
                           00000000005a0086001914031907180b000000191403170624310000000011010100000000
                           00000011010100000000000000110101000000000000000000000000000000000000000000
                           000000000000000000000000000000000000000000000000cca6e1
                 - Growatt processing values for:  CSE193106A
                         - pvserial:       CSE193106A
                         - pvstatus:       1
                         - pvpowerin:      30.8
                         - pvpowerout:     30.2
                         - pvenergytoday:  10.9
                         - pvenergytotal:  163.6
                         - pv1watt:        4.2
                         - pv2watt:        26.6
                         - pvfrequentie:   50.0
                         - pvgridvoltage:  238.3
                         - pv1voltage:     60.3
                         - pv1current:     0.0
                         - pv2voltage:     190.2
                         - pv2current:     0.1
                         - pvtemperature:  21.2
```

Rest seems fine :)

Jeltel, 2020-04-04:
Maybe its the difference between 31 Inverter Temperature and 41 IPM Temperature.
Inverter Temperature is maybe sort of the ambient temperature and 41 the internal temperature of the device?
Then I would like both :)

johanmeijer, 2020-04-04:
It really does seem to be the temperature at this moment (it is at the same location in the record as the 35 in your previous one).  

You are not generating at lot energy so maybe it is lower at this moment (the actual temperature in my inverter is now 26).  

The IPM temperature for your record is 27. I will add that to the JSON tomorrow (can not test it any more today). 

Please look at again tomorrow when it produces more energy.

Jeltel, 2020-04-04:
Ok, I guess you're right. I see just before shutdown the temperature drops from around 35 to 21. So this was probably just before shutdown.

Do you know what the IPM temperature is?

Thanks for your work today. Really nice to see features get in so fast.

johanmeijer, 2020-04-05:
I added pvipmtemperature to the jsonmessage. 

I am not sure if it gives the right value. I Can not test it with my Growatt 1500-s while IPM temperature is not inserted in the record (value = 0). 

IPM stands for inteligent power module. This a major module within the inverter to generate and control the (output) power.

Jeltel, 2020-04-05:
I'm getting a value of 38,1 °C. Seems the general ballpark at least. Will see what it does during the day.

Thanks for all this. I'll let you close this if you think its done.

johanmeijer, 2020-04-06:
For now all relevant values are added to the json string. If more values are needed please open a new issue.

closed 2020-04-06 by johanmeijer