johanmeijer/grott #7: Json markup error?

Jeltel, 2020-04-04:
json i got is:
```
{
    "device": "xxxxxxxxxx",
    "time": "2020-04-04T12:38:06",
    "values": {
        "pvstatus": 1,
        "pv1watt": 6,
        "pv2watt:": 18056,
        "pvpowerout": 17493,
        "pvfrequentie": 5000,
        "pvgridvoltage": 2458,
        "pvenergytoday": 74,
        "pvenergytotal": 1601
    }
}
``` 
But pv2watt has another : in it: `"pv2watt:": 18056,` should be `"pv2watt": 18056,`

Just a bit confusing :)

johanmeijer, 2020-04-04:
Yes! This an error I will change that ( I don not have 2 strings and use only PV1).

Thanks.

Jeltel, 2020-04-04:
Found it in rule 322 of grott.py, changed it myself.
Don't know how to do a pull request though.

johanmeijer, 2020-04-04:
I will change it! Version 1.0.9 is on his way.

Jeltel, 2020-04-04:
The ':' is gone in version 1.0.9. Thanks :)

closed 2020-04-04 by Jeltel