johanmeijer/grott #8: Suggestion: Publish json attributes in seperate MQTT topics

Jeltel, 2020-04-04:
It's pretty hard to extract values from the json in Home Assistant (possibly related to #7 ), where it is easy to extract a value from MQTT in Home Assistant.

Is it an option to publish all the values in a seperate topic?
`energy/growatt/pv1watt` etc?

johanmeijer, 2020-04-04:
I personally prefer to have 1 message with one time stamp so I can handle that in one step in nodered and do further processing.

Jeltel, 2020-04-04:
Well I've got it working, so it's not a high on the list feature. But it might be easier for some.

wilcodl, 2020-04-04:
Do you got example code for Home Assistant, @Jeltel? I want to add the info to HA as well.

Maybe it can also be added to the documentation

johanmeijer, 2020-04-04:
Yes please add home assistent setup to the wiki (or send it me and I will upload).

I use Domoticz so I can add a lot off value with this.

Jeltel, 2020-04-04:
Sure:
```
sensors:
  - platform: mqtt
    state_topic: energy/growatt
    value_template: "{{ value_json['values']['pvenergytoday'] | int / 10 }}" 
    name: Elektra Opgewekt Vandaag
    unit_of_measurement: "kWh"
    icon: mdi:solar-power
```

This is the general format.
* The state_topic is your MQTT topic on which the json is published by Grott
* Value_template converts the specific attribute to the right number.
  * `value_json['values']['pvenergytoday']` will take the pvenergytoday. Replace it with every other attribute in the values attribute to get that one. If you want time for example, you can just do `value_json.time`
  * `| int` converts it into an integer so we can do calculations on it
  * `/10` divides by 10 to get the correct value
* Name: Speaks for itself
* Unit_of_measurement kan be anything you like, as long as its a string
* Icon: taken from https://materialdesignicons.com/ Any icon prefixed with mdi: will work

IMPORTANT! get the right string characters around the value_template. Its a double quote `" "` around the whole template. And a single quote `' '` around the attribute names. 
I was messing around with that before I got it right.

You can put that in the wiki. I cannot seem to do that from my account.

johanmeijer, 2020-04-05:
Thanks! Added to the wiki :)

johanmeijer, 2020-04-08:
At this moment I do not see the added value of having a json message per value. For now I close this issue. If in the future more requests for multiple json messages are received, I will reconsider it.

closed 2020-04-08 by johanmeijer