johanmeijer/grott #9: Grott on Ubuntu Server routing issues

Jeltel, 2020-04-13:
As mentioned before I'm trying to get grott running on my hassnuc with Ubuntu Server 18.04. On that system I have also Home Asssistant and two seperate other docker programs running. Adding grott to this should be great, because the nuc is already running.
I didn't got this working earlier, so I opted for the easy way out by starting my Pizero W and adding grott. That went smoothly :)

Now I've tried to start it on my nuc again by following your tutorial for the Pi. I've got this far:
* grott.py runs without errors and delivers expected output after installing the needed pip3 modules
* Enabling grott to systemctl also works fine
* Rerouting the data however:
  * `/proc/sys/net/ipv4/ip_forward` exists and contains a 0 initially. Changed that to 1
  * `/etc/sysctl.conf` exists and I've rempved the # before the line that says `net.ipv4.ip_forward = 1`
  * Adding `sudo iptables -t nat -A PREROUTING -p tcp --dport 5279 -j DNAT --to-destination 47.91.67.66:5279` is also done nicely
  * `sudo iptables -t nat -A POSTROUTING -j MASQUERADE` can also be added, but slows the response of the NUC way down.

After adding the POSTROUTING rule the NUC response is very slow and the data isn't send through to server.growatt.com. Therefore no communication is set up which can be sniffed on the NUC> Data from the ShineWifi stick is sent to the NUC as can be seen during the -t option output.

Since I have no knowledge of routing the data. I have no idea what's going on and where to search for the solution.
Home Assistant also adds rules to the iptables. Don't know if they might be conflicting. 

Can someone with more knowledge of iptables help me?

Jeltel, 2020-04-13:
Is `sudo iptables -t nat -A POSTROUTING -j MASQUERADE` maybe a bit to wide? Can't we limit this to the ip of the ShineWifi stick?

johanmeijer, 2020-04-15:
As far I understood the masquerade rule enables NAT and should enable the way back from the Growatt server to the Growatt inverter. 

It might be possible to replace this with a more specific Masquerade or SNAT rule. 

I am not very familar with these setting I will try to find out if there is more information availble.

johanmeijer, 2020-04-15:
Sorry my intension was not to close it.

johanmeijer, 2020-04-15:
Do you have both WLAN and Ethernet (Cable) enabled and connected on your NUC?

Jeltel, 2020-04-16:
Both enabled, but only connected on ethernet.

johanmeijer, 2020-04-16:
Oke. I am not sure the masquerade NAT rule is the problem. But I see on internet that it might give some performance delays. I replaced these rule by a more specific SNAT rule. I do not expierence performance issue so I can not really verify if its helps. Maybe you can? 

What I did: 

**Added the SNAT rule:** 
sudo iptables -t nat -A POSTROUTING -o wlan0 -j SNAT --to-source 192.168.0.206
_wlan0, will be eth0 for your config and to source IP, 192.168.0.20, is the IP of your nuc._ 

**After that I delete the masquerade rule:** 

Show de rules with linenummers: 
- sudo iptables -t nat -L --line-numbers

Output: 
[image]

And the deleted the masquerade rule: 
- sudo iptables -t nat -D POSTROUTING 1

Hopefully this works...............

Jeltel, 2020-04-20:
Thanks for the research. I'll try it later and report back.

johanmeijer, 2020-04-20:
Your network might be a little bit more complex (additional virtual adapters) because of the docker implementation. I have not found a lot information yet about the combination of IP forwording and the use of Docker. 

I like to know if this works. Otherwise I finally have to set up a docker environment myself ;)

Jeltel, 2020-04-21:
Since this isn't running in docker, I don't think the dockers running on the same machine should matter. But I'll see, when I get to it ;)

Jeltel, 2020-04-29:
@johanmeijer don't you mean to use the IP of the Growatt Inverter and not the device running grott (raspberry Pi in your case, Nuc in mine?)

> Added the SNAT rule:
> sudo iptables -t nat -A POSTROUTING -o wlan0 -j SNAT --to-source 192.168.0.206
> wlan0, will be eth0 for your config and to source IP, 192.168.0.20, is the IP of your nuc.

johanmeijer, 2020-04-30:
What I understood from the documentation (and what is working for me) is this the IP adrress of the physical netwerkadapter of your NUC. 

The Masquerade options means the system is looking dynamically which adapter should be used.  With the SNAT rule you fix (=make static) that. 

But be aware I am not running Docker containers. I looked at it on a other system with Docker and  Docker containers can have there own virtual network interface (depends on the way you start it). You can see this with ifconfig. Docker also adds rules to iptables (e.g. a masquerade rule).

The network of a Linux environment with docker is more complex.  I am not sure docker networking will work if you change (delete) the Masquerade rule. 

I have created a grott docker container and it  seem to be working (not fully tested yet). I will publish more about this later (in the other topic).  Be aware when running grott in a docker container you still need to setup IP forwarding on base Linux level as described.

Jeltel, 2020-05-04:
I now have these entries:
```
Chain PREROUTING (policy ACCEPT)
num  target     prot opt source               destination
1    ....
2    DNAT       tcp  --  anywhere             anywhere             tcp dpt:5279 to:47.91.67.66:5279

...
Chain POSTROUTING (policy ACCEPT)
num  target     prot opt source               destination
1    ...
21   SNAT       all  --  anywhere             anywhere             to:192.168.x.y

...
```
Where 192.168.x.y is my local server IP. I see Grott is receiving messages, but is not able to process them:

```
Grott Command line parameters processed:
	verbose:     	 True
	config file: 	 grott.ini
	output file: 	 <_io.TextIOWrapper name='/.../growatt/output.txt' mode='w' encoding='UTF-8'>
	nomqtt:      	 False
	trace:       	 False

Grott configuration file processed:

	minrecl:     	 100
	decrypt:     	 True
	compat:      	 True
	valueoffset: 	 26
	inverterid:  	 ...
	mqttip:      	 192.168.x.y
 	mqttport:    	 1883
	mqtttopic:   	 energy/growatt
	mqtttauth:   	 True
	mqttuser:    	 ...
	mqttpsw:     	 ...
 	growattip:   	 47.91.67.66
 	growattport: 	 5279

Grott value location offset:  26 	Compat mode:  True

Grott monitoring started

	 - TCP Segment Growatt:
		 - Source Port: 2937, Destination Port: 5279
		 - Source IP: 192.168.1.29, Destination IP: 47.91.67.66
		 - Sequence: 6509, Acknowledgment: 0
		 - Flags:
			 - URG: 0, ACK: 0, PSH: 0
			 - RST: 0, SYN: 1, FIN:0
	 - TCP Data less then minimum record length, data not processed
	 - TCP Segment Growatt:
		 - Source Port: 2937, Destination Port: 5279
		 - Source IP: 192.168.1.29, Destination IP: 47.91.67.66
		 - Sequence: 6509, Acknowledgment: 0
		 - Flags:
			 - URG: 0, ACK: 0, PSH: 0
			 - RST: 0, SYN: 1, FIN:0

... 
	 - TCP Segment Growatt:
		 - Source Port: 18348, Destination Port: 5279
		 - Source IP: 192.168.x.y, Destination IP: 47.91.67.66
		 - Sequence: 23603, Acknowledgment: 0
		 - Flags:
			 - URG: 0, ACK: 0, PSH: 0
			 - RST: 0, SYN: 1, FIN:0
	 - TCP Data less then minimum record length, data not processed
	 - TCP Segment Growatt:
		 - Source Port: 38089, Destination Port: 5279
		 - Source IP: 192.168.x.y, Destination IP: 47.91.67.66
		 - Sequence: 25231, Acknowledgment: 0
		 - Flags:
			 - URG: 0, ACK: 0, PSH: 0
			 - RST: 0, SYN: 1, FIN:0
	 - TCP Data less then minimum record length, data not processed
```
Also the data is not visible within the Shine server. So the data is not going through.

Are my settings correct?

johanmeijer, 2020-05-04:
No, having the data not going through is not oke. So the settings are not working for you. 

There is no communication between the Inverter and Growatt so the "real data" is not send (and can not be processed bij Grott). 

I do understand that you have more postrouting setting because of the docker environment. I am working at a Docker setup myself so I can test it.

johanmeijer, 2020-05-05:
I setup a Raspberry PI with docker containers enabled but can not reproduce your problem.

I tried it with both the full masquerade rule and with the more specific SNAT rule and they both seems to work on a environment with docker installed (Grott in a container also works but I have to fine-tune that a little bit more). 

Just to be sure I share my settings in the pictures below. 

[image]

[image]

[image]

When I look at your rules I do not see real differences. I do not have a lot of docker containers running (only the grott one) so my configuration is not that complex. I can see that you have more post-routing chain rules then the 2 you shared (1 and 21, so there are 19 more I think). There might be a rule in there that prohibit the forwarding to the growatt server. 

I used the IP address of the wlan0 interface because that is the only network interface being used. I supose you used the eth0 IP address (192.168.x.y) ? I fixed this address in my router (not using dynamic IP address for my servers).

johanmeijer, 2020-05-22:
@Jeltel. 
I published version 2.0.0 of Grott. This version has a proxy mode that does not require IP forwarding and is less resource intensive. 

I think this mode will make it easier for you to run Grott on your Ubuntu server with Docker installed. 

I will also enable docker container support in the near future. See the topic on Docker Support requirement for more information. 

In the readme you can find information on how to use Version 2.0.0 and enabling the proxy mode. 
The wiki will be updated later. 

I am very interested in your opinion about this update,

Jeltel, 2020-05-24:
@johanmeijer installed version 2 on my nuc, copied settings over from the pizero and ran it.
Pointed the shinelan wifi stick to my nuc and voila. Worked immediately.

And if there is a problem with my nuc I will notice very soon (for far more important things stop working), so this proxy thing is a way easier setup. And more robust.

Thanks!

closed 2020-05-24 by Jeltel