johanmeijer/grott #10: Grott as Server ?

windundsterne, 2020-05-02:
very nice tool, setup worked directly with a growatt 1000s shine wifi / Ubuntu 18.03!

Is there a way to setup grott to act as server without redirecting the data to China?

Phil

johanmeijer, 2020-05-02:
Thanks!

At this moment this not possible to use grott as a server. But I do understand the requirement.

I also tought about it. The tool is now based on sniffering the network without interrupting the communtion between the inverter and growatt server. 

If grott needs to act as a server or proxy (allows optional communication with growatt server). I have to know more details about the used protocol (reversed enginering). 

I will put it on the wish list but do not expect it to be ready soon.

johanmeijer, 2020-09-28:
@windundsterne 

In the release 2.2.1 I added an option, if you use proxy mode, to block / filter incomming commands (blockcmd = True in .ini). 
The data will still be send to China but it is not possible to control your inverter (configer, reboot etc) from China. 

Do you think this will help? 

Creating and maintaining a server is a lot of work seeing the fact that new inverters and shinewifi-x / link-x devices have al their own protocol/data records. I now block the known configure and all unknown other records so new or chnaged configure records can not "hurt" the inverters. 

I personally like the Growatt reports and graphics and do not mind sharing the data with Growatt.

johanmeijer, 2020-10-19:
At this moment I will not implement a server option for Grott. At think the current grott implementation can handle most of the requirements (includiding packet filtering and blocking). 

In the future, if there are more request for the server option,  I might reconsider this decission.

closed 2020-11-20 by johanmeijer

Badmonky, 2021-12-01:
I would love this feature, Since I am off grid, and only have internet when i turn on my hotspot, my inverter is not always able to communicat with the growatt server. However I am planning to use grott to build a "dump load feature" that monitors the SOC of my batteries and the PV input, and depending on that state will heat a hottub if the conditions are favorable. 

Right now that will only work if i would have my phone act as the hotspot 100% of the time.

johanmeijer, 2021-12-01:
It is under consideration (start looking at it). It is not very easy to do (more reversed engenering of protocol needed).

Badmonky, 2021-12-01:
That's great to hear. 

Did you see this project:
https://github.com/knowthelist/Growatt-server

It sounds like from the readme, that it could be a standalone server, may be a source of inspiration.

johanmeijer, 2021-12-02:
I will look at it. Thanks.