johanmeijer/grott #11: Option to disable send data to Growatt

pwbrouwers, 2020-05-24:
Is it possible to add a configuration option to disable sending the data to growatt. Now I’m blocking all traffic to the growatt server.

O…. And changing the ip address for the growatt server in the ini file does not work.

Thanks,
Peter.

johanmeijer, 2020-05-25:
Hi Peter,

Thank you for finding the bug ( I have gone a little bit too far with cleaning up).
I repaired and published it (Version 2.0.2)

I do not think that blocking traffic to the growatt server will bring the right results. I think the Growatt shine stick needs communication with the growatt server to be able to send data. If you block that communication no data will be sent at all. 
The shine stick will buffer the data untill the session with the growatt server is established. All buffered data will be sent after a reconnect. 

If you are succesfull in blocking the communication and still see the data I have to change my ideas about how the communication works. Please let me know. 

I am thinking about a third mode (server) in which no connection with Growatt is needed (Grott will act as the Growatt Server) but have to do some more investigation on the protocol used.

See also issue #10.

pwbrouwers, 2020-05-26:
Hi Johan,

Thanks for fixing, the issue! 

I will test what wil happen if i block the access to the growatt server, and i will let you know.

Gr,
Peter.

closed 2020-05-26 by pwbrouwers

ChillXXL, 2020-06-21:
Thanks for the great app. I am wondering if the grott server is down, like for a day, and the rerouting of the data doesn't forward to the Growatt servers does than gaps exist in your month and year totals as like it skipped a day?

johanmeijer, 2020-06-21:
Hi.

Grott doesn't is a server. You are the server so if nothing sent through there is something wrong with your own configuration.

Is your system up? Is grott still runni g? Maybe a restart will help.

ChillXXL, 2020-06-22:
Sorry for the misunderstanding. I don't have grott running yet. I mend by server, the proxy-server. I now monitor the solar panels by the growatt app for year and month totals. 

What I am curious for, if for example, the grott proxy is down for a couple of days for some reasons, is this data lost and unknown by the Growatt app or does the Growatt server and inverter catch up the days once the grott proxy is up again.

johanmeijer, 2020-06-22:
In proxy mode data will not be sent to the growatt server anymore.

The inverter (wifi stick) will store the data and it will be sent after the connection is aivalable again. There will be no data lost. 
However I do not know for what period data can be stored. In my experience at least more then one day.

In sniffer mode (with ip forwarding) data will be sent to the grottserver if grott is not running and your system is still active.

ChillXXL, 2020-06-22:
Thanks. I didn't know that the inverter or wifi-stick has a (small) buffer if the connection goes down for a while. Makes sence and is good to know. 

I want to still use both apps (the official growatt app and yours) so with the automatic backup-feature in the inverter/stick in mind I can experiment freely with your app and use the sniffer mode.

> In sniffer mode (with ip forwarding) data will be sent to the **grottserver** if grott is not running and your system is still active.

With grottserver you mean **growatt** server? (typo?)

johanmeijer, 2020-06-22:
Yes sorry growatt server....

xDAVIDx93, 2020-08-29:
> In proxy mode data will not be sent to the growatt server anymore.
>
> The inverter (wifi stick) will store the data and it will be sent after the connection is aivalable again. There will be no data [...]

I read somewhere the ShineWifi-X can store 30 days of data