johanmeijer/grott #14: Am I doing something wrong ?

Vinisz, 2020-06-07:
Hi,

Do not have much experience with the Rpi and python, but how do I run this ?
I now do: python ./grott.py

but this throws an error as you can see below.
Do I need to change rights (chmod) things or somehting ?
I just did a clone from git on the Rpi.
Hop you can help me out..

pi@raspberrypi:~/grott $ python ./grott.py
  File "./grott.py", line 27
    if conf.verbose: conf.print()
                              ^
SyntaxError: invalid syntax

johanmeijer, 2020-06-07:
Do use python 3 or 2 ? Grott requires python 3.x.

Vinisz, 2020-06-07:
I see I have both..

Should I switch default version as described here :

 

 <https://raspberry-valley.azurewebsites.net/Python-Default-Version/> https://raspberry-valley.azurewebsites.net/Python-Default-Version/

?

Vinisz, 2020-06-07:
Btw, saw you are also working on docker image, I like that, would we great if it can run on synology nas

johanmeijer, 2020-06-08:
A switch is not required, but does not hurt if you do not have python 2 applications running (and are not going to do that in the feature). 

Python3 is the new standard but there is a lot of python2 stuff out there, 

At my RPI's python2 is still standard but use also python3 with the seperate command.  

Be aware I think  that PIP3 (the python installation program is not installed on the RPI by default. 
You needs this to install the paho-mqtt library. Grott requires this library. 

I am working on the docker images. I will at least provide a ARM32 version (for the RPI) and a "standard" version  (e.g to support Ubuntu on  x86/x64 system). I am not sure that one of the images will run on synology but we can try that. 

I will update the information in the seperate issue on docker when it is available.

Vinisz, 2020-06-08:
Thanks Johan,

May I ask what you did exactly on the Pi after scratch install ? (Which os did you install and what did you install after that ?)

johanmeijer, 2020-06-08:
Hmm that that was a while ago.....

I download and installed the minimal raspbian (is called different now I think) I modified the build to supprt SSH and add wifi setting before I booted the RPI from it. 

The only I added (on this rpi) is PIP3: 
sudo apt install python3-pip 

Install paho-mqtt
sudo pip3 install paho-mqtt

After that I created the growatt directory. Download the files (.py and .ini). Modified the .ini to use proxy and the right inverter name. 

Now you can run grott!

python3 grott.py -v  (startin foreground) 

You have to reroute the Growatt inverter data to the rpi (see wiki). 

For run grott at a service you have to install en anble the service. This also described in the wiki. 

I think the wiki can be rather useful. I tried to document as much as possible,

Vinisz, 2020-06-08:
Wiki ? I had no idea there is a wiki (did not see it the readme.md) 

Where can I find it ?

Vinisz, 2020-06-08:
Never mind the wiki question… found it already… not so used to github

Vinisz, 2020-06-08:
Took some work but you really helped… it is working !

 

Via mqtt to node-red (I use node-red for all my automation)

Thanks for the help…

ps. I have the “Shinelink” box together with Growatt 3600 TL-XE , so you can add that to your “tested” list.

 

Config should be on Decrypt = True and Compat = False…

 

Thanks again !

closed 2020-06-08 by Vinisz