johanmeijer/grott #15: Offset wrong ?

Vinisz, 2020-06-08:
Hi,

Closed the other thread, as that one was about running the software, which is ok now...
But... I think there is still an issue.. the offset ? does not look quite ok.
The output is like this and as far as I can see, Wattage of PV1 and PV2 are fine but things like enegry today and energy total make no sense at all...  (should be 11.6kwh and 13.8 kwh)
Here is the data ouput:

 -Growatt plain data:
                 00cd726a757825704e41433439343130414644444630413039313132000000000000020bb80c3
                 40001000006fd07ad0003000002c3077a00050000043a00000000000000000000000000000000
                 00000000000000000000068c1385091b00090000069c000000000000000000000000000000000
                 91b000000000000000000000000000000000001fa7b000000740000008a0000008f0000003500
                 00003f00000043000000500000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000fffa00000000000000000000013b013b0000
                 000001910dff00004e20000400000000000000000000000000000000000000000000000000000
                 03c0000000000000000000000000000000000000c350cb1000000000000000000000000000000
                 0000000000000000000000000000000000000000000000000013850912000a000006960000000
                 00000000000000000000000000000069f00284e20000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 0000000052aa47726f25cb
         -Growatt processing values for:  DDF0A09112
                 - pvserial:          DDF0A09112
                 - pvstatus:          1
                 - pvpowerin:         178.9
                 - pvpowerout:        0.0
                 - pvenergytoday:     15276442.5
                 - pvenergytotal:     169.2
                 - pv1watt:           70.7
                 - pv2watt:           108.2
                 - pvfrequentie:      0.0
                 - pvgridvoltage:     0.0
                 - pv1voltage:        196.5
                 - pv1current:        0.3
                 - pv2voltage:        191.4
                 - pv2current:        0.5
                 - pvtemperature:     0.0
                 - pvipmtemperature:  0.0
         - MQTT jsonmsg:
                         {"device": "DDF0A09112", "time": "2020-06-08T18:16:56", "values":
                         {"pvstatus": 1, "pv1watt": 707, "pv2watt": 1082, "pvpowerin": 1789,
                         "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvenergytoday":
                         152764425, "pvenergytotal": 1692, "pv1voltage": 1965, "pv2voltage": 1914,
                         "pv1current": 3, "pv2current": 5, "pvtemperature": 0, "pvipmtemperature":
                         0}}
         - MQTT message message sent

johanmeijer, 2020-06-08:
Do you have the output at startup (where the parameters are displayed)?

Vinisz, 2020-06-08:
Like this ? (btw , I also tried with compat “True” and offset “26” , does not seem to make any difference

 

Grott Command line parameters processed:

        verbose:         True

        config file:     grott.ini

        output file:     <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>

        nomqtt:          False

        trace:           False

 

Grott settings:

 

_Generic:

        version:         2.0.3

        verbose:         True

        trace:           False

        config file:     grott.ini

        minrecl:         100

        decrypt:         True

        compat:          False

        valueoffset:     0

        offset:          6

        inverterid:      DDF0A09112

        mode:            proxy

        grottport        5280

_MQTT:

        nomqtt           False

        mqttip:          192.168.0.3

        mqttport:        1883

        mqtttopic:       energy/growatt

        mqtttauth:       False

        mqttuser:        grott

        mqttpsw:         growatt2020

_Growatt server:

        growattip:       47.91.67.66

        growattport:     5279

johanmeijer, 2020-06-08:
Yes offset is wrong, this is the content of my ini file: 

#Specify grott monitor configuration

[Generic]
# Specify inverter id
inverterid = XXXXXXXXXX
# Specify mode (sniff or proxy)
mode = proxy
# specify port to listen to (only proxy, default 5279)
port = 5279
#Decrypt is False if growatt communication is not encrypted (older inverters)
decrypt = True
#Compat is True and valoffset needs to be set if offset / growatt protocol has been changed. Since 2020 this needs to be specified (True,26)
compat = True
valueoffset = 26

[Growatt]
#ip address and port of Growatt server
ip = 47.91.67.66
#ip=192.168.0.22
port = 5279

[MQTT]
#mqtt parameters definitions
ip = 192.168.0.206
port = 1883
topic= energy/growatt
auth = False
user = grott
password = growatt2020

johanmeijer, 2020-06-08:
To set right processing values:

decrypt =True
compat = True 
valueoffset = 26 

Are the right setting for the current Growatt shine software.

Vinisz, 2020-06-08:
If I do that, it gives me messages like :

 

-Growatt processing values for:  DDF0A09112

         - No valid monitor data, PV status: : 592

 

Are the right settings for the current Growatt shine software.

johanmeijer, 2020-06-08:
And if you use decrypt = False ? 

Do have a shinelan or shin wifi stick?

johanmeijer, 2020-06-08:
Sorry I does not read it right. It is finding your inverter so that's oke.

johanmeijer, 2020-06-08:
Can you please send me agian the output at startup and the plain data?

Vinisz, 2020-06-08:
> Can you please send me again the output at startup and the plain data?

I have this one:

 <https://stralendgroen.nl/product/shine-link-lanbox/> https://stralendgroen.nl/product/shine-link-lanbox/ (the shinelan was not available for long time)

Here is the data:

 

Grott Command line parameters processed:

        verbose:         True

        config file:     grott.ini

        output file:     <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>

        nomqtt:          False

        trace:           False

 

Grott settings:

 

_Generic:

        version:         2.0.3

        verbose:         True

        trace:           False

        config file:     grott.ini

        minrecl:         100

        decrypt:         True

        compat:          False

        valueoffset:     26

        offset:          6

        inverterid:      DDF0A09112

        mode:            proxy

        grottport        5280

_MQTT:

        nomqtt           False

        mqttip:          192.168.0.3

        mqttport:        1883

        mqtttopic:       energy/growatt

        mqtttauth:       False

        mqttuser:        grott

        mqttpsw:         growatt2020

_Growatt server:

[... 322 lines cut ...]
                 - pv1current:        0.0

                 - pv2voltage:        3600.0

                 - pv2current:        100.0

                 - pvtemperature:     0.0

                 - pvipmtemperature:  822.4

         - MQTT jsonmsg:

                         {"device": "DDF0A09112", "time": "2020-06-08T19:01:23", "values":

                         {"pvstatus": 1, "pv1watt": 655360000, "pv2watt": 1095512366, "pvpowerin":

                         14221312, "pvpowerout": 805329473, "pvfrequentie": 16961, "pvgridvoltage":

                         1, "pvenergytoday": 959525170, "pvenergytotal": 0, "pv1voltage": 100,

                         "pv2voltage": 36000, "pv1current": 0, "pv2current": 1000, "pvtemperature":

                         0, "pvipmtemperature": 8224}}

         - MQTT message message sent

johanmeijer, 2020-06-08:
I am trying to analyse it but that will take some time. 

Can you try it with:

decrypt =True
compat = True
valueoffset = 6

Vinisz, 2020-06-08:
It generates output with offset 6, here is the output with that setting…

 

-Growatt processing values for:  DDF0A09112

                 - pvserial:          DDF0A09112

                 - pvstatus:          1

                 - pvpowerin:         43.3

                 - pvpowerout:        0.0

                 - pvenergytoday:     15289549.5

                 - pvenergytotal:     36.0

                 - pv1watt:           16.0

                 - pv2watt:           27.3

                 - pvfrequentie:      0.0

                 - pvgridvoltage:     0.0

                 - pv1voltage:        162.7

                 - pv1current:        0.1

                 - pv2voltage:        167.3

                 - pv2current:        0.1

                 - pvtemperature:     0.0

                 - pvipmtemperature:  0.0

johanmeijer, 2020-06-08:
I have to look deeper in to this. 

I think the values should be: 

decrypt =True
compat = True
valueoffset = 26

There will always be records with no valid data. (Especially at start-up, there is some kind of sync process). But once in the 5 minutes there should be a valid record as well! 

Can you please try this? 

In the meanwhile I will try to decode  your record and find out if your offset is different. 

Just for my information what is your inverter model and which stick do you use (wifi or lan).

johanmeijer, 2020-06-08:
Sorry I see you already posted the link to the lanbox.

Vinisz, 2020-06-08:
Yes, I did… 

I’m no trying with the setting you suggested, tried it already but let me run it a longer time, see what comes out….

johanmeijer, 2020-06-08:
compat needs to be True as well. In your output above is states that compat = False

======================
_Generic:

        version:         2.0.3

        verbose:         True

        trace:           False

        config file:     grott.ini

        minrecl:         100

        decrypt:         True

        compat:          False

        valueoffset:     26

        offset:          6

        inverterid:      DDF0A09112

        mode:            proxy

        grottport        5280

Vinisz, 2020-06-08:
I now, that was on that try…
Now it’s like this:

 

Grott Command line parameters processed:

        verbose:         True

        config file:     grott.ini

        output file:     <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>

        nomqtt:          False

        trace:           False

 

Grott settings:

 

_Generic:

        version:         2.0.3

        verbose:         True

        trace:           False

        config file:     grott.ini

        minrecl:         100

        decrypt:         True

        compat:          True

        valueoffset:     26

        offset:          26

        inverterid:      DDF0A09112

        mode:            proxy

        grottport        5280

_MQTT:

        nomqtt           False

        mqttip:          192.168.0.3

        mqttport:        1883

        mqtttopic:       energy/growatt

        mqtttauth:       False

        mqttuser:        grott

        mqttpsw:         growatt2020

_Growatt server:

        growattip:       47.91.67.66

        growattport:     5279

johanmeijer, 2020-06-08:
Oke we will wait what the output is. It should be sent out every 5 minutest, if it is not dark :)

Vinisz, 2020-06-08:
Yes, I’ll update you tomorrow…

Vinisz, 2020-06-09:
Good morning Johan…

 

That 26 did not work out very well but I now found out that I get best results with 

 

decrypt:         True

        compat:          True

        valueoffset:     6

        offset:          6

 

with that mode, also amps, voltage, temps for both PV1 and PV2 (I have two strings attached to the inverter) start working again in the mobile app..

Only thing not ok (at least, not in the MQTT output) now are the “Energy today” and “Energy Total”

If you have time you may look into it but for me it is not that important… I can see those values in the mobile app and PV output will calculate them for me, so that is also good….

 

Other news, I have grott now running in a docker container on synology NAS (in proxy mode, sniff would be harder to do…)
It is based on Ubuntu container, only needed to add apt, sudo, git and the things you described:

sudo apt install python3-pip

Install paho-mqtt
sudo pip3 install paho-mqtt

After that, all running fine and uses only 12 mb of ram ! (and no cpu)

 

Greetings,

 

Vincent

johanmeijer, 2020-06-09:
Hi vincent,

I am going to try to debug it (might take some days). 

Just be sure can you sent me the latest trace output inluding the mqtt send information?

Nice that you have it running on docker at the nas. I have it running at docker at the raspberry pi but needed a special os build for that (arm 32 bits bases). I have not been able to build and test a generic build based on an other os build (python:3.7-slim). At this moment I do not have a ubuntu docker environment available. 

I made some changes in grott (version 2.0.4) to make parameter processing with docker easier. I will be possible to specify inverterid at container startup without changing the ini file (and rebuilding the container).

johanmeijer, 2020-06-09:
Do you also have an indication of what the energy total and energy today should be?

Vinisz, 2020-06-09:
On the output you had from yesterday total should be around 15 and today about 11.

johanmeijer, 2020-06-12:
@Vinisz it is a bit more complicate than I thought. I had expected it was a simple search for the right offset but the record layout of your inverter seems to be different than the usual one (which is a dump of the inverter registers in the documented oder). 

I am now searching for the right values in the record (it is like a puzzle) but also try to find the big picture behind it to make it more future proof. 

Your right, the initial setting with a offset of 6 (default) is the most appropiate one. But will not have all the right values in it. 

If possible can you send me more data (invluding the unencrypted data) preferably by mail (I think you will now have my email address). 

I don nothink you already specified your inverter model. Maybe that can help with searching for solutions at the internet.

I have not giving it up yet!

Vinisz, 2020-06-20:
Hi Johan,

 

Sorry, was out for a while..

What data would you like to see ?

 

Greetings,

 

VIncent

Vinisz, 2020-07-08:
Johan,

 

Not trying to rush you, but did you get any further on this ?

 

Gr,

Vincent

johanmeijer, 2020-07-08:
Hi Vincent, 

I think I did find the other values but have to verify it with more data. If you can send me some additional records that will help me.

I will sent you a mail.

ghust, 2020-07-24:
Hi guys, I'm experiencing the same issue.

Is there any news regarding this? :)

ghust, 2020-07-24:
Latest example:

{"device":"YUZ0828205","time":"2020-07-24T09:19:20","values":{"pvstatus":0,"pv1watt":65536,"pv2watt":1179648,"pvpowerin":2,"pvpowerout":287574241,"pvfrequentie":19,"pvgridvoltage":0,"pvenergytoday":0,"pvenergytotal":0,"pv1voltage":0,"pv2voltage":8823,"pv1current":44,"pv2current":2320,"pvtemperature":26,"pvipmtemperature":0}}

Value in app says 2.6KwH today.

verbose log:

- Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.11', 33564), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x01\x00\x02\x00\xd9\x01\x04\x59\x55\x5a\x30\x38\x32\x38\x32\x30\x35\x36
                 \x46\x5a\x4a\x39\x32\x38\x30\x32\x34\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00
                 \x2c\x00\x01\x00\x00\x22\x77\x09\x10\x00\x12\x00\x00\x11\x24\x08\xe1\x00\x13
                 \x00\x00\x11\x53\x00\x00\x21\xc8\x13\x8a\x09\x59\x00\x24\x00\x00\x21\xd7\x00
                 \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1a
                 \x00\x00\x01\x8c\x00\x03\x5c\xa6\x01\x71\x00\x00\x00\x00\x00\x00\x00\x00\x00
                 \x00\x00\x00\x00\x00\x00\x00\x01\x8b\x0e\x57\x00\x00\x00\x00\x00\x2d\x00\x59
                 \x4e\x20\x00\x00\x00\x00\x00\x00\x00\x0d\x00\x00\x00\xd0\x00\x00\x00\x0d\x00
                 \x00\x00\xc8\x00\x00\x01\x98\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
                 \x00\x00\x00\x00\x00\x17\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
                 \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
                 \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
         - Growatt unscrambled data processed for:  YUZ0828205
         -Growatt plain data:
                 0001000200d9010459555a3038323832303536465a4a393238303234000000000000020000002
                 c000100002277091000120000112408e1001300001153000021c8138a09590024000021d70000
                 00000000000000000000000000000000001a0000018c00035ca60171000000000000000000000
                 00000000000018b0e5700000000002d00594e20000000000000000d000000d00000000d000000
                 c8000001980000000000000000000000000000000000170000000000000000000000000000000
                 0000000000000000000000000000000000000000000000000000000000000
         -Growatt processing values for:  YUZ0828205
                 - pvserial:          YUZ0828205
                 - pvstatus:          0
                 - pvpowerin:         0.2
                 - pvpowerout:        28757424.1
                 - pvenergytoday:     0.0
                 - pvenergytotal:     0.0
                 - pv1watt:           6553.6
                 - pv2watt:           117964.8
                 - pvfrequentie:      0.19
                 - pvgridvoltage:     0.0
                 - pv1voltage:        0.0
                 - pv1current:        4.4
                 - pv2voltage:        882.3
                 - pv2current:        232.0
                 - pvtemperature:     2.6
                 - pvipmtemperature:  0.0

johanmeijer, 2020-07-24:
@ghust your problem seem to be different. Can you try it with offset = 16?

johanmeijer, 2020-07-24:
@Vinisz did you receive my mail?

johanmeijer, 2020-07-25:
@ghust Sorry I was not sharp yesterday. Your invertid is: 6FZJ928024 (not YUZ0828205, that is the datalogger id). 
If  you use this in the ini file it should work (be aware: valuoffset is not used if compat = True): 

[Generic]

inverterid = 6FZJ928024 

mode = proxy

port = 5279

decrypt = False

compat = True
valueoffset = 26

johanmeijer, 2020-07-30:
@ghust does it works?

johanmeijer, 2020-08-06:
@Vinisz can you try it with branch 2.1.0? 

I added automated protocol conversion so the ini file can be simplefied (even invertid is not necessary anymore). You have to specify compat = false! (at least not specify compat = true, false is default). 

**Please give me feed back. Without feedback or additional data I can not perform futher debuging and improve Grott................**

Example.ini !!!!! 

#Specify grott monitor configuration
#Updated: 2020-08-06

[Generic]
#Specify inverter id (not necessary in version 2.1.0 if compat = false!)
#inverterid = ABC1234567
#Specify mode (sniff or proxy)(> 2.1.0 proxy is default)
mode = proxy
#specify port to listen to (only proxy, default 5279)
port = 5279  
#Compat is True and valoffset needs to be set if offset / growatt protocol has been changed. 
compat = False
#valueoffset = 26
#Decrypt is False if growatt communication is not encrypted (older inverters), (not necessary in versi0n 2.1.0 if compat = false!)
#decrypt = True

[Growatt] 
#ip address and port of Growatt server (specify only if IP has changed)
ip = 47.91.67.66
port = 5279                                                        

[MQTT]
#mqtt parameters definitions
ip = 192.168.0.206
port = 1883
topic= energy/growatt
auth = False
user = grott
password = growatt2020

johanmeijer, 2020-08-17:
Added branche 2.1.1 (delete 2.1.0). This beta does improved (automatic) record layout recognition and processing. 
Fixed errors with 2.1.0. 

Be aware this is a beta . Testing would be appriciated. I can not simulate all situations.

johanmeijer, 2020-08-26:
@ghust succes with the other tooling. 

@Vinisz I am very curious if the values are right for you now.  Be aware you hae to specify compat = False in .ini! 
With the new version modifing this is not a big problem anymore.

johanmeijer, 2020-09-20:
@Vinisz Did you already have time to test it with the 2.1.2 beta version? 

I works with a 3600mtl-xe with shinelink-x but your record layout seems to be a little different so I am very curious.

Vinisz, 2020-09-23:
Ok, just started testing the 2.1.2

 

All looks fine, data in node-red is ok:

 

 

I have these settings in the .ini file :

 

#Specify grott monitor configuration

#defaults are descriped

#remove # and changed default to change configuration

# Updated: 2020-08-24

# Version 2.1.2

 

[Generic]

# Specify inverter id (not necessary in versi0n 2.1.0 if compat = false!)

#inverterid = ABC1234567

# Specify mode (sniff or proxy)(> 2.1.0 proxy is default)

#mode = proxy

# specify port and IP address to listen to (only proxy, default 5279), 0.0.0.0 $

ip = 192.168.0.50

port = 5280

#Compat is True and valoffset needs to be set if offset / growatt protocol has $

compat = False

#valueoffset = 6

#Decrypt is False if growatt communication is not encrypted (older inverters), $

#decrypt = True

 

[Growatt]

#ip address and port of Growatt server (specify only if IP has changed)

 

#ip = 47.91.67.66

#port = 5279

 

[MQTT]

#mqtt parameters definitions

#be aware nomqtt = True means no MQTT processing will be done!!!!!!

 

#nomqtt = False

ip = 192.168.0.3

port = 1883

topic= energy/growatt

#auth = False

#user = grott

#password = growatt2020

 

[PVOutput]

#PVOutput parameters definitions

 

#pvoutput = True

#apikey = yourapikey

#systemid = 12345

 

Just added pvoutput api key and systemID as well. 
That also seems to work perfect (including voltage now) , so of that works ok, I can remove the node-red mqttt to pvoutput config part.

 

So, good job ! I’ll keep this running for a while, let me know if you want to see something else…

johanmeijer, 2020-09-23:
Nice!

Are all values ok now?

Regards,
Johan

Vinisz, 2020-09-23:
Yes, looks great...

johanmeijer, 2020-09-23:
Super!

Vinisz, 2020-09-24:
One request remains…. It would be great if you could offer this in a small docker container……

johanmeijer, 2020-09-24:
Almost ready......

I am testing a container version at this moment. Ran 1 day without issues I
will publish it soon.

Vinisz, 2020-09-24:
great, let me know if i need to test as well.. (running docker on synology nas)

johanmeijer, 2020-09-25:
Hi @Vinisz, 

I published a docker container on docker hub under: ledidobe/grott:2.2.0. 

The minimal command to run it is: 
docker run -d --network=host --restart on-failure -e gmqttip="yourip" ledidobe/grott:2.2.0

So only IP of your mqtt server needs to be specified. 

It build with grott version 2.2.0 this version has a lot of nw functionality (like docker support, direct output to PVOUTPUT.org  and command blocking/filtering so inverter can not be hacked) 

Most of the .ini settings can be used with the docker to. For a ini setting there is a envrionmental defined that can be set with -e at startup. e.g. if you want to use mqtt security specify also: 
-e gmqttauth = "True" -e gmqttuser = "grott" -e gmqttpassword = "growatt2020" 

I will publish version 2.2.0 and update the documentation in the comming days. 

I am very curious if the this container runs on a synology nas. I tested it on a Ubuntu Linux on x86 machine. I also published an ARMv32 version for the older Raspberry PI's (ledidobe/grottrpi:2.2.0).

Vinisz, 2020-09-26:
Just tested it on synolgy NAS and does run fine, however….

I want to make a persistent volume mapping from grott.ini on the nas to /app/grott.ini in the container (so it using persistent grott.ini)

When I try do that, the container crashes… not quite sure why, I have mappings like this on lots of other containers.

johanmeijer, 2020-09-26:
After  long thinking (and discussing it with more people whith more Docker knowledge)  I decided not go for the volume mapping for .ini. 

I try to eliminate the need for the settings in .ini as less as possible (automate a lot of things) and create for every setting an environmental so it would be possible to chnge settings at startup if needed. 

I have not tested volume mapping but will try it.   I do not think I will use or need it (because of the changes I made to the program).

Vinisz, 2020-09-27:
Ok, so for every ini setting there is an environment setting possible ?

johanmeijer, 2020-09-27:
Yes! I will document it but did not had the time for it.

But..... 

As long compat = False (environmental gcompat), which is the default, most of the settings are not needed anymore. Grott will detect automaticaly inverterid, encryption, offset etc.

The only things in normal situation you have to specify are the mqtt settings (as described) or pvoutput.org settings (new functionality to write directly to pvoutput).

If you not using mqtt authorisation specifing the mqtt ip address (enironmental gmqttip) should be enough.

Vinisz, 2020-09-27:
Well... I like to specify pvoutput parameters, is that also possible ?

johanmeijer, 2020-09-27:
Sure.  At this moment I run with: 

docker run -d --network=host --restart on-failure -e gmqttip="192.168.0.206" -e gblockcmd="True" -e gnoipf="True" -e pvoutput="True" -e gpvapikey="apikey" -e gpvsystemid="systemid" ledidobe/grott:2.2.0

So with the new pvoutput parms: 
pvoutput="True" -e gpvapikey="apikey" -e gpvsystemid="systemid"  

-e gblockcmd = "True"
This blocks all commands comming from outside to prevent hacking of the inverter 

-e gnoipf="True"
This is in addition on blockcmd and allows the change growatt server IP address from the growatt website dialogue. I use this to switch between the servers at home with different Grott implementations without restarting/reconfiguring the ShineWifi each time. 
Be aware this create a hole in the defence so don't use this all te time.

Vinisz, 2020-09-28:
Well Johan, I have to say I’m impressed…. Very good Job !

I have the docker container running on Synology NAS DS415+ (Intel Athom CPU)

It runs perfect, and only uses 16Mb o memory so you did a get job with keeping the memory footprint down.

I just added the variables in the “Environment” setting of the docker containter (that’s the way you need to do that on synology), only needed to adjust “pvoutput” to “gpvoutput” and, one important thing for people to remember, you need to add the “TX” variable as well for the correct timezone (otherwise PV output offset will be wrong) , so I added that one as well with “Europe/Amsterdam” in there…

All working just fine, I’ll leave it running like this for a couple of days.. see if it’s stable…
Screenshot here to show synology docker environment settings (done worry about the api key, it’s truncated):

johanmeijer, 2020-09-28:
I am glad it works well! It was my first docker expierence (I learned a lot from it). 

I indeed deleted the g before gpvoutput, sorry for that. Where do you have to add the TX (do you mean TZ?) parameter? 

I do not have any problems with time zone's but have defined the time zone at PVOUTPUT.org. 

I normally use the time as definied in the data record by the inverter (Shine device) (this is normally set by growatt). Your record does not contain a time set bij the inverter I believe (your record layout is a bit special). Grott then  use the server/container time. In that case you indeed have to do something with time zone's I think. 

I will do some testing with volume mapping in the comming days to see if I can solve that issue as well.

Next steps are:
Create a generic Grott container (without a tagg) so will always use the latest version. (I will leave the tagged version as bckup). 
Document the 2.2.0 version including command blocking, docker and pvoutput support.

Vinisz, 2020-09-28:
Well, I would not put any energy in doing the volume mapping thing of you can all arrange it with parameters like this.
This works very well, so why change that ?

The timezone (TXZ) is stated as a variable in the “environment” setting of the docker container on synology. You can use that or, indeed, use PV output to make it work. 

Normally, the docker container runs in UTC on synology NAS, so that is why I need to change it.

johanmeijer, 2020-09-28:
I think your right. Volume mapping will be put on the nice to have list. I will put my energy in publishing and document the 2.2.0. release.

Thank you for testing!

johanmeijer, 2020-09-28:
I created a new version of docker container based on version 2.2.1. (Some cosmetic changes during startup). 

It has the tag 2.2.1 but also latest. 

You should be able now to run/download  it without using tags: so ledidobe/grott should be enough. This will run the latest version.

Vinisz, 2020-09-29:
Celebrated a little bit too early I think.

Container crashed yesterday, (I think it has to do with inactivity) , here’s the latest log lines…

 

        - Growatt packet received:

stdout

20:03:02

                 <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.3', 39082), raddr=('47.91.67.66', 5279)>

stdout

				

 

20:03:02

        - Data less then minimum record length, data not processed

stdout

20:04:03

	stdout

 

20:04:03

        - Growatt packet received:

stdout

20:04:03

                 <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.3', 39082), raddr=('47.91.67.66', 5279)>

stdout

 

20:04:03

        - Data less then minimum record length, data not processed

stdout

20:05:03

	stdout

 

20:05:03

        - Growatt packet received:

stdout

20:05:03

                 <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.3', 39082), raddr=('47.91.67.66', 5279)>

stdout

 

20:05:03

        - Data less then minimum record length, data not processed

stdout

20:06:03

	stdout

 

20:06:03

        - Growatt packet received:

stdout

20:06:03

                 <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.3', 39082), raddr=('47.91.67.66', 5279)>

stdout

 

20:06:03

        - Data less then minimum record length, data not processed

stdout

20:07:04

	stdout

 

20:07:04

        - Growatt packet received:

stdout

20:07:04

                 <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.3', 39082), raddr=('47.91.67.66', 5279)>

stdout

 

20:07:04

        - Data less then minimum record length, data not processed

stdout

20:08:04

	stdout

 

20:08:04

        - Growatt packet received:

stdout

20:08:04

                 <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.3', 39082), raddr=('47.91.67.66', 5279)>

stdout

 

20:08:04

        - Data less then minimum record length, data not processed

stdout

20:09:04

	stdout

 

20:09:04

        - Growatt packet received:

stdout

20:09:04

                 <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.3', 39082), raddr=('47.91.67.66', 5279)>

stdout

 

20:09:04

        - Data less then minimum record length, data not processed

stdout

20:10:06

	stdout

 

20:10:06

        - Growatt packet received:

stdout

20:10:06

                 <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.3', 39082), raddr=('47.91.67.66', 5279)>

stdout

 

20:10:06

        - Data less then minimum record length, data not processed

stdout

20:11:07

	stdout

 

20:11:07

        - Growatt packet received:

stdout

20:11:07

                 <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.3', 39082), raddr=('47.91.67.66', 5279)>

stdout

 

20:11:07

        - Data less then minimum record length, data not processed

stdout

20:12:07

	stdout

 

20:12:07

        - Growatt packet received:

stdout

20:12:07

                 <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.3', 39082), raddr=('47.91.67.66', 5279)>

stdout

 

20:12:07

        - Data less then minimum record length, data not processed

stdout

20:13:04

        - Grott connection error

stdout

 

20:13:04

	stdout

20:13:04

        - Growatt packet received:

stdout

 

20:13:04

                 <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.3', 5279), raddr=('192.168.0.187', 1029)>

stdout

20:13:04

        - Data less then minimum record length, data not processed

stdout

 

20:13:04

Traceback (most recent call last):

stdout

20:13:04

  File "grott.py", line 35, in <module>

stdout

 

20:13:04

    proxy.main(conf)

stdout

20:13:04

  File "/app/grottproxy.py", line 67, in main

stdout

 

20:13:04

    self.on_close(conf)

stdout

20:13:04

  File "/app/grottproxy.py", line 88, in on_close

stdout

 

20:13:04

    if conf.verbose: print("\t -", self.s.getpeername(), "has disconnected")

johanmeijer, 2020-09-29:
Did it really crashed  and did it restarted? It seems that connection is closed (inverter stops). 

If it didnot restarted can you try to restart it with ---restart allways. 

I saw it before (also when grott run as a service) that when the inverter stops, it ends the network connection unfriendly . 

I will look into my own logs as well.

Vinisz, 2020-09-29:
Yes, it did crash… and it did not restart (but that was because had not turn on “auto restart”) , now it will auto-restart.
I think your assumption is correct, looks like when the inverter stops, it does not end very well…. Which causes a crash.

johanmeijer, 2020-09-29:
I will see if I can make grott more resilient. It should not crash when
this happens.

Challange will be how to test if it occurs occassionally.

johanmeijer, 2020-09-30:
Ok the errno 107 occurs also here regulary. Especially early in the morning (dark weather) and evening when the inverter loses power. 

It does not hurts if grott restarts automatically but it is not very elegant. I made some changes to overcome this (and some potentential other errors). I will test it the comming days and if it works build a new docker image this weekend.

[5 more replies]

closed 2020-10-19 by johanmeijer