johanmeijer/grott #16: step by step manual

sanderlv, 2020-06-15:
Hi, is there a locarion for a step by step manual? I have an XE converter with the ethernet port dongle (wireless to growatt though) and I would lik home assistant to see data.

I can run a vm (ubuntu) but don't know where to start...

johanmeijer, 2020-06-16:
Hi sander,

Did you saw the wiki(https://github.com/johanmeijer/grott/wiki) a lot is documented here.

What you have to do in steps:
- prepare base environment (linux) 
- install mqtt (i use mosquitto)
- install python3 /pip3 
- install paho-mqtt with pip3 (see wiki)
- install grott (see wiki)
- optional enable grott as a service (see wiki)
- reroute the growatt communication via the grott server (see wiki)

Hopefully this helps.

johanmeijer, 2020-06-25:
Hi sander,

The last status seems to be ok!

What do you see when you look at the system log: 

Command:
journalctl -u grott -e 

This show all log entries for grott and jumps to the last page. 

It should show some information about the communcation between the inveter and the growatt server.

sanderlv, 2020-06-25:
Last login: Thu Jun 25 09:55:39 2020 from 192.168.1.159
ubuntu@grott:~$ journalctl -u grott -e
Jun 25 09:14:37 grott grott[2396]:         verbose:              True
Jun 25 09:14:37 grott grott[2396]:         trace:                False
Jun 25 09:14:37 grott grott[2396]:         config file:          grott.ini
Jun 25 09:14:37 grott grott[2396]:         minrecl:              100
Jun 25 09:14:37 grott grott[2396]:         decrypt:              True
Jun 25 09:14:37 grott grott[2396]:         compat:               True
Jun 25 09:14:37 grott grott[2396]:         valueoffset:          26
Jun 25 09:14:37 grott grott[2396]:         offset:               26
Jun 25 09:14:37 grott grott[2396]:         inverterid:           NAC5A1650C
Jun 25 09:14:37 grott grott[2396]:         mode:                 proxy
Jun 25 09:14:37 grott grott[2396]:         grottport             5279
Jun 25 09:14:37 grott grott[2396]: _MQTT:
Jun 25 09:14:37 grott grott[2396]:         nomqtt                False
Jun 25 09:14:37 grott grott[2396]:         mqttip:               192.168.1.142
Jun 25 09:14:37 grott grott[2396]:         mqttport:             1883
Jun 25 09:14:37 grott grott[2396]:         mqtttopic:            energy/growatt
Jun 25 09:14:37 grott grott[2396]:         mqtttauth:            True
Jun 25 09:14:37 grott grott[2396]:         mqttuser:             mqtt
Jun 25 09:14:37 grott grott[2396]:         mqttpsw:              mqtt
Jun 25 09:14:37 grott grott[2396]: _Growatt server:
Jun 25 09:14:37 grott grott[2396]:         growattip:            47.91.67.66
Jun 25 09:14:37 grott grott[2396]:         growattport:          5279
Jun 25 09:14:37 grott grott[2396]: Grott proxy mode started
lines 808-830/830 (END)
Jun 25 09:14:37 grott grott[2396]:         verbose:              True
Jun 25 09:14:37 grott grott[2396]:         trace:                False
Jun 25 09:14:37 grott grott[2396]:         config file:          grott.ini
Jun 25 09:14:37 grott grott[2396]:         minrecl:              100
Jun 25 09:14:37 grott grott[2396]:         decrypt:              True
Jun 25 09:14:37 grott grott[2396]:         compat:               True
Jun 25 09:14:37 grott grott[2396]:         valueoffset:          26
Jun 25 09:14:37 grott grott[2396]:         offset:               26
Jun 25 09:14:37 grott grott[2396]:         inverterid:           NAC5A1650C
Jun 25 09:14:37 grott grott[2396]:         mode:                 proxy
Jun 25 09:14:37 grott grott[2396]:         grottport             5279
Jun 25 09:14:37 grott grott[2396]: _MQTT:
Jun 25 09:14:37 grott grott[2396]:         nomqtt                False
Jun 25 09:14:37 grott grott[2396]:         mqttip:               192.168.1.142
Jun 25 09:14:37 grott grott[2396]:         mqttport:             1883
Jun 25 09:14:37 grott grott[2396]:         mqtttopic:            energy/growatt
Jun 25 09:14:37 grott grott[2396]:         mqtttauth:            True
Jun 25 09:14:37 grott grott[2396]:         mqttuser:             mqtt
Jun 25 09:14:37 grott grott[2396]:         mqttpsw:              mqtt
Jun 25 09:14:37 grott grott[2396]: _Growatt server:
Jun 25 09:14:37 grott grott[2396]:         growattip:            47.91.67.66
Jun 25 09:14:37 grott grott[2396]:         growattport:          5279
Jun 25 09:14:37 grott grott[2396]: Grott proxy mode started

sanderlv, 2020-06-25:
I "keep seeing" updates in the server.growatt.com portal

johanmeijer, 2020-06-25:
Hi Sander,

I supose 192.168.1.140 is the IP of you VM with Grott running?

Have you configured the IP-Forwarding part as described in the Wiki? That will not be working for the proxy modus. 
If you have configured IP-forwarding please specify mode = sniff in the ini file so we can see if data is sent via your VM. 

It is normal that the growatt server info will be updated. The only thing Grott is doing is "listening" to the IP traffic (Sniff mode) or intercept the IP traffic and send it to the Growatt server (proxy mode). But we need the communication between inverter en growatt server to be alive to "see" things. 

I am not sure you can change the intervall. 5 minutes is normally what the inverter use to send to the growatt server.

sanderlv, 2020-06-25:
> I supose 192.168.1.140 is the IP of you VM with Grott running?

YES

> Have you configured the IP-Forwarding part as described in the Wiki? That will not be working for the proxy modus.

NO I have not, I want proxy.

> If you have configured IP-forwarding please specify mode = sniff in the ini file so we can see if data is sent via your VM.

See above.

> It is normal that the growatt server info will be updated. The only thing Grott is doing is "listening" to the IP traffic (Sniff mode) or intercept the IP traffic and send it to the Growatt server [...]

Ok, clear --> thus this means my VM is proxying the traffic correctly?

> I am not sure you can change the intervall. 5 minutes is normally what the inverter use to send to the growatt server.

Ok.

But... I see no MQTT messages coming in on my mqtt mosquitto server (192.168.1.142 - which works very well for all other mqtt stuff. ... where to start troubleshooting?

johanmeijer, 2020-06-25:
If you do no get more info when you display the log with: journalctl -u grott -e there is no IP trafic running via your VM. 

I expect something like: 
[image]

If you see nothing els then the start-up message then there is no data beeing sent via your VM. 

You can test if grott is enabled by typing: 192.168.1.140:5279 in your browser. You will see some IP traffic (from the browser) in your log.

johanmeijer, 2020-06-27:
Sander do you agree that this issue can be closed? I think I is working now.

sanderlv, 2020-06-27:
Yes, agreed. Hope the "simulation" mode is there soon ;-)

closed 2020-06-27 by sanderlv