johanmeijer/grott #18: Im a total noob and need help

xDAVIDx93, 2020-08-20:
Hi just signed up to github downloaded the grott-2.1.1-Beta- and extracted it i have a shinwifi-x and im just wondering if someone can help run me through setting this up end goal is to have pvoutput website running with automatic data update im using windows 10 for now..

johanmeijer, 2020-08-20:
Hi Grott is designed to use mqtt and to distribute it from there to the home automation tooling (eg domoticz, openhab and home assistent). With node red (javascript visualisation) it is posible to retrieve it from mqtt and send it to pvoutput.

I do understand this sound very complex for somebody who is not familiar with this tooling.

I can help you with setting up this complex enironment and help you with mqtt and nodered.

What we also can try to do is enhance grott so it will sent the data directly to pvoutput (next or instead of mqtt).  This might take some weeks but I can try to program it.

Please let me know your preference. If there is somebody else with ideas I like to hear it.

xDAVIDx93, 2020-08-20:
I use Google home in my house but only for a bunch of smart speakers that's about it oh and a xiaomi robot vacuum so I dont have open hub or anything like that just google home?

I was hoping to redirect the ip and info from the shinewifi-x somewhere else to like my win 10 pc then have it send that data to pvoutput and then if possible also to https://server.growatt.com/ but if that's not possible I think pvoutput looks better anyway. So if someone could help me do that please let me know :)

johanmeijer, 2020-08-26:
Oke lets try it: I published Version 2.1.2 (Beta) this will be the base for direct  PVOoutput.org and running as a service under windows. 

Most of the setup details are described in the wiki (https://github.com/johanmeijer/grott/wiki). But here in short what is needed: 

1. Download grott from this github and place in the directory wanted
2. Download Python3: https://www.python.org/downloads/windows/ (I use 3.7, 2.x will not work)
3. I assume PIP3 (Python installation manager is installed with python)
4. Install necessary python packages  
At command line: 
- pip3 install paho-mqtt
- pip3 install requests
5. Modify grott.ini:
(uncomment this statements and modify using your PVOutput.org credentials)   
- nomqtt = True  
- pvoutput = True
- #apikey = yourapikey 
- #systemid = yoursystemid
 
6.. Now you can test grott for the first time:  
- at the command line in the grott directory 
- python3 grott.py -v 
7. Grott will start and listen in proxy modes
8. Stop grott (ctrl pause or ctrl break) 
9. Reroute the growatt logging to user windows system (your windows ip needs to be known)  
- See wiki: https://github.com/johanmeijer/grott/wiki/Rerouting-Growatt-Wifi-TCPIP-data-via-your-Grott-Server  
10. Test Grott again (see above) 
- the Growatt data will processed by Grott and sent to PVOutput.org!
11. Stop grott again (see above)
12. Enable grott as a windows service (so it will run at the background and survive system reboots)  
- See wiki: https://github.com/johanmeijer/grott/wiki/Grott-as-a-service-(Windows) 

Hopefully this helps you with setting up grott. Please let me know your results and expierences.

xDAVIDx93, 2020-08-27:
Download grott from this github and place in the directory wanted
Download Python3: DONE
I assume PIP3 YEP
Install necessary python packages
At command line:
pip3 install paho-mqtt DONE
pip3 install requests DONE

Modify grott.ini:
nomqtt = True
pvoutput = True
#apikey = yourapikey
#systemid = yoursystemid
DONE i think i did this correctly alls i did was remove the # from infront of
nomqtt = True
pvoutput = True 
and put True then added my api and system id next to but left the #
#apikey = yourapikey
#systemid = yoursystemid

6.. Now you can test grott for the first time:

at the command line in the grott directory 
this part i dont understand in cmd what do i type and how do i know where my grott directory is?
python3 grott.py -v
if you could explain this part more that would be fantastic cause thats where im stuck.

xDAVIDx93, 2020-08-27:
[image]

assuming this is the directory? what exactly do i type into cmd? directory and command so i can copy paste and hit enter

johanmeijer, 2020-08-27:
What I ment with the command line is a windows terminal  screen with a command line. 

Go to start en enter CMD and start this applicactions. 

[image]

You have to switch to the directory with: 
cd \users\david\downloads\grott-2.1.2-beta-\grott-2.1.2-beta-\grott-2.1.2-beta-

If it works I think I would change the directory to (move the files to) something more simple like:
c:\users\david\grott or even c:\grott

xDAVIDx93, 2020-08-27:
[image]

sorry i have no idea what im doing i moved the folder location but when i run the command still nothing

xDAVIDx93, 2020-08-27:
do you have teamviewer?

xDAVIDx93, 2020-08-27:
[image]

i think i did it is that correct??

xDAVIDx93, 2020-08-27:
[image]
[image]

hope this is correct

xDAVIDx93, 2020-08-27:
so i put in my desktops ip address into the growatt server set ip address field and used growatt20200828 password and it says saved with a tick 
[image]

i hope thats correct

then i installed grott using nssm in cmd as admin

[image]

but when i start it it fails and pauses

[image]

so i opened services as admin and hit resume on Grott and it says that error??

xDAVIDx93, 2020-08-27:
> [image]
> [image]
> 
> hope this is correct

Forgot to mention I checked pvoutput and it wasn't receiving any output data I assume its because cmd isn't constantly running grott script and I need the nssm service for it to constantly run is that correct?

xDAVIDx93, 2020-08-27:
[image]

removed nssm grott and reinstalled with no augments and it started without pausing this time

xDAVIDx93, 2020-08-27:
still no data to pvoutput though all still goes to server.growatt.com

johanmeijer, 2020-08-28:
It does look oke! Your are almost there I think. 

But no data is rerouted via your windows machine. That seem to be strange. 

You can either grott run viaa the command line or as a service. With both active only one will handle the data (the one that started first). 

I see I made a mistake in the NSSM grott Service. The -c should be removed. That is the reason why the service does not start!. It should be:  **grott.py -v -o grottlog.txt**  
Sorry!!

[image]

Please do not start (stop) or remove the service. Run grott in the windows command screen. After a while you should see that some data is beeing process by grott. If the data is really rerouted via your desktop PC,  the Growatt server should not receive any data anymore if grott is not running!   

You will not lose any data if grott is stopped. The data is bufferd in the shine-lan stick and will be sent as soon there is a connection again. 

Can you change the IP address via the shine-lan dialogue? (I have the shine wifi stick and this only possible when I reset the stick)? Maybe it is not changed via the growatt website (altough it was succesfull). 

I assume you have changed the PVOutput APIKEY and systemid in the .ini file. 

I do not have teamviewer yet but can install it if necessary.

xDAVIDx93, 2020-08-29:
Okay ip address is working now and server.growatt only works if cmd grott.py is running but ive hit this error now..

Grott send data to pvoutput response bad request 400 energy value to high?

[image]

johanmeijer, 2020-08-29:
The good news is, Grott is working.

But it is not processing the data right.

Can you sent me the part with the growatt data? Hopefully you can capture this.

In the meanwhile you can try it with.  

decrypt = yes
compat = true
valueoffset = 26

xDAVIDx93, 2020-08-29:
Ill try in 2 or so hours when I get home from work problem is my inverter goes into stand by when the sun goes down but ill give it a go and see what happens

johanmeijer, 2020-08-29:
For me the day is just started (the Netherlands) ;)

xDAVIDx93, 2020-08-29:
Ok shes come througj with the good hope this captures all the info you need also my shine wifi thing is model x 

ShineWifi-x

[image][image][image][image][image][image][image][image][image][image][image]

johanmeijer, 2020-08-29:
I have some toubles with retrieving the right information from your screen dumps. I think I miss some records. 

Can you start grott with:   

python grott.py -v -o textlog.txt   

You will get only the start mesagge. The rest is logged into the file.  

Let it run for a while (10 minutes or so) and then stop it and sent the textlog.txt? 

Hopefu;lly the Growatt is still working.

johanmeijer, 2020-08-29:
I need the data just before the last screen. I do not think this is in the screen dumps.

xDAVIDx93, 2020-08-29:
[image]

Not that?

xDAVIDx93, 2020-08-29:
Nope growatt is in stand by this os all I got, ill try again tomorrow.

Grott Command line parameters processed:
	verbose:     	 True
	config file: 	 grott.ini
	output file: 	 <_io.TextIOWrapper name='textlog.txt' encoding='cp1252'>
	nomqtt:      	 False
	inverterid:  	 ABC1234567
	pvoutput:    	 False

Grott Proces configuration file

Grott Proces environmental variables

Grott override settings if set in commandline

Grott Proces json layout files:

Grott layout records loaded:
T020104  : 
{'decrypt': False, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T050104  : 
{'decrypt': True, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T060104  : 
{'decrypt': True, 'pvserial': 76, 'date': 116, 'pvstatus': 158, 'pvpowerin': 162, 'pv1voltage': 170, 'pv1current': 174, 'pv1watt': 178, 'pv2voltage': 186, 'pv2current': 190, 'pv2watt': 194, 'pvpowerout': 202, 'pvfrequentie': 210, 'pvgridvoltage': 214, 'pvenergytoday': 262, 'pvenergytotal': 270, 'pvtemperature': 286, 'pvipmtemperature': 322}
T055104  : 
{'decrypt': True, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 170, 'pvfrequentie': 178, 'pvgridvoltage': 182, 'pvenergytoday': 274, 'pvenergytotal': 282, 'pvtemperature': 450, 'pvipmtemperature': 454}

Grott settings:

_Generic:
	version:     	 2.1.2
	verbose:     	 True
	trace:       	 False
	config file: 	 grott.ini
	minrecl:     	 100
	decrypt:     	 True
	compat:      	 False
	valueoffset: 	 6
	offset:      	 6
	inverterid:  	 ABC1234567
	mode:        	 proxy
	grottip      	 0.0.0.0
	grottport    	 5279
_MQTT:
	nomqtt       	 True
	mqttip:      	 localhost
	mqttport:    	 1883
	mqtttopic:   	 energy/growatt
	mqtttauth:   	 False
	mqttuser:    	 grott
	mqttpsw:     	 growatt2020
_Growatt server:
	growattip:   	 47.91.67.66
	growattport: 	 5279
_PVOutput:
	pvoutput:    	 True
	pvurl:       	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:    	 9cac8d83546faa4cb1c82ffb88c14fb586e88aff
	pvsystemid:  	 77962

Grott proxy mode started

johanmeijer, 2020-08-29:
Okay thank you. 

I think the automatic protocol recognition is not totally working fine yet. I based it on to little test records I think. 
The more information I get the better it will be. 

Sorry for that.

xDAVIDx93, 2020-08-29:
There you go thats about 10mins from when I started it this morning.

Grott Command line parameters processed:
	verbose:     	 True
	config file: 	 grott.ini
	output file: 	 <_io.TextIOWrapper name='textlog.txt' encoding='cp1252'>
	nomqtt:      	 False
	inverterid:  	 ABC1234567
	pvoutput:    	 False

Grott Proces configuration file

Grott Proces environmental variables

Grott override settings if set in commandline

Grott Proces json layout files:

Grott layout records loaded:
T020104  : 
{'decrypt': False, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T050104  : 
{'decrypt': True, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T060104  : 
{'decrypt': True, 'pvserial': 76, 'date': 116, 'pvstatus': 158, 'pvpowerin': 162, 'pv1voltage': 170, 'pv1current': 174, 'pv1watt': 178, 'pv2voltage': 186, 'pv2current': 190, 'pv2watt': 194, 'pvpowerout': 202, 'pvfrequentie': 210, 'pvgridvoltage': 214, 'pvenergytoday': 262, 'pvenergytotal': 270, 'pvtemperature': 286, 'pvipmtemperature': 322}
T055104  : 
{'decrypt': True, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 170, 'pvfrequentie': 178, 'pvgridvoltage': 182, 'pvenergytoday': 274, 'pvenergytotal': 282, 'pvtemperature': 450, 'pvipmtemperature': 454}

Grott settings:

_Generic:
	version:     	 2.1.2
	verbose:     	 True
	trace:       	 False
	config file: 	 grott.ini
	minrecl:     	 100
	decrypt:     	 True
	compat:      	 False
	valueoffset: 	 6
	offset:      	 6
	inverterid:  	 ABC1234567
	mode:        	 proxy
	grottip      	 0.0.0.0
	grottport    	 5279
_MQTT:
	nomqtt       	 True
	mqttip:      	 localhost
	mqttport:    	 1883
	mqtttopic:   	 energy/growatt
	mqtttauth:   	 False
	mqttuser:    	 grott
	mqttpsw:     	 growatt2020
_Growatt server:
	growattip:   	 47.91.67.66
	growattport: 	 5279
_PVOutput:
	pvoutput:    	 True
	pvurl:       	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:    	 9cac8d83546faa4cb1c82ffb88c14fb586e88aff
	pvsystemid:  	 77962

Grott proxy mode started

	 - ('192.168.1.5', 36534) has connected

	 - Growatt packet received:
		  <socket.socket fd=532, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 50007), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
[... 1696 lines cut ...]
	 - Grott automatic protocol detection
	 - layout   :  T060150
	 - Growatt data decrypted
	 - Growatt plain data:
		 00160006024101505847443241313036424600000000000000000000000000000000000000005
		 8544436413132304635000000000000000000000000000000000000000014081d10093a020bb8
		 0c340001000024d90cb4000c000012c00cd4000b0000121900000000000000000000000000000
		 00000000000000000000000241f1380098200260000242b000000000000000000000000000000
		 0009820000000000000000000000000000241f001ed6940000016a000019a600001a11000000b
		 700000c67000000b900000daa0000000000000000000000000000000000000000000000000000
		 000000000000000000000000000000000000000000001093009600000000147a0000015c00000
		 000000001a40e8c00004e20001200000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000c350cb1000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000123a
	 - Grott data ack record or data record not defined no processing done

	 - Growatt packet received:
		  <socket.socket fd=500, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 5279), raddr=('192.168.1.5', 36534)>
	 - Data less then minimum record length, data not processed

xDAVIDx93, 2020-08-29:
Changed grott.ini
decrypt = yes
compat = true
valueoffset = 26

Decrypt = True already..?

Other then that it now seems to not know the SN? And it seemed to be closer to working without the changes.

Below grott.ini is 10mins of textlog with changes applied

For now ill leave the grott.ini file until you give me something else to try

#Specify grott monitor configuration
#defaults are descriped
#remove # and changed default to change configuration
# Updated: 2020-08-24
# Version 2.1.2

[Generic]
# Specify inverter id (not necessary in versi0n 2.1.0 if compat = false!)
#inverterid = ABC1234567
# Specify mode (sniff or proxy)(> 2.1.0 proxy is default)
#mode = proxy
# specify port and IP address to listen to (only proxy, default 5279), 0.0.0.0 ==> own ip address
#ip = 0.0.0.0
#port = 5279  
#Compat is True and valoffset needs to be set if offset / growatt protocol has been changed. 
compat = True
valueoffset = 26
#Decrypt is False if growatt communication is not encrypted (older inverters), (not necessary in versi0n 2.1.0 if compat = false!)
decrypt = True

[Growatt] 
#ip address and port of Growatt server (specify only if IP has changed)

#ip = 47.91.67.66
#port = 5279                                                        

[MQTT]
#mqtt parameters definitions
#be aware nomqtt = True means no MQTT processing will be done!!!!!!

nomqtt = True
#ip = local
#port = 1883
#topic= energy/growatt
#auth = False
#user = grott
#password = growatt2020

[PVOutput]
#PVOutput parameters definitions

pvoutput = True
apikey = 9cac8d83546faa4cb1c82ffb88c14fb586e88aff
systemid = 77962

😭😭😭😭😭

Grott Command line parameters processed:
	verbose:     	 True
	config file: 	 grott.ini
	output file: 	 <_io.TextIOWrapper name='textlog.txt' encoding='cp1252'>
	nomqtt:      	 False
	inverterid:  	 ABC1234567
	pvoutput:    	 False

Grott Proces configuration file

[... 535 lines cut ...]
		 00099f000000000000000000000000000058c9001f2bd60000000c000019b600001a210000000
		 600000c6f0000000600000db20000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000102f0104000000001489000000bc00000
		 000000000e40eec00004e20002d00000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000c350cb1000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000084af
	 - No Growatt data processed or SN not found:

	 - Growatt packet received:
		  <socket.socket fd=520, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 5279), raddr=('192.168.1.5', 1239)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=512, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 50058), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=520, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 5279), raddr=('192.168.1.5', 1239)>
	 - Data less then minimum record length, data not processed

johanmeijer, 2020-08-29:
Oh yes sorry. When compat = true is used you also has to specify inverid. 

If valueoffset = 26 doesnot work, can you try it with with valueoffset = 6. 

I will look at the data tomorrow ( it is now 23.46 here).

xDAVIDx93, 2020-08-29:
All good ill give a few things a try where do I find inverterid though?

xDAVIDx93, 2020-08-29:
Link to my data logger so you can check its similar to the S model and also if and when we get it working you can add it to your supported list :)

https://www.ginverter.com/Monitoring/10-630.html

johanmeijer, 2020-08-29:
Your inverterid is XTD6A120F5.

xDAVIDx93, 2020-08-29:
Changed the inverid in the .ini and it shows that in the log file but not right up the top for some reason.. also I tried valueoffset 26 and 6 and neither worked it still has a problem with SN

Grott Command line parameters processed:
	verbose:     	 True
	config file: 	 grott.ini
	output file: 	 <_io.TextIOWrapper name='textlog.txt' encoding='cp1252'>
	nomqtt:      	 False
	inverterid:  	 ABC1234567
	pvoutput:    	 False

Grott Proces configuration file

Grott Proces environmental variables

Grott override settings if set in commandline

Grott Proces json layout files:

Grott layout records loaded:
T020104  : 
{'decrypt': False, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T050104  : 
{'decrypt': True, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T060104  : 
{'decrypt': True, 'pvserial': 76, 'date': 116, 'pvstatus': 158, 'pvpowerin': 162, 'pv1voltage': 170, 'pv1current': 174, 'pv1watt': 178, 'pv2voltage': 186, 'pv2current': 190, 'pv2watt': 194, 'pvpowerout': 202, 'pvfrequentie': 210, 'pvgridvoltage': 214, 'pvenergytoday': 262, 'pvenergytotal': 270, 'pvtemperature': 286, 'pvipmtemperature': 322}
T055104  : 
{'decrypt': True, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 170, 'pvfrequentie': 178, 'pvgridvoltage': 182, 'pvenergytoday': 274, 'pvenergytotal': 282, 'pvtemperature': 450, 'pvipmtemperature': 454}

Grott settings:

_Generic:
	version:     	 2.1.2
	verbose:     	 True
	trace:       	 False
	config file: 	 grott.ini
	minrecl:     	 100
	decrypt:     	 True
	compat:      	 True
	valueoffset: 	 6
	offset:      	 6
	inverterid:  	 XDT6A120F5
	mode:        	 proxy
	grottip      	 0.0.0.0
	grottport    	 5279
_MQTT:
	nomqtt       	 True
	mqttip:      	 localhost
	mqttport:    	 1883
	mqtttopic:   	 energy/growatt
	mqtttauth:   	 False
	mqttuser:    	 grott
	mqttpsw:     	 growatt2020
_Growatt server:
	growattip:   	 47.91.67.66
	growattport: 	 5279
_PVOutput:
	pvoutput:    	 True
	pvurl:       	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:    	 9cac8d83546faa4cb1c82ffb88c14fb586e88aff
	pvsystemid:  	 77962

Grott proxy mode started

	 - ('192.168.1.5', 17726) has connected

	 - Growatt packet received:
		  <socket.socket fd=516, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 49885), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
[... 387 lines cut ...]
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000318a
	 - No Growatt data processed or SN not found:

	 - Growatt packet received:
		  <socket.socket fd=496, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 5279), raddr=('192.168.1.5', 17726)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=516, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 49885), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=496, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 5279), raddr=('192.168.1.5', 17726)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=516, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 49885), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=496, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 5279), raddr=('192.168.1.5', 17726)>
	 - Data less then minimum record length, data not processed

johanmeijer, 2020-08-30:
Okay I have some challenges  with detecting the new generation growatt inverters/data loggers. (What is your inverter type?)

I try to create a temproray fix. 

Can you download the file T060104temp.json from gitthub (2.1.2 Beta, I just uploaded it)?  
Rename this file toT060104.json.   
Restart grott with the original .ini setting you used (only nomqtt = True and the changed pvooutput parms). 

Let see if this helps. I am very curious. 

If this looks okay I will try to fix the automatic detection for the new devices.

johanmeijer, 2020-08-30:
By the way I am very glad with your data,. This helps very much with improving grott!

xDAVIDx93, 2020-08-30:
> Okay I have some challenges  with detecting the new generation growatt inverters/data loggers. (What is your inverter type?)
> 
> I try to create a temporary fix. 
> 
> Can you download the file T060104temp.json from gitthub (2.1.2 Beta, I just uploaded it)?  
> Rename this file toT060104.json.   
> Restart grott with the original .ini setting you used (only nomqtt = True and the changed pvooutput parms). 
> 
> Let see if this helps. I am very curious. 
> 
> If this looks okay I will try to fix the automatic detection for the new devices. 

Ok ill try in the morning and its a Growatt 5000TL-X

johanmeijer, 2020-08-30:
Thanks. 

The rename is not necesarry, It will work with T010604temp.json!

This is what I detect in your record now: 

[image]

xDAVIDx93, 2020-08-30:
Ahhhh! I'm so excited!

johanmeijer, 2020-08-30:
I had to move the file  to a tempfix directory (tempfix/T060104temp.json) so normally processing is not disturbed. 

Please copy this file to your main grott directory.

xDAVIDx93, 2020-08-30:
So alls im doing is putting the new temp fix folder into my grott directory and setting invertid back to having a # and compat = False also with a #  valueoffset dont need the new t055104 and t065104 .jsons? That are on top just the t060104temp.json in the temfix folder?

xDAVIDx93, 2020-08-30:
Okay back to all the old setting in the .ini and the new temp fix folder give this error similar to the old error before setting the invertid

- Grott send data to PVOutput : 
		 -  {'d': '20200831', 't': '06:30', 'v1': 16056321000, 'v2': 0.0, 'v6': 0.0}
	 - Grott PVOutput response: 
		 -  Bad request 400: Energy value [2147483647] too high for [06:30]

Grott Command line parameters processed:
	verbose:     	 True
	config file: 	 grott.ini
	output file: 	 <_io.TextIOWrapper name='textlog.txt' encoding='cp1252'>
	nomqtt:      	 False
	inverterid:  	 ABC1234567
	pvoutput:    	 False

Grott Proces configuration file

Grott Proces environmental variables

Grott override settings if set in commandline

Grott Proces json layout files:

Grott layout records loaded:
T020104  : 
{'decrypt': False, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T050104  : 
{'decrypt': True, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T060104  : 
{'decrypt': True, 'pvserial': 76, 'date': 116, 'pvstatus': 158, 'pvpowerin': 162, 'pv1voltage': 170, 'pv1current': 174, 'pv1watt': 178, 'pv2voltage': 186, 'pv2current': 190, 'pv2watt': 194, 'pvpowerout': 202, 'pvfrequentie': 210, 'pvgridvoltage': 214, 'pvenergytoday': 262, 'pvenergytotal': 270, 'pvtemperature': 286, 'pvipmtemperature': 322}
T055104  : 
{'decrypt': True, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 170, 'pvfrequentie': 178, 'pvgridvoltage': 182, 'pvenergytoday': 274, 'pvenergytotal': 282, 'pvtemperature': 450, 'pvipmtemperature': 454}

Grott settings:

_Generic:
	version:     	 2.1.2
	verbose:     	 True
	trace:       	 False
	config file: 	 grott.ini
	minrecl:     	 100
	decrypt:     	 True
	compat:      	 False
	valueoffset: 	 6
	offset:      	 6
	inverterid:  	 ABC1234567
	mode:        	 proxy
	grottip      	 0.0.0.0
	grottport    	 5279
_MQTT:
	nomqtt       	 True
	mqttip:      	 localhost
	mqttport:    	 1883
	mqtttopic:   	 energy/growatt
	mqtttauth:   	 False
	mqttuser:    	 grott
	mqttpsw:     	 growatt2020
_Growatt server:
	growattip:   	 47.91.67.66
	growattport: 	 5279
_PVOutput:
	pvoutput:    	 True
	pvurl:       	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:    	 9cac8d83546faa4cb1c82ffb88c14fb586e88aff
	pvsystemid:  	 77962

Grott proxy mode started

	 - ('192.168.1.5', 13356) has connected

[... 992 lines cut ...]
	 - Grott automatic protocol detection
	 - layout   :  T060150
	 - Growatt data decrypted
	 - Growatt plain data:
		 000c0006024101505847443241313036424600000000000000000000000000000000000000005
		 8544436413132304635000000000000000000000000000000000000000014081e09342a020bb8
		 0c3400010000b6fc0d74004400005b8d0d65004400005b6f00000000000000000000000000000
		 00000000000000000000000b32d138209d300b40000b26e000000000000000000000000000000
		 0009d30000000000000000000000000000b32d001f704d0000006b00001a1500001a820000003
		 700000ca00000003600000de20000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000102f000a000000001480000001e400000
		 000000001c00efd00004e20005b00000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000c350cb1000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000000d26
	 - Grott data ack record or data record not defined no processing done

	 - Growatt packet received:
		  <socket.socket fd=536, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 5279), raddr=('192.168.1.5', 13356)>
	 - Data less then minimum record length, data not processed

johanmeijer, 2020-08-31:
Okay I was not totally clear I think. 

You have to copy the T060104temp.json from the tempfix folder into the grott main folder (the folder where .py modules are). 

I can not put in into the main folder. Then it would be active for everbody who downloads the 2.1.2 release (also with the old inverters). And give a lot of errors. Only the owners of the -x inverters have to perform the copy.

xDAVIDx93, 2020-08-31:
Ok ill try that now

xDAVIDx93, 2020-08-31:
Do I delete the other .json files?

johanmeijer, 2020-08-31:
No not necessary

xDAVIDx93, 2020-08-31:
i think its reading some other .json (- record layout :  T060104) i dont know how to make it read the new temp one

Grott Command line parameters processed:
	verbose:     	 True
	config file: 	 grott.ini
	output file: 	 <_io.TextIOWrapper name='textlog.txt' encoding='cp1252'>
	nomqtt:      	 False
	inverterid:  	 ABC1234567
	pvoutput:    	 False

Grott Proces configuration file

Grott Proces environmental variables

Grott override settings if set in commandline

Grott Proces json layout files:

Grott layout records loaded:
T020104  : 
{'decrypt': False, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T050104  : 
{'decrypt': True, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T060104  : 
{'decrypt': True, 'pvserial': 76, 'date': 116, 'pvstatus': 158, 'pvpowerin': 162, 'pv1voltage': 170, 'pv1current': 174, 'pv1watt': 178, 'pv2voltage': 186, 'pv2current': 190, 'pv2watt': 194, 'pvpowerout': 202, 'pvfrequentie': 210, 'pvgridvoltage': 214, 'pvenergytoday': 262, 'pvenergytotal': 270, 'pvtemperature': 286, 'pvipmtemperature': 322}
T055104  : 
{'decrypt': True, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 170, 'pvfrequentie': 178, 'pvgridvoltage': 182, 'pvenergytoday': 274, 'pvenergytotal': 282, 'pvtemperature': 450, 'pvipmtemperature': 454}

Grott settings:

_Generic:
	version:     	 2.1.2
	verbose:     	 True
	trace:       	 False
	config file: 	 grott.ini
	minrecl:     	 100
	decrypt:     	 True
	compat:      	 False
	valueoffset: 	 6
	offset:      	 6
	inverterid:  	 ABC1234567
	mode:        	 proxy
	grottip      	 0.0.0.0
	grottport    	 5279
_MQTT:
	nomqtt       	 True
	mqttip:      	 localhost
	mqttport:    	 1883
	mqtttopic:   	 energy/growatt
	mqtttauth:   	 False
	mqttuser:    	 grott
	mqttpsw:     	 growatt2020
_Growatt server:
	growattip:   	 47.91.67.66
	growattport: 	 5279
_PVOutput:
	pvoutput:    	 True
	pvurl:       	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:    	 9cac8d83546faa4cb1c82ffb88c14fb586e88aff
	pvsystemid:  	 77962

Grott proxy mode started

	 - ('192.168.1.5', 6375) has connected

	 - Growatt packet received:
		  <socket.socket fd=488, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 49859), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
[... 234 lines cut ...]
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=488, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 49859), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=596, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 5279), raddr=('192.168.1.5', 6375)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=488, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 49859), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=488, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 49859), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=488, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 49859), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=596, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.16', 5279), raddr=('192.168.1.5', 6375)>
	 - Data less then minimum record length, data not processed

xDAVIDx93, 2020-08-31:
[image]

xDAVIDx93, 2020-08-31:
grott.ini

#Specify grott monitor configuration
#defaults are descriped
#remove # and changed default to change configuration
# Updated: 2020-08-24
# Version 2.1.2

[Generic]
# Specify inverter id (not necessary in versi0n 2.1.0 if compat = false!)
#inverterid = ABC1234567
# Specify mode (sniff or proxy)(> 2.1.0 proxy is default)
#mode = proxy
# specify port and IP address to listen to (only proxy, default 5279), 0.0.0.0 ==> own ip address
#ip = 0.0.0.0
#port = 5279  
#Compat is True and valoffset needs to be set if offset / growatt protocol has been changed. 
#compat = False
#valueoffset = 6
#Decrypt is False if growatt communication is not encrypted (older inverters), (not necessary in versi0n 2.1.0 if compat = false!)
#decrypt = True

[Growatt] 
#ip address and port of Growatt server (specify only if IP has changed)

#ip = 47.91.67.66
#port = 5279                                                        

[MQTT]
#mqtt parameters definitions
#be aware nomqtt = True means no MQTT processing will be done!!!!!!

nomqtt = True
#ip = local
#port = 1883
#topic= energy/growatt
#auth = False
#user = grott
#password = growatt2020

[PVOutput]
#PVOutput parameters definitions

pvoutput = True
apikey = 9cac8d83546faa4cb1c82ffb88c14fb586e88aff
systemid = 77962

johanmeijer, 2020-08-31:
Hmm strange. 

Can you rename T060104temp.json to t060104temp.json ? 

You have to see somethiing like this during start; 

[image]

xDAVIDx93, 2020-08-31:
[image]

xDAVIDx93, 2020-08-31:
it worked!!! :)

xDAVIDx93, 2020-08-31:
[image]

johanmeijer, 2020-08-31:
;) 

Lets see if the values are comparable with the growatt figures. 

If it runs stable from the command line maybe you can try to start it as a WIndows service again. 

If everything thing works well. I will release 2.1.3. with the necessary fixes so the temp fix is not necessary anymore.

xDAVIDx93, 2020-08-31:
Growatt inverter data from growatt.server

[image][image][image][image][image]

xDAVIDx93, 2020-08-31:
Pvoutput.org  data

[image]

xDAVIDx93, 2020-08-31:
> Growatt inverter data from growatt.server
> 
> [image][image][image][image][image]

[image]

xDAVIDx93, 2020-08-31:
Growatt inverter data from growatt.server

[image][image][image][image][image][image]

xDAVIDx93, 2020-08-31:
Pvoutput:
221
150
137
103
72
41
22
5
2
2

Growatt.server:
221.6
150.2
137
103.4
72.9
41.9
22
5.9
2.6
2
1.9

johanmeijer, 2020-08-31:
Looking good I think.

Different rounding but not that bad.

johanmeijer, 2020-09-02:
Hi David,

Is everything still working?

xDAVIDx93, 2020-09-02:
yes got 2 days of good data all accurate and 2 seconds ago i started the nssm service

[52 more replies]

closed 2020-10-20 by johanmeijer