johanmeijer/grott #19: Wrong data values?

a-dekker, 2020-09-15:
Using a Growatt MIN 3600TL-XE with a ShineLink-X here.

Could it be that this combination generated a different payload?

I constantly get the same two read result values, either
```
-Growatt processing values for:  XXXXXXXXXX
        - pvserial:          XXXXXXXXXX
        - pvstatus:          0
        - pvpowerin:         0.0
        - pvpowerout:        0.2
        - pvenergytoday:     80532947.3
        - pvenergytotal:     111155609.7
        - pv1watt:           0.0
        - pv2watt:           0.0
        - pvfrequentie:      0.0
        - pvgridvoltage:     12.4
        - pv1voltage:        0.0
        - pv1current:        0.0
        - pv2voltage:        0.0
        - pv2current:        0.0
        - pvtemperature:     100.0
        - pvipmtemperature:  1233.6
```
as a successful read, or
```
-Growatt processing values for:  XXXXXXXXXX
        - pvserial:          XXXXXXXXXX
        - pvstatus:          0
        - pvpowerin:         0.0
        - pvpowerout:        0.2
        - pvenergytoday:     0.0
        - pvenergytotal:     0.0
        - pv1watt:           0.0
        - pv2watt:           0.0
        - pvfrequentie:      30.0
        - pvgridvoltage:     312.4
        - pv1voltage:        0.0
        - pv1current:        0.0
        - pv2voltage:        0.0
        - pv2current:        0.0
        - pvtemperature:     0.0
        - pvipmtemperature:  234.5
```
as an incomplete read.

The data on the growatt server side is fine.

johanmeijer, 2020-09-15:
Which version of grott are you using?

The 2.1.2 version (beta) should be able to handle the xe inverters.

Some values might be wrong but we can
 figure that out easily

a-dekker, 2020-09-15:
I am using the master branch. Will switch to the 2.1.2 branch and try again tomorrow.

johanmeijer, 2020-09-15:
Yes please. And if it gives strange values for energy today  please copy
T060104temp.json to the main directory and rename it to t060104.json. This
should work.

You do not have to specify anything in the .ini except the ip addess of the
mqtt server if that is not running at the same machine

a-dekker, 2020-09-16:
It doesn't look much different in the 2.1.2-(Beta) branch, (no matter if I have the default t060104.json or overwrite it with T060104temp.json).

```
- Growatt processing values for:  XXXXXXXXXX
        - pvserial:          XXXXXXXXXX
        - pvstatus:          0
        - pvpowerin:         0.0
        - pvpowerout:        0.2
        - pvenergytoday:     80532947.3
        - pvenergytotal:     111155609.7
        - pv1watt:           0.0
        - pv2watt:           0.0
        - pvfrequentie:      0.0
        - pvgridvoltage:     12.4
        - pv1voltage:        0.0
        - pv1current:        0.0
        - pv2voltage:        0.0
        - pv2current:        0.0
        - pvtemperature:     100.0
        - pvipmtemperature:  1233.6
- MQTT jsonmsg:
```
or
```
- Growatt processing values for:  XXXXXXXXXX
        - pvserial:          XXXXXXXXXX
        - pvstatus:          0
        - pvpowerin:         0.0
        - pvpowerout:        0.2
        - pvenergytoday:     0.0
        - pvenergytotal:     0.0
        - pv1watt:           0.0
        - pv2watt:           0.0
        - pvfrequentie:      30.0
        - pvgridvoltage:     312.4
        - pv1voltage:        0.0
        - pv1current:        0.0
        - pv2voltage:        0.0
        - pv2current:        0.0
        - pvtemperature:     0.0
        - pvipmtemperature:  232.7
```

johanmeijer, 2020-09-16:
What did you specify in the .ini? Can you send me the messages during
startup and the growatt data just before the mqtt sent?

a-dekker, 2020-09-16:
grott.ini
```
#Specify grott monitor configuration

[Generic]
# Specify inverter id
inverterid = XXXXXXXX
# Specify mode (sniff or proxy)
mode = proxy
# specify port to listen to (only proxy, default 5279)
port = 5279
#Decrypt is False if growatt communication is not encrypted (older inverters)
decrypt = True
#Compat is True and valoffset needs to be set if offset / growatt protocol has been changed. Since 2020 this needs to be specified (True,26)
compat = True
valueoffset = 26

[Growatt]
#ip address and port of Growatt server
ip = 47.91.67.66
port = 5279

[MQTT]
#mqtt parameters definitions
ip = 192.168.71.101
port = 1883
topic= energy/growatt
auth = False
user = grott
password = growatt2020
```
Startup messages:
```
Grott Growatt logging monitor : 2.1.2
Grott Command line parameters processed:
        verbose:              True
        config file:          grott.ini
        output file:          <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
        nomqtt:               False
        inverterid:           ABC1234567
        pvoutput:             False
Grott Proces configuration file
Grott Proces environmental variables
Grott override settings if set in commandline
Grott Proces json layout files:
t065104.json
t065104.json.old
t055104.json
Grott layout records loaded:
T020104  :
{'decrypt': False, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvene>
T050104  :
{'decrypt': True, 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvener>
T060104  :
{'decrypt': 'True', 'pvserial': 76, 'date': 116, 'pvstatus': 158, 'pvpowerin': 162, 'pv1voltage': 170, 'pv1current': 174, 'pv1watt': 178, 'pv2voltage': 186, 'pv2current': 190, 'pv2watt': 194, 'pvpowerout': 250, 'pvfrequentie': 258, 'pvgridvoltage': 262, 'pvenergytoday': 354,>
T055104  :
{'decrypt': 'True', 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 170, 'pvfrequentie': 178, 'pvgridvoltage': 182, 'pvenergytoday': 274, 'pven>
T065104  :
{'decrypt': 'True', 'pvserial': 76, 'date': 116, 'pvstatus': 158, 'pvpowerin': 162, 'pv1voltage': 170, 'pv1current': 174, 'pv1watt': 178, 'pv2voltage': 186, 'pv2current': 190, 'pv2watt': 194, 'pvpowerout': 250, 'pvfrequentie': 258, 'pvgridvoltage': 262, 'pvenergytoday': 354,>
Grott settings:
_Generic:
        version:              2.1.2
        verbose:              True
        trace:                False
        config file:          grott.ini
        minrecl:              100
        decrypt:              True
        compat:               True
        valueoffset:          26
        offset:               26
[... 86 lines cut ...]
         - Growatt processing values for:  NAC5A166E6
                 - pvserial:          NAC5A166E6
                 - pvstatus:          0
                 - pvpowerin:         0.0
                 - pvpowerout:        0.2
                 - pvenergytoday:     80532947.3
                 - pvenergytotal:     111155609.7
                 - pv1watt:           0.0
                 - pv2watt:           0.0
                 - pvfrequentie:      0.0
                 - pvgridvoltage:     12.4
                 - pv1voltage:        0.0
                 - pv1current:        0.0
                 - pv2voltage:        0.0
                 - pv2current:        0.0
                 - pvtemperature:     100.0
                 - pvipmtemperature:  1233.6
         - MQTT jsonmsg:
                         {"device": "NAC5A166E6", "time": "2020-09-16T14:04:34", "values":
                         {"pvstatus": 0, "pv1watt": 0, "pv2watt": 0, "pvpowerin": 0, "pvpowerout": 2,
                         "pvfrequentie": 0, "pvgridvoltage": 124, "pvenergytoday": 805329473,
                         "pvenergytotal": 1111556097, "pv1voltage": 0, "pv2voltage": 0, "pv1current":
                         0, "pv2current": 0, "pvtemperature": 1000, "pvipmtemperature": 12336}}
         - MQTT message message sent
```

johanmeijer, 2020-09-16:
Oke. I was not clear I think. The .ini file almost be empty in the 2.1.2 it
will automatically detect the record layout (hopefully).

At  least you have to specify compat = False to enable automatic detection.

Just to be sure rename the t060104.json.old to t060104.old (or something
else without json in it).

a-dekker, 2020-09-16:
renamed  t060104.json.old to t060104.old

Content grott.ini
```

[Generic]

[Growatt]

[MQTT]
ip = 192.168.71.101
topic= energy/growatt
auth = False
user = grott
password = growatt2020
```
Startup log
```
Grott Growatt logging monitor : 2.1.2

Grott Proces configuration file

Grott Proces environmental variables

Grott override settings if set in commandline

Grott Proces json layout files:
t065104.json
t055104.json

Grott layout records loaded:
T020104  : 
T050104  : 
T060104  : 
T055104  : 

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.71.101', 39270), raddr=('47.91.67.66', 5279)>

         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.71.101', 5279), raddr=('192.168.71.123', 1026)>

         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.71.101', 5279), raddr=('192.168.71.123', 1026)>

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.71.101', 39270), raddr=('47.91.67.66', 5279)>

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.71.101', 39270), raddr=('47.91.67.66', 5279)>
```
And these lines go on forever. No json log anymore.

a-dekker, 2020-09-16:
Hmm, this was without verbose mode, will see what verbose gives.

johanmeijer, 2020-09-16:
Hmm new type of record. You need a t065004.json I am afraid.
I will try to create one (I am on vacation so this will be a challenge with the limited IT equipement I have here).

a-dekker, 2020-09-16:
No hurry, enjoying your holiday is more important!

johanmeijer, 2020-09-16:
Thanks. 

I published a t065004.json. It is a copy of the record layout of the shinewifi-x (and xe inverter). Lets see if this ones works. 
If not I have to debug it next week.

a-dekker, 2020-09-16:
Just did a git pull, will let you now more tomorrow.

a-dekker, 2020-09-17:
First results look a lot better
```
- Growatt new layout processing
         - decrypt       :  True
         - offset        :  6
         - record layout :  T065004
         - pvserial:          XXXXXXXXX
         - pvstatus:          1
         - pvpowerin:         259.1
         - pvpowerout:        262.7
         - pvenergytoday:     0.1
         - pvenergytotal:     1135.4
         - pv1watt:           259.1
         - pv2watt:           0.0
         - pvfrequentie:      49.99
         - pvgridvoltage:     228.4
         - pv1voltage:        430.8
         - pv1current:        0.6
         - pv2voltage:        14.9
         - pv2current:        0.0
         - pvtemperature:     30.2
         - pvipmtemperature:  30.2
```
I will check a few more times today.

johanmeijer, 2020-09-17:
Looks good :)

a-dekker, 2020-09-17:
Yeah, the total values also do increase as expected.

I do get quite some discarded reads en grott restarts every now and then (ConnectionResetError: [Errno 104] Connection reset by peer), but I do not think Grott is to blame for that.

johanmeijer, 2020-09-17:
I am not aware of the connection reset problem. But to  be honest I never
looked at the log after I start Grott automatically as sevice. But now I
have to ;)

Thanks for the information. I really like to support as much as Growatt
inverter models as possible.

a-dekker, 2020-09-17:
Another model tested for now :-)

I'll close the issue.

closed 2020-09-17 by a-dekker