johanmeijer/grott #20: Support for Growatt MIN 3600-TL XE (layout=T065104X)

poljvd, 2020-10-19:
Hi,

I tried running your application and get the following output:

grott_1  |       - Grott automatic protocol detection
grott_1  |       - Grott data record length 585
grott_1  |       - layout   :  T065104X
grott_1  |       - Growatt data decrypted V2
grott_1  |       - Growatt plain data:
grott_1  |               00250006024151044e41433541323432343600000000000000000000000000000000000000004
grott_1  |               444463241323530424c0000000000000000000000000000000000000000000000000000020bb8
grott_1  |               0c340001000003e206470003000001e9079e0002000001f900000000000000000000000000000
grott_1  |               0000000000000000000000003e6138709490007000003f6000000000000000000000000000000
grott_1  |               000949000000000000000000000000000000000017070200000000000003930000039c0000000
grott_1  |               0000001b400000000000001e80000000000000000000000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000fffa00000000000000000000011701170
grott_1  |               0000000016e0e3c00004e20000200000000000000000000000000000000000000000000000000
grott_1  |               00003c0000000000000000000000000000000000000c350cb1000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000002710000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott_1  |               000000000002c54
grott_1  |       - Grott data ack record or data record not defined no processing done
grott_1  |

My guess (based on looking at the code) is that the detected layout is not supported (yet). Is there anything I can do to make your application work with my setup? Or am I missing something?

Kind regards,
Johan

PS: 
My setup:
- Growatt MIN 36000-TL XE with ShineLan RF module
- I installed Grott (v2.2.1d) on an raspberry pi using the docker image.
- Config file is almost default (enable MQTT, disable pvoutput)

johanmeijer, 2020-10-19:
Hi Johan,

Indeed this record is not coded yet. I will update grott. Can you send me the part off data just before the part you posted (growatt original data)? 

Do you use thr grottrpi docker image or the basis grott (which one I have do to publish first) ?

Regards,
Johan

johanmeijer, 2020-10-19:
The docker container has to be rebuild. If you use the plain version it is posible to create  a T065104X.json  file with the right layout definitions.

poljvd, 2020-10-19:
I assumed the posted part of the log was enough. But here comes also the original message:

grott_1  |       - Growatt packet received:
grott_1  |                <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.53.100', 47008), raddr=('47.91.67.66', 5279)>
grott_1  |       - Growatt original Data:
grott_1  |               \x00\x64\x00\x06\x02\x41\x51\x04\x09\x33\x2c\x42\x20\x46\x40\x75\x46\x59\x77
grott_1  |               \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott_1  |               \x2b\x33\x27\x46\x35\x75\x47\x5f\x35\x2d\x74\x74\x47\x72\x6f\x77\x61\x74\x74
grott_1  |               \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x6a
grott_1  |               \xcc\x78\x73\x72\x6e\x77\x61\x78\xcf\x41\x06\x6f\x7e\x61\x74\x72\x52\x75\xa3
grott_1  |               \x77\x69\x74\x74\x41\xd4\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
grott_1  |               \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x4b\x0b\x7c\xfd\x68\x39
grott_1  |               \x74\x49\x72\x6f\x7b\x1e\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
grott_1  |               \x61\x74\x74\x4e\x3f\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott_1  |               \x6f\x77\x61\x63\x48\x1e\x72\x6f\x77\x64\x74\x74\x44\xea\x6f\x77\x62\xd4\x74
grott_1  |               \x47\x72\x6d\x77\x61\x75\xc2\x47\x72\x6f\x75\x61\x74\x75\xad\x72\x6f\x77\x61
grott_1  |               \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
grott_1  |               \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
grott_1  |               \x72\x6f\x77\x61\x74\x74\xb8\x88\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x75
grott_1  |               \x53\x46\x55\x6f\x77\x61\x74\x75\x3f\x7c\x50\x77\x61\x3a\x54\x47\x7a\x6f\x77
grott_1  |               \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott_1  |               \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
grott_1  |               \x47\x72\x6f\x77\x61\x74\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
grott_1  |               \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
grott_1  |               \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
grott_1  |               \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
grott_1  |               \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
grott_1  |               \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott_1  |               \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
grott_1  |               \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
grott_1  |               \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
grott_1  |               \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
grott_1  |               \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
grott_1  |               \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
grott_1  |               \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott_1  |               \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x61\x22
grott_1  |       - Grott automatic protocol detection
grott_1  |       - Grott data record length 585
grott_1  |       - layout   :  T065104X
grott_1  |       - Growatt data decrypted V2
grott_1  |       - Growatt plain data:
grott_1  |               00640006024151044e41433541323432343600000000000000000000000000000000000000004
grott_1  |               444463241323530424c0000000000000000000000000000000000000000000000000000020bb8
grott_1  |               0c34000100000cbb067400090000061507cc0008000006a600000000000000000000000000000
grott_1  |               000000000000000000000000c79138a094d000e00000c7f000000000000000000000000000000
grott_1  |               00094d0000000000000000000000000000000000173c590000000500000398000003a00000000
grott_1  |               2000001b600000002000001ea0000000000000000000000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000fffa00000000000000000000012701270
grott_1  |               000000001780e3f00004e20000800000000000000000000000000000000000000000000000000
grott_1  |               00003c0000000000000000000000000000000000000c350cb1000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000002710000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott_1  |               00000000000134d
grott_1  |       - Grott data ack record or data record not defined no processing done

poljvd, 2020-10-19:
I use the grottrpi docker image.
I am not sure what you mean with "If you use the plain version ..."?

johanmeijer, 2020-10-19:
With plain I mean run it as a pytho program (service) outside docker. But
do not bother I will update the docker image.

johanmeijer, 2020-10-19:
I had som espare time so created a new build (2.2.2)

you can now use: ledidobe/grottrpi, ledidobe/grottrpi:latest and ledidobe/grottrpi:2.2.2

Hopefully it recognises your records now :)

poljvd, 2020-10-19:
Thank you for your efforts.

I am receiving data now that looks oke. Not sure whether it is of any significance, but I also saw two messages with different layout (T065103X and T065129) after recreating and restarting the container. All messages since then have T065104X. Also, the date parsing may have an issue.

grott_1  |       - Growatt packet received:
grott_1  |                <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.53.100', 47104), raddr=('47.91.67.66', 5279)>
grott_1  |       - Growatt original Data:
grott_1  |               \x00\xc7\x00\x06\x02\x41\x51\x03\x09\x33\x2c\x42\x20\x46\x40\x75\x46\x59\x77
grott_1  |               \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott_1  |               \x2b\x33\x27\x46\x35\x75\x47\x5f\x35\x2d\x74\x74\x47\x72\x6f\x77\x61\x74\x74
grott_1  |               \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x61
grott_1  |               \x74\x74\x3b\x72\x6e\x77\xb8\x74\x74\x47\x16\x6f\x77\x46\x64\x74\x47\xfe\xcf
grott_1  |               \x74\x89\x35\x38\x76\x5c\x5f\x77\x3b\x35\x36\x06\x72\x6d\x77\x60\x74\x74\x44
grott_1  |               \x9a\x6f\x4b\x61\x48\x74\x1d\x72\x35\x77\x61\x30\x30\x01\x40\x2e\x45\x54\x44
grott_1  |               \x36\x0b\x72\x6f\x77\x61\x74\x75\x47\x16\x6f\x77\x61\x74\x54\x67\x52\x3f\x21
grott_1  |               \x41\x3d\x1a\x31\x17\x1d\x03\x04\x06\x54\x67\x72\x6f\x64\x8d\x76\x75\x40\x96
grott_1  |               \x6f\x7d\x61\x67\x74\x48\x72\x73\x77\x56\x74\x75\x40\x42\x65\x22\x73\xfa\x60
grott_1  |               \x59\x75\x5f\x7d\x34\x66\xfa\x53\x6c\x68\x47\x6b\x21\x66\xc9\x66\x71\x70\xc2
grott_1  |               \x7d\x96\x55\xe1\x7c\xfa\x61\x46\x74\x75\x72\x5d\x77\x53\x74\x46\x47\x40\x6f
grott_1  |               \x45\x61\x46\x74\x75\x72\x5d\x77\x69\x74\x7c\x4e\x90\x6f\x77\x20\x38\x35\x06
grott_1  |               \x42\x5d\x47\x50\x54\x54\x77\x40\x6e\x46\x61\x74\x74\x47\x61\xf3\x77\x53\x7d
grott_1  |               \xc0\x4e\x90\x67\x33\x69\x62\x74\x53\x72\x6a\x7e\x0e\x7c\x88\x60\x62\x48\x67
grott_1  |               \x46\x64\x53\x57\x55\x7f\x50\x71\x74\xe2\x47\x72\x6e\x93\x61\x8b\x3a\x67\x72
grott_1  |               \x90\x39\x41\x74\x8b\x09\x52\x6f\x88\x2f\x54\x75\x47\x72\x6f\x78\x60\x74\x50
grott_1  |               \x47\x72\x6f\x77\x61\x74\x7f\xff\x7e\x5b\x77\x61\x30\x30\x01\x40\x2e\x45\x54
grott_1  |               \x44\x36\x0b\x42\x5f\x47\x51\x44\x44\x77\x42\x5f\x47\x51\x44\x44\x77\x42\x5f
grott_1  |               \x47\x51\x44\x44\x47\x72\x6e\x83\x61\x74\x75\xd7\x72\x6e\x77\x61\x74\x74\x47
grott_1  |               \x72\x6d\x2f\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x51\xda\x6f\x77\x61\x74
grott_1  |               \x74\x47\x72\x6f\x77\x61\x74\x10\x47\x77\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
grott_1  |               \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x23\x72\x0b\x77\x61\x74\x74\x47\x72
grott_1  |               \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
grott_1  |               \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
grott_1  |               \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
grott_1  |               \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x37\x2f\x39\x2c\x2c\x1f
grott_1  |               \x2a\x37\x2f\x39\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x54
grott_1  |               \x54\x67\x52\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
grott_1  |               \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott_1  |               \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x2c\x1f\xdf\x73
grott_1  |       - Grott automatic protocol detection
grott_1  |       - Grott data record length 585
grott_1  |       - layout   :  T065103X
grott_1  |       - Growatt data decrypted V2
grott_1  |       - Growatt plain data:
grott_1  |               00c70006024151034e41433541323432343600000000000000000000000000000000000000004
grott_1  |               444463241323530424c0000000000000000000000000000000000000000000000000000020000
grott_1  |               007c000100d9000000640000271000008ca003e8414c312e30005a41424100020001000003e80
grott_1  |               03c003c005a005a00004444463241323530424c00000000000100640000000020202050562049
grott_1  |               6e7665727465722020000013ec020107e4000a0013000f001c0037000107300a55128e141e073
grott_1  |               00a55128e141e07300a55128e141e07a309e21293138d00320032003200320032003200320032
grott_1  |               003200320008000809e20000414c41413032303120203032013100000000139c003209b409e20
grott_1  |               844081600140005096f08fc2710271027102710271027100096000001e400ff4e2000ff4e2000
grott_1  |               ff4e2000ff4e20010000000f0100240000000000000bb80c3400004444463241323530424c303
grott_1  |               0303030303030303030303030303030303030000001f400000190000100000000000002580000
grott_1  |               000000000000000016a8000000000000000000000064000500000000000000000000000000000
grott_1  |               00000000064006400000000000000000000000000000000000000000000000000000000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott_1  |               00000000000005858585858585858585800000000000000000000000000002020202000000000
grott_1  |               00000000000000000000000000000000000000000000000000000000000000000000000000000
[... cut ...]
grott_1  |               4c000000000000000000000000000000000000000013ec0004000135000500013100060002333
grott_1  |               20008001e415a4531413234314436000000000000000000000000000000000000000000090001
grott_1  |               30000a000130000d0002313000140008475453573030303000150007312e302e322e300016000
grott_1  |               456312e30001b000136001c000130001e0005474d542b38001f0013323032302d31302d313920
grott_1  |               31353a32393a3035b726
grott_1  |       - Grott data ack record or data record not defined no processing done

johanmeijer, 2020-10-19:
The T065103X is a announce message from the inverter that is sent to the growatt server at the moment the session is started.  Tis is not interesting for logging data. The T065129 I have not seen yet. I will look if I have equivalent records (T060129) with my configuration and try figure out what they are use for. For now I have not missed them. 

For logging data the Txxxx04(x) Txxxx50(x) records are important the 04 is actual logging and the 50 is historical data coming from the inverter buffer (eg if there connections to the growatt server has been broken). 
In the MQTT json message I added a key buffer=yes (50 record) or no (04) record. The time is the time the record is created by the inverter (not available in older inverters, then the actual time is used). In this way is is possible to handle historical data. 

Thanks for the information and testing. We can support more inverter type now. 

Let me know if something is missing or goes wrong.

poljvd, 2020-10-20:
Hi,

All seems to be running fine now for the last day. Thanks again for your efforts. Feel free to close the issue if you like (or leave it open for the T065129 messages).

Kind regards

johanmeijer, 2020-10-21:
I will put the analysis of the purpose of the T065129 message on the todo list (It doesn't cause any issues at this moment I think).  Can you look into your log how often this occurs (by example: docker logs grott|grep -i T065129) that would give more information. 

For now I think it only occurs after a (re)start and is some kind of sync message. But I can be wrong.

johanmeijer, 2020-10-24:
@poljvd Did you had time to search for the T065129 message?

johanmeijer, 2020-11-12:
This issue is solved and can be closed.

closed 2020-11-12 by johanmeijer