johanmeijer/grott #21: Move grott.ini to grott.ini.sample

a-dekker, 2020-10-24:
As I am running grott directly from the git repo, my personal grott.ini always conflicts with your upstream version.
Moving grott.ini to grott.ini.sample (and putting grott.ini in the .gitignore file) would solve that issue. You only have to tell users to copy grott.ini.sample grott.ini once and then let them edit the file to there own taste.

johanmeijer, 2020-10-24:
@a-dekker Thank you for the tip. I will change this with the next update.

You are aware that it is possible to specify your own .ini file at startup (-c yourfile.ini). This is how  I test with different configurations.

a-dekker, 2020-10-24:
No, I was not aware I could specify my own config file. That would be a solution as well. But then keeping a customed config file in git, it's more clean if it is ignored by git.

johanmeijer, 2020-10-24:
I agree. 

By the way, with the new version and when you running grott at the same server as the mqtt server it should be possible to run with the default ini. Grott will automatically  detect your inverter and record layout.

a-dekker, 2020-10-24:
My config is almost empty. I have a MQTT ip defined, but if it uses localhost by default it should be fine. Also defined the growatt server, but that could be a default value.
The only thing might be the "blockcmd = True", which is not default as I understood.

johanmeijer, 2020-10-24:
Yes it uses locolhost as default and you can use -b startup parm to enable command blocking.

johanmeijer, 2020-11-12:
With version 2.2.4. just publish the default grott.ini has been move to the exampes directory. 

I think this issue can be closed.  Do you agree @a-dekker ?

a-dekker, 2020-11-12:
It's good enough for me, no more conflicts when pulling new code in case of using the a modified default grott.ini.

I'll close the issue, thanks for your effort.

closed 2020-11-12 by a-dekker