johanmeijer/grott #23: Grott data ack record or data record not defined no processing done

Gitborter, 2020-11-11:
Hi Johan,

I've got this message from one of my growatt inverters : it seems that no data is processed ?

Can you help me with this ?  thanks in advance.

#011 - Growatt original Data:
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 \x00\x01\x00\x02\x00\x21\x01\x19\x41\x48\x32\x34\x30\x33\x30\x35\x39\x30\x00
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 \x10\x00\x11\x41\x43\x3a\x43\x46\x3a\x32\x33\x3a\x30\x36\x3a\x35\x33\x3a\x38
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 \x39\x00\x01\x00\x02\x00\x19\x01\x19\x41\x48\x32\x34\x30\x33\x30\x35\x39\x30
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 \x00\x11\x00\x09\x31\x30\x2e\x30\x2e\x33\x2e\x31\x35\x00\x01\x00\x02\x00\x14
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 \x01\x19\x41\x48\x32\x34\x30\x33\x30\x35\x39\x30\x00\x12\x00\x04\x35\x32\x37
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 \x39\x00\x01\x00\x02\x00\x19\x01\x19\x41\x48\x32\x34\x30\x33\x30\x35\x39\x30
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 \x00\x13\x00\x09\x31\x30\x2e\x30\x2e\x33\x2e\x31\x35\x00\x01\x00\x02\x00\x17
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 \x01\x19\x41\x48\x32\x34\x30\x33\x30\x35\x39\x30\x00\x15\x00\x07\x32\x2e\x30
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 \x2e\x30\x2e\x30
Nov 11 09:01:38 growatt-proxy grott[1357]: #011 - Grott automatic protocol detection
Nov 11 09:01:38 growatt-proxy grott[1357]: #011 - Grott data record length 156
Nov 11 09:01:38 growatt-proxy grott[1357]: #011 - layout   :  T020119
Nov 11 09:01:38 growatt-proxy grott[1357]: #011 - Growatt unencrypted data used
Nov 11 09:01:38 growatt-proxy grott[1357]: #011 - Growatt plain data:
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 0001000200210119414832343033303539300010001141433a43463a32333a30363a35333a383
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 90001000200190119414832343033303539300011000931302e302e332e313500010002001401
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 19414832343033303539300012000435323739000100020019011941483234303330353930001
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 3000931302e302e332e313500010002001701194148323430333035393000150007322e302e30
Nov 11 09:01:38 growatt-proxy grott[1357]: #011#011 2e30
Nov 11 09:01:38 growatt-proxy grott[1357]: #011 - Grott data ack record or data record not defined no processing done

johanmeijer, 2020-11-11:
Hi. This is an identify record and not a data record. Processing is not needed (it does not contain any inverter output information). 

Normally this records are supressed but this one is longer (156 bytes) then I have seen until now. By default records below the 100 bytes are supressed (can be set by the minrecl parameter).

Do you think you missing data or does everything looks oke? 

Records ending with 04 or 50 (in your case I think a T020104 / T020150) should be processed.

Gitborter, 2020-11-11:
Hi Johan,

I think I send you the  wrong set of data. one of my inverters is not beeing processed while the other goes smooth. There is a difference in layout ( T020404 not beeing processed and T020104 beeing processed smoothly)  I think this is the data not beeing processed : 

#011 - Growatt original Data:
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x00\x01\x00\x02\x00\xd9\x04\x04\x41\x48\x32\x34\x30\x33\x30\x35\x39\x30\x4c
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x55\x33\x31\x38\x31\x31\x36\x34\x39\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x2c\x00\x00\x00\x00\x00\x00\x05\xfe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x00\x00\x00\x00\x00\x00\x00\x00\x13\x85\x09\x15\x00\x00\x00\x00\x00\x00\x00
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x00\x01\x00\x9e\x0b\xa8\x59\x26\x00\xd9\x00\x00\x00\x00\x00\x00\x00\x00\x00
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x12\x00\x00\x00\x00\x00\x2d\x00\x59
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x4e\x20\x00\x00\x00\x00\x00\x00\x00\x06\x00\x01\x0d\x59\x00\x00\x00\x00\x00
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x00\x00\x00\x00\x01\x0d\x59\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
Nov 11 16:53:19 growatt-proxy grott[2288]: #011 - Grott automatic protocol detection
Nov 11 16:53:19 growatt-proxy grott[2288]: #011 - Grott data record length 223
Nov 11 16:53:19 growatt-proxy grott[2288]: #011 - layout   :  T020404
Nov 11 16:53:19 growatt-proxy grott[2288]: #011 - Growatt unencrypted data used
Nov 11 16:53:19 growatt-proxy grott[2288]: #011 - Growatt plain data:
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 0001000200d90404414832343033303539304c553331383131363439000000000000020000002
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 c00000000000005fe000000000000000000000000000000000000138509150000000000000000
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 0000000000000000000000000000000000060001009e0ba8592600d9000000000000000000000
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 000000000000000061200000000002d00594e20000000000000000600010d5900000000000000
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 0000010d590000000000000000000000000000000000000000000000000000000000000000000
Nov 11 16:53:19 growatt-proxy grott[2288]: #011#011 0000000000000000000000000000000000000000000000000000000000000
Nov 11 16:53:19 growatt-proxy grott[2288]: #011 - Grott data ack record or data record not defined no processing done
Nov 11 16:53:19 growatt-proxy grott[2288]: #011 - Growatt packet received:

Gitborter, 2020-11-12:
I've uploaded a log of this morning output of the problem inverter. This is all it does, proxy is doing it's job but no more than that.

[growatt.txt](https://github.com/johanmeijer/grott/files/5529111/growatt.txt)

johanmeijer, 2020-11-12:
Thanks I will have a look at it. The T020204 is new for me.  Which type of inverters do you have? How are they connected with your network?

johanmeijer, 2020-11-12:
Additional question do you use the docker container or the native python version?

Gitborter, 2020-11-12:
Hi Johan,

It's a Growatt model 1000 and I use the native python version.

Thanks in advance.

johanmeijer, 2020-11-12:
@Gitborter I published version 2.2.4. 

This version should solve your problem also. Can you please test it? 

Thanks.

Gitborter, 2020-11-13:
I had to install python3-distutils to get this version working. But the good news is that all my inverters are exporting data through MQTT now. 

Thanks

johanmeijer, 2020-11-13:
Ah ok.

 I used distutils for solving other issues. I did not realised this is not standard installed in python. 

I will update the information (or remove the coding using it). Thanks for the feedback.

johanmeijer, 2020-11-14:
I removed the distutils requirement (not important for you but for others I think). 

Can we close this issue?

Gitborter, 2020-11-15:
Hi Johan,

This issue is solved and can be closed, thanks for your effort and quick response.

closed 2020-11-15 by johanmeijer