johanmeijer/grott #24: Parsing of Boolean values should be case insensitive

benpeart, 2020-11-12:
When specifying boolean environment variables or command line options, traditionally they are all lower case. When I attempted this, it didn't work and it took me a while to figure out I had to do mixed case.  Making them case insensitive would be more flexible and less error prone.

johanmeijer, 2020-11-12:
The mixed case for boolean (first character uppercase) is a python naming convention. I do understand this can be confusing if you have to specify this in the parameters.

I will try to improve this in the next releases. Thanks.

johanmeijer, 2020-12-18:
Versie 2.3.0 (now available in the beta branche) also allow lower-case boolean in .ini and environmental variables.

johanmeijer, 2021-01-13:
Changes are generic available in master branche.

closed 2021-01-13 by johanmeijer