johanmeijer/grott #25: Docker support wiki entry has typos in sample command line

benpeart, 2020-11-12:
The docker support wiki entry is incorrect (https://github.com/johanmeijer/grott/wiki/Docker-support)

  docker run -d --network=host --restart on-failure -e gnomqtt="True" -e gblockcmd="True" -e pvoutput="True" -e gpvapikey="apikey" -e pvsystemid="systemid" ledidobe/grott
  
should read

  docker run -d --network=host --restart on-failure -e gnomqtt="True" -e gblockcmd="True" -e gpvoutput="True" -e gpvapikey="apikey" -e gpvsystemid="systemid" ledidobe/grott
  
(note the missing leading "g" on pvoutput and pvsystemid).

johanmeijer, 2020-11-12:
Thanks I will change that

johanmeijer, 2020-11-12:
@benpeart

 I changed it. This issue can be closed if you agree?

benpeart, 2020-11-13:
Looks great - thank you

closed 2020-11-13 by benpeart