johanmeijer/grott #26: How to install Grott?

JeroenvanRensen, 2020-11-12:
Hi,

I'm Jeroen van Rensen and I want to use this repo.

The readme says:

> Please look at the wiki for detailed information on installation and use!

But in the wiki there isn't a page called **Installation**.

Where can I find it?

Thank you! Jeroen

johanmeijer, 2020-11-12:
Hi @JeroenvanRensen . You might have a good point here. Maybe I have to put a first how to in the wiki. (I am an engineer so for I tought I did already a good job with what I described).  

There is a step by step description in this issue:
https://github.com/johanmeijer/grott/issues/18

From wiki point of view you have to start with:
https://github.com/johanmeijer/grott/wiki/Running-Grott-(stand-alone)

You also have to understand and implement rerouting of the data:
https://github.com/johanmeijer/grott/wiki/Rerouting-Growatt-Wifi-TCPIP-data-via-your-Grott-Server

After this you can customize Grott more reading the information on the wiki (and ask questions here).

Most important .ini values are the MQTT definitions and/or PVOutput org definitions.

johanmeijer, 2020-11-12:
Hi @JeroenvanRensen 
I added a first how to page to the wiki. 

https://github.com/johanmeijer/grott/wiki/@-First-time-installation

Hopefully this will help.

Remarks and comments are welcome.

benpeart, 2020-11-13:
I have a suggestion. Since I used the docker installation method (and found it difficult to find the 'getting started' information for that as well) how about adding a section on the docker installation process in the 'first time installation' with links off to the 'Docker support' wiki page for more details like the other section?

A related suggestion is to add a reference with a link to the sample docker-compose.yml file in the 'Docker support' wiki page (or just include it directly in the wiki given how simple it is). I didn't know it existed so ended up creating my own from scratch before I found it 'hidden' in the source directory.

Finally, to get more exposure for Grott, I'd recommend you post in the pvoutput.org Growatt forum (https://forum.pvoutput.org/c/inverters/growatt) letting people know about your tool. I searched and was ready to build my own from the remnants of earlier efforts before I found yours. Since you now have native pvoutput.org support, I'm sure everyone using that would love to know about Grott.

johanmeijer, 2020-11-20:
@jeroenvanrensen did you manage to install Grott?

jeroenvanrensen, 2020-11-21:
@johanmeijer yes I forgot to tell you but thank you very very much!!!

closed 2020-11-21 by jeroenvanrensen