johanmeijer/grott #27: Grott data ack record or data record not defined no processing done for T055204

benpeart, 2020-11-13:
I just got my pv panels installed and turned on this morning and am finally generating power! As part of my install, I setup grott to output my generation data to pvoutput but am not seeing any generation data show up at pvoutput.

I configured my Growatt Shinelink to point my my internal docker image. This appears to be working because when I check the logs of my docker image, I'm receiving packets and they appear to be getting parsed. I'm also receiving generation data at server.growatt.com and can monitor production there so it looks like the proxy functionality is working properly.

The docker-compose.yml file I'm using is:

version: "2"

services:
  grott:
    image: ledidobe/grott
    container_name: grott
    network_mode: host
    environment:
       - gblockcmd=False
       - gnomqtt=True
       - gpvoutput=True
       - gpvapikey=**********************
       - gpvsystemid=*****
    ports:
      - 5279:5279
    restart: unless-stopped

The Growatt Shinelink Logger is a GKB4827095;  Hardware Version V1.0, Firmware Version 1.0.6.9.
I have two inverters, both are 10000MTLP-US, S/N: DAA*****

Here is the last 100 lines of the log file:

2020-11-13T16:37:49.449568608Z ,
2020-11-13T16:37:49.449606710Z 	 - Growatt packet received:,
2020-11-13T16:37:49.449739339Z 		  <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 5279), raddr=('192.168.1.146', 1029)>,
2020-11-13T16:37:49.449778208Z 	 - Data less then minimum record length, data not processed,
2020-11-13T16:40:49.751385564Z ,
2020-11-13T16:40:49.751430171Z 	 - Growatt packet received:,
2020-11-13T16:40:49.751488636Z 		  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 58996), raddr=('47.91.67.66', 5279)>,
2020-11-13T16:40:49.751504195Z 	 - Data less then minimum record length, data not processed,
2020-11-13T16:40:49.893665673Z ,
2020-11-13T16:40:49.893683926Z 	 - Growatt packet received:,
2020-11-13T16:40:49.893739657Z 		  <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 5279), raddr=('192.168.1.146', 1029)>,
2020-11-13T16:40:49.893756849Z 	 - Data less then minimum record length, data not processed,
2020-11-13T16:41:55.186392530Z ,
2020-11-13T16:41:55.186438547Z 	 - Growatt packet received:,
2020-11-13T16:41:55.186474763Z 		  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 58996), raddr=('47.91.67.66', 5279)>,
2020-11-13T16:41:55.186552207Z 	 - Growatt original Data:,
2020-11-13T16:41:55.187840927Z 		 \x00\x0f\x00\x05\x01\x37\x51\x04\x00\x39\x2d\x43\x59\x46\x43\x77\x4b\x5a\x33,
2020-11-13T16:41:55.187878863Z 		 \x20\x35\x47\x7f\x46\x5c\x47\x56\x37\x74\x47\x72\x6f\x77\x61\x77\x74\x47\x72,
2020-11-13T16:41:55.187892258Z 		 \x43\x77\x60\x74\x74\xcf\x46\x65\x85\x61\x48\x74\x47\x36\x10\x7c\x37\x74\x6f,
2020-11-13T16:41:55.187904246Z 		 \x47\x72\x4f\xd2\x61\x74\xf1\x3c\x65\x03\x7e\xf3\x74\xff\x47\x72\xea\xfd\x61,
2020-11-13T16:41:55.187916082Z 		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x12,
2020-11-13T16:41:55.187927775Z 		 \x77\x61\x75\xc6\x47\x73\x7d\xf5\x60\x52\x74\x47\x71\xf4\x77\x61\x74\x74\x47,
2020-11-13T16:41:55.187939558Z 		 \x72\x6f\x77\x61\x74\x74\x47\x73\x4a\x79\xa8\x74\x74\x47\x72\x6f\x5a\x61\x2d,
2020-11-13T16:41:55.187952048Z 		 \x3a\x67\x72\x6f\x77\x61\x74\x74\x47\x33\x6f\x77\x61\xab\x74\x47\x72\x70\x77,
2020-11-13T16:41:55.187963898Z 		 \x61\x74\x19\x47\x72\x6e\xcd\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72,
2020-11-13T16:41:55.187975680Z 		 \x6f\x77\x61\x74\x74\x66\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74,
2020-11-13T16:41:55.187987503Z 		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61,
2020-11-13T16:41:55.187999156Z 		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x2e\x47\xf4\x6f,
2020-11-13T16:41:55.188010742Z 		 \x6e\x75\x7e\x65\x55\x60\x6f\x77\x61\x74\x6d\x56\x73\x6b\x74\x64\x69\x74\x47,
2020-11-13T16:41:55.188022338Z 		 \x72\x76\x66\x60\x70\x76\x75\x49\x6f\x77\x61\x6d\x65\x46\x73\x6f\x77\x6e\x74,
2020-11-13T16:41:55.188033849Z 		 \x74\x47\x72\x7e\x76\x60\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77,
2020-11-13T16:41:55.188045333Z 		 \x61\x74\x7f\x10\x72\x4b\x77\x61\x5a\x1f\x47\x72\x6f\x57\x61\x74\x74\x29\x72,
2020-11-13T16:41:55.188056946Z 		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x6a\x90,
2020-11-13T16:41:55.188089910Z 	 - Grott automatic protocol detection,
2020-11-13T16:41:55.188100267Z 	 - Grott data record length 319,
2020-11-13T16:41:55.188109417Z 	 - layout   :  T055104,
2020-11-13T16:41:55.189989192Z 	 - Growatt data decrypted V2,
2020-11-13T16:41:55.190024257Z 	 - Growatt plain data:,
2020-11-13T16:41:55.190431429Z 		 000f000501375104474b423438323730393544414133383433303743000000000000030000002,
2020-11-13T16:41:55.190472412Z 		 c0001000088340af2003c0000447f0b56001b000020a50000857b176c0992008b0000858a0000,
2020-11-13T16:41:55.190485001Z 		 00000000000000000000000000000000007d000001b20001128201260000039b0000000000000,
2020-11-13T16:41:55.190495671Z 		 0000000000001250ec900000000002d00594e200000000000000041000000df0000001f000000,
[... 33 lines cut ...]
2020-11-13T16:41:59.839844514Z 	 - layout   :  T055204,
2020-11-13T16:41:59.841817723Z 	 - Growatt data decrypted V2,
2020-11-13T16:41:59.841853285Z 	 - Growatt plain data:,
2020-11-13T16:41:59.842237277Z 		 0010000501375204474b423438323730393544414133383433313142000000000000030000002,
2020-11-13T16:41:59.842252826Z 		 c000100009c280bc1003f00005256096d0022000023e900008f6f176e0995009700008f7e0000,
2020-11-13T16:41:59.842257797Z 		 00000000000000000000000000000000007800000120000112990127000003690000000000000,
2020-11-13T16:41:59.842262346Z 		 0000000000001250ece00000000002d00594e20000000000000003f0000004e0000001d000000,
2020-11-13T16:41:59.842266913Z 		 6d000001250000000000000000000000000000000000230000000000000000000000000000000,
2020-11-13T16:41:59.842271295Z 		 0000000000000000000000000000000000000000000000000000000000000005a00860019140a,
2020-11-13T16:41:59.842275769Z 		 1112113a00000019110109000d1000000019110105150b110000001911010100000f000000001,
2020-11-13T16:41:59.842280145Z 		 101010000000000000000000000000000000a7e001d0000216f0000001f0000006a0000000000,
2020-11-13T16:41:59.842284525Z 		 0000000000000000002df0,
2020-11-13T16:41:59.842288959Z 	 - Grott data ack record or data record not defined no processing done,
2020-11-13T16:41:59.947063029Z ,
2020-11-13T16:41:59.947092637Z 	 - Growatt packet received:,
2020-11-13T16:41:59.947186267Z 		  <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 5279), raddr=('192.168.1.146', 1029)>,
2020-11-13T16:41:59.947217102Z 	 - Data less then minimum record length, data not processed,
2020-11-13T16:43:50.320288024Z ,
2020-11-13T16:43:50.320345145Z 	 - Growatt packet received:,
2020-11-13T16:43:50.320436459Z 		  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 58996), raddr=('47.91.67.66', 5279)>,
2020-11-13T16:43:50.320478706Z 	 - Data less then minimum record length, data not processed,
2020-11-13T16:43:50.508270327Z ,
2020-11-13T16:43:50.508297583Z 	 - Growatt packet received:,
2020-11-13T16:43:50.508353632Z 		  <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 5279), raddr=('192.168.1.146', 1029)>,
2020-11-13T16:43:50.508385493Z 	 - Data less then minimum record length, data not processed,

johanmeijer, 2020-11-13:
Indeed the t055104 is not defined. Have not seen it yet. I will add it. 

In the native app you can add a t055104.json file (see examples and the layout defs in grottconf.py) , in the grott directory with the layout definitions. I think it wil be the same as the T050104 defs but will test it. With docker I have to create a new image I think. 

In the future I will create a possibility to link to a directory (volume)  outside the container. 

By the way thank you for the very usefull tips to improve Grott!

johanmeijer, 2020-11-14:
@benpeart I have published the ledidobe/grott:2.2.4a image which should solve your issue. 

Can you test if it works? 

**Be aware:** You have to specify the 2.2.4a tag. The grott:latest image still points to the 2.2.2. version. 
I will change that when 2.2.4a is fully tested.

benpeart, 2020-11-14:
The good news is that I pulled the new image and am getting power data sent to pvoutput! The interesting thing was it was from a packet T055104 instead of a T055204. The T055204 still generates a "no processing done" error.

I also noticed I'm only getting data from one of the inverters (serial ending 307C) and not the other (serial ending 311B).

Here is the last 200 lines from my log with the new image:

,
	 - Growatt packet received:,
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 5279), raddr=('192.168.1.146', 1029)>,
	 - Data less then minimum record length, data not processed,
,
	 - Growatt packet received:,
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 59994), raddr=('47.91.67.66', 5279)>,
	 - Growatt original Data:,
		 \x00\x63\x00\x05\x01\x37\x51\x04\x00\x39\x2d\x43\x59\x46\x43\x77\x4b\x5a\x33,
		 \x20\x35\x47\x7f\x46\x5c\x47\x56\x37\x74\x47\x72\x6f\x77\x61\x77\x74\x47\x72,
		 \x43\x77\x60\x74\x74\x7b\x86\x65\x9a\x61\x6e\x74\x47\x6c\x19\x7c\x52\x74\x79,
		 \x47\x72\x60\xe0\x61\x74\x4f\xfb\x65\x1f\x7e\xfc\x74\x4c\x47\x72\x54\xbc\x61,
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x73\x10,
		 \x77\x61\x70\x53\x47\x70\xfe\x88\x60\x36\x74\x47\x71\xc2\x77\x61\x74\x74\x47,
		 \x72\x6f\x77\x61\x74\x74\x47\x73\x28\x79\x83\x74\x74\x47\x72\x6f\x5a\x61\x2d,
		 \x3a\x67\x72\x6f\x77\x61\x74\x74\x47\xb7\x6f\x77\x63\x57\x74\x47\x72\x0f\x77,
		 \x61\x75\x7f\x47\x72\x6b\x4a\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72,
		 \x6f\x77\x61\x74\x74\x49\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74,
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61,
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x2e\x47\xf4\x6f,
		 \x6e\x75\x7e\x65\x55\x60\x6f\x77\x61\x74\x6d\x56\x73\x6b\x74\x64\x69\x74\x47,
		 \x72\x76\x66\x60\x70\x76\x75\x49\x6f\x77\x61\x6d\x65\x46\x73\x6f\x77\x6e\x74,
		 \x74\x47\x72\x7e\x76\x60\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77,
		 \x61\x74\x7f\x6a\x72\x63\x77\x61\x7a\xd8\x47\x72\x6f\x15\x61\x74\x75\x48\x72,
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x48\xbd,
	 - Grott automatic protocol detection,
	 - Grott data record length 319,
	 - layout   :  T055104,
	 - decrypt :  True,
	 - Growatt data decrypted V2,
	 - Growatt plain data:,
		 0063000501375104474b423438323730393544414133383433303743000000000000030000002,
		 c000100003cf40aed001a00001e760b33000d00000f9700003bbc1770099d003800003bcb0000,
		 00000000000000000000000000000000017f00000427000291ff0142000003ad0000000000000,
		 0000000000001470ee200000000002d00594e2000000000000000c50000022300000060000001,
		 0b0000043d00000000000000000000000000000000000e0000000000000000000000000000000,
		 0000000000000000000000000000000000000000000000000000000000000005a00860019140a,
		 111212000000001911010403051d0000001911010402323b0000001911010100000f000000001,
		 101010000000000000000000000000000000b2d000c00000eac000000620000010f0000000000,
		 0000000000000000003ad2,
	 - Growatt new layout processing,
		 - decrypt       :  True,
		 - offset        :  6,
		 - record layout :  T055104,
,
	 - Grott data record date/time processing started,
	 - no or no valid time/date found, grott server time will be used (data records from buffer will not be sent!),
	 - Grott values retrieved:,
		 - pvserial:          DAA384307C,
		 - pvstatus:          1,
		 - pvpowerin:         1560.4,
		 - pvpowerout:        1529.2,
		 - pvenergytoday:     38.3,
		 - pvenergytotal:     106.3,
		 - pv1watt:           779.8,
		 - pv2watt:           399.1,
		 - pvfrequentie:      60.0,
		 - pvgridvoltage:     246.1,
		 - pv1voltage:        279.7,
		 - pv1current:        2.6,
		 - pv2voltage:        286.7,
		 - pv2current:        1.3,
		 - pvtemperature:     32.2,
		 - pvipmtemperature:  32.7,
[... 111 lines cut ...]
	 - Data less then minimum record length, data not processed,
,
	 - Growatt packet received:,
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 59994), raddr=('47.91.67.66', 5279)>,
	 - Growatt original Data:,
		 \x00\x67\x00\x05\x00\x99\x52\x29\x00\x39\x2d\x43\x59\x46\x43\x77\x4b\x5a\x3a,
		 \x38\x37\x44\x7f\x46\x5d\x44\x22\x47\x30\x06\x33\x5c\x4f\x55\x47\x45\x76\x30,
		 \x6f\xa5\x61\x70\x74\x46\x47\x6f\x72\x61\x75\x45\x47\x74\x6f\x75\x52\x46\x74,
		 \x4f\x72\x65\x3a\x38\x37\x44\x7f\x46\x5d\x44\x22\x47\x74\x4e\x72\x6e\x47\x61,
		 \x7e\x74\x46\x42\x6f\x7a\x61\x76\x45\x77\x72\x7b\x77\x69\x33\x20\x14\x25\x5f,
		 \x47\x51\x44\x74\x52\x72\x68\x46\x4f\x44\x5a\x76\x5c\x5b\x77\x77\x74\x70\x11,
		 \x43\x41\x47\x61\x6f\x74\x46\x44\x6f\x6b\x61\x75\x44\x47\x6c\x6f\x72\x26\x39,
		 \x20\x6c\x4a\x6f\x68\x61\x67\x46\x77\x40\x5f\x5a\x50\x45\x59\x76\x46\x4f\x46,
		 \x57\x4e\x44\x74\x48\x5e\x44\xef\xd7,
	 - Grott automatic protocol detection,
	 - Grott data record length 161,
	 - layout   :  T055229,
	 - Growatt data decrypted V2,
	 - Growatt plain data:,
		 0067000500995229474b42343832373039354d594330383432334333444141333834333131420,
		 0d2000400013500050001310006000233320008000a4d5943303834323343330009000130000a,
		 000130000d0002313000140008475453573030303000150007312e302e312e340016000456312,
		 e30001b000136001c000130001e0005474d542b38001f0013323032302d31312d31342031363a,
		 30333a31338ea3,
	 - Grott data ack record or data record not defined no processing done,

johanmeijer, 2020-11-14:
Aah. You have multiple inverters coupled with the shinlink. I focussed on the first record and missed the second one.

I will add that one also. 

Good news is I think I can improve the auto detection  so less updates might be needed in the future.

johanmeijer, 2020-11-14:
Thinking more about it we have a challenge with PVoutput. How did you defined the to inverers in PVOutput?

Each inverer send its own record (hint 51/52). Grott will process and sent the records  but PVOutput wil not see the difference between these records. The values will not be summed or so.

Is it possible to define multiple inverters in pvoutput? Did you used that?

benpeart, 2020-11-14:
Interesting. In pvoutput if you go into "edit system," the 9th field down they have you set the number of inverters (which I have set to 2). This would make me think you should be able to specify which inverter the data is for. I didn't see anything in the API spec that allowed you to specify which inverter the record was for so I kept looking...

It appears I need to setup "Data Aggregation" (see https://pvoutput.org/help.html#general-aggregation) with 2 child systems (one for each inverter) and then a parent system that can aggregate the solar production data from the child systems. This complicates grott because now you would need to configure multiple System IDs and somehow map the different records to the correct System IDs (perhaps by specifying the inverter SN that corresponds to the System ID?).

Note, I'm new to pvoutput so I'm making educated guesses based on my reading of their help file - I don't have any first had experience with any of this.

johanmeijer, 2020-11-15:
Yes I think it is a good idea to allow specifing more pvoutput ssid's with specified inverter s/n.  

I have think about how to introduce that with backwards compitibility for the current users.
  
I saw the aggregate option in pvoutput but did not do anything with it yet. I have a seperate pvoutput account which I use for aggregated data. I use mqtt / nodered to aggregate the values of my two pv systems (the other one is not from Growatt). 

But using the aggregate option for this might be a good idea. Is it feature of the free account or do you need PVOutput pro license for it?

johanmeijer, 2020-11-15:
You have to make a donation to pvoutput.org to enable multiple systems and the aggregate functionality. 

I have done that so now I can test with multiple inverters. To be continued.

benpeart, 2020-11-15:
For what it's worth, I also donated and enabled aggregation and pointed my current grott events to the 'child' I created for inverter 1 and it appears to be working as expected. The grott events are showing up both in the child and in the parent. I am only aggregating the solar data from the 'child/inverter 1' because I'm not getting data from the 2nd inverter yet but I'm assuming they would get aggregated correctly if they existed.

johanmeijer, 2020-11-15:
I will implement the possibility to log to pvoutput for multiple inverters (make a link between pvoutput systemid and inverter s/n).  

I do not promise it will be ready tomorrow ;)

johanmeijer, 2020-11-17:
@benpeart I published a docker image version 2.2.5 . 

This is my first attempt for supporting multiple inverters with a shinelink. 

To enable the new functionality you should specify the following environmental variables at docker run (or in the docker-compose.yaml). 

-e gpvinverters=2 
-e gpvsystemid1="pvoutput systemid for first inverter" (for compatibility reasons also gpvsystemid= can be used for this)
-e gpvsystemid2="pvoutput systemid for second inverter"

Be aware I have not been able to test the container version (it is already dark in Amsterdam) but it has run natively for the whole day now. 

I am performing some test with the PVOutput aggregate functionality and made systemid1 parent of systemid2. But I am not sure if that is giving all the information I want. Maybe I am going to define a dummy systemid3 which is the parent of both systemid1 and systemid2. This might make it more easy to report on the inverters separately and aggregated.

johanmeijer, 2020-11-17:
Just to be sure: you have to use the ledidobe/grott:2.2.5  image

benpeart, 2020-11-18:
Thank you for working on this. I really appreciate it!

I built a new image and the log file looks like it has picked up my 2 inverters system ID's:

_PVOutput:,
	pvoutput:    	 True,
	pvurl:       	 https://pvoutput.org/service/r2/addstatus.jsp,
	pvapikey:    	 ************,
	pvinverters: 	 2,
	pvsystemid:  	 {1: '80119', 2: '80120'},

Unfortunately, I'm only getting data for the first inverter:

	 - Grott data record date/time processing started,
	 - no or no valid time/date found, grott server time will be used (buffer records not sent!),
	 - Grott values retrieved:,
		 - pvserial:          DAA384307C,
		 - pvstatus:          0,
		 - pvpowerin:         0.0,
		 - pvpowerout:        0.0,
		 - pvenergytoday:     31.0,
		 - pvenergytotal:     192.4,
		 - pv1watt:           0.0,
		 - pv2watt:           0.0,
		 - pvfrequentie:      59.98,
		 - pvgridvoltage:     245.0,
		 - pv1voltage:        0.0,
		 - pv1current:        0.0,
		 - pv2voltage:        0.0,
		 - pv2current:        0.0,
		 - pvtemperature:     4.7,
		 - pvipmtemperature:  10.5,
	 - MQTT jsonmsg: ,
			 {"device": "DAA384307C", "time": "2020-11-17T21:52:20", "buffered": "no",,
			 "values": {"pvstatus": 0, "pv1watt": 0, "pv2watt": 0, "pvpowerin": 0,,
			 "pvpowerout": 0, "pvfrequentie": 5998, "pvgridvoltage": 2450,,
			 "pvenergytoday": 310, "pvenergytotal": 1924, "pv1voltage": 0, "pv2voltage":,
			 0, "pv1current": 0, "pv2current": 0, "pvtemperature": 47,,
			 "pvipmtemperature": 105}},
	 - No MQTT message sent, MQTT disabled,
	 - Grott send data to PVOutput : ,
		 -  {'X-Pvoutput-Apikey': '********', 'X-Pvoutput-SystemId': '80119'},
		 -  {'d': '20201117', 't': '21:52', 'v1': 31000, 'v2': 0.0, 'v6': 245.0},
	 - Grott PVOutput response: ,
		 -  OK 200: Added Status,

I'm still seeing other messages come through (ie T055103) and am wondering if they contain the information for the second inverter.

	 - Growatt packet received:,
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 52272), raddr=('47.91.67.66', 5279)>,
	 - Growatt original Data:,
		 \x00\x03\x00\x05\x00\xd9\x51\x03\x00\x39\x2d\x43\x59\x46\x43\x77\x4b\x5a\x33,
		 \x20\x35\x47\x7f\x46\x5c\x47\x56\x37\x74\x47\x72\x6f\x77\x61\x76\x74\x47\x72,
		 \x43\x76\x60\x74\x75\x47\x72\x6f\x13\x61\x10\x53\x57\x72\x6e\xf1\xc1\x7a\x64,
		 \x0c\x31\x5e\x59\x51\x74\x74\x47\x72\x6f\x77\x61\x74\x75\x47\x76\x6a\xab\x61,
		 \x6a\x7c\x07\x78\x3f\x61\x25\x6c\xe8\x03\x33\x2e\x44\x59\x40\x47\x77\x45\x2c,
		 \x77\x61\x8c\xd5\x47\x73\x6f\x77\x61\x74\x74\x47\x72\x6f\x71\xf1\x7f\x34\x54,
		 \xfa\x76\x13\x69\x34\x7e\x17\x64\x2b\x6f\xfd\x74\xa6\x45\x73\x6f\x5a\x61\x2d,
		 \x73\xa3\x72\x64\x77\x70\x74\x62\x47\x46\x6f\x60\x65\x98\x74\x7b\x70\xfb\x77,
		 \x5d\x70\x98\x43\x9e\x6f\x7d\x61\x48\x54\x67\x52\x3f\x21\x41\x3d\x1a\x31\x17,
		 \x1d\x03\x04\x06\x54\x67\x19\x0c\x16\x00\x44\x46\x77\x41\x6f\x77\x61\x76\x75,
		 \x76\x72\x6f\x77\x60\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x65\xeb\x74\x89,
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x19\x74,
	 - Grott automatic protocol detection,
	 - Grott data record length 225,
	 - layout   :  T055103,
	 - decrypt :  True,
	 - Growatt data decrypted V2,
	 - Growatt plain data:,
		 0003000500d95103474b423438323730393544414133383433303743000000000000020000002,
		 c010100010000006400642710000186a00e104b43312e30000000000000000001000405dc001e,
		 08400a501644189c444141333834333037430000f8a10001000000000000000006900b4013881,
		 96408400a501644189c00d20201002d005907e4000b001100160034001704ec003c0294003c04,
		 ec04ec000a003c202020505620496e76657274657220206b63616130323033000000020131000,
		 0000100000000000000000000178403e800000000000000000000000000006d00,
	 - Grott data ack record or data record not defined no processing done,

Please let me know if there is anything else I can do to help.

benpeart, 2020-11-18:
One question I had is if the way you are mapping inverter to system ID is deterministic. That is, it needs to always map the same inverter to the same system ID - even between reboots, upgrades, and new docker images.

benpeart, 2020-11-18:
Now that the sun has come up and I'm generating power - I'm seeing records for both inverters come through and are correctly showing up on pvoutput! I don't know why last night I was only seeing records for inverter 1 but it appears to be working correctly now. Thank you!

Just FYI - I have 3 systems defined in pvoutput, one for each inverter and a 3rd for the 'parent' system. The parent system is correctly adding together the two 'child' systems for the inverters.

johanmeijer, 2020-11-18:
Yes. That sound super. I am glad it is working. 

I am not totally convinced this the way how I want to define and detect the inverters. I now detect your inverter based on the record layout (T055104 for inverter1 and T015204 for inverter2).  This works fine with a shinelink who is aware of the two inverters and can send different records. 

If I have 2 growatt inverters with both shinewifi interfaces to send directly the information, they will use both the same record definition (probably T050104 or T060104) and if you want to make a difference between both for PVOutput you need  2 copies of grott running (with different ports). If I can define a hard link for pvoutput systemid and inverterid it will be possible to do everything with one grott instance. 
This will make the link to pvoutput systemid and inverter id more reliable. Now it is based in the definitions 
you made / the order you have chosen in the shinelink (I think this answers your question about mapping). 

I might ask you in the near future to test a new version which will support the hard systemid/inverterid link.

johanmeijer, 2020-11-21:
As promised I changed it a little bit to support multiple inverter support for both shinelink and shinewan/shinewan connected inverters. I published docker image 2.2.6 (ledidobe/grott:226). 

To use this one you have to specify these environmental variables. 
-e gpvinverters=2
-e gpvsystemid1="pvoutput systemid for first inverter" (
-e gpvinverterid1="s/n of first inverter: ends with 307C" 
-e gpvsystemid2="pvoutput systemid for second inverter"
-e gpvinverterid2="s/n of first inverter: ends with 311B" 

I am curious about the results.

benpeart, 2020-11-22:
Thanks, I'll give that a try here in a few minutes and report back. 

I was just going to let you know I've seen a strange anomaly the past two days with the 2.2.5 image. For some reason, I'm not getting data for 3 events during the day - the events at 1:00 PM, 1:05 PM, and 1:10 PM are missing. I've no idea why these 3 and why it is so consistent but thought I'd let you know.

11/21/2020

21/11/20	1:20PM	39.700kWh	1.827kWh/kW	3,863W	16,800W	0.773kW/kW	18.3C	123.7V	31.148kWh	2,118W	
21/11/20	1:15PM	38.300kWh	1.762kWh/kW	6,792W	12,000W	0.552kW/kW	18.3C	123.5V	30.971kWh	2,155W	
21/11/20	1:10PM	37.300kWh	1.716kWh/kW	0W	0W	0.000kW/kW	18.3C	123.6V	30.792kWh	2,711W	
21/11/20	1:05PM	37.300kWh	1.716kWh/kW	0W	0W	0.000kW/kW	18.3C	123.0V	30.566kWh	2,385W	
21/11/20	1:00PM	37.300kWh	1.716kWh/kW	0W	0W	0.000kW/kW	18.3C	122.8V	30.367kWh	1,876W	
21/11/20	12:55PM	37.300kWh	1.716kWh/kW	13,828W	13,200W	0.607kW/kW	18.3C	122.9V	30.211kWh	1,999W	
21/11/20	12:50PM	36.200kWh	1.666kWh/kW	14,110W	7,200W	0.331kW/kW	18.3C	123.5V	30.044kWh	2,096W	

11/20/2020

20/11/20	1:20PM	60.000kWh	2.761kWh/kW	13,540W	46,800W	2.153kW/kW	19.4C	123.9V	35.803kWh	2,173W	
20/11/20	1:15PM	56.100kWh	2.581kWh/kW	13,982W	13,200W	0.607kW/kW	19.4C	123.3V	35.622kWh	2,090W	
20/11/20	1:10PM	55.000kWh	2.530kWh/kW	0W	0W	0.000kW/kW	18.9C	123.4V	35.448kWh	2,113W	
20/11/20	1:05PM	55.000kWh	2.530kWh/kW	0W	0W	0.000kW/kW	18.9C	122.9V	35.272kWh	2,238W	
20/11/20	1:00PM	55.000kWh	2.530kWh/kW	0W	0W	0.000kW/kW	18.9C	122.8V	35.085kWh	1,973W	
20/11/20	12:55PM	55.000kWh	2.530kWh/kW	14,003W	14,400W	0.663kW/kW	18.3C	123.3V	34.921kWh	2,044W	
20/11/20	12:50PM	53.800kWh	2.475kWh/kW	14,159W	14,400W	0.663kW/kW	18.3C	123.1V	34.751kWh	1,907W

johanmeijer, 2020-11-22:
Hmm strange. Do you see the same gaps at the growatt website?

benpeart, 2020-11-22:
Quick update on the new image. I installed the new 2.2.6 image and it looks like it is working correctly. From the log:

Grott settings:,
	version:     	 2.2.6,
	verbose:     	 True,
	trace:       	 False,
	config file: 	 grott.ini,
	minrecl:     	 100,
	decrypt:     	 True,
	compat:      	 False,
	blockcmd:    	 False,
	noipf:       	 False,
	time:        	 auto,
	sendbuf:     	 True,
	valueoffset: 	 6,
	offset:      	 6,
	inverterid:  	 automatic,
	mode:        	 proxy,
	grottip      	 default,
	grottport    	 5279,
_MQTT:,
	nomqtt       	 True,
	mqttip:      	 localhost,
	mqttport:    	 1883,
	mqtttopic:   	 energy/growatt,
	mqtttauth:   	 False,
	mqttuser:    	 grott,
	mqttpsw:     	 growatt2020,
_Growatt server:,
	growattip:   	 47.91.67.66,
	growattport: 	 5279,
_PVOutput:,
	pvoutput:    	 True,
	pvurl:       	 https://pvoutput.org/service/r2/addstatus.jsp,
	pvapikey:    	 ****,
	pvinverters: 	 2,
	pvsystemid:  	 {1: '80119', 2: '80120'},
	pvinvertid:  	 {1: 'DAA384307C', 2: 'DAA384311B'},

I'm still seeing the data show up on pvoutput for both inverters and at growatt.

benpeart, 2020-11-22:
I don't see any gaps on the growatt website, it looks complete. I got the new image setup in time to catch the 3 timeslots that have been missing and they are missing with 2.2.6 as well:

11/22/20	2:10PM	20.000kWh	0.920kWh/kW	3,697W	0W	0.000kW/kW	-	-	24.679kWh	-	
11/22/20	2:05PM	20.000kWh	0.920kWh/kW	3,428W	0W	0.000kW/kW	-	-	24.679kWh	-	
11/22/20	1:20PM	20.000kWh	0.920kWh/kW	3,588W	13,200W	0.607kW/kW	-	-	24.679kWh	-	
11/22/20	1:15PM	18.900kWh	0.870kWh/kW	3,684W	4,800W	0.221kW/kW	57.9F	-	24.679kWh	-	
11/22/20	1:10PM	18.500kWh	0.851kWh/kW	0W	0W	0.000kW/kW	57.9F	122.9V	24.679kWh	2,208W	
11/22/20	1:05PM	18.500kWh	0.851kWh/kW	0W	0W	0.000kW/kW	57.9F	123.3V	24.495kWh	2,924W	
11/22/20	1:00PM	18.500kWh	0.851kWh/kW	0W	0W	0.000kW/kW	57.9F	124.1V	24.251kWh	2,657W	
11/22/20	12:55PM	18.500kWh	0.851kWh/kW	4,684W	4,800W	0.221kW/kW	57.9F	123.7V	24.030kWh	2,133W	
11/22/20	12:50PM	18.100kWh	0.833kWh/kW	4,459W	2,400W	0.110kW/kW	57.9F	123.3V	23.852kWh	2,036W

I looked through the grott log file and at 12:55, grott received a message and forwarded it on to pvoutput with a "OK 200: Added Status." Then a bunch of packages with "Data less than minimum record length" came through until 1:12 when the log file records two errors:

2020-11-22T18:12:02.127198043Z 	 - Growatt packet received:,
2020-11-22T18:12:02.127205693Z 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 39700), raddr=('47.91.67.66', 5279)>,
2020-11-22T18:12:02.127228855Z 	 - Data less then minimum record length, data not processed,
2020-11-22T18:12:04.130419084Z 	 - Grott connection error,
2020-11-22T18:12:04.130443943Z ,
2020-11-22T18:12:04.130450555Z 	 - Growatt packet received:,
2020-11-22T18:12:04.130506226Z 		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 5279), raddr=('192.168.1.146', 1029)>,
2020-11-22T18:12:04.130530564Z 	 - Data less then minimum record length, data not processed,
2020-11-22T18:12:04.130967051Z 	 - peer has disconnected,
2020-11-22T18:12:17.956728240Z 	 - ('192.168.1.146', 1029) has connected,
2020-11-22T18:12:40.466760606Z ,

I'll attached the grott log as well as the Growatt log from both inverters/systems.

[Grott log 2.5.6.txt](https://github.com/johanmeijer/grott/files/5580037/Grott.log.2.5.6.txt)
[System ID 80119 - DAA384307C history data - 2020-11-22_2020-11-22.xlsx](https://github.com/johanmeijer/grott/files/5580046/System.ID.80119.-.DAA384307C.history.data.-.2020-11-22_2020-11-22.xlsx)
[System ID 80120 - DAA384311B history data - 2020-11-22_2020-11-22.xlsx](https://github.com/johanmeijer/grott/files/5580047/System.ID.80120.-.DAA384311B.history.data.-.2020-11-22_2020-11-22.xlsx)

johanmeijer, 2020-11-23:
Hi Ben,

This seams a little bit complex but I try to explain what I see. 

From the grott log 18:00 (your time 1.00 pm I think). There does not seem to be any communication between grott/inverter and the growatt server any more. 

This is the last message (ping) replied by growatt: 

2020-11-22T17:57:01.272620453Z 	 - Growatt packet received:,
2020-11-22T17:57:01.272633136Z 		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.145', 5279), raddr=('192.168.1.146', 1029)>,
2020-11-22T17:57:01.272664666Z 	 - Data less then minimum record length, data not processed,

(Data less then minimum record length is something I do so no all records, like ping,  are processed as data records). 

From that momen we try from "our" site (grott/shinelink) to ping the growatt server. Thats is done till 18:12:02. At that moment a connection error is detected and a new connection is made. 

2020-11-22T18:12:04.130419084Z 	 - Grott connection error,
2020-11-22T18:12:04.130967051Z 	 - peer has disconnected,
2020-11-22T18:12:17.956728240Z 	 - ('192.168.1.146', 1029) has connected,

Then the negotiation between grott/shinelink starts again (eg theT055x03/T055x29 records) and at: 18:13:32 the first data record is processed and sent to pvoutput (I show the json message because al information is in there, I know you you only use PVOutout). 

2020-11-22T18:13:32.075308918Z 	 - MQTT jsonmsg: ,
2020-11-22T18:13:32.075736142Z 			 {"device": "DAA384307C", "time": "2020-11-22T13:13:32", "buffered": "no",,
2020-11-22T18:13:32.075773246Z 			 "values": {"pvstatus": 1, "pv1watt": 10140, "pv2watt": 5017, "pvpowerin":,
2020-11-22T18:13:32.075787553Z 			 20202, "pvpowerout": 19218, "pvfrequentie": 5999, "pvgridvoltage": 2466,,
2020-11-22T18:13:32.075799696Z 			 "pvenergytoday": 96, "pvenergytotal": 3650, "pv1voltage": 2798,,
2020-11-22T18:13:32.075811502Z 			 "pv2voltage": 2788, "pv1current": 34, "pv2current": 16, "pvtemperature":,
2020-11-22T18:13:32.075855472Z 			 251, "pvipmtemperature": 299}},

This the first inverter and for the other it follows (the T055204). These are actual records (buffered: no) direct from the shinelink. 
The shinelink does not put date/time in the record and date/time from the container is used. 
That seems to be oke. 

After a short while shinelink start sending the buffered records (it buffers the records if no connection is available). This seem to be working also. At 18:13:45 the first buffered record is processed (T055150,  50 = buffered record). 

2020-11-22T18:13:45.933674779Z 	 - MQTT jsonmsg: ,
2020-11-22T18:13:45.933925549Z 			 {"device": "DAA384307C", "time": "2020-11-22T14:06:38", "buffered": "yes",,
2020-11-22T18:13:45.933935662Z 			 "values": {"pvstatus": 1, "pv1watt": 9094, "pv2watt": 4588, "pvpowerin":,
2020-11-22T18:13:45.933942339Z 			 18194, "pvpowerout": 17831, "pvfrequentie": 5998, "pvgridvoltage": 2483,,
2020-11-22T18:13:45.933948612Z 			 "pvenergytoday": 99, "pvenergytotal": 3653, "pv1voltage": 2922,,
2020-11-22T18:13:45.933954687Z 			 "pv2voltage": 2900, "pv1current": 30, "pv2current": 15, "pvtemperature":,
2020-11-22T18:13:45.933960759Z 			 250, "pvipmtemperature": 297}},

You can recognized this by "buffered" :  "yes". The shinelink now puts a time in the data record! That is preverable because it is a record from the past. Now something strange happens:  the time in the datarecord seems to be a hour ahead. It should be 13:06:38 (or 1:06.38 PM) but is 14:06:38. It is sent to pvoutput with a date/time in the future (probably they are overwritten later that day by actual information). 

2020-11-22T18:13:45.933981460Z 	 - Grott send data to PVOutput systemid:  80119 for inverter:  DAA384307C,
2020-11-22T18:13:45.933997263Z 		 -  {'X-Pvoutput-Apikey': '1238f06db567269542cb95c2bccdfbf36b221944', 'X-Pvoutput-SystemId': '80119'},
2020-11-22T**18:13:45**.934003360Z 		 -  {'d': '20201122', 't': '**14:06**', 'v1': 9900, 'v2': 1783.1, 'v6': 248.3},
2020-11-22T18:13:46.235283528Z 	 - Grott PVOutput response: ,
2020-11-22T18:13:46.235323587Z 		 -  OK 200: Added Status,

My first conclusions: 

- Grott / shinelink seems to handle the stall/error in communication well. Data is buffered and sent out later. 
- The shinelink time differs 1 hour with the container time. I think the shinelink time is set by the growatt server at startup / initiation so maybe there is a time definition error in your growatt server or shinelink settings? Normally this is not a problem. It only affects the buffered records at this moment. 
- Every day at 1:00 pm the connection is lost / stalled. That is strange. Is something happening with the server (NAS) or with the internet connection at that time?

benpeart, 2020-11-23:
Thank you - that looks like a very accurate description of what is happening. 

I loaded the web interface for my local Shinelink and in "Datalogger Setting" I confirmed that the time is off by one hour as you predicted. Unfortunately, when I try to set it to the correct local time and click "save" it doesn't give any error but doesn't actually change the time either. I then signed in to server.growatt.com and checked my settings there and it was set to GMT-4. I updated that to be the correct GMT -5 and am hoping that it will push the correct time to the Shinelink. I think if I can get the time set correctly everywhere, I should be fine. Unfortunately, this is caused by us switching into and out of daylight savings time twice a year - hopefully I can find a simple solution or I'll just have to get them on the same timezone and leave it alone (and off by an hour for 1/2 the year).

As to why the connect is lost / stalled every day at 1:00 - I don't know. My internet connection in general doesn't show any loss of connectivity so I'm not sure why Shinelink isn't able to get through. Hopefully I'll get the dates corrected and then it won't really matter.

Thank you for digging into this and figuring out what was going on. From my limited experience, I think it is safe to assume the 2.2.6 image is working correctly including the new support for multiple inverters. Feel free to push out a new official image and close this issue.

benpeart, 2020-11-24:
Just thought I'd let you know that at some point during the night something I did yesterday finally got the time set correctly on my inverters/ShineLan box. The ShineLan device, Growatt and pvoutput are all reporting the correct time which should solve my 'missing' data problem. Thanks again for all your work on providing this tool!

johanmeijer, 2020-11-24:
I am glad the problem seems to be solved. I will promote the 2.2.6 version to the official image. 

Thank your for helping me with improving Grott. 

I think this issue can be closed.

closed 2020-11-25 by benpeart