johanmeijer/grott #30: grott crach with buffert data

harryverkooijen, 2020-12-08:
After an internet failure the inverter started tot issue buffet record. 
This time the time stamp is ok but the date is wrong. 

server.grotwatt.com refuses the record and eventually grott crashes.

I use gtime=server.

maby the check of an correct time record in buffert record also must include the date. the solision for time errors is good for real time values, but for buffert ones it is tricky, i think.
after restart, Grott dit not crach antmore, but buffert records are refused.

2020-12-08 09:14:22	stdout	requests.exceptions.ConnectionError: HTTPSConnectionPool(host='pvoutput.org', port=443): Max retries exceeded with url: /service/r2/addstatus.jsp (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f144eb410d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
2020-12-08 09:14:22	stdout	    raise ConnectionError(e, request=request)
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/requests/adapters.py", line 516, in send
2020-12-08 09:14:22	stdout	    r = adapter.send(request, **kwargs)
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/requests/sessions.py", line 643, in send
2020-12-08 09:14:22	stdout	    resp = self.send(prep, **send_kwargs)
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/requests/sessions.py", line 530, in request
2020-12-08 09:14:22	stdout	    return session.request(method=method, url=url, **kwargs)
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/requests/api.py", line 61, in request
2020-12-08 09:14:22	stdout	    return request('post', url, data=data, json=json, **kwargs)
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/requests/api.py", line 119, in post
2020-12-08 09:14:22	stdout	    reqret = requests.post(conf.pvurl, data = pvdata, headers = pvheader)
2020-12-08 09:14:22	stdout	  File "/app/grottdata.py", line 343, in procdata
2020-12-08 09:14:22	stdout	    procdata(conf,data)    
2020-12-08 09:14:22	stdout	  File "/app/grottproxy.py", line 165, in on_recv
2020-12-08 09:14:22	stdout	    self.on_recv(conf)
2020-12-08 09:14:22	stdout	  File "/app/grottproxy.py", line 85, in main
2020-12-08 09:14:22	stdout	    proxy.main(conf)
2020-12-08 09:14:22	stdout	  File "grott.py", line 35, in <module>
2020-12-08 09:14:22	stdout	Traceback (most recent call last):
2020-12-08 09:14:22	stdout	
2020-12-08 09:14:22	stdout	During handling of the above exception, another exception occurred:
2020-12-08 09:14:22	stdout	
2020-12-08 09:14:22	stdout	urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='pvoutput.org', port=443): Max retries exceeded with url: /service/r2/addstatus.jsp (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f144eb410d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
2020-12-08 09:14:22	stdout	    raise MaxRetryError(_pool, url, error or ResponseError(cause))
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/urllib3/util/retry.py", line 439, in increment
2020-12-08 09:14:22	stdout	    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/urllib3/connectionpool.py", line 727, in urlopen
2020-12-08 09:14:22	stdout	    timeout=timeout
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
2020-12-08 09:14:22	stdout	Traceback (most recent call last):
2020-12-08 09:14:22	stdout	
2020-12-08 09:14:22	stdout	During handling of the above exception, another exception occurred:
2020-12-08 09:14:22	stdout	
2020-12-08 09:14:22	stdout	urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7f144eb410d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution
2020-12-08 09:14:22	stdout	    self, "Failed to establish a new connection: %s" % e
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/urllib3/connection.py", line 172, in _new_conn
2020-12-08 09:14:22	stdout	    conn = self._new_conn()
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/urllib3/connection.py", line 309, in connect
2020-12-08 09:14:22	stdout	    conn.connect()
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/urllib3/connectionpool.py", line 978, in _validate_conn
2020-12-08 09:14:22	stdout	    self._validate_conn(conn)
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/urllib3/connectionpool.py", line 381, in _make_request
2020-12-08 09:14:22	stdout	    chunked=chunked,
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/urllib3/connectionpool.py", line 677, in urlopen
2020-12-08 09:14:22	stdout	Traceback (most recent call last):
2020-12-08 09:14:22	stdout	
2020-12-08 09:14:22	stdout	During handling of the above exception, another exception occurred:
2020-12-08 09:14:22	stdout	
2020-12-08 09:14:22	stdout	socket.gaierror: [Errno -3] Temporary failure in name resolution
2020-12-08 09:14:22	stdout	    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/socket.py", line 752, in getaddrinfo
2020-12-08 09:14:22	stdout	    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/urllib3/util/connection.py", line 61, in create_connection
2020-12-08 09:14:22	stdout	    (self._dns_host, self.port), self.timeout, **extra_kw
2020-12-08 09:14:22	stdout	  File "/usr/local/lib/python3.7/site-packages/urllib3/connection.py", line 160, in _new_conn
2020-12-08 09:14:22	stdout	Traceback (most recent call last):
2020-12-08 09:14:22	stdout	 -  {'d': '20010108', 't': '08:36', 'v1': 0, 'v2': 0.0, 'v6': 412.6}
2020-12-08 09:14:22	stdout	 -  {'X-Pvoutput-Apikey': '7df8101bae80b9b5e256340d3f96240448770597', 'X-Pvoutput-SystemId': '28371'}
2020-12-08 09:14:22	stdout	 - Grott send data to PVOutput : 
[... 111 lines cut ...]
2020-12-08 09:13:51	stdout	 00020006010101504a50433641323731583200000000000000000000000000000000000000004
2020-12-08 09:13:51	stdout	 - Growatt plain data:
2020-12-08 09:13:51	stdout	 - Growatt data decrypted V2
2020-12-08 09:13:51	stdout	 - decrypt :  True
2020-12-08 09:13:51	stdout	 - Record layout used :  T06NNNN
2020-12-08 09:13:51	stdout	 - layout   :  T060150
2020-12-08 09:13:51	stdout	 - Grott data record length 265
2020-12-08 09:13:51	stdout	 - Grott automatic protocol detection
2020-12-08 09:13:51	stdout	 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x3a\xfa
2020-12-08 09:13:51	stdout	 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
2020-12-08 09:13:51	stdout	 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x46\x63\x1f\x77\x61\x74\x74\x47\x72\x6f
2020-12-08 09:13:51	stdout	 \x47\x72\x6f\x7d\xb5\x74\x74\x51\x71\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
2020-12-08 09:13:51	stdout	 \x6f\x2e\x46\x64\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x64\x58\x61\x74\x74
2020-12-08 09:13:51	stdout	 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x9d\x74\x6b\x71\x4e\x74\x74\x47\x5f
2020-12-08 09:13:51	stdout	 \x74\x47\x72\x6f\x63\xf1\x74\x4f\x4f\xf8\x6f\xa3\x61\x74\x74\x47\x72\x6f\x77
2020-12-08 09:13:51	stdout	 \x72\x7f\x41\x61\x74\x74\x47\x72\x6f\x78\x9a\x74\x74\x47\x72\x6f\x77\x61\x74
2020-12-08 09:13:51	stdout	 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x67\xcf\x62\x4f\x77\x61\x74\x74\x47
2020-12-08 09:13:51	stdout	 \x74\x74\x6b\x72\x6f\x77\x61\x74\x74\x4b\x35\x6f\x77\x61\x74\x74\x47\x7e\x3a
2020-12-08 09:13:51	stdout	 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x60\x75\x7c\x4f\x51\x6f\x75\x61
2020-12-08 09:13:51	stdout	 \x27\x21\x24\x47\x35\x75\x43\x5f\x43\x50\x74\x74\x47\x72\x6f\x77\x61\x74\x74
2020-12-08 09:13:51	stdout	 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
2020-12-08 09:13:51	stdout	 \x00\x02\x00\x06\x01\x01\x01\x50\x0d\x22\x2c\x41\x20\x46\x43\x76\x2a\x5d\x77
2020-12-08 09:13:51	stdout	 - Growatt original Data:
2020-12-08 09:13:51	stdout	  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.3', 59606), raddr=('47.91.67.66', 5279)>
2020-12-08 09:13:51	stdout	 - Growatt packet received:

johanmeijer, 2020-12-08:
Hi Harry,

Your date/time stays challenging. I (Grott) test if it is a valid data/time and formally: 2001-01-08T08:35:00 is a valid date and time. It is not what we expected and not what PVOutput.org likes (to far back in time). 

PVOutput will refuse data that is to old. And apparently it also does not like a second record that soon after it blocks the first one (to much retries). 

I have to think how I can solve your problem. Maybe I will add a parm to bypass buffered data records processing totally. 
This might be the only option in your case. (g)time = server means it will always use server time for real time records. For buffered records it will use the date/time in the record if it is there and otherwise it will not be sent. 
Sending buffer records with the actual server time is not desirable I think. 

I can also test for a date  window. E.g. it should not be older then xx days (eg 90 like pvoutput does) and not in the future.  But I can not assure the date/time that is coming from the record (if it has a valid format) is the correct date. 

Do you have more information on the "crash" caused by the Growatt server communication? I do not see it here I think. If the Growatt server is also refusing the data and Grott crashed I have to look deeper into that (I do not know how to test it yet). 
  
Is it possible for you to try to set date/time via the growatt server. First you have to set your timezone right (probably already done during installation). This can be done energy/plantmanagement
[image]

And the set the date/time via inverter /settings: 

[image]

Key is growatt+date e.g.: growatt20201209 

Date/time change should be allowed by Gott. If not it might be necessary to start Grott with gblockcmd=False.

harryverkooijen, 2020-12-09:
I set time and time zone correct. gblockcmd=False and gtime=server I removed.

see wat happens.

I think that when gtime=server that it is better to skip buffered records altogether, every time stamp is possible wrong (in my case), so it is impossible to control it.

johanmeijer, 2020-12-09:
let's see what happens.

I did made so much changes to Grott that I had forgotten that their is also a parameter sendbuf (gsendbuf environmental). The default is sendbuf=True. And the buffered data will be sent if the date/time is valid. When you specify (g)sendbuf=False. Buffered data will not be sent at all.

harryverkooijen, 2020-12-09:
ok,

I wait a few days. When the time problem reocures I use time=server and sendbuf=False

harryverkooijen, 2020-12-13:
grott works fine, but after a few days;

2020-12-13 10:30:39 | stdout | - Grott values retrieved:
-- | -- | --
2020-12-13 10:30:39 | stdout | - no or no valid time/date found, grott server time will be used (buffer records not sent!)
2020-12-13 10:30:39 | stdout | - Grott data record date/time processing started

the time safety filter is working but I have the risk that incorrect time record slip throe your filter when the format is correct (but the time is wrong).
I going to use time=server and sendbuf=False to be sure.

I set the time in the inverter correct, that is not the issue.

Hopefully sometime we understand what the TS3 is doing with the time, for now it is a stable solution.

johanmeijer, 2020-12-14:
Ok. It is a pitty we  can not find the cause of your problem. I will keep trying to find a solution. Your data is part of my test set. Maybe I detect something in it in the future.

johanmeijer, 2020-12-18:
@harryverkooijen do you want to keep this issue open or can it be closed?

harryverkooijen, 2020-12-19:
with sendbuf=False no more problems arise, issue can be closed, It is an acceptable workaround.

johanmeijer, 2020-12-31:
Workaround available

closed 2020-12-31 by johanmeijer