johanmeijer/grott #31: How to configure grott.ini?

jeroenvanrensen, 2020-12-30:
Hello,

I'm trying to configure my `grott.ini` file. My IP Address is `192.168.178.53` and this is my configuration file:

```
[MQTT]
#mqtt parameters definitions
#be aware nomqtt = True means no MQTT processing will be done!!!!!!

#nomqtt = False
ip = 192.168.178.53
port = 1883
#topic= energy/growatt
auth = True
user = admin
password = admin
```

But when I run `python3 grott.py -v`, I get this output:

```
Grott Growatt logging monitor : 2.2.6

Grott Command line parameters processed:
        verbose:         True
        config file:     grott.ini
        output file:     <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
        nomqtt:          False
        inverterid:      automatic
        pvoutput:        False
        blockcmd:        False
        noipf:           False

Grott process configuration file

Grott process environmental variables

Grott override settings if set in commandline

Grott process json layout files

Grott layout records loaded
T02NNNN  :
{'decrypt': 'False', 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T05NNNN  :
{'decrypt': 'True', 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 122, 'pvfrequentie': 130, 'pvgridvoltage': 134, 'pvenergytoday': 182, 'pvenergytotal': 190, 'pvtemperature': 206, 'pvipmtemperature': 242}
T06NNNN  :
{'decrypt': 'True', 'pvserial': 76, 'date': 136, 'pvstatus': 158, 'pvpowerin': 162, 'pv1voltage': 170, 'pv1current': 174, 'pv1watt': 178, 'pv2voltage': 186, 'pv2current': 190, 'pv2watt': 194, 'pvpowerout': 202, 'pvfrequentie': 210, 'pvgridvoltage': 214, 'pvenergytoday': 262, 'pvenergytotal': 270, 'pvtemperature': 286, 'pvipmtemperature': 322}
T05NNNNX  :
{'decrypt': 'True', 'pvserial': 36, 'date': 56, 'pvstatus': 78, 'pvpowerin': 82, 'pv1voltage': 90, 'pv1current': 94, 'pv1watt': 98, 'pv2voltage': 106, 'pv2current': 110, 'pv2watt': 114, 'pvpowerout': 170, 'pvfrequentie': 178, 'pvgridvoltage': 182, 'pvenergytoday': 274, 'pvenergytotal': 282, 'pvtemperature': 450, 'pvipmtemperature': 454}
T06NNNNX  :
{'decrypt': 'True', 'pvserial': 76, 'date': 126, 'pvstatus': 158, 'pvpowerin': 162, 'pv1voltage': 170, 'pv1current': 174, 'pv1watt': 178, 'pv2voltage': 186, 'pv2current': 190, 'pv2watt': 194, 'pvpowerout': 250, 'pvfrequentie': 258, 'pvgridvoltage': 262, 'pvenergytoday': 354, 'pvenergytotal': 362, 'pvtemperature': 530, 'pvipmtemperature': 534}

Grott external record whitelist 'recwl.txt' not found

Grott records whitelisted :  {'5050', '0116', '5150', '5019', '5016', '5103', '5004', '0104', '5119', '5003', '0119', '0150', '5116', '0103', '5104'}

Grott settings:

_Generic:
        version:         2.2.6
        verbose:         True
        trace:           False
        config file:     grott.ini
        minrecl:         100
        decrypt:         True
        compat:          False
        blockcmd:        False
        noipf:           False
        time:            auto
        sendbuf:         True
        valueoffset:     6
        offset:          6
        inverterid:      automatic
        mode:            proxy
        grottip          default
        grottport        5279
_MQTT:
        nomqtt           False
        mqttip:          192.168.178.53
        mqttport:        1883
        mqtttopic:       energy/growatt
        mqtttauth:       True
        mqttuser:        admin
        mqttpsw:         admin
_Growatt server:
        growattip:       47.91.67.66
        growattport:     5279
_PVOutput:
        pvoutput:        False
        pvurl:           https://pvoutput.org/service/r2/addstatus.jsp
        pvapikey:        yourapikey
        pvinverters:     1
        pvsystemid:      systemid1

Grott proxy mode started
Hostname : DESKTOP-VS6P90V
IP :  192.168.178.234 , port :  5279
```

And it does not look like the example output as [described here](https://github.com/johanmeijer/grott/wiki/MQTT-JSON-Message-format).

So what am I doing wrong?

Thank you! Jeroen

johanmeijer, 2020-12-30:
At first sight It doesnot look wrong to me. Did you rerouted the data from the inverter?

Did you saw the first time installation description in the wiki?

https://github.com/johanmeijer/grott/wiki/@-First-time-installation

jeroenvanrensen, 2020-12-31:
@johanmeijer Yes I saw that article and it really helped.

You say it looks good, but where can I find the data?

Thank you! Jeroen

johanmeijer, 2020-12-31:
Hi Jeroen. The data can be written to 3 places: 

- MQTT (to be processed with nodered, or passed to home automation tooling like domoticz, openhab or home assistent)
- PVOutput.org 
- Influxdb (from version 2.3.x) (to save and process the data by e.g. grafana)  

You have to do something additional to use the data,

jeroenvanrensen, 2020-12-31:
@johanmeijer Ok thanks for your answer. I would like to send an HTTP request with the data. Is that in some way possible?

johanmeijer, 2020-12-31:
Not yet directly. It is always possible via MQTT and nodered (ar an other program that can read message from the MQTT queue and translate it to a http/request, like telegraf) but then you have to learn (and install) other things. 

Where do you want to sent your data. Do you have an example of the http request (put/get). 

I sent a json string to mqtt and influxdb. It should not be very difficult to send a http request with the json string. The complexity is always in the security  and that kind of things. I should be a very generic request that can be used for different use cases otherwise it will be hard to support.

jeroenvanrensen, 2021-01-01:
@johanmeijer I have set up this endpoint:

```
192.168.178.192:83/grott
```

It is a post endpoint.

How can I send the data to that endpoint?

Thank you! Jeroen

johanmeijer, 2021-01-02:
@jeroenvanrensen as described above at this moment it is not an option to write directly to an http end point. 

I will open an issue for adding HTTP request support to Grott. 

At this moment I think of an add on that any one can add their own code for additional processing (an exit point in grott). We can then create an example with a http request.

jeroenvanrensen, 2021-01-02:
@johanmeijer Yeah that sounds smart. Thanks for your help, I will have a look at the three options you mentioned. Thank you very much!

jeroenvanrensen, 2021-01-02:
@johanmeijer I have one last question: Do I have to configure something in the `[Generic]` and `[Growatt]` sections of `grott.ini`, and if yes, what? Thank you!

johanmeijer, 2021-01-02:
@jeroenvanrensen I tried to describe the configuration of grott in: 

https://github.com/johanmeijer/grott/wiki/Grott-Configuration

If you want to use grott only for mqtt and mqtt is running at the same machine then you donot have to specify anything in grott.ini.

johanmeijer, 2021-01-09:
@jeroenvanrensen I created a new version off grott (2.4.0) that support extensions.

With this you can add your own data processing to Grott. I created an example extension that writes(Http put) the Grott data to a http server. 

You can found 2.4.0 in the 2.4 branche. 

Short how to use: (I will document at in the wiki soon). 

Copy grottext.py from the example directory into the Grott main directory (change the coding if you want). 

Add in the grott ini: 

    [extension]
    extension = True
    extname = grottext
    extvar = {"ip": "<your server url>", "port": "<your server port>"}

Please let me know if this is where you looking for. 

Be aware: It is under development and might not work as expected.

johanmeijer, 2021-01-13:
@jeroenvanrensen is this where you were looking for?

johanmeijer, 2021-03-27:
Extension is available and works and can be used for sending data to a server. 

See: https://github.com/johanmeijer/grott/wiki/Extensions

This issue will be closed.

closed 2021-03-27 by johanmeijer