johanmeijer/grott #34: Load consumption values are also recorded?

warlockpt, 2021-01-06:
Hi,
I have a SPH5000 inverter installed, including also a meter in my main electrical board, which is also measuring by load consumption. I'm able to see the measured load consumption values in growatt server.
Is grott able to also record those values from load consumption or only the ones regarding energy production?

Thanks

johanmeijer, 2021-01-06:
At this moment only the energy production is logged. 

I have not seen data from a hybrid inverter yet so I do not now if the information is in it (and I can find it), 

What are you using for network interface (Shinelan, Shinewife or shinelink)? Is the energy production sent to the growatt website (can you see it there)?

We can try to find the information. I will need a data record (logged by Grott) and indication of the load / production at that moment so I can validate the values found. 

It will not be easy to find. There will be probably around 1500 value's in the data record and I have a not a complete (most actual) layout description of the record.

warlockpt, 2021-01-06:
I'm using Shine Wifi.
Both the energy production and energy consumption data are sent to growatt website and I can see it there... at any given moment, I can see how much kW I'm producing, consuming and/or injecting in the grid.
This is an example (during the night, so production is zero):
[Example](https://pasteboard.co/JIqHilq.png)

When you say "data record (logged by Grott)", you mean by sniffing or redirecting the traffic?

johanmeijer, 2021-01-07:
For both sniffing and the proxy functionality of Grott yuo need to redirect the growatt data via the sever Grott is running on. 

Setting up the proxy is easier because you do have to make  network changes (IP - Forwarding). It is described here: 

https://github.com/johanmeijer/grott/wiki/Rerouting-Growatt-Wifi-TCPIP-data-via-your-Grott-Server

What  need is the growatt original data, like this: 

Jan 07 16:37:21 raspitest grott[5739]:          - Growatt original Data:
Jan 07 16:37:21 raspitest grott[5739]:                  \x00\x5d\x00\x06\x01\x01\x01\x04\x0d\x22\x2c\x45\x59\x45\x4c\x74\x41\x2d\x77
Jan 07 16:37:21 raspitest grott[5739]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Jan 07 16:37:21 raspitest grott[5739]:                  \x3e\x3a\x23\x46\x4c\x75\x41\x5d\x41\x50\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Jan 07 16:37:21 raspitest grott[5739]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x75\x73\x57\x57\x78\x75\x61
Jan 07 16:37:21 raspitest grott[5739]:                  \x74\x74\x6b\x72\x6f\x77\x61\x74\x74\x40\xe4\x6f\x77\x61\x74\x74\x47\x72\x6f
Jan 07 16:37:21 raspitest grott[5739]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x67\xc1\x7b\x5e\x77\x61\x74\x74\x47
Jan 07 16:37:21 raspitest grott[5739]:                  \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Jan 07 16:37:21 raspitest grott[5739]:                  \x74\x42\x72\x6f\xd2\x3b\x77\xb3\x96\x8a\x6f\xbb\x61\x74\x74\x47\x72\x6f\x77
Jan 07 16:37:21 raspitest grott[5739]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x75\xc9\x77\x61\x74\x74\x47\x5f
Jan 07 16:37:21 raspitest grott[5739]:                  \x6f\x2e\x2f\x54\x74\x47\x72\x6f\x77\x61\x74\x71\x47\x72\xc2\xcb\x61\x74\x74
Jan 07 16:37:21 raspitest grott[5739]:                  \x47\x72\x6f\x77\x61\x74\x74\xea\xce\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Jan 07 16:37:21 raspitest grott[5739]:                  \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x78\x6f
Jan 07 16:37:21 raspitest grott[5739]:                  \x76\x68\x2e\x70\xd9\x76\xc7\x77\x75\x7d\xca\x07\x06\x6f\x77\x6f\x15\x74\x47
Jan 07 16:37:21 raspitest grott[5739]:                  \x7f\xf6\x77\x61\x74\x76\x47\x62\x6f\x77\x61\x74\x74\x47\x72\x6f\x1d\xba
Jan 07 16:37:21 raspitest grott[5739]:          - Grott automatic protocol detection
Jan 07 16:37:21 raspitest grott[5739]:          - Grott data record length 265
Jan 07 16:37:21 raspitest grott[5739]:          - layout   :  T060104
Jan 07 16:37:21 raspitest grott[5739]:          - Record layout used :  T06NNNN
Jan 07 16:37:21 raspitest grott[5739]:          - decrypt :  True
Jan 07 16:37:21 raspitest grott[5739]:          - Growatt data decrypted V2
Jan 07 16:37:21 raspitest grott[5739]:          - Growatt plain data:
Jan 07 16:37:21 raspitest grott[5739]:                  005d0006010101044a50433238313833334200000000000000000000000000000000000000005
Jan 07 16:37:21 raspitest grott[5739]:                  14d42323832333236310000000000000000000000000000000000000000150107102517020000
Jan 07 16:37:21 raspitest grott[5739]:                  002c0000000000000796000000000000000000000000000000000000138609310000000000000
Jan 07 16:37:21 raspitest grott[5739]:                  0000000000000000000000000000000000000050000a55a03c7d1f800cc000000000000000000
Jan 07 16:37:21 raspitest grott[5739]:                  00000000000000000007a600000000002d00594e2000000000000000050000adbc00000000000
Jan 07 16:37:21 raspitest grott[5739]:                  000000000adbc0000000000000000000000000000000000000000000000000000000a0001095a
Jan 07 16:37:21 raspitest grott[5739]:                  049e04a8001409be407400000e6100000d9900000002001000000000000000006adb
Jan 07 16:37:21 raspitest grott[5739]:          - Growatt new layout processing
Jan 07 16:37:21 raspitest grott[5739]:                  - decrypt       :  True
Jan 07 16:37:21 raspitest grott[5739]:                  - offset        :  6
Jan 07 16:37:21 raspitest grott[5739]:                  - record layout :  T06NNNN

The original data is in principle enough. I can create the plain data.

johanmeijer, 2021-03-27:
Issue closed because of lack of reaction.

warlockpt, 2021-04-26:
Hi,
Sorry, but been unable to handle this before... 
Meanwhile, I've just installed Grott and getting data... obviously, the data I'm getting in MQTT is not correct. as I believe the record definition isn't correct.
Any pointers on how the record definition can be adjusted?

johanmeijer, 2021-04-26:
Can you send me the data as described above. Then I can determine what info we need to create a layout file. 

In the meanwhile we are able to tackle the SPF data record (off grid converter with battery). I think we will able to create something for your inverter as wel. 
 
I do not have a record description document for SPH yet.

warlockpt, 2021-04-27:
Here is the logs/growatt's export data.
I took samples at two moments: One where I was importing from grid (as solar production was low) and another one where I was exporting to grid (as production was higher).
I still haven't adjusted Grott's TZ, so grott's log is one hour behind my own TZ.
[grott_log_20210427T091206.txt](https://github.com/johanmeijer/grott/files/6383565/grott_log_20210427T091206.txt)
[grott_log_20210427T102700.txt](https://github.com/johanmeijer/grott/files/6383566/grott_log_20210427T102700.txt)
[NRCIA14015 mix data - 2021-04-27_2021-04-27.xls](https://github.com/johanmeijer/grott/files/6383567/NRCIA14015.mix.data.-.2021-04-27_2021-04-27.xls)

johanmeijer, 2021-04-27:
@warlockpt thanks. I will try to decipher it!

johanmeijer, 2021-04-28:
@warlockpt progress status. I think I manage to decode the record. See information below. 

I will build a test  RPI docker container for you (probably this weekend; lack of spare time).

[image]

warlockpt, 2021-04-28:
Looks good. 
I imagine that example is when production is higher than load, as I see pactouser with 0 (and pactogrid with 640).
Great work!!

johanmeijer, 2021-05-01:
Yes. This is the other one: 

[image]

johanmeijer, 2021-05-01:
@warlockpt 

I created a new docker  container for you to test with on docker hub: ledidobe/grottrpi:2.5.4sph

If everything works fine I will create a new grott version (2.6.x) with sph and spf support. 

To map the sph record layout you can find a description this document (not totally correct/complete but usable).  

http://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf 
You can read here growatt's description of the fields. 

Your record is a combination of the "main" record description started at page 25 (values no: 0 - 124) and "storage power" started  at page 37 (value no 1000- 1066). 

Let me know if it works!

johanmeijer, 2021-05-01:
If you want to test the SPH layout it with a "plain"  version of Grott please copy the t060104sph.json file from the example/record layout directory into the main Grott directory. 

After a restart of Grott the new layout is loaded.

warlockpt, 2021-05-01:
Tks a lot.
I plan to install the new docker tomorrow and will let you know how it goes

johanmeijer, 2021-05-05:
@warlockpt did you already test it?

warlockpt, 2021-05-07:
Yes, I was finally able to install it. I've just stopped the "normal" container and installed the updated one.
Things seem to be running smoothly.
I believe there are some fields in the mqtt record which seem not to be populated (e.g., #pactogrids,#pactogridt) or duplicated (e.g. plocaloadr, plocaloadtot).
Appart from it, tumbs up and 5-star work !!!

johanmeijer, 2021-05-08:
Nice!

The values with # in front I had my doubt about. They are part of the record layout description but might indeed have no added value. For know I tried to map all fields that were shown in the description. So there might be doubles or not interesting fields in it. 

Thanks for testing and helping to make Gott also usable for the new inverters!

johanmeijer, 2021-06-01:
@warlockpt can I close this issue?

warlockpt, 2021-06-01:
@johanmeijer , yeah, I've closed it.
Grott has been working without problems :)

closed 2021-06-01 by warlockpt

johanmeijer, 2021-06-01:
Thans!