johanmeijer/grott #35: Home Assistant yaml files to expose the Grott entities

jerommeke-bos, 2021-01-10:
Dear Johan,

Thank you for your great work.

Attached you'll find a zip file that contains yaml 
files that will expose all the entities from Grott to HA, inculding some dummy sensors for the type of the inverter and the type and serial number of the datalogger (These should be set by hand):

- sensors_grott_eng.yaml (english version)
- sensors_grott_eng.yaml (dutch version)

The only thing that has to be done in HA is:

- Copy the files to the homeassistant folder, of your Home Assistant installation (the same folder where the configuration.yaml file is located).
- Add the line "sensor: !include sensors_growatt_eng.yaml" or "sensor: !include sensors_growatt_nl.yaml" to the configuration.yaml (Multiple "sensor:" includes are allowed in the configuration.yaml for those that have already one or more sensor: includes in the configuration.yaml like myself) .
- Restart HA and the new sensors will be available, you'll find them on within the entities list (search for growatt).

All has been tested with the latest docker version of Grott , a MIN 4200TL-XE inverter and ShineLink X datalogger.

Best regards,

Jeroen Bos

[sensors_growatt.zip](https://github.com/johanmeijer/grott/files/5793135/sensors_growatt.zip)

johanmeijer, 2021-01-11:
Hello Jeroen. 

Thank you for the input. I will add the files to github repository and will add the information to the Wiki.

johanmeijer, 2021-01-11:
@jerommeke-bos 

I add the sensor yaml file to the repository
https://github.com/johanmeijer/grott/tree/master/examples/Home%20Assistent

The wiki is updated: 
https://github.com/johanmeijer/grott/wiki/Home-Assistant-configuration-examples

johanmeijer, 2021-01-13:
@jerommeke-bos I think this issue can be closed?

jerommeke-bos, 2021-01-13:
Thank you, Johan

closed 2021-01-13 by jerommeke-bos