johanmeijer/grott #36: Version 2.4 supports self created extensions. Submit your contribution!

johanmeijer, 2021-01-11:
I created a new version: 2.4.0 that support extensions.

With this you can add your own data processing to Grott. 

I created an example extension that writes(HTTP put) the Grott data to a http server. 

You can be found 2.4.0 in the 2.4 branche. 

Short how to use: 

Copy grottext.py from the github example directory into the Grott main directory (change the coding if you want). 

Add in the grott ini: 

    [extension]
    extension = True
    extname = grottext
    extvar = {"ip": "<your server url>", "port": "<your server port>"}

In the extension the following data is available: 

- Grott MQTT JSON containing all output variables that are available within Grott
  - you can print(jsonmsg) to see which variables are available
- Growatt Raw data record. The unscrambled data record that is send by the inverter to the Growatt server
  - you can print(data) to see which variables are available
- Grott configuration information
  - you can print(dir(conf)) to see which config variables are available
  - e.g. conf.verbose variable to indicate if verbose is enabled or not. 
- It is also possible to add your own variables to the config file. In the example: 
  -  extvar = {"ip": "<your server url>", "port": "<your server port>"} 
  -  this creates 2 variables: conf.extvar["ip"] and conf.extvar["port"] 

Be aware: It is under development and might not work as expected. 

I am really curious if this is something that is going to be used. I like to see your extensions and will add them to a  sample directory in the github repository.

johanmeijer, 2021-01-13:
Version 2.4.0 is promoted to the master branche and general available.

wdonker, 2021-06-14:
Well, it looks like I am the first to submit my self-created extension.
It is meant to send the Growatt data to Domoticz by MQTT in the Domoticz-specific format on topic domoticz/in. So no node-red translation needs to be done. My extension supports 2 Domoticz virtual devices, one to show the status (on/off) of the solar panels and another to keep track of the energy delivered.
At first I got some strange energy values now and then but then I saw you check on pvstatus 257 in grottdata.py. Maybe this check should also be applied before starting the extension logic. I know added it to the extension code itself.

The extension is now running in my setup for some weeks without any error. Thank you for providing this nice way to get rid of the Growatt server dependency!

[grott extension.zip](https://github.com/johanmeijer/grott/files/6650639/grott.extension.zip)

johanmeijer, 2021-06-15:
Nice! 

Can I add the extension to the repository and describe it in the wiki?

I do not think it would be a good idea to filter the status  values. In the past I thought that only 0 or 1was possible. With the new inverters also supporting grid monitoring and batteries there are lot more status values possible. Altough I still do not know what status 257 means ;)

wdonker, 2021-06-16:
> Can I add the extension to the repository and describe it in the wiki?

Yes, of course.

> I do not think it would be a good idea to filter the status values.

No problem, it works this way too.

mvank, 2021-06-17:
Hi @johanmeijer, @wdonker,

maybe dump question, but what is the benefit of the extension?
I am using the example file "sensors_growatt_nl.yaml" within HA, which is working fine!

what options will bring the extensions?

wdonker, 2021-06-18:
Well, the yaml is for use with HA, my extension is for use with Domoticz.

johanmeijer, 2021-06-25:
@mvank I created the possibility of extension to enable other use of Grott than I have planned until know (MQTT, PVOutput, InfluxDB). 

To sent an other type of message to Domoticz is an good example (I use Nodered to convert the MQTT message but that is more complex). 

Somebody else is using it to send a HTTP request (API-call) to his own server (I created an example for that).

If Grott works fine for you don't have to create  a extension.

mvank, 2021-06-25:
@johanmeijer I only noticed difference in pvoutput and growatt self. Just reached the 1000KWh according growatt, but in pvoutput this s is 6KWh less. 

Also in pvoutput noticed that averages are show as rounded values

[image]

johanmeijer, 2021-06-25:
Hmm I will have a look at it. 

Will take some some time while I am going on holidays and it will not be apriciated if I am going to put a lot of time in Grott.

Maybe it is better to open another issue for this. Let keep things seperated. Which value do you use within Grott? There is difference  between the power the solar panels generate and what is delivered to the grid (lost in the inverer).

mvank, 2021-06-26:
Hi @johanmeijer,

I will open other issue! 
I appreciate the work what you have done and still do!
Enjoy your holiday!

pierstitus, 2022-01-11:
I just made a pull request for a CSV export extension, #91.

johanmeijer, 2022-01-27:
@pierstitus: I have merged the pull request and will update the documentation. Thank you very much for this nice add on. 
Grott is now able to export the data in CSV files! 

For more details see closed pull request #91.

johanmeijer, 2023-08-13:
Closed this one to clean-up the Issue list. Extensions are being used and available for CSV (excel) files creation, Home Extension automatic configuration and simple HTTP request communication. 

You are always welcome to create new extensions and share them with Grott community.

closed 2023-08-13 by johanmeijer