johanmeijer/grott #37: grott as home-assistant add-on?

wjansenw, 2021-01-28:
HI @johanmeijer ,

Thanks for this script! Works nicely here. 
Would it be possible to run this as an home-assistant add-on? Should be doable I guess, as you already have it running in docker? But I have no idea how to do this...

(and removing the need to send the data to growatt server would be nice as well)

Tnx,
Wim

wjansenw, 2021-01-28:
That went pretty OK.
https://developers.home-assistant.io/docs/add-ons/tutorial

* created a folder grott
* created a simple config file; make sure to expose the port 5279 (cf. above example, but change port 8000 to port 5279)
* put the grott-files in that same folder

Install & start the addon. Make sure to point to the correct IP.

Seems to work. Ideally, config params to be configurable via supervisor, but that is for later...

Now, move my virtualbox that is hosting home assistant to a raspberry (which I do not have yet).

johanmeijer, 2021-01-28:
That is nice! 

I am going to test it myself(will start with a RPI) and add it to the wiki. 

I do understand the requirement for removing the need of sending the data to the growatt Server but then Grott has to act as the Growatt server. That is a little bit complex to program (for now). But it stays on the wish-list. 

To have some kind in between solution I add the option to block the commands coming from Growatt and protect your inverter. see: https://github.com/johanmeijer/grott/wiki/Command-blocking-(Protect-your-inverter) 
The solar performance data (nothing confidential I think) is still written to the Growatt server but they can not control your inverter (reset, adjust settings values etc) from outside.

wjansenw, 2021-02-01:
FYI - created this repo. 
https://github.com/wjansenw/home-assistant-add-on

You can add the repo in your home assistant add-on store, and install Grott as an add-on in home assistant.
It will pull the latest files from your repository, and launch it.

It uses the environment variables, and they can be set via the configuration of the add-on. Some options have already been set.

wjansenw, 2021-02-01:
I noticed now that I'm not yet exposing all options as environment variables. But the idea is there, and just needs some small changes.

johanmeijer, 2021-02-02:
Good work! Thanks. 

I will update the wiki and add a link to your github (if it is okay with you).

johanmeijer, 2021-02-04:
Closed by accident.

johanmeijer, 2021-02-13:
I add a link to your github in the Grott home-assistent wiki page. 

I think this issue can be closed.

closed 2021-03-27 by johanmeijer

charliesjc, 2021-04-07:
I've setup grott as a home-assistant add-on and it seems to be working. It is definitely proxying my inverter data and I can see that in the logs and the growatt server is still showing everything. I can't for the life of me get it to work with MQTT though. I keep seeing "MQTT connection refused by target" in the grott-addon log file. I have tested the broker from the MQTT integration and send it packets and it works. The MQTT logs also show the connection and username from the MQTT integration. But using grott it just doesn't work (also doesn't ever show the username I've setup for grott in the log). I have obviously restarted it to troubleshoot (grott and the HA server) and it hasn't helped.

I did notice some weird things though in the grott-addon config if you switch from UI to YAML edit, the values selected in the UI don't necessarily align with what's in the YAML.

Any ideas?

johanmeijer, 2021-04-08:
@wjansenw can you help with this one?

charliesjc, 2021-04-11:
An update on this: I ran some tests using `mosquitto_pub` and found that any reference to localhost, 127.0.0.1, or hassio (I tried this because it's in the hosts file) results in connection refused or address not available. If I use homeassistant in the `mosquitto_pub` command then it works. But if I use homeassistant in the grott config it shows in the log that it is still using localhost and then the connection is refused. What DOES work though is using the IP of the instance on my local network which in this case is 192.168.1.124. If I use that the messages are sent just fine.

I'm using a VM with the HA OS, that likely has something to do with it. I looked in the hosts file and didn't see anything that stands out as odd.

I can only imagine that the built-in HA MQTT integration knows about this network quirk and acts accordingly.

johanmeijer, 2021-04-11:
Does it works now?

I see in the documentation of @wjansenw (https://github.com/wjansenw/home-assistant-add-on/blob/main/grott/DOCS.md) that you can specify mqtt ip / port etc during HA add-on installation. Maybe you can specify your IP address (in stead of 127.0.0.0)?

charliesjc, 2021-04-11:
Yes it works now. I was just very surprised that localhost and 127.0.0.1 just don't work. It works when I specify the VM's IP.

blackbird0589, 2022-08-29:
Hi @all,
I have a problem with the installation-process. The Version of @wjansenw can be installed sucessfully, but if i want to start, there is still red light and an error message "s6-overlay-suexec: fatal: can only run as pid 1".
Second try was to generate a local add-on as descripted with the newest version of @johanmeijer - if i want to start installation in HA-addon store, only error message "could not be installed".
Could you please give me some hints what's wrong?!

Sdahl1234, 2022-08-29:
You can try this fork
https://github.com/Emacee/grott-home-assistant-add-on

blackbird0589, 2022-08-30:
Hi Sdahl1234,
thanks for your help! That worked partially for me. Installation and forwarding to growatt server works perfekt. In my opinion handshake between grott and MQTT works fine. Also I've integrated sensors via "sensors_grott_eng.yaml" coming out of wiki.

Last problem is, that i can found sensors in HA, but there are no values (no update of the sensors).
Do you know what problem this could be?!

blackbird0589, 2022-09-24:
> Hi Sdahl1234, thanks for your help! That worked partially for me. Installation and forwarding to growatt server works perfekt. In my opinion handshake between grott and MQTT works fine. Also I've integrated sensors via "sensors_grott_eng.yaml" coming out of wiki.
> 
> Last problem is, that i can found sensors in HA, but there are no values (no update of the sensors). Do you know what problem this could be?!
> 
> Edit: I've determined, that this one is v 2.3.1 - I'm using a Growatt SPF 5000 ES Offgrid Inverter. Maybe this is the problem with the sensors. How can I update to the newest Version?

@Sdahl1234 do you have some input to solve my problem?