johanmeijer/grott #38: date not processed correctly when using buffered data

wjansenw, 2021-02-05:
HI @johanmeijer ,
I see that if grott has not been running, it will send buffered data --- that is OK.

However, it looks like it does not process the data field correctly. It sends the current date/time, in stead of the buffered date/time.

It appears correct on the growatt servers, but not in other tools.

I see in the logs:
```
- Grott data record date/time processing started
- No or no valid date found
```

An example of data block:
```
                 002a0006024151504e41433541323731414800000000000000000000000000000000000000004
                 445463141323130393300000000000000000000000000000000000000001502050a1039020bb8
                 0c3400010000049e0a0d00020000028109cb00020000021d00000000000000000000000000000
                 0000000000000000000000005b3138709350008000005be000000000000000000000000000000
                 000935000000000000000000000000000000000052087d000000010000168a000015d00000000
                 000000b1d0000000000000ab30000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000fffa0000000000000000000000db00db0
                 0000000013b0e2100004e20000300000000000000000000000000000000000000000000000000
                 00003c0000000000000000000000000000000000000c350cb1000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000002710000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 0000000000003ac
```

Regards,
Wim

johanmeijer, 2021-02-06:
Hi Wim,

You found a bug! I was looking at the wrong offset for date in your record layout. 

I published 2.4.1 (only grottconf.py has changed) which should fix it.  

Date processing was wrong for both buffered and  non-buffered records. For non-buffered record you do not notice this while the date in the record is almost the same as the actual date. 

You should see a message like this now: 

     Grott data record date/time processing started 
     date-time:  2021-02-06T16:49:38

johanmeijer, 2021-02-06:
@wjansenw hopefully it is fixed now

wjansenw, 2021-02-07:
Tnx for the fix. Will test soon.

wjansenw, 2021-02-08:
HI @johanmeijer - difficult to test with the amount of snow on the roof - but one data point seems to indicate the processing does not work as it should:

```
         - Growatt plain data:
                 00380006024151044e41433541323731414800000000000000000000000000000000000000004
                 44546314132313039330000000000000000000000000000000000000000000000000000020bb8
                 0c340000000000000aa20000000000000a9f00000000000000000000000000000000000000000
                 000000000000000000000000000138a091f000000000000000000000000000000000000000000
                 00091f000000000000000000000000000000000053fc2c00000000000016db0000161e0000000
                 000000b450000000000000ad90000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000fffa00000000000000000000006800680
                 000000000d30a9700004e20000000000000000000000000000000000000000000000000000000
                 0000090000000000000000000000000000000000000c350cb1000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000002710000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 000000000008fe4
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T06NNNNX

         - Grott data record date/time processing started
         - no or no valid time/date found, grott server time will be used (buffer records not sent!)
```

johanmeijer, 2021-02-08:
@wjansenw 

Hmm. In your case it seems that  the unbuffered records does not contain any date/time information. 

This is probably not a problem while the buffered records seems to contain the date (what is really essentiental otherwise it would record with the wrong date/time information). 

How does grott detect the difference between buffered and unbuffered: 

In the header of the record (the first 8 bytes eg in your last record: 0038000602415104)  the type of record is specified by the 
bytes: 
4: 06 (means encrytped and specifies record start offset)
7: 51 (give more information on inverter type / network interface) 
8: 04 (record content: 04 =  normal record, 50 =  buffered) 

This is why you see record type t065104x (x means the extend version, there is a normal and an extended version). 

With a record type t065104x the date starts at position 68.

In your first record (065150) the date/time is:  1502050a1039. The format is yymmddhhmmss in hex, so 15 = 21 (date/time  of this record is 2021-02-05T10:16:57 ).  

In your second record (065104) the date/time at that position is: 000000000000. That is not a valid value and Grott uses the current server time. 

It always surprises me how Growatt handles date/time. In my case it is always specified. In your case it seems only to be there with buffered records. In the older configurations it is never specified (but there are also no buffered records) and I have seen one configuration where the data/time was crippled after a while. 

I like to see if your buffered records now give the right date/time information. I do understand if you do not want to test that but it would be nice to know.

wjansenw, 2021-02-08:
i'll check once the snow is gone... it's not providing any power right now it seems...

johanmeijer, 2021-03-27:
@wjansenw is it oke? Can we close this one?

wjansenw, 2021-03-27:
Yes, it is working OK now.

closed 2021-03-27 by johanmeijer