johanmeijer/grott #40: integer vs. float in data

Jeebeevee, 2021-03-09:
I was trying to get the data into grafana (with influxdb) but I did not get the correct data (i was thinking)
after a good time seaching i found that the data was saved as integer and not as float. 
This gives the same data but multiply bij 10.

When i look at the log from Grott i see the correct data from the inverter. 
in de jsonmsg for MQTT and influxdb is see the changed data.

```
         - Grott values retrieved:
                 - pvserial:          XXXXXXXXXXXX
                 - pvstatus:          1
                 - pvpowerin:         58.5
                 - pvpowerout:        57.5
                 - pvenergytoday:     1.5
                 - pvenergytotal:     3236.2
                 - pv1watt:           30.7
                 - pv2watt:           27.8
                 - pvfrequentie:      49.98
                 - pvgridvoltage:     233.3
                 - pv1voltage:        106.4
                 - pv1current:        0.2
                 - pv2voltage:        135.1
                 - pv2current:        0.2
                 - pvtemperature:     24.8
                 - pvipmtemperature:  30.3
         - MQTT jsonmsg: 
                         {"device": "XXXXXXXXXXXX", "time": "2021-03-09T15:23:20", "buffered": "no",
                         "values": {"pvstatus": 1, "pv1watt": 307, "pv2watt": 278, "pvpowerin": 585,
                         "pvpowerout": 575, "pvfrequentie": 4998, "pvgridvoltage": 2333,
                         "pvenergytoday": 15, "pvenergytotal": 32362, "pv1voltage": 1064,
                         "pv2voltage": 1351, "pv1current": 2, "pv2current": 2, "pvtemperature": 248,
                         "pvipmtemperature": 303}}
         - MQTT connection refused by target
         - Grott Send data to PVOutput disabled 
         - Grott InfluxDB publihing started
         - Timezone local specified default timezone used
         - Grott original time :  2021-03-09T15:23:20 adjusted UTC time for influx :  2021-03-09T14:23:20
         - Grott influxdb jsonmsg: 
                         [{'measurement': 'XXXXXXXXXXXX', 'time': '2021-03-09T14:23:20', 'fields':
                         {'device': 'XXXXXXXXXXXX', 'buffered': 'no', 'pvstatus': 1, 'pv1watt': 307,
                         'pv2watt': 278, 'pvpowerin': 585, 'pvpowerout': 575, 'pvfrequentie': 4998,
                         'pvgridvoltage': 2333, 'pvenergytoday': 15, 'pvenergytotal': 32362,
                         'pv1voltage': 1064, 'pv2voltage': 1351, 'pv1current': 2, 'pv2current': 2,
                         'pvtemperature': 248, 'pvipmtemperature': 303}}]
```

Is this because MQTT/inlfuxdb don't work with float or is it a small bug??

For now i devide the data with 10 in grafana, it's a bit of a work arround, but it works

johanmeijer, 2021-03-09:
Hi @Jeebeevee it is not a bug. 

Grott sends the original values used (and sent) by the growatt inverter to influxdb (to be used by grafana) and MQTT. The means by example that powerout is specified in 0.1W, voltage in 0.1v, frequency in 0.01hz and temperature in 0.1 degree celcius. 

The provided values are described in more detail here: https://github.com/johanmeijer/grott/wiki/MQTT.

To improve the readability of the verbose output I (re)formated the values, but I think is beter to send the original values to influxdb (and MQTT)

I will add a reference  at the influxdb wiki page to the MQTT pages to make it more clear. 

Thank your for your remarks!

Jeebeevee, 2021-03-09:
Thanks for the response.
In the logs I missing the dot. As example. In my log look at pv1watt it get 30.7 in the log but also in the influxDB it says 307 (no dot).
In my idea is it not the original value. It is a small thing and can be fixed on a later step. It is only good to know that this is a thing.

johanmeijer, 2021-03-09:
Sorry but I disagree.

It are the real values. The inverter sent 307  (Integer value witout .) and
this in 0.1 watts format (accoording the growatt documentation) so that is
indeed 30.7 watts.

So no bug or error only maybe a difference in opion  how to format. But I
prefer to stay with the original values.

Jeebeevee, 2021-03-10:
The verbose output did it for me then. 
You are right to keep de data as original as posible.

Thx for the reply, explanation and this script.

johanmeijer, 2021-03-11:
Thank you for the remarks,  keeps me sharp and improves Grott and the documenation!

closed 2021-03-13 by Jeebeevee