johanmeijer/grott #42: Grott for SPF 5000ES with metrics for battery, output, load etc.

charliesjc, 2021-04-12:
I'm running Grott on Home Assistant OS and have managed to get some of the metrics to show as needed, but I need some customization to show more than just info for solar. In my setup I don't even use solar at all and have my Growatt SPF 5000ES inverter set up with a PylonTech 3000C 3.5kWh Lithium battery for off-grid backup due to rolling blackouts.

I need to customize the T050104.json file for my needs as that seems to be the header for my inverter but I'll need to work out the values to send to MQTT, comparing the CSV download from the growatt server to the output I receive is the most obvious way.

Anyone else have some more information or some values/indexes from some more recent Growatt inverters to get the ball rolling?

johanmeijer, 2021-04-12:
Yes! lt would be nice if can do this together (the "grott"community). 

Links to the growatt record layout I have used: 

Old definitions: 
http://www.photovoltaicsolar.in/Growatt_Manual/1KW-50KW%20Modbus%20RS485%20RTU%20Protocol.pdf 

More recent ones: 
http://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf  

It is really a challenge to map the fields to the data record. But I would be very nice if we can managed it to enhance the  Grott output with battery load information as well. 

See a short instruction on how to create your own record layout at: https://github.com/johanmeijer/grott/wiki/Grott-advanced-(customize-behaviour). 
Record layout samples can be found in the sample directory.

johanmeijer, 2021-04-14:
The bad new is that the record layout is totally different for an off-grid inverter. 

The good news is I found a record layout on internet and it is very easy to create a new grott layout definition json file for this. 

I already started with making a t050104o.json and will put the results until now in the example directory. 
If you rename this file to t050104.json put this file in the grott home directory (or see use your own HA solution for placing the file) it should work. 

If you want you can complete the file. I will do it but this will take some days (need some spare time). 

You can find the layout document at: 
https://docplayer.net/199800433-Growatt-offgrid-spf5000-modbus-rs485-rtu-protocol.html

Maybe you are already able to test it and see if the result are as expected. 

I am also going to think how grott can automatically detect (or set a switch in the ini/environmentals) it is a off-grid inverter.

charliesjc, 2021-04-15:
Oh my goodness, that new document is a winner.

I haven't been able to look at it thoroughly yet but I would guess that the first part of the document might deal with the 0103 header or DATA3 per J Vromans discoveries and the second part is the stream we generally want for the log data, DATA4 with the 0104 header.

I'm just assuming the first part of the doc refers to "DATA3" because there's an easily visible firmware number in my data with that header. 

I'll get on this some more from tomorrow. Good find, Johan

johanmeijer, 2021-04-15:
I would not focus on the 0103. That record is only sent at initialization time (at least with grid attached inverters). 

Maybe I can use it later to retrieve more information on the inverter / network device but that needs some additional investigation (and redesign of grott). 

The 0104 is sent every 5 minutes (it should be possible to set time but that did not worked for me until  know) and contains de actual status information. 

I already started with mapping the fields in the t050104o.json records hopefully this is a good start for you.

charliesjc, 2021-04-15:
Lo and behold:

- Grott data record date/time processing started
	 - date-time:  2021-04-15T21:08:22
	 - Grott values retrieved:
		 -  pvserial             :  RKG3B02007
		 -  pvstatus             :  10
		 -  vpv1                 :  0.0
		 -  vpv2                 :  0.0
		 -  ppv1                 :  0.0
		 -  ppv2                 :  0.0
		 -  buck1curr            :  0.0
		 -  buck2curr            :  0.0
		 -  op_watt              :  300.0
		 -  op_va                :  322.0
		 -  acchr_watt           :  702.0
		 -  acchr_VA             :  702.0
		 -  bat_Volt             :  50.5
		 -  batterySoc           :  94
		 -  bus_volt             :  425.6
		 -  grid_volt            :  230.2
		 -  line_freq            :  50.0
		 -  outputvolt           :  230.2
		 -  outputfreq           :  339.2
		 -  outputDCV            :  6553.4
		 -  invtemp              :  28.2
		 -  dcdctemp             :  28.2
		 -  loadpercent          :  6.0
		 -  bat_s_volt           :  0.0
		 -  bat_volt_dsp         :  0.0
		 -  timetot              :  0.0
		 -  buck1_ntc            :  25.9
		 -  buck2_ntc            :  0.0
		 -  OP_Curr              :  1.3
		 -  Inv_Curr             :  5.9
		 -  AC_InWatt            :  1185.0
		 -  AC_InVA              :  1174.0
		 -  faultBit             :  0.0
		 -  warningBit           :  0.0
		 -  faultValue           :  0.0
		 -  warningValue         :  0.0
		 -  DTC                  :  2010.5
		 -  checkStep            :  4.5
		 -  productionLineMode   :  8.9
		 -  constantPowerOK      :  0.1
		 -  epv1tod              :  0.0
		 -  epv1tot              :  0.0
		 -  epv2tod              :  0.0
		 -  epv2tot              :  0.0
		 -  eacCharToday         :  0.9
		 -  eacCharTotal         :  49.6
		 -  ebatDischarToday     :  0.0
		 -  ebatDischarTotal     :  1.8
		 -  ACCharCurr           :  0.0
		 -  ACDischarWatt        :  308019.2
		 -  ACDischarVA          :  12215924.4
		 -  BatDischarWatt       :  274.0
		 -  BatDischarVA         :  296.0
		 -  BatWatt              :  0.0
		 -  BatOverCharge        :  0
		 -  MpptFanSpeed         :  65535
		 -  InvFanSpeed          :  58476
		 - 
Some values are looking a bit weird, but some, like output frequency, are actually fine, most times it shows as 50Hz and I've seen the weird high numbers even in the server log files, I think it's a general bug.

Some values still need to be tweaked for decimal places.

johanmeijer, 2021-04-15:
Ok. Not bad for the first itteration. 

Indeed some high values looks a little bit weird. 

I think we have to compare it with the values from the growatt website and see what the differences are.

charliesjc, 2021-04-22:
Right. Some more progress. I have a load of useful, correct and meaningful data now: 

Growatt new layout processing,
		 - decrypt       :  True,
		 - offset        :  6,
		 - record layout :  T05NNNN,
	 - Grott data record date/time processing started,
	 - date-time:  2021-04-22T17:42:37,
	 - Grott values retrieved:,
		 -  datalogserial        :  XXXXXXXXXX,
		 -  pvserial             :  XXXXXXXXXX,
		 -  pvstatus             :  11,
		 -  vpv1                 :  0.0,
		 -  vpv2                 :  0.0,
		 -  ppv1                 :  0.0,
		 -  ppv2                 :  0.0,
		 -  buck1curr            :  0.0,
		 -  buck2curr            :  0.0,
		 -  op_watt              :  313.0,
		 -  op_va                :  342.0,
		 -  acchr_watt           :  0.0,
		 -  acchr_VA             :  0.0,
		 -  bat_Volt             :  50.1,
		 -  batterySoc           :  97,
		 -  bus_volt             :  428.0,
		 -  grid_volt            :  226.8,
		 -  line_freq            :  50.2,
		 -  outputvolt           :  226.8,
		 -  outputfreq           :  50.2,
		 -  invtemp              :  25.9,
		 -  dcdctemp             :  25.8,
		 -  loadpercent          :  6.2,
		 -  buck1_ntc            :  24.0,
		 -  OP_Curr              :  1.3,
		 -  Inv_Curr             :  0.2,
		 -  AC_InWatt            :  320.0,
		 -  AC_InVA              :  317.5,
		 -  faultBit             :  0,
		 -  warningBit           :  0,
		 -  faultValue           :  0,
		 -  warningValue         :  0,
		 -  constantPowerOK      :  1,
		 -  epv1tod              :  0.0,
		 -  epv1tot              :  0.0,
		 -  epv2tod              :  0.0,
		 -  epv2tot              :  0.0,
		 -  eacCharToday         :  0.9,
		 -  eacCharTotal         :  57.9,
		 -  ebatDischarToday     :  0.0,
		 -  ebatDischarTotal     :  1.8,
		 -  eacDischarToday      :  4.1,
		 -  eacDischarTotal      :  227.5,
		 -  ACCharCurr           :  0.0,
		 -  ACDischarWatt        :  355.0,
		 -  ACDischarVA          :  391.0,
		 -  BatDischarWatt       :  0.0,
		 -  BatDischarVA         :  0.0,
		 -  BatWatt              :  -20.0,

There is still rubbish at the end which appears to relate to BMS but I'm not winning with the data. I don't really care either. It doesn't seem to be anything that gets used and/or it's not actually the BMS data because it's coming from the datalogger and not the inverter itself. There's even some data in the middle above that "could" be fault bit, warning bits/codes but it may not be what actually comes down the datalogger.

geoffles, 2021-04-23:
Any chance you have a dump of the BMS data portion and if Growatt reports anything on their shine portal?

The datalogger may be forwarding BMS info to the server so would be cool if we cool match?

charliesjc, 2021-04-23:
Unfortunately I don't. That could be the next project. I'll probably need to look into an RS485/CAN to Usb cable?

I don't think the BMS info comes down the datalogger anyway. Looking at the Modbus protocol document and the number of registers that are available under the "Input Register" section, for them to all be on the datalogger the datalogger stream would need to be almost 800 bytes long, currently it is only 225 bytes.

Are you running batteries in your setup?

charliesjc, 2021-04-23:
Here's one where it's running in status 10 (AC Charge and Bypass) instead of 11 (Just plain Bypass):

-  pvstatus             :  10
		 -  vpv1                 :  70.2
		 -  vpv2                 :  0.0
		 -  ppv1                 :  1.0
		 -  ppv2                 :  0.0
		 -  buck1curr            :  0.0
		 -  buck2curr            :  0.0
		 -  op_watt              :  308.0
		 -  op_va                :  318.0
		 -  acchr_watt           :  709.0
		 -  acchr_VA             :  709.0
		 -  bat_Volt             :  51.0
		 -  batterySoc           :  94
		 -  bus_volt             :  433.0
		 -  grid_volt            :  227.5
		 -  line_freq            :  50.0
		 -  outputvolt           :  227.5
		 -  outputfreq           :  50.1
		 -  invtemp              :  26.1
		 -  dcdctemp             :  24.7
		 -  loadpercent          :  6.1
		 -  buck1_ntc            :  22.8
		 -  OP_Curr              :  1.3
		 -  Inv_Curr             :  5.6
		 -  AC_InWatt            :  1145.0
		 -  AC_InVA              :  1137.5
		 -  faultBit             :  0
		 -  warningBit           :  0
		 -  faultValue           :  0
		 -  warningValue         :  0
		 -  constantPowerOK      :  1
		 -  epv1tod              :  0.0
		 -  epv1tot              :  0.0
		 -  epv2tod              :  0.0
		 -  epv2tot              :  0.0
		 -  eacCharToday         :  0.5
		 -  eacCharTotal         :  58.6
		 -  ebatDischarToday     :  0.0
		 -  ebatDischarTotal     :  1.8
		 -  eacDischarToday      :  2.6
		 -  eacDischarTotal      :  231.7
		 -  ACCharCurr           :  14.0
		 -  ACDischarWatt        :  331.0
		 -  ACDischarVA          :  343.0
		 -  BatDischarWatt       :  0.0
		 -  BatDischarVA         :  0.0
		 -  BatWatt              :  -714.0

The only funny one in there is the vpv1 sitting at 70.1V. It matches the data exactly and is also what it says on the inverter screen. It's apparently an anomaly because there is no solar at all. I suspect a firmware update would eliminate it.

geoffles, 2021-04-23:
Yes, I have batteries although the BMS is not linked to my inverter at the moment on the advice of my installer (It will just switch to bypass if the voltage drops below a certain point).

I have a CANBus module although haven't yet been brave enough to connect it to the BMS or the inverter :P

It's possible that the voltage across your PV terminals really is the 70v if nothing is connected to the terminals and it's just basically static charge.

charliesjc, 2021-04-23:
What batteries are you running? A lithium set?

I've seen some weird things on my side of the world where installers just don't know how to get the BMS to interface with the Growatt inverters so instead of using it as intended they simply run the inverter in User Mode and set thresholds based on voltages. A complete waste of time if you ask me. 

My installer had issues with mine and didn't really understand certain things about ethernet/similar cables. The cable he brought, that came with the battery, was for use with the RS485 port but had the pins on the wrong end cut off so it was incredibly useless. What worked nicely was a normal straight ethernet cable plugged into the CAN sockets with Lithium program 52 set on the inverter. Worked like a charm.

From looking at the manual I could gather Growatts plan was program 1 to 15 for the RS485 bus and program 51 to 65 are possibly the same but just on the CAN bus. (The installer was trying program 2 on RS485 with a dodgy cable). So 52 on CAN is the same? Meh, it works. So now the BMS runs the show with regards to charge current demand and cutoff etc.

If you send a WhatsApp message to Xing Yang at Growatt, he could probably tell you what profile would work for you. (He seems to be the new support contact there)

geoffles, 2021-04-23:
Yes, I've got a stack of Dyness Lithium batteries.

Installer said I could connect the BMS and its clear from the manual what the settings should be, but he mentioned that he's not confident about the programming on the inverter and what it should do when the charge is low - said he was waiting for software updates (maybe they're available now?).

The cables that came with the batteries for the RJ45 RS485 port seem to be exactly ethernet cables (same wiring arrangment).

Thanks for the reference at Growatt - might check it out.

the-butterfry, 2021-04-23:
Watching this as it will probably be the path for my Wifi-S USB module on a SPF3000 TL on #46

johanmeijer, 2021-04-24:
@charliesjc I merged your pull request so your changes are now part of the t050104o.json in the /examples/Record Layout directory. 

Thanks for the good work! I hope you are happy  with the result (I am). 

I will document a howto on the SPF offgrid inverter in the wiki. 

Next release I will integrate this in grott so no additional json will be needed. At this moment I am not sure how to detect the difference between a t050104 for a SPF and non-SPF inverter. Maybe I will get an idea on that later. On the short term I will build in a "switch" (parameter in .ini and environmental) to indicated it is a SPF inverter. 

Do we need a 050150 record as well? Normally the 0150 buffer records are created if the inverter has no netwerk contact with the growatt server. These historical records are being sent later when network is available again . The layout of the record is the same that is why I created the T05NNNN record layout covering both records. 

Just noticed that is is not necessary to rename the t050104o.json only copy it in the main directory is enough. Grott "eats" all files starting with t and having the .json suffix. That was programming with  future use in mind :) (of just lucky). 

I am thinking of project with a rs485/modbus connection as will using a ESP8266 / ESP32 micro controller. This is also used in the  shinewifi stick  (maybe we can even flash this stick in the future). 

The only thing is that I  have to stop with this project. I only have one Growatt Inverter and want to keep that one in use for using and testing Grott. Maybe I can replace the inverter of my old (20+ years) string of panels with a Growatt inverter in the future. Then I will have to inverters to play with. 

If somebody knows a cheap 750 or 1000 watt inverter please let me know (2000 is also okay I  will swap it with my current inverter).

geoffles, 2021-04-27:
I have considered plugging into the modbus, particularly after the conversations here - although I'm in a similar situation to yourself.

Myself and my installer are both engineers - he's quite busy and my time is quite limited, but if I get any info, I'll definitely share it here. I have a suspicion, though, that the Shine WiFi sticks on the Growatt literally just IP extensions on the modbus with maybe interval control and obfuscation.

I think this is the case because of the binary format, and the seeming close alignment with the actual modbus documentation which is intended for the RS485.

charliesjc, 2021-04-27:
I had thought about this as well. I tried that PVKeeper nonsense that you can download from Growatt for monitoring via USB and although the program is rubbish, the fact that the installed drivers create a comm port means you could probably sniff the data coming out of the inverter for comparison with the wifi dongle. Rather than having to get an RS485 to USB converter or something similar. You should just be able to get away with the USB printer cable that comes with the inverter.

johanmeijer, 2021-11-14:
@charliesjc can we close this one?

charliesjc, 2021-11-14:
@johanmeijer Yes I think we can.

closed 2021-11-14 by johanmeijer