johanmeijer/grott #44: SPF 5000TL HVM-P

geoffles, 2021-04-22:
Hi, 

I'm trying to decode the records for my growatt inverter using the information you guys have collected so far. I've sniffed out some data using Wireshark (I knew how already so was easier than using the Grott sniffer).

Assuming the first 8 bytes are still the header, I'm getting '10 13 31 77 78 24 ec fa' - which would make your format code something like 77ecfa.

The message format appears to also be obfuscated using the Growatt XOR on the message body since I can see the repeating "Growatt" pattern starting from byte 8 (where the header ends).

I'm a little uneasy about posting the entire message on a public forum given the potentially sensitive information in the stream - but is there some other guidance or advice you could provide to assist in helping to extract the record format?

I'd be very willing to contribute the data and record format once I've sanitized it.

Many thanks.

johanmeijer, 2021-04-22:
The header is not encrypted. So start with decrypting from the 9th byte. 

Some header information: 
ss ss 00 xx ll ll tt tt data
ss ss  = sequence number
xx  = additional data format/record information (e.g.: 02 unencrypted, 05 or 06 both encrypted)
ll ll = length of tt tt + data
tt tt = message type  (eg: 01 04: actual data record, or 01 50: buffered data record ). 

I already started with decoding the SPF data (see t050104o.json in the example/layout directory). See also issue42 (also for additional documentation links). 

I am not sure if there is a lot of difference between SPF 5000ES or SPF 5000TL.  

I think it is easy to use Grott to test new record layout (using the additional .json records as described in: https://github.com/johanmeijer/grott/wiki/Grott-advanced-(customize-behaviour). 

I do not advise you to use the sniffer mode of Grott. Use the proxy mode is more advisable. Setting up sniffering needs some (complex) ipforwarding rules.

charliesjc, 2021-04-22:
I've made progress with this. I've made a pull request with my updates to the relevant file.

geoffles, 2021-04-22:
Ok, so I'm a bit silly - the content I copied from wireshark included the ethernet frame header. Growatt Header is 060104 so it seems there is an existing format.

I've now succesfully decoded the device serial number.

I'll upload some sanitised data and records probably tomorrow.

I also seem to have a bunch of the buffered frames (0150) - which get emitted while the growatt site is down. I'm hoping I'll be able to at least continue reporting consumption on my LAN even thuogh their service is out.

Thanks for the technical assistance.

closed 2021-04-22 by geoffles