johanmeijer/grott #46: ShineWiFi-S - So close

the-butterfry, 2021-04-23:
I'm working with a ShineWiFi-S USB on a SPF 3000 TL LVM - 24P. You don't have it listed here, but its so close as to probably just needing a different definition or offset.

as you can see below some of the information is correct. Can I help debug this to add a definition for this WiFi module?

Here's the output from grott start:

Apr 23 10:01:22 Farm-pi grott[12089]: #011organization:#011 grottorg
Apr 23 10:01:22 Farm-pi grott[12089]: #011bucket:      #011 grottdb
Apr 23 10:01:22 Farm-pi grott[12089]: #011token:       #011 **secret**
Apr 23 10:01:22 Farm-pi grott[12089]: _Extension:
Apr 23 10:01:22 Farm-pi grott[12089]: #011extension:   #011 False
Apr 23 10:01:22 Farm-pi grott[12089]: #011extname:     #011 grottext
Apr 23 10:01:22 Farm-pi grott[12089]: #011extvar:      #011 {'none': 'none'}
Apr 23 10:01:22 Farm-pi grott[12089]: Grott proxy mode started
Apr 23 10:01:22 Farm-pi grott[12089]: Hostname : Farm-pi
Apr 23 10:01:22 Farm-pi grott[12089]: IP :  127.0.1.1 , port :  5279
Apr 23 10:01:27 Farm-pi grott[12089]: #011 - ('192.168.55.3', 44503) has connected
Apr 23 10:01:27 Farm-pi grott[12089]: #011 - Growatt packet received:
Apr 23 10:01:27 Farm-pi grott[12089]: #011#011  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 47420), raddr=('47.251.11.172', 5279)>
Apr 23 10:01:27 Farm-pi grott[12089]: #011 - Data less then minimum record length, data not processed
Apr 23 10:01:27 Farm-pi grott[12089]: #011 - Growatt packet received:
Apr 23 10:01:27 Farm-pi grott[12089]: #011#011  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 5279), raddr=('192.168.55.3', 44503)>
Apr 23 10:01:27 Farm-pi grott[12089]: #011 - Data less then minimum record length, data not processed
Apr 23 10:01:27 Farm-pi grott[12089]: #011 - Growatt packet received:
Apr 23 10:01:27 Farm-pi grott[12089]: #011#011  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 47420), raddr=('47.251.11.172', 5279)>
Apr 23 10:01:27 Farm-pi grott[12089]: #011 - Data less then minimum record length, data not processed
Apr 23 10:01:27 Farm-pi grott[12089]: #011 - Growatt packet received:
Apr 23 10:01:27 Farm-pi grott[12089]: #011#011  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 5279), raddr=('192.168.55.3', 44503)>
Apr 23 10:01:27 Farm-pi grott[12089]: #011 - Data less then minimum record length, data not processed
Apr 23 10:01:28 Farm-pi grott[12089]: #011 - Growatt packet received:
Apr 23 10:01:28 Farm-pi grott[12089]: #011#011  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 47420), raddr=('47.251.11.172', 5279)>
Apr 23 10:01:28 Farm-pi grott[12089]: #011 - Data less then minimum record length, data not processed
Apr 23 10:01:29 Farm-pi grott[12089]: #011 - Growatt packet received:
Apr 23 10:01:29 Farm-pi grott[12089]: #011#011  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 47420), raddr=('47.251.11.172', 5279)>
Apr 23 10:01:29 Farm-pi grott[12089]: #011 - Data less then minimum record length, data not processed
Apr 23 10:01:30 Farm-pi grott[12089]: #011 - Growatt packet received:
Apr 23 10:01:30 Farm-pi grott[12089]: #011#011  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 47420), raddr=('47.251.11.172', 5279)>
Apr 23 10:01:30 Farm-pi grott[12089]: #011 - Data less then minimum record length, data not processed
Apr 23 10:01:31 Farm-pi grott[12089]: #011 - Growatt packet received:
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 47420), raddr=('47.251.11.172', 5279)>
Apr 23 10:01:31 Farm-pi grott[12089]: #011 - Growatt original Data:
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x00\x09\x00\x06\x01\x5f\x01\x03\x03\x36\x2b\x47\x20\x47\x4d\x76\x40\x3e\x77
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x3a\x3c\x25\x35\x35\x72\x43\x5f\x44\x56\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x70\x63\x4e\x72\x5d\x74\x61
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x74\x74\x6b\x72\x6e\x77\x60\x74\x75\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x77\x63\x41\x44\x75\x5c\x5f\x40\x51\x44\x46\x69\x42\x5b\x77\x61\x8b\x8b\x47
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x72\x6f\x71\x61\x75\x74\x47\x72\x6f\x77\x61\x21\x3f\x03\x33\x2e\x42\x50\x44
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x47\x70\x72\x6f\x77\x61\x74\x75\x46\x72\x6f\x77\x61\x74\x74\x17\x73\x74\x76
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x71\x74\x92\x47\x6c\x6f\x75\x61\x74\x74\x47\x72\x6f\x39\x44\x74\x74\x47\x5f
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x6f\x2e\x66\x91\x74\x43\x72\x78\x77\x69\x74\x4c\x47\x63\x6f\x77\x61\x74\x74
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x01\x44\x47\x72\x1a\x47\x61\x74\x74\x47
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x72\xbd\x77\x61\x70\xc4\x45\x2a\x6e\x97\x63\x2c\x75\x6b\x76\xdf\x77\x3b\x74
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \xf2\x45\x2a\x64\xcf\x61\x77\x74\x59\x72\xbd\x76\x7b\x74\x74\x47\x72\x6f\x77
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x73\x6f\x77\x61\x74\x74\x46\x72
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x6f\x76\x45\x74\x48\x47\x0a\x6f\x69\x61\x0c\x74\x47\x72\x6f\x77\x61\x74\x74
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x12\x2e
Apr 23 10:01:31 Farm-pi grott[12089]: #011 - Grott automatic protocol detection
Apr 23 10:01:31 Farm-pi grott[12089]: #011 - Grott data record length 359
Apr 23 10:01:31 Farm-pi grott[12089]: #011 - layout   :  T060103
Apr 23 10:01:31 Farm-pi grott[12089]: #011 - no matching record layout found, try generic
Apr 23 10:01:31 Farm-pi grott[12089]: #011 - Record layout used :  T060103
Apr 23 10:01:31 Farm-pi grott[12089]: #011 - Growatt data decrypted V2
Apr 23 10:01:31 Farm-pi grott[12089]: #011 - Grott Growatt data decrypted
Apr 23 10:01:31 Farm-pi grott[12089]: #011 - Growatt plain data:
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 00090006015f01034444443041333931325100000000000000000000000000000000000000005
Apr 23 10:01:31 Farm-pi grott[12089]: #011#011 54b44414135313033370000000000000000000000000000000000000000150417090032030000
[... 127 lines cut ...]
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011 -  pvtemperature        :  16.7
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011 -  pvipmtemperature     :  0.0
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011 -  epv1today            :  0.2
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011 -  epv1total            :  9.6
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011 -  epv2today            :  0.0
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011 -  epv2total            :  0.0
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011 -  epvtotal             :  0.0
Apr 23 10:01:47 Farm-pi grott[12089]: #011 - MQTT jsonmsg:
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011#011 {"device": "UKDAA51037", "time": "2021-04-23T09:01:46", "buffered": "no",
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011#011 "values": {"pvstatus": 12, "pvpowerin": 71303168, "pv1voltage": 0,
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011#011 "pv1current": 2560, "pv1watt": 0, "pv2voltage": 94, "pv2current": 0,
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011#011 "pv2watt": 0, "pvpowerout": 50, "pvfrequentie": 0, "pvgridvoltage": 0,
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011#011 "pvgridcurrent": 0, "pvgridpower": 2715, "pvgridvoltage2": 100,
[... cut ...]
Apr 23 10:01:47 Farm-pi grott[12089]: #011 - Grott Send data to Influx disabled
Apr 23 10:01:47 Farm-pi grott[12089]: #011 - Grott extension processing disabled
Apr 23 10:01:47 Farm-pi grott[12089]: #011 - Growatt packet received:
Apr 23 10:01:47 Farm-pi grott[12089]: #011#011  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 5279), raddr=('192.168.55.3', 44503)>
Apr 23 10:01:47 Farm-pi grott[12089]: #011 - Data less then minimum record length, data not processed

charliesjc, 2021-04-24:
Considering your inverter is also an off grid type, I would try using the content of the t050104o.json file that I've helped to modify. I see the changes I've made aren't merged yet but maybe @johanmeijer may implement it soon if he's happy with the updates? Otherwise if you want to copy the contents of the file over now you can look at my fork of this project in the offgrid layout branch.

johanmeijer, 2021-04-24:
Jeah I will merge it today :)

johanmeijer, 2021-04-24:
@the-butterfry you need a t060104o.json that is a slighty different then the t050104o.json that is developed with issue #42.
 05 has a different offset then the 06 records. 

Let me see if I can find the start offset and then we can modify the 050104 spf record and see if it works for you and change if needed.

It would be create if we manage to create this one as well. Your help is really appreciated.

the-butterfry, 2021-04-24:
Yup, I saw that change and was following. I'm definitely in for doing the legwork to diagnose this one and get it right. I'm afk till Monday, but let me know what info you need.

johanmeijer, 2021-04-25:
Hi @the-butterfry,  

I created t060104o.json (modified t060104.json created with @charliesjc). 

You can copy this file form the example directory:  https://github.com/johanmeijer/grott/tree/master/examples/Record%20Layout  to the Grott Main directory. 

Hopefully this give more information on your solar system. I am not totally convinced on all values but let see. Can you compare this with the values at the Growatt Website? 

I download the information in excel format (xls) from the website at the  http://server.growatt.com/index (history data button).

the-butterfry, 2021-04-26:
hah, @johanmeijer you got alot closer than I did with my version. I figured there was a shift somewhere up at the top that I was missing.  I'm digging into it now to see what differs from the server data.  Would it help to have my .xls export?

A couple of questions on fields: What is  'op_va' an 'op_watt' represent? I think op_va is another temp sensor with a shifted decimal. There also seem to be another temp sensor in the data download. 
Bus voltage is an odd one, unless that is a setting and not representative of anything. Only thing AC I've got on this system is the output at 120.

Here's the output now:

Apr 26 06:47:07 Farm-pi grott[8670]: #011 - Grott automatic protocol detection
Apr 26 06:47:07 Farm-pi grott[8670]: #011 - Grott data record length 359
Apr 26 06:47:07 Farm-pi grott[8670]: #011 - layout   :  T060104
Apr 26 06:47:07 Farm-pi grott[8670]: #011 - Record layout used :  T060104
Apr 26 06:47:07 Farm-pi grott[8670]: #011 - Growatt data decrypted V2
Apr 26 06:47:07 Farm-pi grott[8670]: #011 - Grott Growatt data decrypted
Apr 26 06:47:07 Farm-pi grott[8670]: #011 - Growatt plain data:
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 04f10006015f01044444443041333931325100000000000000000000000000000000000000005
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 54b4441413531303337000000000000000000000000000000000000000015041a052f07030000
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 002c000c016c0000000000140000000000000000000000140000003c00000000000000000a630
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 06409560000000004b11770000000a0009f000000000000000000000076007d0000000b000000
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 000000000000000000000000004e25002d0059000100000000000000000000008a00000000000
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 00000000000000000003300000000000000040000000000000000000000000000000000000000
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 000000000000fffffff600000000000000100000000000000000000000070000005a008600000
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 00100000000000000000000000000000000000000000000000000000000000000000000000000
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 00000000000000000000000000000000000000000000000000000000000000000000000001000
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 100000001000000000001209d
Apr 26 06:47:07 Farm-pi grott[8670]: #011 - Growatt new layout processing
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 - decrypt       :  True
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 - offset        :  6
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 - record layout :  T060104
Apr 26 06:47:07 Farm-pi grott[8670]: #011 - Grott data record date/time processing started
Apr 26 06:47:07 Farm-pi grott[8670]: #011 - date-time:  2021-04-26T05:47:07
Apr 26 06:47:07 Farm-pi grott[8670]: #011 - Grott values retrieved:
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  datalogserial        :  DDD0A3912Q
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  pvserial             :  UKDAA51037
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  pvstatus             :  12
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  vpv1                 :  36.4
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  vpv2                 :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  ppv1                 :  2.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  ppv2                 :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  buck1curr            :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  buck2curr            :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  op_watt              :  2.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  op_va                :  1536.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  acchr_watt           :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  acchr_VA             :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  bat_Volt             :  26.6
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  batterySoc           :  100
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  bus_volt             :  239.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  grid_volt            :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  line_freq            :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  outputvolt           :  120.1
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  outputfreq           :  60.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  invtemp              :  16.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  dcdctemp             :  15.9
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  loadpercent          :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  buck1_ntc            :  11.8
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  buck2_ntc            :  12.5
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  OP_Curr              :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  Inv_Curr             :  1.1
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  AC_InWatt            :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  AC_InVA              :  0.0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  faultBit             :  0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  warningBit           :  0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  faultValue           :  0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  warningValue         :  0
Apr 26 06:47:07 Farm-pi grott[8670]: #011#011 -  constantPowerOK      :  1
[... cut ...]
Apr 26 06:47:49 Farm-pi grott[8670]: #011#011  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 42172), raddr=('47.251.11.172', 5279)>
Apr 26 06:47:49 Farm-pi grott[8670]: #011 - Data less then minimum record length, data not processed
Apr 26 06:47:49 Farm-pi grott[8670]: #011 - Growatt packet received:
Apr 26 06:47:49 Farm-pi grott[8670]: #011#011  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 5279), raddr=('192.168.55.3', 28221)>
Apr 26 06:47:49 Farm-pi grott[8670]: #011 - Data less then minimum record length, data not processed

the-butterfry, 2021-04-26:
Ok scratch that, I found the  two other temp status 'buck1_ntc' and 2 are what I'm guessing is temps on the DC<>DC circuits. 

Looks like current net battery charge is set at 'BatWatt'.  Is it supposed to be a negative number? Maybe I'm not understanding here.

I think these are swapped:
Apr 26 07:55:44 Farm-pi grott[8670]: #011#011 -  epv1tod              :  0.0
Apr 26 07:55:44 Farm-pi grott[8670]: #011#011 -  epv1tot              :  0.0
Apr 26 07:55:44 Farm-pi grott[8670]: #011#011 -  epv2tod              :  13.8
Apr 26 07:55:44 Farm-pi grott[8670]: #011#011 -  epv2tot              :  0.0

charliesjc, 2021-04-26:
Hi Cory, 
This document will help a lot with your troubleshooting: [https://docplayer.net/199800433-Growatt-offgrid-spf5000-modbus-rs485-rtu-protocol.html](https://docplayer.net/199800433-Growatt-offgrid-spf5000-modbus-rs485-rtu-protocol.html).

It is the modbus protocol that was used to come up with the extra data needed for my offgrid inverter. The chances are this document actually suits your inverter better than mine because you have a SPF 3000 TL I believe, and this is for the 5000 TL. Look under the "Input Registers" to see the order of things and remember that because we're dealing with a Wifi dongle, the starting index for everything won't be the same. Coming straight off the inverter it would start at 0 but not in our case @johanmeijer started the file at 76 for you because thats where the "serial number" starts from, after that everything is generally in the order that the manual has. I left out ones where my inverter (or dongle rather) doesn't show good data. 

Also remember the manual shows every register as 2 bytes, and where a power (W or VA) is involved there is usually a H and an L so you combine the lot into one long 4 byte register. Hence the length 2's and 4's in the .json files. But to make it a little more complicated, the way grott processes the data is as a string which means you need to reference characters, not bytes. So 2 bytes is 4 characters, 4 bytes is 8, which is why there are gaps in .json file that might not seem right at first glance.

The amount you need to "divide" by is also hinted at for each register in the manual. I'll leave you to discover what the negative number for BatWatt means in the manual.

P.S. you can also call these fields whatever you like, I followed the naming in the input registers but for me some things like buck temp and dc dc temp etc. are actually portrayed as Temp1/2/3 at server.growatt.com

johanmeijer, 2021-04-26:
By the way according the document a negative "batwatt" means your charging (and  not discharge) the battery .

johanmeijer, 2021-04-26:
@the-butterfry and @charliesjc let me know f you are happy with the layout. I will then make it part of the next Grott release.

charliesjc, 2021-04-26:
@johanmeijer I'm happy with my layout for the 050104 off-grid. All the values seem to be tracking correctly with what's available on the growatt server.

the-butterfry, 2021-04-26:
thanks guys, yeah just catching up to speed here. I'll review the doc when I get a sec.
honestly, there's nothing weird about the data in the .xls or syslog. my internal IPs, inverter serial etc.. who cares. happy to share. seeing us work it out live will probably help others too.

Here's the concurrent dump

Apr 26 09:43:07 Farm-pi grott[8670]: #011 - Grott automatic protocol detection
Apr 26 09:43:07 Farm-pi grott[8670]: #011 - Grott data record length 359
Apr 26 09:43:07 Farm-pi grott[8670]: #011 - layout   :  T060104
Apr 26 09:43:07 Farm-pi grott[8670]: #011 - Record layout used :  T060104
Apr 26 09:43:07 Farm-pi grott[8670]: #011 - Growatt data decrypted V2
Apr 26 09:43:07 Farm-pi grott[8670]: #011 - Grott Growatt data decrypted
Apr 26 09:43:07 Farm-pi grott[8670]: #011 - Growatt plain data:
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 00130006015f01044444443041333931325100000000000000000000000000000000000000000
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 0000000000000000000000000000000000000000000000000000000000015041a082b07030000
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 002c000c0342000000000c260000000000720000000000000000003c00000000000000000a8a0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 06409840000000004b0176f000000b300af00000000000000000000009c00b50000000b000000
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 000000000000000000000000004e25002d0059000100000000000000010000008b00000000000
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 00000000000000000003300000000000000040000000000000000000000000000000000000000
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 000000000000fffff40c00000000000000190072000000000000000000070000005a008600720
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 00100000000000000000000000000000000000000000000000000000000000000000000000000
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 00000000000000000000000000000000000000000000000000000000000000000000000001000
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 100000001000000000001aa35
Apr 26 09:43:07 Farm-pi grott[8670]: #011 - Growatt new layout processing
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 - decrypt       :  True
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 - offset        :  6
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 - record layout :  T060104
Apr 26 09:43:07 Farm-pi grott[8670]: #011 - Grott data record date/time processing started
Apr 26 09:43:07 Farm-pi grott[8670]: #011 - date-time:  2021-04-26T08:43:07
Apr 26 09:43:07 Farm-pi grott[8670]: #011 - Grott values retrieved:
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  datalogserial        :  DDD0A3912Q
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  pvserial             :
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  pvstatus             :  12
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  vpv1                 :  83.4
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  vpv2                 :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  ppv1                 :  311.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  ppv2                 :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  buck1curr            :  11.4
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  buck2curr            :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  op_watt              :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  op_va                :  1536.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  acchr_watt           :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  acchr_VA             :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  bat_Volt             :  27.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  batterySoc           :  100
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  bus_volt             :  243.6
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  grid_volt            :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  line_freq            :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  outputvolt           :  120.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  outputfreq           :  60.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  invtemp              :  17.9
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  dcdctemp             :  17.5
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  loadpercent          :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  buck1_ntc            :  15.6
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  buck2_ntc            :  18.1
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  OP_Curr              :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  Inv_Curr             :  1.1
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  AC_InWatt            :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  AC_InVA              :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  faultBit             :  0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  warningBit           :  0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  faultValue           :  0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  warningValue         :  0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  constantPowerOK      :  1
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  epv1tod              :  0.0
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  epv1tot              :  0.1
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  epv2tod              :  13.9
Apr 26 09:43:07 Farm-pi grott[8670]: #011#011 -  epv2tot              :  0.0
[... cut ...]
Apr 26 09:43:36 Farm-pi grott[8670]: #011 - Growatt packet received:
Apr 26 09:43:36 Farm-pi grott[8670]: #011#011  <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.55.2', 5279), raddr=('192.168.55.3', 6348)>
Apr 26 09:43:36 Farm-pi grott[8670]: #011 - Data less then minimum record length, data not processed

[UKDAA51037 storage data - 2021-04-26_2021-04-26.xls](https://github.com/johanmeijer/grott/files/6378712/UKDAA51037.storage.data.-.2021-04-26_2021-04-26.xls)

MrGingerT, 2021-04-26:
Hello Guys

I have a GroWatt 4KW Inverter wiith Wifi Dongle attached.  I tried last night to use this code latest as of 26/4/21 on a Ubuntu VM. Previously have very old code on a raspberryPI (pre github) which works fine using wireshark type packet capture. I am using this new code in proxy mode  and have the same "Data less then minimum record length, data not processed". I do not pretend to be an expert - any help would be appreciated. I enclose a copy of the output

Thank you very much for your help in advance

from management dongle interface

Version | 4.0
-- | --
Serial Number | AH55340024
Check Code | 48111

from grott verbose logging

Grott Growatt logging monitor : 2.5.4

Grott Command line parameters processed:
        verbose:         True
        config file:     grott.ini
        output file:     <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
        nomqtt:          False
        inverterid:      automatic
        pvoutput:        False
        blockcmd:        False
        noipf:           False

Grott process configuration file

Grott process environmental variables

Grott override settings if set in commandline

Grott process json layout files

Grott layout records loaded
T02NNNN  :
{'decrypt': {'value': 'False'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'num', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'pvipmtemperature': {'value': 242, 'length': 2, 'type': 'num', 'divide': 10}, 'epv1today': {'value': 278, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 294, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 310, 'length': 4, 'type': 'num', 'divide': 10}}
T05NNNN  :
{'decrypt': {'value': 'True'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'numx', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'pvipmtemperature': {'value': 242, 'length': 2, 'type': 'num', 'divide': 10}, 'epv1today': {'value': 278, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 286, 'length': 4, 'type': 'num [... cut ...]
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.34.27.247', 5279), raddr=('10.34.27.85', 26359)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.34.27.247', 46798), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed
^CCtrl C - Stopping server
         - ('10.34.27.85', 26359) has disconnected

johanmeijer, 2021-04-26:
@MrGingerT This does not look bad. I think you have to look for it a little bit longer. This are still initialization records. At some point in time the data records will come I suppose (T020104). 

If the problem stays can you please open a separate issue for this? Is easier to track I think.

johanmeijer, 2021-04-26:
@the-butterfry This does not looks to bad. 

If I look in the excel then: 

epv1tot = epvtoday (and eChargeToday) 
epv1tod = epvtotal  (and eChargeTotal)

Seeing the fact that in the record the values for epvtoday am depvtotal are accuring only once I assume growatt use this fields also for eChargeToday and eChargeTotal. 

@charliesjc do you recognize this?

the-butterfry, 2021-04-26:
@johanmeijer yep, i'll swap those.

One thing i'm a bit hazy on, is why does the syslog sometimes fall back to the Generic T06NNN profile?

NApr 26 13:00:46 Farm-pi grott[596]: #011 - Grott automatic protocol detection
Apr 26 13:00:46 Farm-pi grott[596]: #011 - Grott data record length 359
Apr 26 13:00:46 Farm-pi grott[596]: #011 - layout   :  T060150
Apr 26 13:00:46 Farm-pi grott[596]: #011 - no matching record layout found, try generic
Apr 26 13:00:46 Farm-pi grott[596]: #011 - Record layout used :  T06NNNN
Apr 26 13:00:46 Farm-pi grott[596]: #011 - Growatt data decrypted V2
Apr 26 13:00:46 Farm-pi grott[596]: #011 - Grott Growatt data decrypted
Apr 26 13:00:46 Farm-pi grott[596]: #011 - Growatt plain data:
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 00180006015f01504444443041333931325100000000000000000000000000000000000000005
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 54b4441413531303337000000000000000000000000000000000000000015041a0b2733030000
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 002c000c035f000000000a9600000000005e0000000000000000000000000000000000000a9b0
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 064098a0000000004af1770000000c300c10000000000000000000000b400ca0000000b000000
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 000000000000000000000000004e25002d0059000100000000000000000000009000000000000
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 00000000000000000003300000000000000040000000000000000000000000000000000000000
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 000000000000fffff61e0000000000000017005e000000000000000000070000005a008600620
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 00100000000000000000000000000000000000000000000000000000000000000000000000000
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 00000000000000000000000000000000000000000000000000000000000000000000000001000
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 100000001000000000001ba5e
Apr 26 13:00:46 Farm-pi grott[596]: #011 - Growatt new layout processing
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 - decrypt       :  True
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 - offset        :  6
Apr 26 13:00:46 Farm-pi grott[596]: #011#011 - record layout :  T06NNNN
Apr 26 13:00:46 Farm-pi grott[596]: #011 - Grott data record date/time processing started
Apr 26 13:00:46 Farm-pi grott[596]: #011 - date-time:  2021-04-26T11:39:51
Apr 26 13:00:46 Farm-pi grott[596]: #011 - Grott values retrieved:

johanmeijer, 2021-04-26:
@the-butterfry Ah yes! We need a additional layout :)

The record  sent is a T060150. This means (50) it is a buffered record and is sent as soon there is contact with the Growatt server. 

We have not defined that one yet for the SPF. The content is exactly the same as the T060104 record so no problem. I will add this. You can also make a copy of the T060104o.json record rename it to T060150o.json and change the field with T060104 in the file in T060150. 

This should work. But no worries I will add this also to the next release.

johanmeijer, 2021-04-26:
I updated the t060104o.json record in https://github.com/johanmeijer/grott/tree/master/examples/Record%20Layout 

If you copy this one to the Grott main directory the t060150 issue should be solved. 

Also changed /added the epvtoday and epvtotal fields (deleted the epv1tod/tod and epv2tot/tod)

the-butterfry, 2021-04-26:
ahhha! that explains why it 'smooths' out after a bit. it was trying to do backlog. thanks! i'll update my setup.

charliesjc, 2021-04-27:
> @the-butterfry This does not looks to bad.
> 
> If I look in the excel then:
> 
> epv1tot = epvtoday (and eChargeToday)
> epv2tod = epvtotal (and eChargeTotal)
> 
> Seeing the fact that in the record the values for epvtoday am depvtotal are accuring only once I assume growatt use this fields also for eChargeToday and eChargeTotal.
> 
> @charliesjc do you recognize this?

I don't have the excel spreadsheet for this data so I don't know exactly. But yes, I suspect there is some basic math going on there and epv1tot and epv2tot would just be added together to epvtotal for logging purposes. 

When I was sorting mine out I didn't even bother too much with the excel spreadsheet. I just kept the growatt server window open and used the values straight from the site. There are a few in there that don't get output to the excel log. I found my missing values by changing the scale on the graphs from say "time" to "year" and then other data was shown. The Battery Information graph on the left with all the yellow bars is a bit misleading though. It says Discharged (as in battery) but it isn't the battery, for me it was the eacDischarToday in the bars, then the eacDischarTotal is in the Energy Trend graph above on the right when you switch to "year" view. "Discharge" above that is then the sum of eacDischarTotal and ebatDischarTotal. These 2 values are also in my excel log as eopDischrTotal

I don't have any solar though so I don't have much insight as to what the energy data should look like with solar charging in the mix.

MrGingerT, 2021-04-29:
Hello @johanmeijer - I just wanted to thank you for your support. I can confirm the sofwtare is working now - I dont knwo what I did differently it didnt work for a day and then started working ! regardless Thank you
I do still get what looks like a warning but it works nevertheless
     - Growatt packet received:
              <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.34.27.247', 46798), raddr=('47.91.67.66', 5279)>
     - Data less then minimum record length, data not processed

johanmeijer, 2021-04-29:
@MrGingerT Your welcome. 

The data less then minimum record length is normal. 
I have put in a filter (can be set with minrecl= parm in .ini, or gminrecl= parm environmental) so not al records (especially the ack/nack etc) are fully processed. If you specify minrecl=1 all processing is always done and you will see all records in the conversation between your inverter and the growatt server.
I only use this for debugging purposes.

the-butterfry, 2021-04-30:
Just an update as to how I'm putting Grott to use. 
I'm using an Overkill Solar BMS in my 24v LiFepo4 battery. It has a nice Bluetooth LE interface to it. From the Raspberry pi i'm pulling data in from both Grott and the BMS. 

Over the past few days, @tgalarneau and I adapted his code at https://github.com/tgalarneau/bms to accept my 8cell version of the BMS. This was the first I'd seen of anyone getting all that telemetry off of these into a useful format from BT. 

[image]

As you can see, i'm running it all within Home Assistant. There is a bunch more to fill out and automate now, but the proof of concept is done. Growatt SPF 3000 and Overkill BMS with full working telemetry. 

I want to thank you guys for the help in getting this going for me. Let me know if I can be of further assistance in tweaking the json configs for Grott.

johanmeijer, 2021-05-08:
Wow! Looks good. 

If you thinks values are not ok or missing please let me know.

the-butterfry, 2021-05-09:
Ok, after a while running grott there seems to be something fishy with the buffered file. I'm not in front of the unit at the moment but I'll dig into it in a few days and post up about it.

johanmeijer, 2021-05-09:
Ok. I am curious what goes wrong.

charliesjc, 2021-05-23:
Well I suppose at this point if you make use of the json files for the data, it only works for the header specified in the json file. When the buffered file comes along, in my case as a 050150 header it doesn't use the data from the json file but rather the default in the grott_conf.py which for me is T05NNNN and won't have the correct mapping at all. My buffered data used to work because I modified the grott_conf file instead of using the json file as I was using the Home Assistant implementation and hadn't worked out how to import the external config files. I since have worked that out and my buffered incoming data should also now be messed up, although I haven't seen it because I've been up for weeks now without a problem.

johanmeijer, 2021-05-23:
I will publish 2.6 release soon with support for spf and sph inverters without configuring seperate .json files.

the-butterfry, 2021-06-03:
ok, FINALLY getting around to this. And yeah, it was the buffered data, exactly as you said. A side benefit of that is I can always tell when I loose internet in my graphs though. I'll get a bunch of out of bounds data

Aside from that I've been up and running, what a month now and the normal .json data is just fine.