johanmeijer/grott #47: KeyError: 'pvenergytoday'

farsonic, 2021-05-15:
Running this with a new 5000TL-X and have the following error message. 

As an aside, how do I also revert the inverter to point back to growatt until this issue is resolved? I can't seem to reset it back to default parameters! 

	 - MQTT jsonmsg:
			 {"device": "HMG3A510SC", "time": "2021-05-15T13:18:24", "buffered": "no",
			 "values": {"#recortype": 3124, "pvstatus": 1, "pvpowerin": 46505,
			 "pv1voltage": 3725, "pv1current": 81, "pv1watt": 30447, "pv2voltage": 1824,
			 "pv2current": 89, "pv2watt": 16058, "pvpowerout": 0, "pvfrequentie": 0,
			 "pvgridvoltage": 2418, "pvgridcurrent": 0, "pvgridpower": 0,
			 "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 2211446784,
			 "pvgridvoltage3": 11828, "pvgridcurrent3": 5, "pvgridpower3": 2844852224,
			 "totworktime": 679, "eactoday": 1137, "eactotal": 100, "epvtotal": 6684672,
			 "epv1today": 88, "epv1total": 458, "epv2today": 0, "epv2total": 0,
			 "pvtemperature": 581, "pvipmtemperature": 0, "pvboosttemp": 0, "bat_dsp":
			 526, "pbusvolt": 4020, "#nbusvolt": 0, "#ipf": 20000, "#realoppercent": 91,
			 "#opfullwatt": 0, "#deratingmode": 0, "eacharge_today": 0, "eacharge_total":
			 0, "batterytype": 0, "uwsysworkmode": 0, "systemfaultword1": 0,
			 "systemfaultword2": 0, "systemfaultword3": 0, "systemfaultword4": 0,
			 "systemfaultword5": 0, "systemfaultword6": 0, "systemfaultword7": 0,
			 "pdischarge1": 0, "p1charge1": 0, "vbat": 0, "SOC": 0, "pactouserr": 0,
			 "#pactousers": 0, "#pactousert": 0, "pactousertot": 0, "pactogridr": 0,
			 "#pactogrids": 0, "#pactogridt": 0, "pactogridtot": 0, "plocaloadr": 0,
			 "#plocaloads": 0, "#plocaloadt": 0, "plocaloadtot": 0, "#ipm": 0,
			 "#battemp": 0, "spdspstatus": 0, "spbusvolt": 0, "etouser_tod": 0,
			 "etouser_tot": 0, "etogrid_tod": 0, "etogrid_tot": 0, "edischarge1_tod": 0,
			 "edischarge1_tot": 0, "eharge1_tod": 0, "eharge1_tot": 0, "elocalload_tod":
			 0, "elocalload_tot": 0}}
	 - MQTT connection refused by target
	 - Grott send data to PVOutput systemid:  9144 for inverter:  HMG3A510SC
Traceback (most recent call last):
  File "grott.py", line 35, in <module>
    proxy.main(conf)
  File "/home/pi/grott/grottproxy.py", line 85, in main
    self.on_recv(conf)
  File "/home/pi/grott/grottproxy.py", line 165, in on_recv
    procdata(conf,data)
  File "/home/pi/grott/grottdata.py", line 382, in procdata
    "v1"    : definedkey["pvenergytoday"]*100,
KeyError: 'pvenergytoday'

johanmeijer, 2021-05-15:
Interesting error. How do you use Grott? Docker or standalone and which version are you using? 

You running a 5000TL-X but get values if your using a SPH. Did you copy a .json file from the example directory in the main directory or do you use the grott:2.5.4sph RPI docker container (only ment for testing with a SPH device). 

Falling back with a failing Grott is a challenge. You can disable pvoutput in Grott (pvoutput = False in .ini or gpvoutput = false environmental) and the error will not occur (grott runs and will keep the communication with Growatt alive). Then you can reset your IP adres via the growatt web server. 

Resetting and reinstall the growatts shinewifi (shinelan or other device) will also be a possibility. 

This does not seems a error that can not be solved. Grott using the wrong layout definitions but that can be repaired. 

First repair: in case of plain Grott on Linux/windows do not copy .json files in main directory. If you use grottrpi container use ledidobe/grottrpi:latest or ledidobe/grottrpi:2.5.2 (no tag should also refer to latest I think). 

It seems a little bit weird but thanks for having this error. I point me to an other challenge in Grott (pvoutput can now not be used with an SPH device). 

For a 5000TL-X using Grott with PVoutput.org should not give any issues.

farsonic, 2021-05-15:
This is just running natively on a raspberry Pi. I cloned the GitHub and
modified the .ini file and that’s about it.

Will get more for you tomorrow if needed. I also do have a power meter
attached also. Not sure if this is a problem?

F

johanmeijer, 2021-05-15:
An additional power meter should not be a problem. 

I made a mistake, sorry for that. The t060104sph.json should not be installed in the main directory by default (only when using a SPH type of inverter). In the next version of Grott (2.6 now in tested) this should not be a problem any more. 

I deleted the file from Gitt hub master/main directory. You can clone Grott again or delete the t060104spf.json from your RPI Grott directory. After that is should work fine.

farsonic, 2021-05-15:
Thanks, I’ll test this now.

Re

Francois

farsonic, 2021-05-15:
ok, that fixed the problem :) I now have data going into pvoutput.org. How do I go about getting the consumption from the smart meter captured at the same time?

farsonic, 2021-05-16:
- Grott data record date/time processing started
	 - date-time:  2021-05-16T10:02:04
	 - Grott values retrieved:
		 -  pvserial             :  HMG3A510SC
		 -  pvstatus             :  1
		 -  pvpowerin            :  3450.1
		 -  pv1voltage           :  396.0
		 -  pv1current           :  4.2
		 -  pv1watt              :  1687.1
		 -  pv2voltage           :  188.5
		 -  pv2current           :  9.4
		 -  pv2watt              :  1763.0
		 -  pvpowerout           :  3381.2   <- Current solar panel generation
		 -  pvfrequentie         :  50.0
		 -  pvgridvoltage        :  240.5
		 -  pvgridcurrent        :  13.9
		 -  pvgridpower          :  3383.8
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  0.0
		 -  pvgridvoltage3       :  0.0
		 -  pvgridcurrent3       :  0.0
		 -  pvgridpower3         :  0.0
		 -  totworktime          :  59.2
		 -  pvenergytoday        :  4.8
		 -  pvenergytotal        :  128.4
		 -  epvtotal             :  126.9
		 -  epv1today            :  1.6
		 -  epv1total            :  76.0
		 -  epv2today            :  3.1
		 -  epv2total            :  50.9
		 -  pvtemperature        :  42.2
		 -  pvipmtemperature     :  37.8
	 - MQTT jsonmsg:
			 {"device": "HMG3A510SC", "time": "2021-05-16T10:02:04", "buffered": "no",
			 "values": {"pvstatus": 1, "pvpowerin": 34501, "pv1voltage": 3960,
			 "pv1current": 42, "pv1watt": 16871, "pv2voltage": 1885, "pv2current": 94,
			 "pv2watt": 17630, "pvpowerout": 33812, "pvfrequentie": 4998,
			 "pvgridvoltage": 2405, "pvgridcurrent": 139, "pvgridpower": 33838,
			 "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0,
			 "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime":
			 426022, "pvenergytoday": 48, "pvenergytotal": 1284, "epvtotal": 1269,
			 "epv1today": 16, "epv1total": 760, "epv2today": 31, "epv2total": 509,
			 "pvtemperature": 422, "pvipmtemperature": 378}}
	 - MQTT message message sent
	 - Grott send data to PVOutput systemid:  9144 for inverter:  HMG3A510SC

johanmeijer, 2021-05-16:
> ok, that fixed the problem :) I now have data going into pvoutput.org. How do I go about getting the consumption from the smart meter captured at the same time?

I believe it is not really necessary to send the consumption information from the smart meter at the same time as the energy generation information . PVOutput.org will try to match generation and consumption over an specific interval (I think 5 minutes because that is the interval PVOutput uses). 

I sent my smart meter info via a interface (we have a P1 interface port at the meters in the Netherlands) I created myself. It is based on a esp8266 that sent a MQTT message and I use Nodered to read this message and sent it to PVOoutput. 

What kind of  interface do you use to measure the consumption?

farsonic, 2021-05-16:
I use this meter, single phase version.

https://www.ginverter.com/Accessories/11-608.html

It is connected to the inverter via modbus and when the values for the
meter should be present in the data packet pushed to the shine server.

They are visible in the IOS app for example

F

johanmeijer, 2021-05-17:
Ok. I Haven't seen this device before. I am not sure if the information is in the same data record as the inverter information. We can try to look for that. I might take some time. I have to find out where this information is stored in the record. 
Or maybe there is separate record generated for this. 

Can you see the values in the inverter history data report on the growatt website (dashboard/history data spreadsheet that can be exported to CSV file)?  

To find where if and where it is stored I need the Grott trace data, example: 
- Growatt original Data:
      \x00\x65\x00\x06\x01\x01\x01\x04\x0d\x22\x2c\x45\x59\x45\x4c\x74\x41\x2d\x77
      \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
      \x3e\x3a\x23\x46\x4c\x75\x41\x5d\x41\x50\x74\x74\x47\x72\x6f\x77\x61\x74\x74
      \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x71\x65\x49\x7e\x6a\x75\x61
      \x74\x74\x6b\x72\x6e\x77\x61\x4f\xfd\x4f\x44\x6f\x3f\x61\x74\x4f\xce\x72\x6f
      \x77\x61\x74\x74\x47\x72\x6f\x77\x58\xb4\x67\xc0\x7b\x04\x77\x5a\x74\x74\x7e
      \xb2\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
      \x74\x50\x72\x6f\xc8\x8a\x70\x17\x2f\xca\x6e\x53\x61\x74\x74\x47\x72\x6f\x77
      \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x7c\xcc\x77\x61\x74\x74\x47\x5f
      \x6f\x2e\x2f\x54\x74\x47\x72\x6f\x77\x61\x74\x6c\x47\x72\xa6\xdb\x61\x74\x74
      \x47\x72\x6f\x77\x61\x74\x74\x8e\xde\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
      \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x78\x6f
      \x77\x68\xf3\x7c\x48\x7a\x6a\x77\x60\x7d\x9f\x07\x06\x6e\xa2\x6f\xb5\x70\x1a
      \x7c\xae\x77\x61\x74\x74\x47\x59\x6f\x77\x61\x74\x74\x47\x72\x6f\x40\xf3

And the history data from the growatt server at the same time (in csv format would be nice). 

It will be some kind of puzzle while I do not see this specified in the data record descriptions I have.

farsonic, 2021-05-17:
Here is the data....I'll check on the spreadsheet. 

 - Growatt original Data:
		 \x00\x09\x00\x06\x00\xe6\x01\x20\x1f\x35\x2b\x43\x20\x41\x45\x75\x40\x56\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x27\x3a\x26\x47\x35\x72\x43\x5f\x24\x22\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x71\x65\x40\x68\x40\x77\xc1
		 \x74\x74\x47\xf6\x6f\x77\x68\x55\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6e
		 \x71\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\xab\xdc\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x99\xc8\x47\x72\x6f\x77\x61\x74\x74\x47\x8d\x90\x9e\xc3\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x71\x8c\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x98\xe9\x72\x6f\x9a\xdd\x8b\x8b\xae\xd0\x6f\x77\x62\x97\x74\x47\x73
		 \x9b\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x7c\xc1\x74\x74
		 \x44\x13\x6f\x77\x61\x66\x74\x47\x73\xa9\x77\x61\x74\x74\x47\x72\x60\x76\x61
		 \x74\x75\x9e\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\xf9\x57

farsonic, 2021-05-17:
[XGD4A51229_1 Electric-meter data - 2021-05-17_2021-05-18.xls](https://github.com/johanmeijer/grott/files/6496789/XGD4A51229_1.Electric-meter.data.-.2021-05-17_2021-05-18.xls)

farsonic, 2021-05-17:
[HMG3A510SC tlx data - 2021-05-17_2021-05-18.xls](https://github.com/johanmeijer/grott/files/6496793/HMG3A510SC.tlx.data.-.2021-05-17_2021-05-18.xls)

johanmeijer, 2021-05-18:
Thanks for the info. I did not release there is a separate record being sent to growatt. In Grott terminology 060120 with the values from the smart monitor. 

The bad news is a do not have any description of the record layout of this. The good news is I almost were able to decode it using the xls. At this moment I miss some values (not being in the record) and some values are calculated I think. Next to it it will be hard to determine values that are zero. 

I will go on with processing of the record. Then we can verify it and see if it usable to send to PVOUTPUT (which value do we need to sent?). I think it might be a problem that Positive active today energy(kWh) and Reverse active today energy(kWh) are calculated). That will be a challenge with Grott (that does not have any memory functionality at this moment) but maybe PVOUTPUT can solve that for us. 

If you can sent a grott record and xls from during the day (you generate more power then use) the will be very helpfull to determine values (e.g. reverse active power).

johanmeijer, 2021-05-18:
The  current status: 

[image]

x1-x10 I still have to label but can mostly retrieved of the spreadsheet. For the reverse active power I need an other record as described above.

farsonic, 2021-05-18:
[XGD4A51229_1 Electric-meter data - 2021-05-19_2021-05-19.xls](https://github.com/johanmeijer/grott/files/6504745/XGD4A51229_1.Electric-meter.data.-.2021-05-19_2021-05-19.xls)
[HMG3A510SC tlx data - 2021-05-19_2021-05-19.xls](https://github.com/johanmeijer/grott/files/6504746/HMG3A510SC.tlx.data.-.2021-05-19_2021-05-19.xls)

farsonic, 2021-05-18:
what exactly are you after from the spreadsheet. I've just uploaded the spreadsheets from yesterday.

farsonic, 2021-05-18:
- Growatt original Data:
		 \x00\x03\x00\x06\x02\x41\x01\x03\x1f\x35\x2b\x43\x20\x41\x45\x75\x40\x56\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x27\x3a\x26\x47\x35\x72\x43\x5f\x24\x22\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x71\x67\x4f\x6f\x75\x75\x61
		 \x74\x74\x3b\x72\x6e\x77\x38\x74\x74\x47\x16\x6f\x77\x46\x64\x74\x47\xb1\x3f
		 \x74\x89\x35\x3f\x76\x5c\x5f\x77\x3b\x35\x35\x06\x72\x66\x77\x60\x74\x74\x44
		 \x9a\x6f\x36\x61\x35\x74\xe0\x72\xc8\x77\x61\x3c\x39\x00\x41\x2e\x42\x50\x44
		 \x27\x04\x72\x6f\x77\x61\x74\x75\x47\x16\x6f\x77\x61\x74\x54\x67\x52\x3f\x21
		 \x41\x3d\x1a\x31\x17\x1d\x03\x04\x06\x54\x67\x72\x6f\x63\x31\x76\x75\x40\x97
		 \x6f\x72\x61\x67\x74\x4f\x72\x71\x77\x41\x74\x77\x40\x7a\x65\x5f\x73\x28\x60
		 \x17\x75\x67\x7d\x3b\x66\x28\x53\x22\x6f\x77\x61\x74\x74\x47\x72\x6f\x70\x9e
		 \x7d\x96\x55\xfc\x7c\xe0\x61\x46\x74\x75\x72\x5d\x77\x64\x74\x46\x47\x77\x6f
		 \x45\x61\x71\x74\x75\x72\x5d\x77\x61\x74\x74\x4d\x66\x6f\x77\x20\x3f\x35\x06
		 \x42\x56\x47\x50\x54\x54\x77\x4b\x6e\x46\x61\x74\x74\x47\x61\xce\x77\x42\x7d
		 \x14\x4d\x66\x67\xef\x69\x62\x74\x53\x72\x6a\x7e\x0e\x7c\x88\x60\x62\x48\x67
		 \x46\x64\x53\x57\x55\x7f\x50\x71\x74\x74\x47\x72\x6e\xcf\x61\x8b\x3a\x67\x72
		 \x90\x39\x41\x74\x8b\x09\x52\x6f\x88\x2f\x54\x5a\x46\x72\x6f\x78\x61\x74\x46
		 \x47\x73\x6c\x9f\x61\x74\x7f\xff\x7e\x5b\x77\x61\x3c\x39\x00\x41\x2e\x42\x50
		 \x44\x27\x04\x42\x5f\x47\x51\x44\x44\x77\x42\x5f\x47\x51\x44\x44\x77\x42\x5f
		 \x47\x51\x44\x44\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xc0\x01

farsonic, 2021-05-18:
[image]

farsonic, 2021-05-18:
[image]

farsonic, 2021-05-18:
Here is the next record five minutes after that last one

- Growatt original Data:
		 \x00\x1d\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x43\x20\x41\x45\x75\x40\x56\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x27\x3a\x26\x47\x35\x72\x43\x5f\x24\x22\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x71\x67\x4f\x56\x41\x75\x6a
		 \xcc\x78\x73\x72\x6e\x77\x61\x51\xf7\x49\x89\x6f\x7d\x61\x74\x64\xf9\x75\xb5
		 \x77\x78\x74\x74\x53\xb7\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x63\xb3\x7c\xfa\x68\x19
		 \x74\x6e\x72\x6f\x52\x50\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x4e\x1f\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x71\x40\x47\x72
		 \x70\xfa\x61\x7d\x9e\xc9\x72\x6f\x77\x69\x74\x74\x4f\x71\x6f\x77\x66\x9a\x74
		 \x47\x72\x6c\x77\x61\x70\xbd\x47\x72\x6f\x72\x61\x74\x77\x62\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x7b\x55\x16\x6f\xe1\x61\x74\x74\x47\x72\x01\x77\x61\x75
		 \x7e\x47\x72\x6f\x77\x61\x74\x75\x06\x62\x74\x77\x61\x3a\x54\x47\x60\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\xd4\x74\x47\x72\x6f\x77\x61\x74\x30\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x4b\x3a
	 - Grott automatic protocol detection

farsonic, 2021-05-18:
Here is one just now where consumption is more than generation

	 - Growatt original Data:
		 \x00\x05\x00\x06\x02\x41\x01\x03\x1f\x35\x2b\x43\x20\x41\x45\x75\x40\x56\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x27\x3a\x26\x47\x35\x72\x43\x5f\x24\x22\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x71\x67\x4f\x5e\x66\x75\x61
		 \x74\x74\x3b\x72\x6e\x77\x38\x74\x74\x47\x16\x6f\x77\x46\x64\x74\x47\xb1\x3f
		 \x74\x89\x35\x3f\x76\x5c\x5f\x77\x3b\x35\x35\x06\x72\x66\x77\x60\x74\x74\x44
		 \x9a\x6f\x36\x61\x35\x74\xe0\x72\xc8\x77\x61\x3c\x39\x00\x41\x2e\x42\x50\x44
		 \x27\x04\x72\x6f\x77\x61\x74\x75\x47\x16\x6f\x77\x61\x74\x54\x67\x52\x3f\x21
		 \x41\x3d\x1a\x31\x17\x1d\x03\x04\x06\x54\x67\x72\x6f\x63\x31\x76\x75\x40\x97
		 \x6f\x72\x61\x67\x74\x4f\x72\x43\x77\x5a\x74\x77\x40\x7a\x65\x5f\x73\x28\x60
		 \x17\x75\x67\x7d\x3b\x66\x28\x53\x22\x6f\x77\x61\x74\x74\x47\x72\x6f\x70\x9e
		 \x7d\x96\x55\xfc\x7c\xe0\x61\x46\x74\x75\x72\x5d\x77\x64\x74\x46\x47\x77\x6f
		 \x45\x61\x71\x74\x75\x72\x5d\x77\x61\x74\x74\x4d\x66\x6f\x77\x20\x3f\x35\x06
		 \x42\x56\x47\x50\x54\x54\x77\x4b\x6e\x46\x61\x74\x74\x47\x61\xce\x77\x42\x7d
		 \x14\x4d\x66\x67\xef\x69\x62\x74\x53\x72\x6a\x7e\x0e\x7c\x88\x60\x62\x48\x67
		 \x46\x64\x53\x57\x55\x7f\x50\x71\x74\x74\x47\x72\x6e\xcf\x61\x8b\x3a\x67\x72
		 \x90\x39\x41\x74\x8b\x09\x52\x6f\x88\x2f\x54\x5a\x46\x72\x6f\x78\x61\x74\x46
		 \x47\x73\x6c\x9f\x61\x74\x7f\xff\x7e\x5b\x77\x61\x3c\x39\x00\x41\x2e\x42\x50
		 \x44\x27\x04\x42\x5f\x47\x51\x44\x44\x77\x42\x5f\x47\x51\x44\x44\x77\x42\x5f
		 \x47\x51\x44\x44\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x68\xd0

johanmeijer, 2021-05-19:
Hi,  Thank you for the records (it is help full for testing). But what I really need is the same record as you send me te first day only at an other time. 

I need a record were grott after de - Growatt orignal data (the \x data above) write the messages: 
- Grott automatic protocol detection
- Grott data record length 265
- layout   :  T060120

The records you sent above are other (not T060120) records with other information (but I can use the T060104X for testing the solar for the 5000TL-X output information). 

The Excel output you sent was ok. But I need it also a the time the new record is created.

farsonic, 2021-05-19:
- Growatt original Data:
		 \x00\x1d\x00\x06\x00\xe6\x01\x20\x1f\x35\x2b\x43\x20\x41\x45\x75\x40\x56\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x27\x3a\x26\x47\x35\x72\x43\x5f\x24\x22\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x71\x60\x4f\x53\x45\x77\xc1
		 \x74\x74\x47\xf6\x6f\x77\x68\x23\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x47\x61\x74\x74\x47\x72\x6f\x77\x61\x8b\x8b\x92\xa0\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x58\x3d\x47\x72\x6f\x77\x61\x74\x74\x47\x8d\x90\x85\xe5\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x8b\xb8\x8e\x27\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x9e\x8b\xa1\x95\x72\x6f\x5b\x28\x8b\x8b\xb5\xf6\x90\x88\x9d\x3c\x74\x47\x73
		 \x9c\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x67\x23\x74\x74
		 \x42\x7a\x6f\x77\x61\x45\x74\x47\x70\xec\x77\x61\x74\x74\x47\x72\x7a\x3c\x61
		 \x74\x76\xf3\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x18\x7b
	 - Grott automatic protocol detection
	 - Grott data record length 238
	 - layout   :  T060120
	 - no matching record layout found, try generic
	 - Record layout used :  T060120
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 001d000600e601205847443441353132323900000000000000000000000000000000000000004
		 84d4733413531305343000000000000000000000000000000000000000015051408212a00a000
		 000084000009570000000000000000000000300000000000000000ffffd5d2000000000000000
		 000002c490000000000000000fffff2840000000000000000fffffc480000000000000000ffff
		 d5d200002c49fffff284fffffc48000001f300000000000000000000000000001042000005080
		 000003100000283000000000000154b000002b400000000000000000000000000000000000000
		 0000000000790f

farsonic, 2021-05-19:
will upload the next capture also...

farsonic, 2021-05-19:
- Growatt original Data:
		 \x00\x1e\x00\x06\x00\xe6\x01\x20\x1f\x35\x2b\x43\x20\x41\x45\x75\x40\x56\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x27\x3a\x26\x47\x35\x72\x43\x5f\x24\x22\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x71\x60\x4f\x54\x40\x77\xc1
		 \x74\x74\x47\xf6\x6f\x77\x68\x39\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x43\x61\x74\x74\x47\x72\x6f\x77\x61\x8b\x8b\x95\x01\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x5b\x80\x47\x72\x6f\x77\x61\x74\x74\x47\x8d\x90\x86\x62\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x8b\xb8\x8e\x24\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x9e\x8b\xa6\x34\x72\x6f\x58\x95\x8b\x8b\xb6\x71\x90\x88\x9d\x3f\x74\x47\x73
		 \x9b\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x67\x23\x74\x74
		 \x42\x7b\x6f\x77\x61\x45\x74\x47\x70\xeb\x77\x61\x74\x74\x47\x72\x7a\x3b\x61
		 \x74\x76\xf2\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x4e\x17
	 - Grott automatic protocol detection
	 - Grott data record length 238
	 - layout   :  T060120
	 - no matching record layout found, try generic
	 - Record layout used :  T060120
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 001e000600e601205847443441353132323900000000000000000000000000000000000000004
		 84d4733413531305343000000000000000000000000000000000000000015051408262f00a000
		 0000840000094d0000000000000000000000340000000000000000ffffd273000000000000000
		 000002ff40000000000000000fffff1030000000000000000fffffc4b0000000000000000ffff
		 d27300002ff4fffff103fffffc4b000001f400000000000000000000000000001042000005090
		 000003100000284000000000000154c000002b500000000000000000000000000000000000000
		 00000000002f63

farsonic, 2021-05-19:
[XGD4A51229_1 Electric-meter data - 2021-05-20_2021-05-20.xls](https://github.com/johanmeijer/grott/files/6511840/XGD4A51229_1.Electric-meter.data.-.2021-05-20_2021-05-20.xls)
[HMG3A510SC tlx data - 2021-05-20_2021-05-20.xls](https://github.com/johanmeijer/grott/files/6511841/HMG3A510SC.tlx.data.-.2021-05-20_2021-05-20.xls)

johanmeijer, 2021-05-20:
Thanks! I almost there. I will publish version 2.6  with support for the smart monitor soon.

johanmeijer, 2021-05-23:
I will publish  2.6 release te comming days (running in test now) which will be able to handle the smart meter data and send it to mqtt, inluxdb and PVOutput.

For PVOutput we have to test if grott is sending the write data. Choosing the right values and sent this in right way to PVOutput.org is a little bit complex.

farsonic, 2021-05-23:
Thanks!!

Im Happy to help test

Re

Francois

johanmeijer, 2021-05-24:
Hi Francois,

I created branch 2.6 that should handle the Smart Meter info. 

I am very curious at the results. 

Johan

farsonic, 2021-05-24:
install 2.6...seeing the folllowing.

92.168.0.218', 44888), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x04\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x43\x20\x41\x45\x75\x40\x56\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x27\x3a\x26\x47\x35\x72\x43\x5f\x24\x22\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x71\x6d\x41\x51\x5c\x75\x6a
		 \xcc\x78\x73\x72\x6e\x77\x61\x74\x69\x48\xad\x6f\x77\x61\x74\x74\x4e\x74\x73
		 \x77\x61\x74\x74\x47\x66\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x68\x7c\xf2\x68\x5b
		 \x74\x43\x72\x6f\x77\x49\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x4e\x5d\x6f\x77\x61\x74\x74\x47\xce\x9e\x77\x61\x74\x74\x47\x72
		 \xd2\x7c\x61\x64\xbc\x04\x72\x6f\x77\x61\x74\x74\x4a\x57\x6f\x77\x6c\x72\x74
		 \x47\x72\x6f\x77\x61\x73\xad\x47\x72\x6f\x77\x61\x74\x71\x6a\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x7b\x48\xd4\x64\xbb\x61\x74\x74\x47\x72\x1b\x77\x61\x74
		 \xea\x47\x72\x6f\x77\x61\x74\x74\x9c\x63\x79\x77\x61\x3a\x54\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\xd4\x74\x47\x72\x6f\x77\x61\x74\x15\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xf0\x2d
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060104X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00040006024101045847443441353132323900000000000000000000000000000000000000004
		 84d47334135313053430000000000000000000000000000000000000000150519062333020bb8
		 0c3400010000001d0fdf000000000009061c00000000001400000000000000000000000000000
		 00000000000000000000000001a1385092f000400000028000000000000000000000000000000
		 00092f000000000000bcf1000000000000bd0b0010c8430000000000000d2500000d060000000
		 0000007d9000000000000052d0000000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000f0fa60bcc0000000000740000009e00000
		 000000000db111600004e20000000000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000c350cb1000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000a00000000000000061000000000000000000000000000000000
		 000000000008242
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNX

	 - Grott data record date/time processing started
	 - date-time:  2021-05-25T06:35:51
	 - Grott values retrieved:
		 -  pvserial             :  HMG3A510SC
		 -  pvstatus             :  1
		 -  pvpowerin            :  2.9
[... 96 lines cut ...]
	 - Grott values retrieved:
		 -  datalogserial        :  XGD4A51229
		 -  pvserial             :  HMG3A510SC
		 -  voltage_l1           :  234.8
		 -  Current_l1           :  20.5
		 -  act_power_l1         :  4796.0
		 -  app_power_l1         :  4810.1
		 -  react_power_l1       :  -367.9
		 -  powerfactor_l1       :  1.0
		 -  pos_rev_act_power    :  4796.0
		 -  app_power            :  4810.1
		 -  react_power          :  -367.9
		 -  powerfactor          :  1.0
		 -  frequency            :  50.0
		 -  pos_act_energy       :  538.4
		 -  rev_act_energy       :  204.4
	 - MQTT jsonmsg:
			 {"device": "XGD4A51229", "time": "2021-05-25T06:35:53", "buffered": "no",
			 "values": {"voltage_l1": 2348, "Current_l1": 205, "act_power_l1": 47960,
			 "app_power_l1": 48101, "react_power_l1": -3679, "powerfactor_l1": 997,
			 "pos_rev_act_power": 47960, "app_power": 48101, "react_power": -3679,
			 "powerfactor": 997, "frequency": 500, "pos_act_energy": 5384,
			 "rev_act_energy": 2044}}
	 - MQTT message message sent
	 -

farsonic, 2021-05-25:
Looks great ! 

[image]

johanmeijer, 2021-05-25:
Nice :)

farsonic, 2021-05-31:
Getting some real bogus values into the system at the moment.

[image]

johanmeijer, 2021-05-31:
Oh you used a lot of energy ;). 

Can you find the grott output at that time or is that already gone? Maybe we can see what happened.

johanmeijer, 2021-06-02:
@farsonic

 I looked if I made a mistake in the layout definitions. I have not found it yet. 

Does it occurs more often? Are you able to capture the Grott log data at that time?

SebastianHarnau, 2021-08-13:
[image]

Same problem here... here are some parts of log file:

`Aug 13 05:29:12 grott grott[278182]: #011 - Growatt packet received:
Aug 13 05:29:12 grott grott[278182]: #011#011  <socket.socket fd=14, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.170.98', 34236), raddr=('47.91.67.66', 5279)>
Aug 13 05:29:12 grott grott[278182]: #011 - Growatt original Data:
Aug 13 05:29:12 grott grott[278182]: #011#011 \x00\x10\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x44\x20\x46\x4d\x75\x24\x5e\x77
Aug 13 05:29:12 grott grott[278182]: #011#011 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Aug 13 05:29:12 grott grott[278182]: #011#011 \x28\x36\x29\x44\x35\x73\x44\x5f\x42\x2f\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Aug 13 05:29:12 grott grott[278182]: #011#011 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x7c\x79\x40\x6f\x63\x75\x6a
Aug 13 05:29:12 grott grott[278182]: #011#011 \xcc\x78\x73\x72\x6e\x77\x61\x77\xbe\x43\x75\x6f\x7e\x61\x74\x77\x8d\x72\x6f
Aug 13 05:29:12 grott grott[278182]: #011#011 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Aug 13 05:29:12 grott grott[278182]: #011#011 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x44\xcb\x7c\xfd\x69\xac
Aug 13 05:29:12 grott grott[278182]: #011#011 \x74\x4c\x72\x6f\x74\xaa\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
Aug 13 05:29:12 grott grott[278182]: #011#011 \x61\x74\x74\x4f\xaa\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Aug 13 05:29:12 grott grott[278182]: #011#011 \x6f\x77\x61\xcf\x89\x9e\x72\x6f\x77\x61\x74\x74\x57\x0f\x6f\x77\x71\x11\x74
Aug 13 05:29:12 grott grott[278182]: #011#011 \x47\x72\x6f\x77\x61\x64\x11\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Aug 13 05:29:12 grott grott[278182]: #011#011 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Aug 13 05:29:12 grott grott[278182]: #011#011 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Aug 13 05:29:12 grott grott[278182]: #011#011 \x72\x6f\x77\x61\x74\x74\xb8\x8d\x65\x19\x61\x74\x74\x47\x72\x0b\x77\x61\x75
Aug 13 05:29:12 grott grott[278182]: #011#011 \x5e\x47\x72\x6f\x77\x61\x74\x75\x36\x7f\x90\x77\x61\x3a\x54\x47\x7d\x6f\x77
Aug 13 05:29:12 grott grott[278182]: #011#011 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Aug 13 05:29:12 grott grott[278182]: #011#011 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Aug 13 05:29:12 grott grott[278182]: #011#011 \x47\x72\x6f\x77\x61\x74\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Aug 13 05:29:12 grott grott[278182]: #011#011 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Aug 13 05:29:12 grott grott[278182]: #011#011 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Aug 13 05:29:12 grott grott[278182]: #011#011 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Aug 13 05:29:12 grott grott[278182]: #011#011 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
Aug 13 05:29:12 grott grott[278182]: #011#011 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Aug 13 05:29:12 grott grott[278182]: #011#011 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Aug 13 05:29:12 grott grott[278182]: #011#011 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Aug 13 05:29:12 grott grott[278182]: #011#011 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Aug 13 05:29:12 grott grott[278182]: #011#011 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Aug 13 05:29:12 grott grott[278182]: #011#011 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Aug 13 05:29:12 grott grott[278182]: #011#011 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
Aug 13 05:29:12 grott grott[278182]: #011#011 \x61\x7c\x74\x05\x72\x6f\x74\x94\x7a\x64\x47\x72\x6d\x77\x61\x51\x74\x47\x72
Aug 13 05:29:12 grott grott[278182]: #011#011 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x5b\x0c
Aug 13 05:29:12 grott grott[278182]: #011 - Grott automatic protocol detection
Aug 13 05:29:12 grott grott[278182]: #011 - Grott data record length 585
Aug 13 05:29:12 grott grott[278182]: #011 - layout   :  T060104X
Aug 13 05:29:12 grott grott[278182]: #011 - no matching record layout found, try generic
Aug 13 05:29:12 grott grott[278182]: #011 - Record layout used :  T06NNNNX
Aug 13 05:29:12 grott grott[278182]: #011 - Growatt data decrypted V2
Aug 13 05:29:12 grott grott[278182]: #011 - Grott Growatt data decrypted
Aug 13 05:29:12 grott grott[278182]: #011 - Growatt plain data:
Aug 13 05:29:12 grott grott[278182]: #011#011 00100006024101045847443341323932563100000000000000000000000000000000000000004
Aug 13 05:29:12 grott grott[278182]: #011#011 741483041343630354e000000000000000000000000000000000000000015080d071d0c020bb8
Aug 13 05:29:12 grott grott[278182]: #011#011 0c340001000003ca04070009000003ca000000000000000000000000000000000000000000000
Aug 13 05:29:12 grott grott[278182]: #011#011 0000000000000000000000003b9138a08d8000b000003cb000000000000000000000000000000
Aug 13 05:29:12 grott grott[278182]: #011#011 0008d80000000000000000000000000000000000bbfdd9000000000000107d000010650000000
[... cut ...]
Aug 13 05:29:19 grott grott[278182]: #011#011 -  {'d': '20210813', 't': '07:29', 'v4': 195083926.6, 'v6': 69233598.7, 'n': 1}
Aug 13 05:29:20 grott grott[278182]: #011 - Grott PVOutput response SM1:
Aug 13 05:29:20 grott grott[278182]: #011#011 -  OK 200: Added Status
Aug 13 05:29:20 grott grott[278182]: #011 - Grott PVOutput response SM2:
Aug 13 05:29:20 grott grott[278182]: #011#011 -  OK 200: Added Status
Aug 13 05:29:20 grott grott[278182]: #011 - Grott Send data to Influx disabled
Aug 13 05:29:20 grott grott[278182]: #011 - Grott extension processing disabled
`

farsonic, 2021-08-15:
Let me know if I need to capture any more information...I'm still having this issue.

johanmeijer, 2021-08-16:
@farsonic @SebastianHarnau 

This might be a difficult one. 

I found a error in the record. There seems to be a scramble (encrypt) error but this is already made before it is being sent to Grott (and the Growatt server) while the CRC (cyclic redundancy check to test if record is corrupted) is ok. 
This means that the error is made in the inverter (shine stick) / smart monitor (I am not sue where encryption is done). 

It could be mistake in the Grott decrypt routine but until know I could not find that and most of the time it works fine. 

Can you see in the data at the Growatt Server if at the time of the error a Smart Meter record is created? If they ignore it maybe the best way for Grott to do is to ignore these "faulty" records as well.  
If there is a record I have to try to debug the decrypt routine.

farsonic, 2021-08-17:
Let me see if I can dig up some additional logs or output

Re

Francois

farsonic, 2021-08-29:
I've been logging every capture for the last 5 days and have a few examples in the logs, this entry for example shows up in PVOUTPUT as power used of 195,085,140W (timestamp 2021-08-28T15:56:11)

I don't see PVUSED date in the logs though? 

         - Grott data record date/time processing started
         - date-time:  2021-08-28T15:56:11
         - Grott values retrieved:
                 -  pvserial             :  HMG3A510SC
                 -  pvstatus             :  1
                 -  pvpowerin            :  927.8
                 -  pv1voltage           :  344.7
                 -  pv1current           :  2.4
                 -  pv1watt              :  862.1
                 -  pv2voltage           :  181.0
                 -  pv2current           :  0.3
                 -  pv2watt              :  65.7
                 -  pvpowerout           :  903.1
                 -  pvfrequentie         :  50.0
                 -  pvgridvoltage        :  238.8
                 -  pvgridcurrent        :  4.0
                 -  pvgridpower          :  953.3
                 -  pvgridvoltage2       :  0.0
                 -  pvgridcurrent2       :  0.0
                 -  pvgridpower2         :  0.0
                 -  pvgridvoltage3       :  0.0
                 -  pvgridcurrent3       :  0.0
                 -  pvgridpower3         :  0.0
                 -  totworktime          :  1181.5
                 -  pvenergytoday        :  28.2
                 -  pvenergytotal        :  2364.2
                 -  epvtotal             :  2344.6
                 -  epv1today            :  17.3
                 -  epv1total            :  1411.5
                 -  epv2today            :  10.7
                 -  epv2total            :  933.1
                 -  pvtemperature        :  39.9
                 -  pvipmtemperature     :  45.1
         - MQTT jsonmsg:
                         {"device": "HMG3A510SC", "time": "2021-08-28T15:56:11", "buffered": "no",
                         "values": {"pvstatus": 1, "pvpowerin": 9278, "pv1voltage": 3447,
                         "pv1current": 24, "pv1watt": 8621, "pv2voltage": 1810, "pv2current": 3,
                         "pv2watt": 657, "pvpowerout": 9031, "pvfrequentie": 5000, "pvgridvoltage":
                         2388, "pvgridcurrent": 40, "pvgridpower": 9533, "pvgridvoltage2": 0,
                         "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0,
                         "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 8506492,
                         "pvenergytoday": 282, "pvenergytotal": 23642, "epvtotal": 23446,
                         "epv1today": 173, "epv1total": 14115, "epv2today": 107, "epv2total": 9331,
                         "pvtemperature": 399, "pvipmtemperature": 451}}
         - MQTT connection refused by target
         - Grott send data to PVOutput systemid:  9144 for inverter:  HMG3A510SC
                 -  {'X-Pvoutput-Apikey': '9f1df7b9d99c33f2232c2cfd1b03cc287a7bbe33', 'X-Pvoutput-SystemId': '9144'}
                 -  {'d': '20210828', 't': '15:56', 'v1': 28200, 'v2': 903.1, 'v6': 238.8}
         - Grott PVOutput response:
                 -  OK 200: Added Status
         - Grott Send data to Influx disabled
         - Grott extension processing disabled

         - Growatt packet received:
                  <socket.socket fd=13, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.218', 5279), raddr=('192.168.0.37', 30620)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.218', 41660), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x96\x00\x06\x00\xe6\x01\x20\x1f\x35\x2b\x43\x20\x41\x45\x75\x40\x56\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x27\x3a\x26\x47\x35\x72\x43\x5f\x24\x22\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x7c\x68\x48\x4a\x63\x77\xc1
                 \x74\x74\x47\xf6\x6f\x77\x68\x39\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x6f\x61\x74\x74\x47\x72\x6f\x77\x61\x8b\x8b\xb8\x57\x6f\x77\x61\x74\x74\x47
[... 1 lines cut ...]
                 \x74\x47\x72\x6f\x77\x61\x74\x8b\xb8\x8d\xb6\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x9e\x8b\x8b\x62\x72\x6f\x62\xd4\x8b\x8b\xad\x22\x90\x88\x9e\xad\x74\x47\x73
                 \x9b\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\xb2\xfb\x74\x74
                 \x6a\x3f\x6f\x77\x69\xaf\x74\x47\x6f\xd2\x77\x61\x74\x74\x47\x72\x9d\x90\x61
                 \x74\x52\xdf\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\xac\xe7
         - Grott automatic protocol detection
         - Grott data record length 238
         - layout   :  T060120
         - Record layout used :  T060120
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 0096000600e601205847443441353132323900000000000000000000000000000000000000004
                 84d4733413531305343000000000000000000000000000000000000000015081c0f380c00a000
                 0000840000094d0000000000000000000000180000000000000000ffffff25000000000000000
                 0000015b50000000000000000ffffea500000000000000000ffffffd90000000000000000ffff
                 ff25000015b5ffffea50ffffffd9000001f40000000000000000000000000000c59a00002d4d0
                 00008db00001dbd000000000000f2e70000269800000000000000000000000000000000000000
                 0000000000cd93
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T060120
                 -

farsonic, 2021-08-30:
this one is even more messed up! 

         - Grott data record date/time processing started
         - date-time:  2021-08-30T09:39:49
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T060120

         - Grott data record date/time processing started
         - date-time:  2021-08-30T09:39:49
         - Grott values retrieved:
                 -  datalogserial        :  XGD4A51229
                 -  pvserial             :  HMG3A510SC
                 -  voltage_l1           :  64220094.6
                 -  Current_l1           :  187009470.8
                 -  act_power_l1         :  119868197.5
                 -  app_power_l1         :  195377765.4
                 -  react_power_l1       :  200287550.8
                 -  powerfactor_l1       :  1919907.7
                 -  pos_rev_act_power    :  195083940.7
                 -  app_power            :  200286913.5
                 -  react_power          :  119868197.5
                 -  powerfactor          :  1635021.9
                 -  frequency            :  237506118.6
                 -  pos_act_energy       :  200287550.8
                 -  rev_act_energy       :  119868197.5
         - MQTT jsonmsg:
                         {"device": "XGD4A51229", "time": "2021-08-30T09:39:49", "buffered": "no",
                         "values": {"voltage_l1": 642200946, "Current_l1": 1870094708,
                         "act_power_l1": 1198681975, "app_power_l1": 1953777654, "react_power_l1":
                         2002875508, "powerfactor_l1": 1919907681, "pos_rev_act_power": 1950839407,
                         "app_power": 2002869135, "react_power": 1198681975, "powerfactor":
                         1635021895, "frequency": 2375061186, "pos_act_energy": 2002875508,
                         "rev_act_energy": 1198681975}}
         - MQTT connection refused by target
         - Grott send data to PVOutput systemid:  9144 for inverter:  HMG3A510SC
                 -  {'X-Pvoutput-Apikey': '9f1df7b9d99c33f2232c2cfd1b03cc287a7bbe33', 'X-Pvoutput-SystemId': '9144'}
                 -  {'d': '20210830', 't': '09:39', 'v3': 200287550800, 'c1': 3, 'v6': 64220094.6}
                 -  {'d': '20210830', 't': '09:39', 'v4': 195083940.7, 'v6': 64220094.6, 'n': 1}
         - Grott PVOutput response SM1:
                 -  OK 200: Added Status
         - Grott PVOutput response SM2:
                 -  OK 200: Added Status
         - Grott Send data to Influx disabled
         - Grott extension processing disabled

         - Growatt packet received:
                  <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.218', 5279), raddr=('
192.168.0.37', 11466)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.218', 47568), raddr=(
'47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.218', 5279), raddr=('192.168.0.37', 11466)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.218', 47568), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.218', 5279), raddr=('192.168.0.37', 11466)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.218', 47568), raddr=('47.91.67.66', 5279)>
[... 35 lines cut ...]
         - no matching record layout found, try generic
         - Record layout used :  T06NNNNX
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 002e0006024101045847443441353132323900000000000000000000000000000000000000004
                 84d4733413531305343000000000000000000000000000000000000000015081e092c2f020bb8
                 0c340001000087db0f37002c000043270722005f000044b400000000000000000000000000000
                 00000000000000000000000852613880960008d00008532000000000000000000000000000000
                 0009600000000000000000000057b900002d6d0083aa590000002c00005d7e00005cb80000001
                 1000037d60000001a000024e20000000000000000000000000000000000000000000000000000
                 0000000000000000000000000000000000000000000f113a0082000000000068000001d400000
                 0000000019a106c00004e20004400000000000000000000000000000000000000000000000000
                 0000000000000000000000000000000000000000000c350cb1000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000a00000000000000119000000000000000000000000000000000
                 000000000001e19
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T06NNNNX

johanmeijer, 2021-09-01:
@farsonic, the first one does not look that bad. My test shows that Grott probably (you have not capped that output) has sent this information to PVOutput: 

 - date-time:  2021-08-28T15:56:12
 - Grott values retrieved:
         -  datalogserial        :  XGD4A51229
         -  pvserial             :  HMG3A510SC
         -  voltage_l1           :  238.1
         -  Current_l1           :  2.4
         -  act_power_l1         :  -21.9
         -  app_power_l1         :  555.7
         -  react_power_l1       :  -555.2
         -  powerfactor_l1       :  -0.0
         -  pos_rev_act_power    :  -21.9
         -  app_power            :  555.7
         -  react_power          :  -555.2
         -  powerfactor          :  -0.0
         -  frequency            :  50.0
         -  pos_act_energy       :  5058.6
         -  rev_act_energy       :  1159.7
 - MQTT jsonmsg:
                 {"device": "XGD4A51229", "time": "2021-08-28T15:56:12", "buffered": "no",
                 "values": {"voltage_l1": 2381, "Current_l1": 24, "act_power_l1": -219,
                 "app_power_l1": 5557, "react_power_l1": -5552, "powerfactor_l1": -39,
                 "pos_rev_act_power": -219, "app_power": 5557, "react_power": -5552,
                 "powerfactor": -39, "frequency": 500, "pos_act_energy": 50586,
                 "rev_act_energy": 11597}}
 - No MQTT message sent, MQTT disabled
 - Grott send data to PVOutput systemid:  83106 for inverter:  HMG3A510SC
         -  {'X-Pvoutput-Apikey': 'yyyyyy', 'X-Pvoutput-SystemId': 'xxxxx'}
         -  {'d': '20210828', 't': '15:56', 'v3': 5058600, 'c1': 3, 'v6': 238.1}
         -  {'d': '20210828', 't': '15:56', 'v4': -21.9, 'v6': 238.1, 'n': 1}

Maybe there is another record T060120 at the same timeframe with the meshed up values.  Within a 5 minutes timeframe it is possible to overide values in PVOutput. 

The second one show exact the same problem as @SebastianHarnau. 

I am going to try if I can detect the encrypt / decrypt error if not the only solution will be to filter this records so no wrong values are sent to PVoutput.org.

farsonic, 2021-09-01:
This is from around the same time 

 - Grott automatic protocol detection
         - Grott data record length 238
         - layout   :  T060120
         - Record layout used :  T060120
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 0097000600e601205847443441353132323900000000000000000000000000000000000000004
                 84d4733413531305343000000000000000000000000000000000000000015081c0f380d00a047
                 72273a26473572435f2422747447726f7761747447726f7761747447726f77747c68484a6377c
                 1747447f66f7768397447726f7761747447726f6f61747447726f77618b8bb8576f7761747447
                 726f776161c147726f77617474478d909d31747447726f7761748bb88db67761747447726f779
                 e8b8b62726f62d48b8bad2290889ead7447739b7761747447726f7761747447726fb2fb74746a
                 3f6f7769af807d
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T060120

         - Grott data record date/time processing started
         - date-time:  2021-08-28T15:56:13
         - Grott values retrieved:
                 -  datalogserial        :  XGD4A51229
                 -  pvserial             :  HMG3A510SC
                 -  voltage_l1           :  64220094.6
                 -  Current_l1           :  187009470.8
                 -  act_power_l1         :  119868197.5
                 -  app_power_l1         :  195377765.4
                 -  react_power_l1       :  200287550.8
                 -  powerfactor_l1       :  1919907.7
                 -  pos_rev_act_power    :  195083940.7
                 -  app_power            :  200287072.1
                 -  react_power          :  119868197.5
                 -  powerfactor          :  1635021.9
                 -  frequency            :  237506488.1
                 -  pos_act_energy       :  200287550.8
                 -  rev_act_energy       :  119868197.5
         - MQTT jsonmsg:
                         {"device": "XGD4A51229", "time": "2021-08-28T15:56:13", "buffered": "no",
                         "values": {"voltage_l1": 642200946, "Current_l1": 1870094708,
                         "act_power_l1": 1198681975, "app_power_l1": 1953777654, "react_power_l1":
                         2002875508, "powerfactor_l1": 1919907681, "pos_rev_act_power": 1950839407,
:

johanmeijer, 2021-09-04:
@farsonic @SebastianHarnau 

I updated the 2.6 branche (2.6.1b) to filter the invalid smart meter monitor. 

I hope this will work for you and am curious for the results!

farsonic, 2021-09-06:
ok, I have updated the code and will leave this running today. Have already spotted this in my PVOUTPUT results...is there a specific entry in the logs when it drops a packet? 

[image]

SebastianHarnau, 2021-09-06:
Updated code seems to work for me, too... thanks a lot!

[image]

johanmeijer, 2021-09-06:
@farsonic yes that is the message!  Grott invalid 0120 record processing stopped,

johanmeijer, 2021-11-14:
@farsonic can we close this one?

farsonic, 2021-11-14:
Yes, seems to be working fine as a work around :) 

re

Francois

johanmeijer, 2021-11-15:
According to the picture you send pvenergytoday is in the message but is zero. Is that the problem? Can it be there is no energy delivered to the power grit? . 

[image]

gabrielpc1190, 2021-11-18:
Well, then I was understanding it incorrectly. I though it was PhotoVoltaic Energy produced Today.
Is there any key for the energy produced during the day? I can see it on the Growatt App and server but not on the JSON

johanmeijer, 2021-11-18:
I am not totally convinced you are wrong. It is my interpretation but I can be wrong also of course (I do not have a SPF so have to do it with information from others). 

I try to decode the records based on the information I had. I can also be wrong. Can you please send me the output started from: 
- Growatt original data: 
to 
- MQ message sent. 

It would also be nice to have an exported .csv file with data from that time from the growatt server website (see below). 
So I can try to match the data. 

[image]

gabrielpc1190, 2021-11-19:
[UKD9A4100Y storage data - 2021-11-18_2021-11-18.xls](https://github.com/johanmeijer/grott/files/7567242/UKD9A4100Y.storage.data.-.2021-11-18_2021-11-18.xls)
Can't get a CSV but a XLS. Here's it
[ledidobe-grott-2.6.1c.html.txt](https://github.com/johanmeijer/grott/files/7567260/ledidobe-grott-2.6.1c.html.txt)
Also I'm attaching the logs I get from Docker on my Synology. It's an html file, just remove the .txt at the end to open it.

johanmeijer, 2021-11-20:
Thanks for the files. 

I think I found the bug and fixed it in 2.6.1f (now the 2.6 branch). 

The fields  pvenergytoday and pvenergytotal should give the right values (Grott needs this field names for pvoutput processing). 

In the growatt spreadsheet this fields are named: epvToday, epvTotal. I added those fields also to the mqtt message. They will have the same values as  pvenergytoday and pvenergytotal. 

You can choose which names to use ;) 

Please let me know if this fixes your problem.

gabrielpc1190, 2021-11-22:
Thank you very much. 

Can you update the docker hub with the 2.6.1f so I can test it?
If not, I'll find my way into implementing the Linux version, currently I'm using a docker container.

johanmeijer, 2021-11-22:
Oh yes. I did not realised you used the docker container

I have create the 2.6.1f container and put it on docker hub. I have not be able to test it (it runs but it is dark now here in Amsterdam). Maybe you can test it also tomorrow and if it works I will promote it to the Beta version (and to the latest after that).

gabrielpc1190, 2021-11-22:
[image]
[image]

As you can see on these two images, yes, I can see epv1today to be exactly as shown on server.growatt.com.
I can confirm the values epv1today and epv1total are working. Can't say the same about pvenergytotal and pvenergytoday, maybe I was wrong and now I'm totaly sure the values I need are epv1today and epv1total.

Thank you for taking the time to fix this.

johanmeijer, 2021-11-23:
Now  I am a little puzzled. The values are totally from what I see in my tests (not docker). Can you again send more log information with the data and information with which layout definition the data is processed.

johanmeijer, 2021-11-23:
This is what is see when I process your record: 

[image]

Is looking very different from what you show, so something is going wrong. I will set up a test environment with docker and see what happens if I inject your record. 
(Normally I test docker only in real life with my own inverter, while testing every combination is a lot of work).

[10 more replies]

closed 2021-12-05 by johanmeijer