johanmeijer/grott #49: UTC conversion for influxdb

a-dekker, 2021-05-22:
For a longer time In notices my Grafana graphics are not in sync with reality. It is about an hour late.
Finally took some time to have a look at it, and the conversion to UTC seems to be the problem, this sets the time one hour in the past.
As soon as I change `utc_dt.strftime` in https://github.com/johanmeijer/grott/blob/224f0b019cc48f46184e411956761ef19b376d80/grottdata.py#L369
to `local_dt.strftime` it is in sync again.

So why this conversion to UTC time? Is this related to some Influxdb timezone config setting?

a-dekker, 2021-05-22:
This change has some other implications it seems. The following change could be better:
```
-           utc_dt = datetime.strptime (jsondate, "%Y-%m-%dT%H:%M:%S") + timedelta(seconds=curtz)
+          utc_dt = datetime.strptime (jsondate, "%Y-%m-%dT%H:%M:%S") - timedelta(seconds=curtz)
```
It actually sets the time one hour ahead localtime, but works out ok in the grafana graph.

johanmeijer, 2021-05-22:
Hmm Timezone stays difficult. My experience is that using the UTC time for influx is the best way to keep it clean. 
Then you do not have to bother about summer and winter time etc. 

The graphical software (grafana in my case) does the conversion to local time while displaying. 

So I tried to correct the time from the inverter to UTC and use the offset (time zone) from the grott server to correct this. 
(that is the time delta). If the inverter and the grott server does not have the same time zone that might be a problem. 

The challenge is that the inverter time should have the same timezone as the Grott server (or the docker container if you use Grott container) and the Grafana server. 

But we can correct things. In the .ini you can specify time zone with the timezone parameter in the [Generic] section.  
by example specify: timezone = europe/amsterdam (I use this one). 

For timezones see: https://en.wikipedia.org/wiki/List_of_tz_database_time_zones 

Again keeping time / timezones consistent across platforms is difficult that is the reason why I try (maybe it does not work always right) to write to influx in an universal time.

a-dekker, 2021-05-22:
Timezones and daylight savings are a real pain when it comes to interaction. UTC helps, but if possible I even prefer to use the Unix time.
Locally I have everything on Europe/Amsterdam time (CEST). That includes the Shinelink device time and server running InfluxDB and Grafana.
I will play some more with the settings and do some more testing tomorrow. Will report back.

johanmeijer, 2021-05-22:
Hi @a-dekker, I did some testing and indeed the timezone=default does not seem to work well (will only subtracts 1 hour in stead of 2) 

If you specify timezone = europe/amsterdam in the grott.ini then it should work fine.   

I will look what is wrong with timezone =default.

johanmeijer, 2021-05-22:
Ok! when using timezone=local (default if specifying nothing). The time will be corrected with the timezone difference between the current time and utc based on a non-daylight saving time. 

So it the Netherlands it will substract one hour and this will work fine in the winter. In summer it does correct the daylight saving.time.
I will correct that (after I know how). 

I think specifying a timezone is a good workaround. 

Thanks for noticing the error and opening an issue for this.

a-dekker, 2021-05-22:
Makes sense. I have set the timezone in grott.ini and will look at the result tomorrow.

a-dekker, 2021-05-23:
Setting TZ Europe/Amsterdam explicitly does indeed fix the one hour delay issue.

johanmeijer, 2021-05-23:
Ok then we have a workaround. I will try to fix the local default in a next release!

the-butterfry, 2021-06-03:
I was playing with the .ini and had set time = server and I was getting no data or any messaging about time offsets. Setting it back to auto resumed normal operation. Maybe add some messaging in the log about the time conversion since MQTT jsonmsg reflects accurate local time whereas the Influx jsonmsg does not in the logs.

But if the timezone feature you're talking about above addresses this.. then great!

johanmeijer, 2021-06-03:
The influxdb time should reflect the utc time. Standard  I use the server timezone offset to convert to utc. 
As you can read above this works ok during winter but with daylight saving time active it is one hour off. I will try fix that in the next release (2.6.2 I think). 

For now this can be solved with setting a timezone in the .ini 

Sometimes the date/time in growatt data record is corrupted (seems to be inverter depended). Grott will skip that and use the server date/time instead if it is a normal record. If it is a buffered record Grott will skip it. It make no sense to write a historical with the current time.

The date/time in the mqtt message is either the time in the data record or server time (invalid date/time in data record or time=server)

I do understand date/time processing is complex and I have not really document it very well. WIll try to update the wiki for it.

ZonPV, 2021-08-24:
Sorry to say that I have the same issue in Grafana.
In the Grott-container on my Synology I add the setting TZ Europe/Amsterdam. That is accepted in the logfile.
In the logfile I read:
[[image]<br /><a target='_blank' href='https://nl.imgbb.com/'>photo free image</a><br />](url)
I do not understand that third line.. Why adjust the time? No need to do that I should say..

In Grafana I see 30 minutes difference!!! All data from the Grott-container is half hour later compared to other graphs...

[[image]<br /><a target='_blank' href='https://nl.imgbb.com/'>photo free image</a><br />](url)
I don't understand why..

closed 2021-11-14 by johanmeijer