johanmeijer/grott #50: Docker with Synology no data

fgerrits, 2021-05-23:
Hi,

Finally got the docker with grott running, after many try-outs and fixes on my side the docker has something strange.

When i restart the synology nas the docker/grott is working, but the readings arent send to pvoutput 
`	 - MQTT jsonmsg: 
			 {"device": "TDG1A3701B", "time": "2021-05-23T11:46:26", "buffered":
			 "nodetect", "values": {"pvstatus": 1, "pvpowerin": 21140, "pv1voltage":
			 1924, "pv1current": 16, "pv1watt": 3219, "pv2voltage": 2314, "pv2current":
			 77, "pv2watt": 17921, "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage":
			 0, "pvenergytoday": 151388249, "pvenergytotal": 20822, "pvtemperature": 0,
			 "pvipmtemperature": 0}}
	 - No MQTT message sent, MQTT disabled
	 - Grott send data to PVOutput systemid:  83865 for inverter:  TDG1A3701B
		 -  {'X-Pvoutput-Apikey': 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx', 'X-Pvoutput-SystemId': '83865'}
		 -  {'d': '20210523', 't': '11:46', 'v1': 15138824900, 'v2': 0.0, 'v6': 0.0}
	 - Grott PVOutput response: 
		 -  Bad request 400: Energy value [2147483647] too high for system size [4355]
	 - Grott Send data to Influx disabled 
	 - Grott extension processing disabled 

	 - Growatt packet received:
		  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.1', 5279), raddr=('192.168.1.34', 43169)>
	 - Data less then minimum record length, data not processed`

So there is no data uploaded, after restarting the grott container nothing is recorded from the tcpdump, so i must restart the synology nas again to get it running again.
What am i doing wrong ?

johanmeijer, 2021-05-24:
Can you also sent me the data before this message (with grott original data and layout used information in it?) 

This are not the right values being sent, I think,

johanmeijer, 2021-05-26:
@fgerrits do you have more information so I can look at it?

fgerrits, 2021-05-26:
Sorry could not respond sooner,

The Log from the grott docker
`Grott Growatt logging monitor : 2.5.2

Grott Command line parameters processed:
	verbose:     	 True
	config file: 	 grott.ini
	output file: 	 <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
	nomqtt:      	 False
	inverterid:  	 automatic
	pvoutput:    	 False
	blockcmd:    	 False
	noipf:       	 False

Grott process configuration file

Grott process environmental variables

Grott override settings if set in commandline

Grott process json layout files

Grott layout records loaded
T02NNNN  : 
{'decrypt': {'value': 'False'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'num', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'pvipmtemperature': {'value': 242, 'length': 2, 'type': 'num', 'divide': 10}, 'epv1today': {'value': 278, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 294, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 310, 'length': 4, 'type': 'num', 'divide': 10}}
T05NNNN  : 
{'decrypt': {'value': 'True'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'num', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'pvipmtemperature': {'value': 242, 'length': 2, 'type': 'num', 'divide': 10}, 'epv1today': {'value': 278, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 294, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 310, 'length': 4, 'type': 'num', 'divide': 10}}
T06NNNN  : 
{'decrypt': {'value': 'True'}, 'pvserial': {'value': 76, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 136, 'divide': 10}, 'pvstatus': {'value': 158, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 162, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 174, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1w [... cut ...]
	growattip:   	 47.91.67.66
[... 81 lines cut ...]
#systemid2 = 67890
#inverterid2 = inverter2

[influx]
# Influxdb parameters definitions

influx = False
influx2 = False
#dbname = grottdb
#ip = localhost
#port = 8086
#user = grott
#password = growatt2020
#token  = "influx_token"
#org  = "grottorg"
#bucket = "grottdb" 

[extension] 
# grott extension parameters definitions

#extension = True
#extname = grottext
#extvar = {"var1": "var1_content", "var2": "var2_content"}
#extvar = {"ip": "192.168.1.26", "port": "8000"}
`

johanmeijer, 2021-05-26:
Ok thanks. I see you use version 2.5.2. You might need version 2.5.4 (the current master) or 2.6.1 (new branche),

I am not sure if this will solve te problem (2.5.4. has some additional support for negative values the newer inverters but I can not see if that is your problem). 

But  to know for sure I need more information from the Grott log. I need a data record.. By example:

May 26 21:18:45 raspitest grott[3133]:          - Growatt original Data:
May 26 21:18:45 raspitest grott[3133]:                  \x00\xc9\x00\x06\x01\x01\x01\x04\x0d\x22\x2c\x45\x59\x45\x4c\x74\x41\x2d\x77
May 26 21:18:45 raspitest grott[3133]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
May 26 21:18:45 raspitest grott[3133]:                  \x3e\x3a\x23\x46\x4c\x75\x41\x5d\x41\x50\x74\x74\x47\x72\x6f\x77\x61\x74\x74
May 26 21:18:45 raspitest grott[3133]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x71\x6e\x52\x60\x43\x75\x61
May 26 21:18:45 raspitest grott[3133]:                  \x74\x74\x6b\x72\x6f\x77\x61\x74\x74\x40\xd0\x6f\x77\x61\x74\x74\x47\x72\x6f
May 26 21:18:45 raspitest grott[3133]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x67\xc1\x7b\x54\x77\x61\x74\x74\x47
May 26 21:18:45 raspitest grott[3133]:                  \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
May 26 21:18:45 raspitest grott[3133]:                  \x74\x00\x72\x6f\xb5\x22\x70\x06\xb3\x5e\x6f\x81\x61\x74\x74\x47\x72\x6f\x77
May 26 21:18:45 raspitest grott[3133]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x75\xdd\x77\x61\x74\x74\x47\x5f
May 26 21:18:45 raspitest grott[3133]:                  \x6f\x2e\x2f\x54\x74\x47\x72\x6f\x77\x61\x74\x3f\x47\x72\xa3\x5f\x61\x74\x74
May 26 21:18:45 raspitest grott[3133]:                  \x47\x72\x6f\x77\x61\x74\x74\x8b\x5a\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
May 26 21:18:45 raspitest grott[3133]:                  \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x78\x6f
May 26 21:18:45 raspitest grott[3133]:                  \x76\x6b\x46\x73\xa1\x75\xb3\x77\x60\x7e\xe2\x07\x06\x6f\x77\x6f\x09\x74\x47
May 26 21:18:45 raspitest grott[3133]:                  \x7f\xc8\x77\x61\x74\x75\x47\x59\x6f\x77\x61\x74\x74\x47\x72\x6f\x13\x3b
May 26 21:18:45 raspitest grott[3133]:          - Grott automatic protocol detection
May 26 21:18:45 raspitest grott[3133]:          - Grott data record length 265
May 26 21:18:45 raspitest grott[3133]:          - layout   :  T060104
May 26 21:18:45 raspitest grott[3133]:          - no matching record layout found, try generic
May 26 21:18:45 raspitest grott[3133]:          - Record layout used :  T06NNNN

fgerrits, 2021-05-28:
this is the complete picture when the tcpdump is working.

,,,
- Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.1', 34856), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\xbd\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x44\x20\x46\x47\x7f\x46\x5e\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3b\x33\x26\x45\x35\x74\x45\x5f\x46\x23\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x71\x63\x4c\x5c\x78\x75\x6a
		 \xcc\x78\x73\x72\x6e\x77\x61\x26\xe0\x40\xf6\x6f\x67\x61\x74\x78\xd4\x7b\x65
		 \x77\x2c\x74\x74\x01\x73\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x16\x26\x7c\xfe\x68\x72
		 \x74\x1e\x72\x6f\x26\x37\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x4e\x74\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x48\xef\x91\x72\x6f\x77\x44\x74\x74\x53\x4d\x6f\x77\x75\xc4\x74
		 \x47\x72\x65\x77\x61\x7f\x76\x47\x72\x6f\x6c\x61\x74\x7d\xe9\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\xb8\x88\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x75
		 \x01\x46\x07\x6f\x77\x61\x74\x75\x20\x7c\x6f\x77\x61\x3a\x54\x47\x37\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x75\x74\x46\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x66\xe7\x7d\x44\x7c
		 \x7f\x45\xa9\x52\x91\x47\x7b\x6f\x71\x61\x76\x74\x47\x62\x5e
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00bd0006024101045847443341323338343100000000000000000000000000000000000000005
		 444473141333730314200000000000000000000000000000000000000001505170b2e17020bb8
		 0c340001000052940784001000000c93090a004d0000460100000000000000000000000000000
		 00000000000000000000000515413890906005900005156000000000000000000000000000000
		 00090600000000000000000000000000000000003c9bd6000000250000143f000014b00000000
		 a00000b020000001b000009ae0000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000fffa00000000000000000000017501750
		 000000001670e0000004e20004500000000000000000000000000000000000000000000000000
		 00003c0000000000000000000000000000000000000c350cb1000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000002710000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000000000000000000001000100000000000000000000079309030e1032c826e500090
		 006000200001031
	 - Growatt processing values for:  TDG1A3701B
	 - Grott values retrieved:
		 - pvserial:          TDG1A3701B
		 - pvstatus:          1
		 - pvpowerin:         2114.0
		 - pvpowerout:        0.0
		 - pvenergytoday:     15138824.9
		 - pvenergytotal:     2082.2
		 - pv1watt:           321.9
		 - pv2watt:           1792.1
		 - pvfrequentie:      0.0
		 - pvgridvoltage:     0.0
[... 9 lines cut ...]
			 1924, "pv1current": 16, "pv1watt": 3219, "pv2voltage": 2314, "pv2current":
			 77, "pv2watt": 17921, "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage":
			 0, "pvenergytoday": 151388249, "pvenergytotal": 20822, "pvtemperature": 0,
			 "pvipmtemperature": 0}}
	 - No MQTT message sent, MQTT disabled
	 - Grott send data to PVOutput systemid:  83865 for inverter:  TDG1A3701B
		 -  {'X-Pvoutput-Apikey': '8e1d73034ba7cc926bfdff8bdebf6fa8d07f55f3', 'X-Pvoutput-SystemId': '83865'}
		 -  {'d': '20210523', 't': '11:46', 'v1': 15138824900, 'v2': 0.0, 'v6': 0.0}
	 - Grott PVOutput response: 
		 -  Bad request 400: Energy value [2147483647] too high for system size [4355]
	 - Grott Send data to Influx disabled 
	 - Grott extension processing disabled 

	 - Growatt packet received:
		  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.1', 5279), raddr=('192.168.1.34', 43169)>
[... cut ...] processed

	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.1', 34856), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.1', 5279), raddr=('192.168.1.34', 43169)>
	 - Data less then minimum record length, data not processed
,,,

johanmeijer, 2021-05-30:
I did some investigation. I am not sure what is happening but you do not run Grott in de automatic mode. You need to specify compat = False in the grott.ini (or gcompat = false enironmental or docker at docker run). 
Not specifying it at all should also work because compat= False is the default. 

If you use the standard docker container on Docker Hub (ledidobe/grott) it should work without changes in this area (you still have to specify MQTT settings, see also: https://github.com/johanmeijer/grott/wiki/Docker-support). 

The comment in the grott.ini you use is not very clear about that indeed. I will change the comment in the grott.ini in the example directory. 

The record above  will give the following values if Grott is running in standard (automatic detect) mode: 

[image]

johanmeijer, 2021-06-16:
@fgerrits is your problem solved?

johanmeijer, 2021-08-08:
This issue is closed by lack of reaction. I assume it is working now.

closed 2021-08-08 by johanmeijer