johanmeijer/grott #51: Large current wattage output at end of day

mvank, 2021-05-31:
Hi noticed the last few days that every night there is a very big value for current wattage output.
[
[image]
](url)

I am just using the sensor information for home assistant.
During the day no issue at all, only with the last message something goes wrong

johanmeijer, 2021-06-01:
Hi @mvank,

Oke. Probably this  means that your Inverter is using energy and not producing. Newer inverts seems to work on the grid power (not the solar power) and keep working when no energy is produced, then it is possible that power is used and not generated.   

Since Grott 2.5.4. negative value's should be handled ok for Powerout. If not please let me know (then probably a layout template definition error). 

Which version of Grott do you use? Can you upgrade tot 2.5.4+? Which can you see in the Grott which layout template is used ? 

Example output: 
May 26 21:18:45 raspitest grott[3133]: - layout : T060104
May 26 21:18:45 raspitest grott[3133]: - no matching record layout found, try generic
May 26 21:18:45 raspitest grott[3133]: - Record layout used : T06NNNN

mvank, 2021-06-02:
Hi @johanmeijer,

I am running 'latest' version, but don't see a version.....
REPOSITORY       TAG       IMAGE ID       CREATED        SIZE
ledidobe/grott   latest    5d49c4492b61   2 months ago   170MB

This is what I see in the docker logs

- Grott automatic protocol detection
- Grott data record length 585
- layout   :  T060104X
- no matching record layout found, try generic
- Record layout used :  T06NNNNX

mvank, 2021-06-02:
updated: found version in grott.py file:
# Updated: 2021-02-27'
verrel = "2.5.2"

Which is strange as I just started using this container a week ago!

johanmeijer, 2021-06-02:
@mvank ok I waited to long with creating a new docker image that supports the negative values. 2.5.2 is stable but need an update.

I will create one comming days!

johanmeijer, 2021-06-02:
I have published ledidobe/grott:2.5.4 container and made this also the latest. 

Hopefully this will solve your problem.

mvank, 2021-06-03:
Hi @johanmeijer,

Just updated to latest image, so this would fix the issue tonight.
thank you again!

mvank, 2021-06-04:
Hi @johanmeijer ,

With the latest update issue have been resolved! 
Thnx!

johanmeijer, 2021-06-04:
Super! Can we close this issue?

mvank, 2021-06-04:
Yes it can be closed!

closed 2021-06-04 by johanmeijer