johanmeijer/grott #52: Offset in data and not correctly shown average in pvoutput

mvank, 2021-06-26:
Hi @johanmeijer,

I noticed two things with grott and pvoutput

1. there seems to be an offset in output KWh in Growatt server and PVoutput
Both screenshots of today
[image]
[image]

2. average information is not correctly shown. first is my own live output, second screenshot shows which is compared with a solaredge output.
[image]
[image]

mvank, 2021-07-19:
Hi @johanmeijer 

It seems that pvoutput is not always up to date (showing last info) unless when I select year overview. 
so eventually the information is available only not directly in daily/monthly output.

The other thing is that the energy is only showing rounded numbers in pvoutput, instead of output from solaredge, is this something from growatt itself?

nicauswu, 2021-09-11:
I have exactly same problem, I only modified grott.ini with 
pvoutput = True 
apikey = ***************************************
systemid = *****

johanmeijer, 2021-09-15:
@mvank , @nicauswu 

Ok. I looked deeper in to this. It is not very easy to understand and explain.  

1. 
Grott sends the power  generated in 1 decimal accuracy (e.g. 123.9) to PVOutput. It seems PVoutput ignore the decimal and does not rounds up. So 123.9 will be 123 in PVOutput.
This might be improved by doing a round up in Grott if you want that. 

2. 
The average values are calculated by PVOutput bij taken the difference in total energy generated for that day and then calculate the average power needed in the interval to generate the difference in energy. 
If at t0 the energy is 1.4 kwh and at t1 1.5kwh the difference is 0.1. The t0-t1 interval is 5 minutes (standard PVOutput). You need   the full 5 minutes 1200 watt power to generate 0.1 kwh. (1200 watt power continuously  will generate 1.2kw in an hour that results in 1200/12 = 0.1 in 5 minutes).  

So what you see is not wrong I think. 

The only thing is that the Growatt inverter provides daily value in Kwh with 1 digit accuracy. And that is not very granulair.  So you see 5 minutes intervals where the energy value has not changed and the average power is 0. And then suddenly there is one interval with a 0.1 kwh energy change and then the average  will be 1200. 

The Solaredge energy values in the graphs above are much more granulair and will give a better average value. 

I do not think we can do better with values we get from the Growatt Inverter right now. 

But I really like to be challenged for this.

gavintweedie, 2021-09-29:
I recently switched to Grott from another script that scraped the Growatt portal and too noticed this. I did some digging and have a workaround/fix that may work for you.

grott sends a few values to pvoutput API (https://pvoutput.org/help/api_specification.html#add-status-service)
(d) Date, Required
(t) Time, Required
(v1) Energy Generation (Wh) ** see note
(v2) Power Generation (W) ** See note
(v6) Voltage (V), Optional

While v1 and v2 are both not required, at least ONE of v1 OR v2 are required. "Similarly when only energy values v1 and v3 are sent, PVOutput will calculate the average power."

As noted earlier in this issue Growatt supplies quite rounded numbers in the Wh field (maps to v1) while the live W field is accurate (maps to v2).

Workaround I have tested is just removing v1 sending to PVOutput from Grott.

From docker logs (with some grepping..):
> 		 -  {'d': '20210929', 't': '11:10', 'v1': 10300, 'v2': 5006.6, 'v6': 247.9}
> 		 -  {'d': '20210929', 't': '11:15', 'v1': 10700, 'v2': 3780.6, 'v6': 245.0}
> 		 -  {'d': '20210929', 't': '11:20', 'v1': 11100, 'v2': 5006.5, 'v6': 247.5}
> 		 -  {'d': '20210929', 't': '11:24', 'v2': 5008.2, 'v6': 247.9}
> 		 -  {'d': '20210929', 't': '11:29', 'v2': 5004.3, 'v6': 247.8}
> 		 -  {'d': '20210929', 't': '11:34', 'v2': 5001.9, 'v6': 247.9}

[image]

Yes the PVOutput screenshot shows no issue before I removed it, but it was there. It appears when you stop feeding v1 values pvoutput works to recalculate the whole current day v1 instead of switching part way through the day. So it actually fixed my data for today.

My change:
/app # diff grottdata.py.bak grottdata.py
--- grottdata.py.bak
+++ grottdata.py
@@ -375,7 +375,6 @@
             pvdata = { 
                 "d"     : pvodate,
                 "t"     : pvotime,
-                "v1"    : definedkey["pvenergytoday"]*100,
                 "v2"    : definedkey["pvpowerout"]/10,
                 "v6"    : definedkey["pvgridvoltage"]/10
                 }
/app # 

Not sure the impact on other inverters/dataloggers, maybe others prefer to send v1 so just removing it may not be the best. Maybe just an ini or runtime flag to say "send live/send average/send both"?

My inverter is a Growatt 5kW, datalogger is a ShineWiFi-X.

mvank, 2021-12-26:
Hi @johanmeijer, solution for me is still changing grottdata.py and mark out 
`#        "v1"    : definedkey["pvenergytoday"]*100,` like described above: https://github.com/johanmeijer/grott/issues/52#issuecomment-950201995

this change the output information to more detailed info:

[image]

johanmeijer, 2021-12-26:
I will add this functionality as a configurable option in the next update!

johanmeijer, 2022-02-06:
@mvank , @gavintweedie @nicauswu 

I added the possibility  to disable sending energytoday to pvoutput (disable V1 input).

You can enable this by specifying:

* .ini [PVOutput section] pvdisv1 = True
* environmental gpvdisv1 = "True" (docker: -e gpvdisv1 = "True")

Docker 2.7.1 build is not available yet. I will build it soon (after testing the regular version for a couple of days).

After testing I will also decide if True might become the default in te future.

cspiby, 2022-02-24:
I've enabled gpvdisv1 to true which shows correct averages now, but when the connection to the growatt server is interrupted (which seems quite regularly at the moment) the data in those periods is lost and the days total doesn't match that of the inverter.

johanmeijer, 2022-02-24:
Do you have the grott log after an interrupt? Normally the inverter buffered the records and those will be sent later.

The only problem might be that the inverter is sending to much buffered records in a short time and pvoutput is refusing the input (max 60 per hour). 

Day totals will be a problem while not al information is received and pvoutput can not calculate it right.

cspiby, 2022-02-24:
This is what I see where data is missing on pvoutput

`	 - Grott data record date/time processing started
	 - no or no valid time/date found, grott server time will be used (buffer records not sent!)
	 - Grott values retrieved:`

cspiby, 2022-02-25:
Same issue today and as far as I can tell monitoring hasn't been down
[grott (1).txt](https://github.com/johanmeijer/grott/files/8140354/grott.1.txt)
, I've attached the log with my personal info redacted.

johanmeijer, 2022-02-26:
Is see a couple of challenges. 

Some inverters/datalogger combination does not send data in the Grott record (i have still not find out why). This is normally not a problem because Grott will use the time of the server to set date/time stamp (used in mqtt, influx and pvoutput). This can be an issue if the connection is lost and records are buffered and sent later. If Grott sees buffered records (50x) and there is no valid time it ignores this records. In your data I do not see buffered records so that is no problem. 

In your data a see a couple of connection errors. It looks likes the connection with the growatt server has been lost. I see this also in my log (a couple times a day). I was not aware of that. Maybe this introduced by setting the report frequency to 1 minute and Growatt does not can or want to keep up with this.  I will lower it and see if this improves.  

The limit of 60 pvoutput updates should be enough is your report frequency is >2 minutes. With 1 minute this might be a problem while I see that sometimes a record is sent within the 1 minute range. I see sometimes that I do not have enough requests left in 1 hour time frame and PVOutput rejects my updates. Be aware a query with the PVOutput mobile app is also seen as a request! So maybe it is a good idea to lower the report frequency or make a donation to PVOutput. You are then allowed to do more request (120). 
Be aware PVoutput will always update the values 1x per 5 minutes (I think the latest update is used). So for PVOutput setting report time frequency lower then 5 minutes has not added value (but will have for the other tooling your using). 
There is an enhancement request for Grott to filter/limit records send to PVOutput. I am still thinking about that one because I do not know (yet) how to implement this in Grott. 

I do see the inverter is attemping to re-establish the connection. This does not result in sending more data records (and output to pvoutput). The data records are only sent after the connection is back again. Until that only ping  records and the 03x announcement records are sent. This will not result in sending data to PVOutput. 

I can understand that if PVOutput does not get al records and there is not a daily total sent, the daily values does not match. PVOutput has to calculate this value but does not have all information. That might result in a wrong values. 

I am very curious if lower the report interval makes the environment more stable (I will try this out my self as well) and improves the PVOutput readings. Otherwise we will have to live with strange averages when Grott sends the daily value.

cspiby, 2022-02-26:
I've lowered the report frequency back to 5 minutes, today has been absolutely fine, I wonder if the growatt server also limits the amount of API requests also which is causing the failed connection.

johanmeijer, 2022-02-26:
At least it can not keep up with it. I will try it with the beta grottserver. Let's see what for report frequency we can achieve.

johanmeijer, 2022-02-27:
I did some tests with grottserver.py (see discussion #98) as destination  in stead of a connection to the growatt server. My conclusion :  The growatt server is the limitation when specifying a report frequency <= 1 minute. I can understand that the communication error are caused by the growatt server. 

With the beta grotserver.py running at the same machine as grott 1 minute is now problem. I even manged to run with 0.5 (half a minute) and 0.1 (every 6 seconds) without any problems. 

Be aware this will increase the PVOutput problem of course. A reason more to look for a solution for that ;)

cspiby, 2022-02-27:
Thanks for looking in to it, Growatt servers are so rubbish, no data recorded on their side until 10:25am this morning but thanks to your proxy, I have complete data in pvoutput and HA :)

thorstenfleischmann, 2023-03-29:
I would love to see `pvdisv1 = True` as the default. Looks way better now.

closed 2023-08-13 by johanmeijer