johanmeijer/grott #53: Multiple inverter issue with PVOutput

Pretmaker, 2021-06-30:
There seems to be a config reader issue with multiple inverters. 

The issue is that both converters get set to the single same systemid on PVOutput.org. I have solved the issue for now manually by editing one file, but more on that below...

Environment Vars:
```
Grott settings:

_Generic:
	version:     	 2.5.4
	verbose:     	 True
	trace:       	 False
	config file: 	 grott.ini
	minrecl:     	 100
	decrypt:     	 True
	compat:      	 False
	blockcmd:    	 False
	noipf:       	 False
	time:        	 auto
	sendbuf:     	 True
	timezone:    	 local
	valueoffset: 	 6
	offset:      	 6
	inverterid:  	 automatic
	mode:        	 proxy
	grottip      	 0.0.0.0
	grottport    	 5279
_MQTT:
	nomqtt       	 True
	mqttip:      	 localhost
	mqttport:    	 1883
	mqtttopic:   	 energy/growatt
	mqtttauth:   	 False
	mqttuser:    	 grott
	mqttpsw:     	 **secret**
_Growatt server:
	growattip:   	 47.91.67.66
	growattport: 	 5279
_PVOutput:
	pvoutput:    	 True
	pvurl:       	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:    	 <REDACTED>
	pvinverters: 	 1
	pvsystemid:  	 <REDACTED> to 0001
_Influxdb:
	influx:      	 False
	influx2:     	 False
	database:    	 grottdb
	ip:          	 localhost
	port:        	 8086
	user:        	 grott
	password:    	 **secret**
	organization:	 grottorg
	bucket:      	 grottdb
	token:       	 **secret**
_Extension:
	extension:   	 False
	extname:     	 grottext
	extvar:      	 {'none': 'none'}

Grott proxy mode started
```

The inverters have been setup to send the data via the proxy option as visible in the logs:
```
- ('192.168.178.52', 14374) has connected
- ('192.168.178.171', 1032) has connected
```

Data is received from both inverters, BUT it's added to a single systemid, while it is configured as should in the example:

```
	 - Grott values retrieved:
		 -  pvserial             :  <INVERTER1>
		 -  pvstatus             :  1
		 -  pvpowerin            :  104.9
		 -  pv1voltage           :  316.1
		 -  pv1current           :  0.3
		 -  pv1watt              :  104.9
		 -  pv2voltage           :  0.0
		 -  pv2current           :  0.0
		 -  pv2watt              :  0.0
		 -  pvpowerout           :  102.7
		 -  pvfrequentie         :  50.0
		 -  pvgridvoltage        :  229.0
		 -  pvgridcurrent        :  1.0
		 -  pvgridpower          :  104.6
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  0.0
		 -  pvgridvoltage3       :  0.0
		 -  pvgridcurrent3       :  0.0
		 -  pvgridpower3         :  0.0
		 -  totworktime          :  239.4
		 -  pvenergytoday        :  0.1
		 -  pvenergytotal        :  171.1
		 -  epvtotal             :  169.3
		 -  epv1today            :  0.1
		 -  epv1total            :  169.3
		 -  epv2today            :  0.0
		 -  epv2total            :  0.0
		 -  pvtemperature        :  34.2
		 -  pvipmtemperature     :  42.5
	 - MQTT jsonmsg: 
			 {"device": "<INVERTER1>", "time": "2021-06-30T08:14:56", "buffered": "no",
			 "values": {"pvstatus": 1, "pvpowerin": 1049, "pv1voltage": 3161,
			 "pv1current": 3, "pv1watt": 1049, "pv2voltage": 0, "pv2current": 0,
			 "pv2watt": 0, "pvpowerout": 1027, "pvfrequentie": 5000, "pvgridvoltage":
			 2290, "pvgridcurrent": 10, "pvgridpower": 1046, "pvgridvoltage2": 0,
			 "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0,
			 "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 1723484,
			 "pvenergytoday": 1, "pvenergytotal": 1711, "epvtotal": 1693, "epv1today": 1,
			 "epv1total": 1693, "epv2today": 0, "epv2total": 0, "pvtemperature": 342,
			 "pvipmtemperature": 425}}
	 - No MQTT message sent, MQTT disabled
	 - Grott send data to PVOutput systemid:  0001 for inverter:  <INVERTER1>
		 -  {'X-Pvoutput-Apikey': '<REDACTED>', 'X-Pvoutput-SystemId': '0001'}
		 -  {'d': '20210630', 't': '08:14', 'v1': 100, 'v2': 102.7, 'v6': 229.0}
	 - Grott PVOutput response: 
		 -  OK 200: Added Status
```

```
	 - Grott values retrieved:
		 -  pvserial             :  <INVERTER2>
		 -  pvstatus             :  1
		 -  pvpowerin            :  91.8
		 -  pv1voltage           :  253.8
		 -  pv1current           :  0.3
		 -  pv1watt              :  91.8
		 -  pv2voltage           :  0.0
		 -  pv2current           :  0.0
		 -  pv2watt              :  0.0
		 -  pvpowerout           :  90.0
		 -  pvfrequentie         :  50.0
		 -  pvgridvoltage        :  228.1
		 -  pvgridcurrent        :  0.9
		 -  pvgridpower          :  91.7
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  0.0
		 -  pvgridvoltage3       :  0.0
		 -  pvgridcurrent3       :  0.0
		 -  pvgridpower3         :  0.0
		 -  totworktime          :  232.0
		 -  pvenergytoday        :  0.0
		 -  pvenergytotal        :  111.7
		 -  epvtotal             :  110.4
		 -  epv1today            :  0.0
		 -  epv1total            :  110.4
		 -  epv2today            :  0.0
		 -  epv2total            :  0.0
		 -  pvtemperature        :  33.5
		 -  pvipmtemperature     :  42.0
	 - MQTT jsonmsg: 
			 {"device": "<INVERTER2>", "time": "2021-06-30T07:14:57", "buffered": "no",
			 "values": {"pvstatus": 1, "pvpowerin": 918, "pv1voltage": 2538,
			 "pv1current": 3, "pv1watt": 918, "pv2voltage": 0, "pv2current": 0,
			 "pv2watt": 0, "pvpowerout": 900, "pvfrequentie": 5002, "pvgridvoltage":
			 2281, "pvgridcurrent": 9, "pvgridpower": 917, "pvgridvoltage2": 0,
			 "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0,
			 "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 1670524,
			 "pvenergytoday": 0, "pvenergytotal": 1117, "epvtotal": 1104, "epv1today": 0,
			 "epv1total": 1104, "epv2today": 0, "epv2total": 0, "pvtemperature": 335,
			 "pvipmtemperature": 420}}
	 - No MQTT message sent, MQTT disabled
	 - Grott send data to PVOutput systemid:  0001 for inverter:  <INVERTER2>
		 -  {'X-Pvoutput-Apikey': '<REDACTED>', 'X-Pvoutput-SystemId': '0001'}
		 -  {'d': '20210630', 't': '07:14', 'v1': 0, 'v2': 90.0, 'v6': 228.1}
	 - Grott PVOutput response: 
		 -  OK 200: Added Status
```

The config file has the following (relevant) values:

```
[PVOutput]
# PVOutput parameters definitions

pvoutput = True

# Use this if you have one inverter
apikey = <REDACTED>
#systemid = 0000

# Use this if you have multiple inverters
systemid1 = 0001
inverterid1 = <INVERTER1>
systemid2 = 0002
inverterid2 = <INVERTER2>
#systemid99 = 99999
#inverterid99 = inverter99
```

As visible from the logs, either inverter gets it's data send to systemid1 on pvoutput. I've redacted some values to keep it a bit universal. If you want I can also supply raw data for testing to fix the issue permanently somewhere. 

For now I've solved it by changing the grottconf.py file:
https://github.com/johanmeijer/grott/blob/master/grottconf.py

```
        #pvoutput default 
        self.pvoutput = False
        self.pvinverters = 2
        self.pvurl = "https://pvoutput.org/service/r2/addstatus.jsp"
        self.pvapikey = "yourapikey"
        self.pvsystemid = {}
        self.pvinverterid = {}
        self.pvsystemid[1] = "systemid1"
        self.pvinverterid[1] = "inverter1"
        self.pvsystemid[2] = "systemid2"
        self.pvinverterid[2] = "inverter2"
```

So in short, I've changed the self.pvinverters value to 2, and added the self.pvsystemid[2] and self.pvinverterid[2] to the config file. Then restarted the script and it works fine and pushes data to both the systems on PVOutput

johanmeijer, 2021-06-30:
If you defined it as described here it should work : 

https://github.com/johanmeijer/grott/wiki/PVOutput.org-support

I am sure it worked in the passed ad people are using it. I am not aware of any changes that could affect this but you never know ;) 

I have to look deeper in to it. I am not able to test it at this moment (are not at home for the next weeks). But I will do it when I am at home. If you have raw data for mee it will help!

Pretmaker, 2021-07-01:
Yeah I tried it, but it kept showing as 1 PV system and just ignored the values that I entered. 

It kept saying pvinverters: 1 with the grott.py -v output.

johanmeijer, 2021-09-20:
@Pretmaker 

This is a documentation error. 

In the [pvoutput] section of the grott.ini the statement pvinverters = 2  (or any amount of inverters you want to specify) should be added. 

I updated the wiki and the example of grott.ini in the 2.5,2 and 2.6 branche.

Sorry for the inconvenience.

Pretmaker, 2021-09-29:
Nice, seems like this is sufficient for now. Worst case people can still find this old issue and open a new one and refer to this one if it's still an issue, but this solved it for me personally.

closed 2021-09-29 by Pretmaker