johanmeijer/grott #56: Feature request: variable for rate limits from pvoutput.org

alco28, 2021-07-02:
Feature request,  in the pvoutput script, [the pvoutput api](https://pvoutput.org/help/api_specification.html) only allows 60 requests per hour to[/url]e if you're using the free version (300 with donation mode). There is no option in grott.ini to limit this a bit is there? 

I have solved it for now by having the datalogger only send data on every 3 minutes, but that's not the best way. I would like to send it every 5 minutes to Pvoutput and every 1 minute to MQTT.

johanmeijer, 2021-07-03:
Oke. Lets think about how to implement.

For my configuration it is not a problem while I can not change the send
frequency of the growatt inverter (it is 1x 5 minutes). I did not realised
that it was possible for other inverters/network device implementation.

The only moment pvoutput gives problems when there has not been a network
connection for a long time and the inverter starts to send the buffered
data.

I think we have to take care of both situations. We want the buffered data
also to be sent after a network failure.

If you have good idea (timer/counter/.....) please let me know!

johanmeijer, 2021-07-21:
Still racking my brains on this one. If somebody has an idea how to implement it would be very welcome.

gabrielpc1190, 2021-11-15:
Set a buffer (or wait time ?) for the values going out to pvoutput, and send once every 5 minutes?

johanmeijer, 2021-11-15:
Yes I was thinking of something like this. The challenge is that Grott has no "memory" and can not sent no data  to the growatt server by it self (after timer complete) at this moment. It only does something when data comes along. 

I think that keeping track of the last values sent (with time info) and then sent the next value when the last value is sent longer then (about) 5 minutes ago is the best solution. 

I am not sure if this works for buffered records (if they are sent sequential). I at least keep track of two different values and times I think (Normal and buffered).

jkkataja, 2022-08-13:
Hullo, if this part has been discussed in some point apologies for me missing it.

Would this functionality also allow setting the inverter collector setting to be 1 minute but keep sending data towards Growatt in 5 minute intervals?

That is, locally one would get data every 1 minute but Growatt servers would still receive data every 5 minutes?

johanmeijer, 2022-10-31:
see also issue #197 and discussion #93

johanmeijer, 2023-08-13:
Is available.

closed 2023-08-13 by johanmeijer