johanmeijer/grott #60: Synology NAS Setup Instructions

Paytah232, 2021-07-15:
Hi. I'm finding it hard to follow any instructions here really. Is anyone able to give details step by step instructions, perhaps with screenshots that I could follow along with? Thanks!

johanmeijer, 2021-07-19:
@Paytah232 did you looked at the wiki (https://github.com/johanmeijer/grott/wiki) ? 

I do not have created specific instructions for NAS ( I do not have one) but you can find a generic instruction and also information about how to set up a docker instance (that is I think what you need for a NAS). 

There are people using Grott in combination with a NAS maybe they can help.

Paytah232, 2021-07-22:
Would you be able to tag someone who might be able to help here? Like I said, I'm not sure confident trying by myself, but if we can document gettingit to work on my system hopefully it will help someone else one day!

mvank, 2021-07-22:
If your Synology also support docker, you can use the docker container instead. 
Have you though about that? I am using Grott as docker container, which is working fine for me!

Paytah232, 2021-07-22:
@mvank Where are the instructions to set up a docker container?

mvank, 2021-07-22:
@Paytah232 How much do you need to know?
Have you every worked with docker containers?
Have you already installed docker on you synology?

Paytah232, 2021-07-24:
I have docker installed and running on Synology. I can follow code and ssh into it if necessary, but ideally all setup done through the docker environment would be better. I just don't know what to run with what command and when to run it.

johanmeijer, 2021-07-24:
How to start Grott with docker is describe here: https://github.com/johanmeijer/grott/wiki/Docker-support

But I might also a good idea to look for some dockers basics.  Maybe this something: https://www.baitando.com/it/2019/09/22/using-docker-on-synology-nas

Paytah232, 2021-07-24:
Okay so I took a look. I can set up using an external grott.ini file, however, what format does my .ini information have to be. Also, what should I be using; what is the difference between MQTT and PVOuput.org.
For MQTT, there is an IP address, what IP address of mine would that be? The wireless monitor or somehow the inverter itself?
For PVOuput.org, there is a gpvapikey and systemid; where do I get those from.
Screenshot for easy reference:
[image]

johanmeijer, 2021-07-25:
Grott is an interface between the Growatt inverter and the system/application you want to use to monitor the performance of your solar panels. 

It can write information to: 

- MQTT (a message queuing system that can be used to send message to other applications like Domoticz, Openhab or Home Assistent). 
- InfluxDB (standard database that can be used to store data, that can be used by reporting software like Grafana), 
- PVOutput.org (website: https://pvoutput.org/ which will create reports for you) 

You have to choose one (or more)  and can configure Grott for the use of it. 

In the wiki (https://github.com/johanmeijer/grott/wiki) almost everything is documented so please use the wiki for more information how to configure. 

The key/systemid can retrieve at the pvoutput.org website.

Paytah232, 2021-07-26:
Okay so I think I would like to use InfluxDB so I can send it to a grafana docker. I have an influxDB/grafana docker setup following this video: https://www.youtube.com/watch?v=YtGSOLOg-LA

How do I link the grott docker to this influx db docker?

johanmeijer, 2021-07-26:
How you can configure Grott for use with influxdb is described in the wiki: 

https://github.com/johanmeijer/grott/wiki/InfluxDB-Support

johanmeijer, 2021-07-29:
Via influxdb configuration eg: 
https://docs.influxdata.com/influxdb/v2.0/organizations/

johanmeijer, 2021-08-08:
@Paytah232 I have to reinstall my influxdb environment (system crash after power outage) so if you still need it I can document some things.  

Please let me know. We can then open a new issue for this.

Paytah232, 2021-08-08:
Hi Johan,

Yes please if you could make some instructions I could follow that would be greatly appreciate.

Cheers,
Peter

Reopened #60<https://github.com/johanmeijer/grott/issues/60>.

ZonPV, 2021-08-24:
This is how I configured the container for Grott on my Synology.[[image]<br /><a target='_blank' href='https://nl.imgbb.com/'>photo free image</a><br />](url)

ZonPV, 2021-08-24:
I succeeded in uploading some values to PVoutput.org. Only value that is not configured in the container is exporting the temperature, V5. Didn't figure out how to do that. Any suggestions?

Paytah232, 2021-08-29:
@johanmeijer I appreciate the effort, but my problem was I didn't really understand doing these steps from start to finish. You've shown the finished product, what would be best for this questions is something like this:
Step 1: - Instruction - Link/Image
Step 2: as above, etc etc etc etc.
The problem is I cannot bridge certain gaps as I don't know what to do, so a literal step by step instruction guide would be helpful, rather than the final settings.

johanmeijer, 2022-02-06:
@ZonPV I made it possible to sent inverter temperature as temp value (V5) to pvoutput. 

You can enable this by specifying:

* .ini [PVOutput section] pvtemp = True
* environmental gpvtemp = "True" (docker: -e gpvtemp = "True")

Docker 2.7.1 build is not available yet. I will build it soon (after testing the regular version for a couple of days).

ZonPV, 2022-02-11:
@johanmeijer Thank you Johan! As soon as the new Docker version is available I will install it and let you know. Still very glad with todays version!

johanmeijer, 2022-02-13:
@ZonPV I have published the ledidobe/grott:beta (2.7.1) iamage!

ZonPV, 2022-02-13:
Thank you Johan. 

I will upgrade tomorrow and let you know!

ZonPV, 2022-02-14:
@johanmeijer 
Just upgraded to latest version 2.7.1 and I can confirm that PVoutput has written the temperature of the inverter. 

Thanks a lot for all your good work and communication!

mrairworthy, 2022-06-12:
Hi!

Looks like I'm just as bad in combining docker + grott + Synology Diskstation + pvoutput.org as Paytah232. :)
My setup: I have docker running on my Diskstation 1512+ (using it for openhab). Then I downloaded ledidobe/grott:latest to host the grott. My Growatt MIC 1000 TX-L is connected to my WIFI with the ShineWIFI-X stick. Last but not least I created an account on pvoutput.org and added my PV. 
Docker settings:
[image]
[image]
[image]
Do I miss anything? PVoutput has not recorded any data so far. 

Thanks a lot for your great work!

johanmeijer, 2022-06-13:
I have no experience with a synology (can not have everything although I would like it). 

Can you look in the docker log and see any Grott message's? That would make debugging a lot easier.

To be honest I am not sure we have already tested with a TL-X. If I look at the documentation it looks for the solar energy generating part to most at TL3-X. You can try that by specifing ginvtype=tl3. But we miss additional information if you also have a battery connected. 

I need some information from you to create a layout record specific for the TL-X (from the grott log).

ZonPV, 2022-06-17:
I can confirm that my Growatt MIC 2000TL-X with ShineWifi-X are doiing very well in a docker container on my Synology. It runs on Grott version 2.7.1.
@johanmeijer : You can add the model 2000TL-X with ShineWifi-X to the list of tested models

johanmeijer, 2022-06-17:
Nice! Thanks for the feedback :)

mrairworthy, 2022-06-17:
Can you share your settings? I know u did before but unfortunately they don’t work for me. (I tried to copy all your environment settings)

closed 2023-08-13 by johanmeijer