johanmeijer/grott #63: Grott only saving the pvserial variable to InfluxDB

lspaula, 2021-07-25:
Unfortunately Grott is only saving the pvserial to the database on InfluxDB. The log file shows the following:

```
Jul 25 12:53:47 openhabianpi grott[25631]:                  - record layout :  T06NNNNX
Jul 25 12:53:47 openhabianpi grott[25631]:          - Grott data record date/time processing started
Jul 25 12:53:47 openhabianpi grott[25631]:          - date-time:  2021-07-25T12:53:47
Jul 25 12:53:47 openhabianpi grott[25631]:          - Grott values retrieved:
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvserial             :  XXXXXXXXXX
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvstatus             :  1
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvpowerin            :  482.4
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pv1voltage           :  303.6
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pv1current           :  1.5
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pv1watt              :  482.4
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pv2voltage           :  0.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pv2current           :  0.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pv2watt              :  0.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvpowerout           :  472.6
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvfrequentie         :  60.1
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvgridvoltage        :  215.6
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvgridcurrent        :  2.1
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvgridpower          :  474.6
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvgridvoltage2       :  0.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvgridcurrent2       :  0.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvgridpower2         :  0.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvgridvoltage3       :  0.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvgridcurrent3       :  0.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvgridpower3         :  0.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  totworktime          :  163.2
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvenergytoday        :  5.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvenergytotal        :  190.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  epvtotal             :  192.4
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  epv1today            :  5.1
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  epv1total            :  192.4
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  epv2today            :  0.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  epv2total            :  0.0
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvtemperature        :  44.2
Jul 25 12:53:47 openhabianpi grott[25631]:                  -  pvipmtemperature     :  50.8
Jul 25 12:53:47 openhabianpi grott[25631]:          - MQTT jsonmsg:
Jul 25 12:53:47 openhabianpi grott[25631]:                          {"device": "XXXXXXXXXX", "time": "2021-07-25T12:53:47", "buffered": "no",
Jul 25 12:53:47 openhabianpi grott[25631]:                          "values": {"pvstatus": 1, "pvpowerin": 4824, "pv1voltage": 3036,
Jul 25 12:53:47 openhabianpi grott[25631]:                          "pv1current": 15, "pv1watt": 4824, "pv2voltage": 0, "pv2current": 0,
Jul 25 12:53:47 openhabianpi grott[25631]:                          "pv2watt": 0, "pvpowerout": 4726, "pvfrequentie": 6009, "pvgridvoltage":
Jul 25 12:53:47 openhabianpi grott[25631]:                          2156, "pvgridcurrent": 21, "pvgridpower": 4746, "pvgridvoltage2": 0,
Jul 25 12:53:47 openhabianpi grott[25631]:                          "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0,
Jul 25 12:53:47 openhabianpi grott[25631]:                          "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 1174895,
Jul 25 12:53:47 openhabianpi grott[25631]:                          "pvenergytoday": 50, "pvenergytotal": 1900, "epvtotal": 1924, "epv1today":
Jul 25 12:53:47 openhabianpi grott[25631]:                          51, "epv1total": 1924, "epv2today": 0, "epv2total": 0, "pvtemperature": 442,
Jul 25 12:53:47 openhabianpi grott[25631]:                          "pvipmtemperature": 508}}
Jul 25 12:53:47 openhabianpi grott[25631]:          - MQTT message message sent
Jul 25 12:53:47 openhabianpi grott[25631]:          - Grott Send data to PVOutput disabled
Jul 25 12:53:47 openhabianpi grott[25631]:          - Grott InfluxDB publihing started
Jul 25 12:53:47 openhabianpi grott[25631]:          - Grott original time :  2021-07-25T12:53:47 adjusted UTC time for influx :  2021-07-25T15:53:47
Jul 25 12:53:47 openhabianpi grott[25631]:          - Grott influxdb jsonmsg:
Jul 25 12:53:47 openhabianpi grott[25631]:                          [{'measurement': 'XXXXXXXXXX', 'time': '2021-07-25T15:53:47', 'fields':
Jul 25 12:53:47 openhabianpi grott[25631]:                          {'pvserial': 'XXXXXXXXXX', 'pvstatus': 1, 'pvpowerin': 4824, 'pv1voltage':
Jul 25 12:53:47 openhabianpi grott[25631]:                          3036, 'pv1current': 15, 'pv1watt': 4824, 'pv2voltage': 0, 'pv2current': 0,
Jul 25 12:53:47 openhabianpi grott[25631]:                          'pv2watt': 0, 'pvpowerout': 4726, 'pvfrequentie': 6009, 'pvgridvoltage':
Jul 25 12:53:47 openhabianpi grott[25631]:                          2156, 'pvgridcurrent': 21, 'pvgridpower': 4746, 'pvgridvoltage2': 0,
Jul 25 12:53:47 openhabianpi grott[25631]:                          'pvgridcurrent2': 0, 'pvgridpower2': 0, 'pvgridvoltage3': 0,
Jul 25 12:53:47 openhabianpi grott[25631]:                          'pvgridcurrent3': 0, 'pvgridpower3': 0, 'totworktime': 1174895,
Jul 25 12:53:47 openhabianpi grott[25631]:                          'pvenergytoday': 50, 'pvenergytotal': 1900, 'epvtotal': 1924, 'epv1today':
Jul 25 12:53:47 openhabianpi grott[25631]:                          51, 'epv1total': 1924, 'epv2today': 0, 'epv2total': 0, 'pvtemperature': 442,
Jul 25 12:53:47 openhabianpi grott[25631]:                          'pvipmtemperature': 508}}]
Jul 25 12:53:47 openhabianpi grott[25631]:          - Grott write to influxdb v1

```

At influx, if I query the measurements on the database, only the serial is shown:

```
 Connected to http://localhost:8086 version 1.8.3
InfluxDB shell version: 1.8.3
> SHOW MEASUREMENTS ON growatt_db
name: measurements
name
----
XXXXXXXXXX
>

```

johanmeijer, 2021-07-25:
That is oke I think. You have to look at what is in the measurement with eg:

select * from xxxxxxxxx limit desc 10

This shows the last 10 entries.

lspaula, 2021-07-25:
Yes, you are right. But then in Grafana, I can only see this top level serial number. 

How do I get the internal values?
[image]

johanmeijer, 2021-07-25:
If you select default xxxxxxxx then select filed(value) should give a list of  the growatt values.

lspaula, 2021-07-25:
Ok. Sorry, that was my fault not looking into the field value. Thank you very much for the quick response. 

By the way, I am having to divide all the data by 10 either in grafana as well as on the MQTT message. I understand that this is the way Growatt inverters send the data to their server (as integers). Is there a way to natively divide by 10 and get all values correct with Grott? 

I use Openhab and would like to get all inverter parameters as items and not as a JSON message. I am doing this manually with transformations for each parameter. Is this the only way of doing this?

johanmeijer, 2021-07-26:
I am not using openHab so I do not know what and how to specify/configure things. On the openhab forum there is an item on Grott maybe that helps?

https://community.openhab.org/t/binding-request-growatt-server/96288/15

At this moment Grott only supports formatting the values for the Grott verbose output message (what you see at the screen/syslog). I have thought to made it general available (so also in the MQTT message and  influxddb) but that has to be something next to the current growatt provided values (already used by the current Grott users). 

At this moment I think passing the values the Growatt inverter sends is the best (basic) solution. It is not that hard to do the formatting in e.g. Grafana. But maybe I wil add the possibility in the future. Convince me it is needed ;)

johanmeijer, 2021-08-08:
@Ispaula, 

Did you mange it to get the items in OpenHab?

lspaula, 2021-08-10:
Hello,

Yes, I did the following to include just the pvgridpower:

Included a Thing:

```
    Thing topic Growatt "Solar Inverter" {
    Channels:
        Type string : GrowattLog [ stateTopic="stat/power/growatt", commandTopic="cmnd/power/growatt" ]
        Type number : GrowattPowerToGrid [ stateTopic="stat/power/growatt", transformationPattern="JSONPATH:$.values.pvgridpower"]
    }

```
Included the Items:

```
// Growatt
String GrowattLog "Growatt Inverter: MQTT return message [%s]" { channel="mqtt:topic:MyBroker:Growatt:GrowattLog" }
String GrowattPowerToGrid "Injected Power [%s W]" { channel="mqtt:topic:MyBroker:Growatt:GrowattPowerToGrid"[profile="transform:JS", function="divideByTen.js"]}
```
And the Sitemap:
```
  Group item=gSolar label="Consumos" icon="energy" {
          Frame label="Potências" {
                Text item=GrowattPowerToGrid valuecolor=[>1000="orange",>200="green",<=200="blue"] icon="energy"
                Text item=CurrentPowerConsumption valuecolor=[>2000="orange",>500="green",<=500="blue"] icon="energy"
          }
       }

```

But I also managed to capture all the data in Grafana, which together with my house consumption is showing me all I wanted:

[image]

Thank you very much for the help.

Regards,

Luciano

johanmeijer, 2021-08-22:
@lspaula 

Nice!

Can we close this issue?

lspaula, 2021-08-22:
Sure, thank you. It was sorted out with respect to grafana.

closed 2021-08-22 by lspaula