johanmeijer/grott #65: Grott (docker) server keeps receiving buffered messaged

gerthass, 2021-08-07:
I'm encounter the following issue. My grott server keeps receiving buffered messages from 2 days ago. It looks like they are coming from the growatt server but I do not have any port opened. the result is that my Homeassistant keeps getting both MQTT messaged, the live ones from the inverter via the Grott proxy and the buffered ones from ???. 

[image]
[image]
[image]

Does anyone else have this issue or knows where to look for the cause?

johanmeijer, 2021-08-08:
@KBraham the buffered ones are comming from the growatt inverter. 

The inverter (shine stick) will sent the buffer a soon as communicating is started. You write you do not have opened a port does this means that you do not communicate with the growatt server on the internet(47.91.67.66, port 5279) ? Normally the records are deleted from the buffer in the inverter wen they are sent (and confirmed) by the growatt server. 
Grott is a proxy a or normal communication flow a connection with the growatt server is needed.

You can disable processing buffered data by specifing: sendbuf = False in the [Generic] of the .ini file.

gerthass, 2021-08-09:
Thanks for the answer. I set sendbuf = False so I will have to wait untill the buffer is empty I suppose because I still receive buffed messages from yesterday. 

The grott server worked just fine for 2 months, but since my inverter didnt have an shinelink connected for a couple of days its looks like strange things are happening. It seems it does not receive any date/time anymore, is that correct?

The Grott server communicates fine with the Growatt server since I can see all the data in the shinelink app/site. I don't have to open any port in my router for that right? (at least my installation firm didn't ask for that and everything seems to work just fin)

just some more background information of my configuration:
[image]

```
 - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.2.3', 5279), raddr=('10.0.2.5', 1033)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.2.3', 55572), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x06\x00\x06\x02\x41\x51\x04\x09\x33\x2c\x42\x20\x47\x47\x77\x3a\x59\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x3b\x32\x26\x46\x35\x72\x42\x5f\x43\x31\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x6a
                 \xcc\x78\x73\x72\x6f\x77\x61\x74\x74\x45\x37\x6f\x77\x61\x74\x74\x47\x70\x26
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x7c\xf0\x69\x81
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x4f\x87\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x77\x52\xcd\x72\x6f\x77\xfe\x74\x74\x46\x64\x6f\x77\x60\x69\x74
                 \x47\x72\x3d\x77\x61\x74\xfa\x47\x72\x6f\x25\x61\x74\x74\xc8\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x75\x1b\x73\x9b\x77\x61\x3a\x54\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x75\x74\x46\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x58\x74\x6b\x73
                 \x07\x79\x71\x53\x64\x47\x7b\x6f\x77\x61\x74\x74\x47\x95\x36
         - Grott automatic protocol detection
         - Grott data record length 585
         - layout   :  T065104X
         - no matching record layout found, try generic
         - Record layout used :  T06NNNNX
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 00060006024151044e41433541333330483600000000000000000000000000000000000000005
                 44547324135303034500000000000000000000000000000000000000000000000000000020bb8
                 0c340000000000000245000000000000024900000000000000000000000000000000000000000
                 000000000000000000000000000138708f5000000000000000000000000000000000000000000
                 0008f5000000000000000000000000000000000003268a0000009f000001160000011d0000005
                 20000008e000000520000008f0000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 0000000015c01f400004e20000000000000000000000000000000000000000000000000000000
                 00003c0000000000000000000000000000000000000c350cb1000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000002710000000000000000000000000000000000
[... cut ...]
                         "epv2today": 82, "epv2total": 143, "pvtemperature": 0, "pvipmtemperature":
                         0}}
         - MQTT message message sent
         - Grott Send data to PVOutput disabled 
         - Grott Send data to Influx disabled 
         - Grott extension processing disabled 

         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.2.3', 5279), raddr=('10.0.2.5', 1033)>
         - Data less then minimum record length, data not processed
```

gerthass, 2021-08-10:
Edit: I changed my environmental _sendbuf_ to _gsendbuf_, now all communication is blocked. I think the issue is coming from the inverter which doesn't send the right date/time? could that be it?

on the shinelink app and server website everything seems to be fine

johanmeijer, 2021-08-11:
@gerthass interesting. If it is data from the buffer it is detected by the record type, not time. 

I would be strange if your inverter is only sending buffered data. In the example you sent above it is not buffered data (buffered": "no", in MQTT message). 

It looks like your inverter does not pass time to in the record:   no or no valid time/date found, grott server time will be used (buffer records not sent!). Grott will use the time of the server (in your case probably the docker container). 

Normally a docker container is started with GMT as time zone. You have to add the TZ parameter to the docker run to set this in your local time zone.

I am not sure this will helps. It is strange no data is sent at all if you specify gsendbuf = false. I will test that.

gerthass, 2021-08-20:
Oke, I've added the TZ of my location (Europe/Amsterdam) and the time is send correctly now.

The original issue has been magically solved. I've no clue, but the inverted did stop sending buffed messages. I think its cache is now clear again or something like that :/

Thanks for you responses and this great solution for catching the messages :) I hope that someday we are able to do this without sending anything to an cloud server!

gerthass, 2021-09-05:
Re-opening this since the issue is back. 

My inverter has been offline for 2-3 hours because the provider did some network maintenance. THat was also the cause of my issue the 1st time. 

So I keep receiving buffed messaged with a time in the passed, but GROTT processes them as if they are now. 
MY regular messages received from the inverter have no valid date/time. 

When I set 'gsendbuf' to False, Grott does not process anything at all..

please help :)

```
         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.2.3', 43026), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x01\x00\x06\x02\x41\x01\x50\x00\x22\x28\x47\x20\x47\x47\x76\x2a\x58\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x3b\x32\x26\x46\x35\x72\x42\x5f\x43\x31\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x7d\x71\x43\x50\x60\x75\x6a
                 \xcc\x78\x73\x72\x6f\x77\x61\x74\x74\x45\xf0\x6f\x77\x61\x74\x74\x47\x70\xea
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x7c\xfe\x69\x96
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x4f\x90\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x5f\x30\x2a\x72\x6f\x77\x61\x74\x74\x4a\x29\x6f\x77\x6c\xbe\x74
                 \x47\x72\x6f\x77\x61\x72\xa8\x47\x72\x6f\x77\x61\x74\x72\xa9\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\xb8\x88\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x75
                 \x5b\x46\x5d\x6f\x77\x61\x74\x75\x6d\x73\xa6\x77\x61\x3a\x54\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x75\x74\x46\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x46\x74\x75\x73
                 \x07\x79\x71\x53\x64\x47\x7b\x6f\x77\x61\x74\x74\x47\x9f\x04
         - Grott automatic protocol detection
         - Grott data record length 585
         - layout   :  T060150X
         - no matching record layout found, try generic
         - Record layout used :  T06NNNNX
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 00010006024101504750473041333331583700000000000000000000000000000000000000005
                 4454732413530303450000000000000000000000000000000000000000015090504220f020bb8
                 0c340000000000000282000000000000028500000000000000000000000000000000000000000
                 000000000000000000000000000138908e2000000000000000000000000000000000000000000
                 0008e200000000000000000000000000000000002b446d0000000000000d5b00000dca0000000
                 0000006dc00000000000006ee0000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000fffa00000000000000000000012f012f0
                 0000000012a01c900004e20000000000000000000000000000000000000000000000000000000
                 00003c0000000000000000000000000000000000000c350cb1000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000002710000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 00000000000000000000000000010001000000000000000000000032003201680e10271000090
                 00000000000ed6b
[... 139 lines cut ...]
                 -  pvgridcurrent3       :  0.0
                 -  pvgridpower3         :  0.0
[... cut ...]
                         "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0,
                         "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime":
                         2862422, "pvenergytoday": 64, "pvenergytotal": 3483, "epvtotal": 3595,
                         "epv1today": 32, "epv1total": 1788, "epv2today": 33, "epv2total": 1807,
                         "pvtemperature": 488, "pvipmtemperature": 488}}
         - Buffered record not sent: sendbuf = False or invalid date/time format
```

johanmeijer, 2021-09-06:
@gerthass It is normal that a lot of buffered records are sent after a connection error of a couple of hours. 

Your first record is a buffered record with suprisingly a readable time record.
I am not sure this is the real time record is created:  time": "2021-09-05T04:34:15". Do your solar panels produce power at 04:34:15?
But you specified buffered = False so this one is not sent:  
- Buffered record not sent: sendbuf = False or invalid date/time format

This seems to be okay. 

The second record is a non buffered record but with no time information in it, so it get the server date/time. This one should not be filtered. 
This is a bug! It seems that if sendbuf = False is specified no records are beeing sent at all. 

I created a fix and am testing it. I will publish this fix (if it works) the comming days. 

Do you use the ledidobe/grott docker container from  docker hub? If so I will create a new docker container as well.

gerthass, 2021-09-07:
> I am not sure this is the real time record is created: time": "2021-09-05T04:34:15". Do your solar panels produce power at 04:34:15?
If the time of the buffered record is ok, why don't sent it? Maybe not specifying sendbuf (or sendbuf = true) is not really a problem?

Strange thing is that there is no power produced at that time. This is also visible in the message, no energy (0W). Not sure why the inverter is sending messages at that time, but GROTT handles the messages as if these are now. (in my previous example the message had the date/time of"2021-09-05T04:34:15" but the real time was 10:43AM.
Also the internet connection has not been down around that time, and to be complete, the networkconnection has never been down. I have no clue where those messages are coming from... :/

johanmeijer, 2021-09-07:
Strange. Grott is doing nothing with the time it is the time set in the record by the growatt inverter. 

The Growatt inverter get its time from the Growatt server. So if you specified your timezone at the growatt website it should be ok.

But you are not the only one with time problems so sometimes it does not work as expected . Probably a bug in the inverter firmware. 

I will create a new docker image with the bug fix so the sendbuf = false will work.

johanmeijer, 2021-09-08:
@gerthass I created a new docker image. You can refer to it by ledidobe/grott:beta or ledidobe/grott:2.6.1c. 

Please let me know if this solves your buffered record problem (records should be suppressed if you specify gsendbuf=False!)

gerthass, 2021-09-14:
Hi @johanmeijer ,

The gsendbuf = False did solve the issue:

```
 - Grott data record date/time processing started
         - date-time:  2021-09-13T20:27:51
         - Grott values retrieved:
                 -  pvserial             :  TEG2A5004P
                 -  pvstatus             :  0
                 -  pvpowerin            :  0.0
                 -  pv1voltage           :  0.0
                 -  pv1current           :  0.0
                 -  pv1watt              :  0.0
                 -  pv2voltage           :  0.0
                 -  pv2current           :  0.0
                 -  pv2watt              :  0.0
                 -  pvpowerout           :  0.0
                 -  pvfrequentie         :  0.0
                 -  pvgridvoltage        :  0.0
                 -  pvgridcurrent        :  0.0
                 -  pvgridpower          :  0.0
                 -  pvgridvoltage2       :  0.0
                 -  pvgridcurrent2       :  0.0
                 -  pvgridpower2         :  0.0
                 -  pvgridvoltage3       :  0.0
                 -  pvgridcurrent3       :  0.0
                 -  pvgridpower3         :  0.0
                 -  totworktime          :  511.4
                 -  pvenergytoday        :  8.0
                 -  pvenergytotal        :  452.5
                 -  epvtotal             :  467.1
                 -  epv1today            :  4.1
                 -  epv1total            :  232.2
                 -  epv2today            :  4.2
                 -  epv2total            :  234.9
                 -  pvtemperature        :  0.0
                 -  pvipmtemperature     :  35.7
         - MQTT jsonmsg: 
                         {"device": "TEG2A5004P", "time": "2021-09-13T20:27:51", "buffered": "yes",
                         "values": {"pvstatus": 0, "pvpowerin": 0, "pv1voltage": 0, "pv1current": 0,
                         "pv1watt": 0, "pv2voltage": 0, "pv2current": 0, "pv2watt": 0, "pvpowerout":
                         0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvgridcurrent": 0, "pvgridpower":
                         0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0,
                         "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime":
                         3681977, "pvenergytoday": 80, "pvenergytotal": 4525, "epvtotal": 4671,
                         "epv1today": 41, "epv1total": 2322, "epv2today": 42, "epv2total": 2349,
                         "pvtemperature": 0, "pvipmtemperature": 357}}
         - Buffered record not sent: sendbuf = False or invalid date/time format

         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.2.3', 5279), raddr=('10.0.2.5', 1034)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.2.3', 60438), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\xac\x00\x06\x02\x41\x01\x04\x00\x22\x28\x47\x20\x47\x47\x76\x2a\x58\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x3b\x32\x26\x46\x35\x72\x42\x5f\x43\x31\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x6a
                 \xcc\x78\x73\x72\x6e\x77\x61\x76\xf6\x41\xa5\x6f\x76\x61\x74\x75\x03\x74\xc4
                 \x77\x60\x74\x74\x46\x4c\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x45\x15\x7c\xf1\x69\xa5
                 \x74\x41\x72\x6f\x75\x16\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x4f\xa3\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x4d\x1b\x28\x72\x6f\x77\xe6\x74\x74\x55\x46\x6f\x77\x73\xb3\x74
                 \x47\x72\x2c\x77\x61\x7d\x21\x47\x72\x6f\x32\x61\x74\x7d\x35\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
[... 183 lines cut ...]
                 -  pvenergytoday        :  13.5
                 -  pvenergytotal        :  466.0
                 -  epvtotal             :  480.7
                 -  epv1today            :  6.7
                 -  epv1total            :  238.9
                 -  epv2today            :  6.9
                 -  epv2total            :  241.8
                 -  pvtemperature        :  32.5
                 -  pvipmtemperature     :  41.3
         - MQTT jsonmsg: 
                         {"device": "TEG2A5004P", "time": "2021-09-14T19:08:55", "buffered": "no",
                         "values": {"pvstatus": 1, "pvpowerin": 642, "pv1voltage": 1751,
                         "pv1current": 1, "pv1watt": 324, "pv2voltage": 1707, "pv2current": 1,
                         "pv2watt": 318, "pvpowerout": 615, "pvfrequentie": 4998, "pvgridvoltage":
                         2257, "pvgridcurrent": 6, "pvgridpower": 631, "pvgridvoltage2": 0,
                         "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0,
                         "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 3764079,
                         "pvenergytoday": 135, "pvenergytotal": 4660, "epvtotal": 4807, "epv1today":
                         67, "epv1total": 2389, "epv2today": 69, "epv2total": 2418, "pvtemperature":
                         325, "pvipmtemperature": 413}}
         - MQTT message message sent
         - Grott Send data to PVOutput disabled 
         - Grott Send data to Influx disabled 
         - Grott extension processing disabled 
```

johanmeijer, 2021-09-15:
Niet. Can we close this issue?

gerthass, 2021-09-15:
Yes, thanks a lot for the great work!

closed 2021-09-15 by gerthass