johanmeijer/grott #67: Growatt SPH 3000

macfloripa, 2021-08-29:
Hello,
First of all, thanks for your efforts with this project. I have just purchased and installed a SPH 3000 model and need some assistance.

I'm running the 2.6.1 version, as per your instructions. Standalone, proxy (not docker) on a raspberry pi, together with influx.

I've specified inverter=spf  in the ini file, and am getting some good and bad values from the program.

I tried to specify inverter=sph, but got back bad values, so switched back to spf, where i get most of the values i need.

When I specified sph I got this message:
--------------
#011 - Growatt packet received:
Aug 29 09:17:28 zm grott[3835]: #011#011  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.206', 4358\
4), raddr=('47.91.67.66', 5279)>
Aug 29 09:17:28 zm grott[3835]: #011 - Growatt original Data:
Aug 29 09:17:28 zm grott[3835]: #011#011 \x00\x46\x00\x05\x00\xd9\x01\x04\x0a\x25\x28\x47\x20\x40\x44\x77\x38\x5f\x33
Aug 29 09:17:28 zm grott[3835]: #011#011 \x23\x33\x44\x06\x46\x5e\x47\x52\x39\x61\x4f\x6f\x66\x67\x54\x76\x74\x47\x72
Aug 29 09:17:28 zm grott[3835]: #011#011 \x43\x77\x64\x77\x8f\x47\x72\x6f\x77\x67\xa2\x74\x47\x72\x6f\x77\x43\x74\x74
Aug 29 09:17:28 zm grott[3835]: #011#011 \x47\x72\x68\x93\x61\x74\x7c\x21\x72\x6f\x77\x61\x74\x74\x47\x72\x7b\x7d\x61
Aug 29 09:17:28 zm grott[3835]: #011#011 \x10\x7b\x3b\x7a\xa0\x60\x0f\x7c\xec\x50\x02\x6f\x77\x60\x66\x74\xb6\x72\x28
Aug 29 09:17:28 zm grott[3835]: #011#011 \x77\x61\x74\x74\x47\x72\x6f\x77\x60\xf6\x74\x47\x72\x66\x77\x68\x74\x74\x47
Aug 29 09:17:28 zm grott[3835]: #011#011 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x09\xf5\x6f\x5a\x61\x2d
Aug 29 09:17:28 zm grott[3835]: #011#011 \x74\x46\x72\x6f\x77\x61\x74\x74\x47\x70\x6f\x77\x61\x2d\x74\x47\x72\x6f\x77
Aug 29 09:17:28 zm grott[3835]: #011#011 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x66\x72\x6f\x77\x63\x74\x74\x47\x48
Aug 29 09:17:28 zm grott[3835]: #011#011 \x6f\x77\x61\x70\x74\x47\x72\x50\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Aug 29 09:17:28 zm grott[3835]: #011#011 \x47\x7a\x9d\x77\x61\x7c\x12\x47\x72\x6d\x2f\x61\x74\x74\x47\x72\x6f\x77\x61
Aug 29 09:17:28 zm grott[3835]: #011#011 \x74\x55\x47\x42\x6f\x77\x61\x72\x74\x47\x72\x1a\x77\x61\x86\x84
Aug 29 09:17:28 zm grott[3835]: #011 - Grott automatic protocol detection
Aug 29 09:17:28 zm grott[3835]: #011 - Grott data record length 225
Aug 29 09:17:28 zm grott[3835]: #011 - layout   :  T050104SPH
Aug 29 09:17:28 zm grott[3835]: #011 - no matching record layout found, try generic
Aug 29 09:17:28 zm grott[3835]: #011 - no matching record layout found, standard processing performed
Aug 29 09:17:28 zm grott[3835]: #011 - Record layout used :  none
Aug 29 09:17:28 zm grott[3835]: #011 - Growatt data decrypted V2
Aug 29 09:17:28 zm grott[3835]: #011 - Grott Growatt data decrypted
Aug 29 09:17:28 zm grott[3835]: #011 - Growatt plain data:
Aug 29 09:17:28 zm grott[3835]: #011#011 0046000500d901044d574730413430304a304442473041343130334d15081d091035020000002
Aug 29 09:17:28 zm grott[3835]: #011#011 c000503fb0000000006d60000000000220000000007e4000008660000000000000000140a0064
Aug 29 09:17:28 zm grott[3835]: #011#011 0f7c08cf176e089817700000011200f1004700000000000000000182000000090009000000000
Aug 29 09:17:28 zm grott[3835]: #011#011 000000000000000000000004e87002d0059000100000000000000020000005900000000000000
Aug 29 09:17:28 zm grott[3835]: #011#011 000000000000000021000000020000003a000000040000003f00000000000000000000000008f
Aug 29 09:17:28 zm grott[3835]: #011#011 2000008660000025800000000000000000021003000000006000000750000f2f0
Aug 29 09:17:28 zm grott[3835]: #011 - Grott data ack record or data record not defined no processing done

----------

When I specified SPF, I get back:

Aug 29 09:19:41 zm grott[3906]: #011 - Grott automatic protocol detection
Aug 29 09:19:41 zm grott[3906]: #011 - Grott data record length 225
Aug 29 09:19:41 zm grott[3906]: #011 - layout   :  T050104SPF
Aug 29 09:19:41 zm grott[3906]: #011 - no matching record layout found, try generic
Aug 29 09:19:41 zm grott[3906]: #011 - Record layout used :  T05NNNNSPF
Aug 29 09:19:41 zm grott[3906]: #011 - Growatt data decrypted V2
Aug 29 09:19:41 zm grott[3906]: #011 - Grott Growatt data decrypted
Aug 29 09:19:41 zm grott[3906]: #011 - Growatt plain data:
Aug 29 09:19:41 zm grott[3906]: #011#011 0047000500d901044d574730413430304a304442473041343130334d15081d091328020000002
Aug 29 09:19:41 zm grott[3906]: #011#011 c000503cd00000000065e00000000001f0000000007d000000866000000000000000014080064
Aug 29 09:19:41 zm grott[3906]: #011#011 0f7a08c0176b0898176a0000011200f2004700000000000000000181000000090009000000000
Aug 29 09:19:41 zm grott[3906]: #011#011 000000000000000000000004e87002d0059000100000000000000020000005900000000000000
Aug 29 09:19:41 zm grott[3906]: #011#011 000000000000000021000000020000003a000000040000003f00000000000000000000000008d
Aug 29 09:19:41 zm grott[3906]: #011#011 400000866000002620000000000000000002000300000000600000075000099b2
Aug 29 09:19:41 zm grott[3906]: #011 - Growatt new layout processing
Aug 29 09:19:41 zm grott[3906]: #011#011 - decrypt       :  True
Aug 29 09:19:41 zm grott[3906]: #011#011 - offset        :  6
Aug 29 09:19:41 zm grott[3906]: #011#011 - record layout :  T05NNNNSPF
Aug 29 09:19:41 zm grott[3906]: #011 - Grott data record date/time processing started
Aug 29 09:19:41 zm grott[3906]: #011 - date-time:  2021-08-29T09:19:40
Aug 29 09:19:41 zm grott[3906]: #011 - Grott values retrieved:
.... (list of values).

Questions:

1) How can I specify a log file just for grott?  (to help with debugging).

2) What information do you need from me to help me debug?

3) Does the program distinguish between SPH and SPF? I only saw that I could use spf in the init file...

4) I'd like to get the SPH Record Layout working. Is there a way to force the program to use a specific layout? I know in previous versions moving the json file to the working directory did that...

Thanks in advance.

macfloripa, 2021-08-31:
ok. Some additional information. I upgraded my shinewifi firmware version from 1.7.66 to 3.0.0.2 and now grott is not able to read the data from the unit. Seems like a new format?

Aug 31 11:32:37 zm grott[28397]:          - Growatt original Data:
Aug 31 11:32:37 zm grott[28397]:                  \x00\x02\x00\x06\x01\x5f\x01\x50\x0a\x25\x28\x47\x20\x40\x44\x77\x38\x5f\x77
Aug 31 11:32:37 zm grott[28397]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Aug 31 11:32:37 zm grott[28397]:                  \x2b\x35\x26\x44\x35\x73\x43\x5f\x44\x2c\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Aug 31 11:32:37 zm grott[28397]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x74\x7c\x6b\x4c\x68\x46\x74\x61
Aug 31 11:32:37 zm grott[28397]:                  \x74\x74\x6b\x72\x6a\x74\x82\x74\x74\x47\x72\x66\xe5\x61\x74\x74\x47\x72\x5f
Aug 31 11:32:37 zm grott[28397]:                  \x77\x61\x74\x74\x4b\x48\x6f\x77\x6d\x02\x74\x47\x72\x6f\x77\x61\x74\x74\x54
Aug 31 11:32:37 zm grott[28397]:                  \xaf\x6f\x13\x6e\x2c\x7c\x18\x65\x1b\x7f\xf9\x63\x00\x47\x72\x6e\x5b\x61\x8a
Aug 31 11:32:37 zm grott[28397]:                  \x74\x2d\x72\x6f\x77\x61\x74\x74\x47\x72\x6e\x0c\x61\x74\x74\x49\x72\x61\x77
Aug 31 11:32:37 zm grott[28397]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x3a\xf3\x47\x5f
Aug 31 11:32:37 zm grott[28397]:                  \x6f\x2e\x61\x75\x74\x47\x72\x6f\x77\x61\x74\x71\x47\x72\x6f\x09\x61\x74\x74
Aug 31 11:32:37 zm grott[28397]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x71\x6f\x77\x61\x3c\x74\x47\x72\x66\x77\x61
Aug 31 11:32:37 zm grott[28397]:                  \x74\x0f\x47\x72\x6f\x71\x61\x74\x74\x3e\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Aug 31 11:32:37 zm grott[28397]:                  \x77\x61\x74\x78\x97\x72\x6f\x7c\x8b\x74\x74\x45\x8a\x6f\x77\x61\x74\x74\x47
Aug 31 11:32:37 zm grott[28397]:                  \x72\x6f\x77\x50\x74\x33\x47\x72\x6f\x79\x61\x74\x74\xad\x72\x6f\x77\x3b\x74
Aug 31 11:32:37 zm grott[28397]:                  \xf2\x47\x5f\x6f\x76\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
Aug 31 11:32:37 zm grott[28397]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Aug 31 11:32:37 zm grott[28397]:                  \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Aug 31 11:32:37 zm grott[28397]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Aug 31 11:32:37 zm grott[28397]:                  \x74\x74\x46\x72\x6e\x77\x60\x74\x74\x47\x72\x6f\x77\x61\x74\x66\x98
Aug 31 11:32:37 zm grott[28397]:          - Grott automatic protocol detection
Aug 31 11:32:37 zm grott[28397]:          - Grott data record length 359
Aug 31 11:32:37 zm grott[28397]:          - layout   :  T060150SPF
Aug 31 11:32:37 zm grott[28397]:          - no matching record layout found, try generic
Aug 31 11:32:37 zm grott[28397]:          - no matching record layout found, standard processing performed
Aug 31 11:32:37 zm grott[28397]:          - Record layout used :  none
Aug 31 11:32:37 zm grott[28397]:          - Growatt data decrypted V2
Aug 31 11:32:37 zm grott[28397]:          - Grott Growatt data decrypted
Aug 31 11:32:37 zm grott[28397]:          - Growatt plain data:
Aug 31 11:32:37 zm grott[28397]:                  00020006015f01504d574730413430304a3000000000000000000000000000000000000000004
Aug 31 11:32:37 zm grott[28397]:                  442473041343130334d000000000000000000000000000000000000000015081f0b1a29030000
Aug 31 11:32:37 zm grott[28397]:                  002c000503e3000000000992000000000030000000000c3a00000c76000000000000000013dd0
Aug 31 11:32:37 zm grott[28397]:                  0640f58085f1774089817740000012c00fe006a0000000000000000017b0000000e000e000000
Aug 31 11:32:37 zm grott[28397]:                  000000000000000000000000004e87002d0059000100000000000000050000007e00000000000
Aug 31 11:32:37 zm grott[28397]:                  000000000000300000048000000090000007b0000000600000079000000000000000000000000
Aug 31 11:32:37 zm grott[28397]:                  0cd000000bea000002f80000000000000000003100470000000e000000ea0000005a0086002d0
Aug 31 11:32:37 zm grott[28397]:                  00100000000000000000000000000000000000000000000000000000000000000000000000000
Aug 31 11:32:37 zm grott[28397]:                  00000000000000000000000000000000000000000000000000000000000000000000000001000
Aug 31 11:32:37 zm grott[28397]:                  10001000000000000000012df
Aug 31 11:32:37 zm grott[28397]:          - Grott data ack record or data record not defined no processing done

johanmeijer, 2021-08-31:
T060150SPF is a standard buffered record. Should not give a problem but I see thee is a bug in the coding for this record. 
I Will repair it. 

Do you run Grott natively or in a docker container?

macfloripa, 2021-08-31:
Hi Johan,
I run it Native on a raspberry pi.
Thanks.

johanmeijer, 2021-09-01:
@macfloripa, I have made a change to 2.6 branch (2.6.1a). Hopefully it works now!

macfloripa, 2021-09-01:
Nice. I'm getting data again, and at a first quick glance, it looks good. I'll keep an eye on it for a time and see if I notice any issues. Thanks so much!

macfloripa, 2021-09-07:
All seems good. Question, does your docker image contain the changes you made? I want to move to a docker installation on my NAS (QNap). Thanks

johanmeijer, 2021-09-07:
No I have not created a  new  docker image yet. Will do it the comming days,

gabrielpc1190, 2021-11-11:
I can confirm for SPF 3000 TL-LVM 24 volts the image ledidobe/grott:2.6.1c works really good!

gabrielpc1190, 2021-11-11:
I have the examples for Home Assistant already.
[mqtt-for-jbd-and-grott.zip](https://github.com/johanmeijer/grott/files/7518184/mqtt-for-jbd-and-grott.zip)
It's a file that contains a yaml file I included on my configuration file.

closed 2021-11-14 by macfloripa