johanmeijer/grott #72: how to install Grott on rpi (openhab 3)

TurboMOD, 2021-10-31:
hi,
I already try to find a "how to" since a few weeks now. my issue is, iam a total noob. python, mqtt, json, influxdb, docker ect. just some kind of magic to me, :/
I got openhab running on a rpi 4, iam able to connect via putty (ssh) or winSCP to have some kind of fileexplorer. I just started creating some rules in openhab ui using javascript, which wasnt that hard...just to give you an idea how noob iam right now ;)

I was looking for the "First time installation" guide. but its only for windows and I want to have grott running on the rpi.
I found https://github.com/johanmeijer/grott/issues/18 but he also want to have it on windows.
https://github.com/johanmeijer/grott/issues/16 he want it on home assistent

but first post from johanmeijer might be a starting point for me:

> What you have to do in steps:

prepare base environment (linux)
install mqtt (i use mosquitto)
install python3 /pip3
install paho-mqtt with pip3 (see wiki)
install grott (see wiki)
optional enable grott as a service (see wiki)
reroute the growatt communication via the grott server (see wiki)
configure home assistent (see wiki)
Hopefully this helps. <

Maybe I have to change the last point (configure home assistent) to configure openhab. But I already stuck at the first point (prepare base environment (linux)) not knowing if I have to do that or what I have to prepare.

I dont wanna waste anyones time. Just need some points what I should look for.

johanmeijer, 2021-11-01:
The first time installation was not ment for windows installation only. 

The only relation with windows is the installation of python because that is not standard enabled in Windows (and is in most Linux distributions). 

I indeed assumed that the user had some Linux experience and know how to download Grott and install additional software like Mosquitto. I did not realised that the might not be te case if you use a prepared OpenHab (Or other home automation tool) . Sorry for that 

Give me some days and I will try to make a more detailed (and tested) installation description. I am not an openhab expert (does not run it myself) so that is something you have to find on the internet (or maybe somebody can help here). 
In this post there is some description on how to: 
-   https://community.openhab.org/t/binding-request-growatt-server/96288/13 

If you only want to interface with OpenHab or PVOutput.org you only need Grott and Mosquitto (or other MQTT package).  
In theory MQTT is not needed for PVOutputorg but Grott needs it by startup at this moment. 
If you want more (e.g. influxdb  and grafana reporting) then more packages need to be installed.

TurboMOD, 2021-11-13:
@johanmeijer thx for trying to help. I just got some time to try again. I will write down every step I did, hopefully it will help others :)

1. create Folder: /etc/openhab/scripts/grott
2. copy all files from grott-master into /etc/openhab/scripts/grott/ since I dont really know what exactly I need

TurboMOD, 2021-11-13:
[quote="treesink, post:14, topic:96288"]
In the PV installation I changed the server address to the ip adress of my RPI, according to the instructions in your wiki.
[/quote]
Not sure what this means. Is this the "Rerouting Growatt Wifi TCPIP data via your Grott Server" part in your wiki?

macfloripa, 2021-11-14:
> [quote="treesink, post:14, topic:96288"]
> In the PV installation I changed the server address to the ip adress of my RPI, according to the instructions in your wiki.
> [/quote]
> Not sure what this means. Is this the "Rerouting Growatt Wifi TCPIP data via your Grott Server" part in your wiki?

Yes, that's correct.

johanmeijer, 2021-11-14:
It is every difficult to create a generic installation instruction. Grott is able to run on windows, several Linux builds and different hardware like RPI, NAS, X86 server etc. I have even been able to run it in IBM zSeries Mainframe (Linux on z). 

I think the @first time installation page (https://github.com/johanmeijer/grott/wiki/@-First-time-installation) and the rest of the wiki https://github.com/johanmeijer/grott/wiki) should give a good overview. 

Hopefully you do understand it will never be complete for every combination but I try to update it with my personal experiences and the information I get from the grott user community. 

In short: 

1. Verify if Python3 (and PIP3) is installed and if not install it. 
see python.org pages how to install,
For RPI and other debian Linux builds: 
> sudo apt-get install python3
> sudo apt-get install python3-pip

2. Install necessary python packages: 
> sudo pip3 install paho-mqtt 
> sudo pip3 install requests

3. Install Mosquitto (or other MQTT package) 
You did this with openHab that is off course ok. 
For RPI and other debian Linux builds: 
> sudo apt-get install mosquitto

4. Install Grott
 create grott directory 
download package from git 

via git tool: 
> sudo apt install git
> git clone -b 2.6 https://github.com/johanmeijer/grott grott
(2.6 branch used in this example). 

Other options: WGET or CURL. 
> wget https://github.com/johanmeijer/grott/archive/2.6.zip -O grott.zip 
or
> curl -L https://github.com/johanmeijer/grott/archive/2.6.zip -o grott.zip

Unpack zip
> unzip grott.zip (this will create directory grott-2.6)
> rename grott dir (if branche used) : mv grott-2.6 grott

5. Create grott.ini
> Go to grott directory : CD grott
> cp examples/grott.ini grott.ini
for standard MQTT usage no changes are needed for more advantage usage see wiki
https://github.com/johanmeijer/grott/wiki/MQTT

6. Create grott service (see wiki:)
> https://github.com/johanmeijer/grott/wiki/Grott-as-a-service-(Linux) 
> https://github.com/johanmeijer/grott/wiki/Grott-as-a-service-(Windows)

7. reroute the Growatt inverter data via Grott (see wiki): 
> https://github.com/johanmeijer/grott/wiki/Rerouting-Growatt-Wifi-TCPIP-data-via-your-Grott-Server 

From here you have to enable you product (like OpenHab or active home) for reading die MQTT messages. As described above I do not have an OpenHab example for other packages descriptions are documented in the wiki or described in the issues. 

Interfacing to PVOutput.org and InfluxDB is also described in the wiki. 

Hopefully this helps a bit.

TurboMOD, 2021-11-20:
trying again with your new "howto"

1. 
-sudo apt-get install python3 
gave me python3 is already the newest version...0 upgraded, 0 installed ect.
-sudo apt-get install python3-pip
same here

2.
-sudo pip3 install paho-mqtt
successfully installed paho-mqtt-1.6.1
-sudo pip3 install requests
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (2.21.0)

3. skipped because did it with openhab

4. skipped because already copied all files to /etc/openhab/scripts/grott

5.
-cd grott
gave me: -bash: cd: grott: No such file or directory

so back to 4.
-sudo apt install git
gave me git is already the newest version...0 upgraded, 0 installed ect.
-git clone -b 2.6 https://github.com/johanmeijer/grott grott
gave a Cloning into `grott`...
...
Resolving deltas: 100% (420/420), done.
-unzip grott.zip
gave unzip: cannot find or open grott.zip...

not sure what to do yet. but now its possible to use 
-cd grott

[image]

TurboMOD, 2021-11-20:
-cp examples/grott.ini grott.ini
didnt do anything. just got a new row (openhabian@openhabian:~/grott $)
so going to the next step 6. Create grott service (see wiki:)

TurboMOD, 2021-11-20:
[image]

TurboMOD, 2021-11-20:
opened grott.service and changed:
`WorkingDirectory=/home/pi/grott/
ExecStart=-/usr/bin/python3 -u /home/pi/grott/grott.py -v`
to
`WorkingDirectory=/home/openhabian/grott/
ExecStart=-/usr/bin/python3 -u /home/openhabian/grott/grott.py -v`

johanmeijer, 2021-11-20:
Do you understand how difficult it is to create a generic how to. Every implementation is a little bit different. 

I think grott is running now? 

you can display the syslog information with journalctl - u grott (or journalctl -u grott -e). 

If it runs. you have to configure openhab to process mqtt messages.

TurboMOD, 2021-11-20:
yeah, it is crazy xD
I just did again the "Rerouting Growatt Wifi TCPIP data via your Grott Server" part and `journalctl -u grott` is showing :
[image]

TurboMOD, 2021-11-20:
already on the mqtt part. this way iasnt working for me:
[https://community.openhab.org/t/binding-request-growatt-server/96288/13](url)
maybe its just something about names. there are many "bbp295" in treesink`s config. not sure where it comes from

TurboMOD, 2021-11-20:
I added a MQTT broker which is online
`UID: mqtt:broker:MQTT_Broker
label: MQTT Broker
thingTypeUID: mqtt:broker
configuration:
  lwtQos: 0
  publickeypin: true
  keepAlive: 60
  clientid: a81ddf58-d642-4e49-b211-bdc0e13539a1
  qos: 0
  reconnectTime: 60000
  port: 1883
  host: localhost
  secure: false
  certificatepin: true
  lwtRetain: true
  enableDiscovery: true
`

johanmeijer, 2021-11-20:
I think the bbp295 is the name of the mqtt broker. I am not sure why it has this name and what your name should be (probably mosquitto if you use mosquitto). 

If I google on: openhab mqtt broker name

I see a couple of examples like: 

Bridge mqtt:broker:mosquitto "Mosquitto" @ "System" [ host="localhost", secure=false, username="wwwwwww", password="XXXXXX" ]

johanmeijer, 2021-11-20:
It might also me possible to configure the MQTT bridge via the OpenHAB menu's?

TurboMOD, 2021-11-30:
yes, bbp295 is the name of his broker, I got it running after changing names. thank you very much for your help :)

johanmeijer, 2022-01-30:
@TurboMOD Can I close this one?

TurboMOD, 2022-02-06:
@johanmeijer yes. thanx :)

closed 2022-02-06 by johanmeijer