johanmeijer/grott #76: Save / Download data locally to csv possible?

cyberdiamond, 2021-11-19:
Is it possible to save data the data from Inverter / Battery / Power Meter data to csv file the same as the server.growatt.com cloud History Data download?

Using thie grott program or if anyone knows how to get it directly from the inverter / shinelink?

Reason I am looking at this is I just saw this message on the server.growatt.com login page:

Dear users: In order to bring you a more convenient product experience, we will update the system service platform on November 23th. The specific updates are as follows:
1. New Disclaimer for device setting System Dashboard / Device List, the Disclaimer is added in the device setting, please tick to agree;
2. Stop accessing historical data The system will stop accessing the Dashboard / Device List / Historical Data function; Thank you for your understanding and support.

I am looking at options for storing the data database or something but always having the raw data stored in this way is good (eg re-import it to a different database or anything else in the future).

gabrielpc1190, 2021-11-20:
I'm using grott with MQTT connecting to Home Assistant, then store it on a MySQL database. I think you can do the same.

cyberdiamond, 2021-11-20:
Yes I think I could do the same as I want to combine the Growatt data and the data from my other older system and eventually automate some decisions / settings for the battery but I think the first thing is just make sure I can save all data locally.

johanmeijer, 2021-11-20:
Look at the possibility of creating extensions (https://github.com/johanmeijer/grott/wiki/Extensions). 

Maybe we can create an extensions for exporting to a .csv file  or MySQL directly. I do not think this will be rocket science a couple of python statements should do it.

pierstitus, 2022-01-11:
Hi,
I wanted this functionality too, and have just made a pull request with a CSV export extension. To get the same format as the Growatt server you should give the corresponding fields as "csvheader" in extvar.

To be able to convert the data to standard units the detected data layout is added to the `conf` variable, so the plugin is no compatible with the released grott version. It would make sense to convert the units already in csvdata.py and pass the converted values to the extension, but for now I just copied the unit conversion code to the extension.

I'm planning to also add the option so save the daily totals to a separate CSV file, not sure when that will happen though.

johanmeijer, 2022-01-27:
@cyberdiamond This extension is now available in Grott (master and soon in 2.7), 

For more details see closed pull request #91 and wiki: https://github.com/johanmeijer/grott/wiki/Extensions. 

Credits to: @pierstitus

PR #91 (2022-01-27): CSV export extension
Add a CSV export extension in the examples/extensions dir. Closes #76 

One CSV file per day is saved.

extvar configuration:
"outpath": path where to save CSV files, default: "/home/pi/grottlog"
"csvheader": comma separated string with fields to store, defaults to all available fields

To be able to convert the data to standard units the detected data layout is added to the `conf` variable as `conf.layout`.
commit 01d693d3 (2022-02-06): 2.7.1

2.7.1
- Add coding from Pull req #91 (csv export exentsion)
  Add a CSV export extension in the examples/extensions dir. Closes #76

  One CSV file per day is saved.

  extvar configuration:
  "outpath": path where to save CSV files, default: "/home/pi/grottlog"
  "csvheader": comma separated string with fields to store, defaults to all available fields
- Add parameter to enable message retain in MQTT (#84)
  - .ini [MQTT section] retain = True
  - environmental gmqttretain = "True" (docker: -e gmqttretain = "True") 
- Add parameter to enable sent inverter temperature as temperature val [... cut]
closed 2022-01-30 by johanmeijer