johanmeijer/grott #77: Hybrid Inverter / Batteries / Smart Meter - Is it possible to access all the available data sent?

cyberdiamond, 2021-11-20:
Background:

I have a Growatt SPH 5000 hybrid inverter (6.6kw panels facing east) with 2x 6.5 kw batteries and a Growatt Smart Meter both attached and communicating with the inverter.
I also have an older Aurora inverter connected to panels on the west side of the roof.

I just recently saw the message on server.growatt.com:

```
Dear users: In order to bring you a more convenient product experience, we will update the system service platform on November 23th. The specific updates are as follows:
1. New Disclaimer for device setting System Dashboard / Device List, the Disclaimer is added in the device setting, please tick to agree;
2. Stop accessing historical data The system will stop accessing the Dashboard / Device List / Historical Data function; Thank you for your understanding and support
```

Because they are removing access to the Download History Data I want to save all of the available data locally for historical and future troubleshooting purposes.

I do currently have all the daily csv files from install date until now saved locally (downloaded from server.growatt.com)

Is there any way to save the raw data direct from the inverter / batteries / smart meter directly?

I have just installed cloned / installed (version 2.6.1d for the Hybrid inverter support) as standalone on my Raspberry Pi that I have been using with rs485 to monitor my other Aurora solar system.

The only modifications I have made to grott.ini (so far):
invtype=sph
blockedcmd=false

Seems to be working ok in proxy mode, data is being sent to server.growatt.com correctly and I can still change the battery charging and time export mode settings from server.growatt.com "Settings" and the inverter accepts those changes.

I have not got it saving locally yet.

Should I also be seeing the Growatt smart meter information getting decoded?

When I compare the data Grott shows vs the Downloaded csv file from server.growatt.com there is a lot of information missing so my configuration might not be correct yet, so any pointers welcome :)

I am happy to help as much as possible with changes &/or testing needed to get all the infomation accessed through Grott.
I can also set a tempory password to share to be able to log in and see / download data direct.
We might be time limited if Growatt are going to remove access to the downloaded information very soon.

Sample output from Grott:

```
pi@RPi:~/grott $ 
pi@RPi:~/grott $ sudo systemctl stop grott
pi@RPi:~/grott $  
pi@RPi:~/grott $ 
pi@RPi:~/grott $ sudo python3 grott.py -v
Grott Growatt logging monitor : 2.6.1d

Grott Command line parameters processed:
	verbose:     	 True
	config file: 	 grott.ini
	output file: 	 <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
	nomqtt:      	 False
	inverterid:  	 automatic
	pvoutput:    	 False
	blockcmd:    	 False
	noipf:       	 False

Grott process configuration file

Grott process environmental variables

Grott override settings if set in commandline

Grott process json layout files

Grott layout records loaded
T02NNNN  : 
{'decrypt': {'value': 'False'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'recortype1': {'value': 70, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 74, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'num', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'isof': {'value': 210, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'gfcif': {'value': 214, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'dcif': {'value': 218, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vpvfault': {'value': 222, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vacfault': {'value': 226, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'facfault': {'value': 230, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'tmpfault': {'value': 234, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'faultcode': {'value': 238, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'pvipmtemperature': {'value': 242, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvolt': {'value': 246, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'nbusvolt': {'value': 250, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'epv1today': {'value': 278, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 294, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 310, 'length': 4, 'type': 'num', 'divide': 10}, 'rac': {'value': 318, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractoday': {'value': 326, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractotal': {'value': 334, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}}
T05NNNN  : 
{'decrypt': {'value': 'True'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'recortype1': {'value': 70, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 74, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'numx', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'isof': {'value': 210, 'length': 2, [... cut ...] 128 | 170 | 169.5 | 231 | 22086 | 1 | 241 | 100 | 56.8 | 0 | 0 | 22086 | 16720 | 16720 | 0 | 3.276 | 3.274 | 0.002 | 0 | 3.276 | 3.276 | 3.276 | 3.276 | 3.276 | 3.276 | 3.276 | 3.275 | 3.275 | 3.275 | 3.276 | 3.276 | 16544 | 330 | 340 | 0 | 1064.1 | 0.1 | 3233 | 3.276 | 3.274 | 2 | 0 | 11 | 18 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 27365 | 202 | 192 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |  

</body>

</html>

johanmeijer, 2021-11-20:
In principle we can show the same data that is in the growatt historical csv data. That is no the problem.
I decide to show only the relevant fields (what I think is relevant)  while otherwise it will generate a very big json messages (and influxdb row if you use that). 

But we extend that easily or you can create your own decode schema (see https://github.com/johanmeijer/grott/wiki/Grott-advanced-(customize-behaviour)). 

You have then to create a t065104XSPH.json file that contains your own decoding schema. If you put this file in the grott home directory it will be loaded at start up and will override the normal decoding. 
I can create an example. I have to dive deeper in this for issue #47 as well so that will not be a problem. 

I have not seen the message from Growatt about removing historical data yet, but be aware I detected there are two login sites! An old one (https://server.growatt.com/login.do) and a new one (https://server.growatt.com/login). I know they want to decomission the old site at some point in time.  

For saving historical data I created an interface with influxdb (that can easily be installed at your rpi as well): https://github.com/johanmeijer/grott/wiki/InfluxDB-Support  But I can understand you do not want to use that (a tip from my personal experience you have to backup the database at regulair basis). 
I decided to go for influxDB while it easy to be used with Grafana to create your own dashboad/reports. 

There is also a possibility to create your own extensions (see isue: https://github.com/johanmeijer/grott/wiki/Extensions). It would not be that difficult to create an extension that writes the data to a csv file (e.g. per day or something like that). 

The Growatt smart meter information should be decoded as well. It will be sent in a separate mqtt message. If you do not see that we have to dive deeper in this (see also issue #47).

cyberdiamond, 2021-11-21:
> In principle we can show the same data that is in the growatt historical csv data. That is no the problem.
That is great news, I would prefer to capture everything available. Storage is cheap and worse case it will compress well.

The the first thing, advanced customise behaviour looks like what we need to get all the fields to be available.
So copy the existing t065104XSPH.json json file as a new file name say t065104XSPH-all.json, copy it the the examples directory, add all of the available fields to it.

I am about to look at this after this message so your offer of an example to start might really help.

I am about to look at issue #47 after this message as well.

Then it can be used by me (and other people) by just: cp examples/t065104XSPH-all.json t065104XSPH.json

I had forgotten about the old site.
The message I see from the new Growatt site is shown below:
[image]

Secondly for saving historical data I initially thought influxDB and Grafana (still want to use Grafana) but spoke with someone about their own solar monitoring and they mentioned that infux summarises and expires data and that you could not do the normal database joins so that turned me off a bit so I need to investigate further after I am able to save all the fields first.

Because of the short time frame of Growatt taking away the history data It also might be easier and quicker to store all the data in influx for now providing it can be adjusted to not summarise and expire the data so I always have all the full raw data and if I want / need it some other way then convert it at that time (and / or look into extension later?)

For the Growatt smart meter decoding I don't see any separate mqtt messages anywhere in the original first post verbose capture?
What steps do I need to do to troubleshoot this?

cyberdiamond, 2021-11-21:
Actually influxdb might not run well on my old raspberry pi 1 model B with 512gb ram.

johanmeijer, 2021-11-22:
Ok. I missed that message totally. A little bit strange from Growatt but
oke. Then we have to see how we can save historical data.

I encourage everybody to download the historical data as much as possible
(now it is still available). If you have it in XLS or CSV format we can
always look later how we can put it in a central repository later.

InlfuxDB is indeed capable of automatically aggregating/summarizing and
expiring data but as long as you do not define the rules it will store and
keep all data. But I can understand if you do not want to use InfluxDB and
indeed it will be a challenge, but not impossible to run it at a RPI1.
I stated above I am not in favour of storing data on the SD card of a RPI.
I lost a couple of SD cards already (mainly because of power outages) so I
would prefer to store (or backup i) somewhere else.

I will create the initial json file with more information  and put in the
example directory. For me a basic programming rule is do not process and
store the data you do not need.  That costs a lot of unnecessary resources
(CPU, Memory, Storage) and is expensive and not sustainable. But be aware I
work in an environment where we process thousands of records per second and
not 1 per (couple of) minutes.

I do not understand why you do not see the growatt smart meter information.
It seems to work for others (as stated in issue #47).  Do you see the smart
meter information at the growatt site? Which type of smart meter do you
have and is it connected with the inverter?

johanmeijer, 2021-11-22:
Ok. I am only able to download from July 2021 in CSV from online.

I am trying to use the API but that has not very detailed information (not al fields).  

The challenge wil be how to recognize the fields if we do not have the detailed information.

johanmeijer, 2021-11-22:
I have uploaded a t06NNNNXSPH.json layout file to the examples/record layout. 

If you copy this file in the grott main directory and stop/start grott the definitions in the file will be loaded. 

I try to make a mix of how you can define / exclude fields. 

It is not that simple to find the right fields. I use the definition described in this document: 

https://www.scribd.com/document/489508943/Growatt-Inverter-Modbus-RTU-Protocol-II-V1-20-English 

As described in the instruction. 

Storage
(SPH Type 03 register range 0~124,1000~1124, 04 register range 0~124,1000~1124)

This means that SPH is using fields 0 - 124 and 1000-1124 as defined in the input reg section. It took a while for me to find out where range 1xxx started in the data. But we can now build on what we have.

cyberdiamond, 2021-11-23:
Growatt did remove the button on the webpage to download the historical data so I now have lost access to that which might make troubleshooting harder but I did save all the daily csv files until today.
Maybe the data is still there but I didn't think to save the webpage source code to see if the backend to download the csv files is still there and working or not?

I installed influxdb on the rpi and grott is saving the data it currently gets to it now and it seems to be running ok so far.
Will probably get a newer one before running grafana on it.

I did save the verbose output from grott for the other day so I also have the historical csv with the data for that same day which might help with translating all the fields if needed.

It is a Growatt SPME (From what I can tell it is re-branded Eastron SDM-230) single phase meter,.
The same smart meter as the other persons in #47 and it is connected to (and working with) the sph inverter.

I didn't see the growatt smart meter data on the web site as separate data that I could download or where to look for it but it did see the export and consumption data on the main screen so it was getting the data through somehow.
Maybe it gets added to the sph data record and gets all sent together?

In the journalctl output I do see a smaller record sometimes that I do not know what it is eg:
`Nov 23 17:52:50 RPi grott[21643]:          - Growatt packet received:
Nov 23 17:52:50 RPi grott[21643]:                   <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.21', 5279), raddr=('192.168.1.22', 1037)>
Nov 23 17:52:50 RPi grott[21643]:          - Data less then minimum record length, data not processed
Nov 23 17:54:37 RPi grott[21643]:          - Growatt packet received:
Nov 23 17:54:37 RPi grott[21643]:                   <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.21', 33652), raddr=('47.91.67.66', 5279)>
Nov 23 17:54:37 RPi grott[21643]:          - Data less then minimum record length, data not processed
Nov 23 17:54:38 RPi grott[21643]:          - Growatt packet received:
Nov 23 17:54:38 RPi grott[21643]:                   <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.21', 5279), raddr=('192.168.1.22', 1037)>
Nov 23 17:54:38 RPi grott[21643]:          - Data less then minimum record length, data not processed
Nov 23 17:57:38 RPi grott[21643]:          - Growatt packet received:
Nov 23 17:57:38 RPi grott[21643]:                   <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.21', 33652), raddr=('47.91.67.66', 5279)>
Nov 23 17:57:38 RPi grott[21643]:          - Data less then minimum record length, data not processed
Nov 23 17:57:41 RPi grott[21643]:          - Growatt packet received:
Nov 23 17:57:41 RPi grott[21643]:                   <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.21', 33652), raddr=('47.91.67.66', 5279)>
Nov 23 17:57:41 RPi grott[21643]:          - Data less then minimum record length, data not processed
Nov 23 17:57:44 RPi grott[21643]:          - Growatt packet received:
Nov 23 17:57:44 RPi grott[21643]:                   <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.21', 33652), raddr=('47.91.67.66', 5279)>
Nov 23 17:57:44 RPi grott[21643]:          - Data less then minimum record length, data not processed
Nov 23 17:57:48 RPi grott[21643]:          - Growatt packet received:
Nov 23 17:57:48 RPi grott[21643]:                   <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.21', 33652), raddr=('47.91.67.66', 5279)>
Nov 23 17:57:48 RPi grott[21643]:          - Data less then minimum record length, data not processed
Nov 23 17:57:53 RPi grott[21643]:          - Growatt packet received:
Nov 23 17:57:53 RPi grott[21643]:                   <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.21', 33652), raddr=('47.91.67.66', 5279)>
Nov 23 17:57:53 RPi grott[21643]:          - Data less then minimum record length, data not processed
Nov 23 17:57:53 RPi grott[21643]:          - Growatt packet received:
Nov 23 17:57:53 RPi grott[21643]:                   <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.21', 5279), raddr=('192.168.1.22', 1037)>
Nov 23 17:57:53 RPi grott[21643]:          - Data less then minimum record length, data not processed
Nov 23 17:57:53 RPi grott[21643]:          - Growatt packet received:
Nov 23 17:57:53 RPi grott[21643]:                   <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.21', 5279), raddr=('192.168.1.22', 1037)>
Nov 23 17:57:53 RPi grott[21643]:          - Growatt original Data:
Nov 23 17:57:53 RPi grott[21643]:                  \x00\xcf\x00\x06\x00\x20\x01\x16\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
Nov 23 17:57:53 RPi grott[21643]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Nov 23 17:57:53 RPi grott[21643]:                  \x50\xdf\x00\xd0\x00\x06\x00\x20\x01\x16\x09\x33\x2c\x42\x20\x45\x40\x75\x45
[... cut ...]
Nov 23 17:57:53 RPi grott[21643]:                  00cf0006002001164e41433541313432373800000000000000000000000000000000000000003
Nov 23 17:57:53 RPi grott[21643]:                  fa861a47441724f76777d476b304f3221013110050e0315330628050e0315330628050e031533
Nov 23 17:57:53 RPi grott[21643]:                  0626ce6fa66172746773797e52583667372f0224230326061b30131b0326061b30131b0326061
Nov 23 17:57:53 RPi grott[21643]:                  b301385d4
Nov 23 17:57:53 RPi grott[21643]:          - Grott data ack record or data record not defined no processing done`

cyberdiamond, 2021-11-23:
Ok I pulled your latest changes then: 

```
pi@RPi:~/grott $ ls -latrh 'examples/Record Layout'/
total 124K
-rw-r--r-- 1 root root  11K Nov 18 22:05  t060104o.json
-rw-r--r-- 1 root root 4.5K Nov 18 22:05 't060104 .json'
-rw-r--r-- 1 root root 5.4K Nov 18 22:05  t050104o.json
-rw-r--r-- 1 root root 4.5K Nov 18 22:05  t050104.json
-rw-r--r-- 1 root root 4.5K Nov 18 22:05  t020104.json
-rw-r--r-- 1 root root 1.7K Nov 18 22:05  t06NNNNX.json
-rw-r--r-- 1 root root 3.3K Nov 18 22:05  t065104x.json
-rw-r--r-- 1 root root 3.3K Nov 18 22:05  t065004x.json
-rw-r--r-- 1 root root 3.3K Nov 18 22:05  t060104x.json
-rw-r--r-- 1 root root  11K Nov 18 22:08  t050104spf.json
-rw-r--r-- 1 root root 3.8K Nov 18 22:08 't060120 .json'
-rw-r--r-- 1 root root  16K Nov 18 22:08  t060104sph.json
-rw-r--r-- 1 root root  11K Nov 18 22:08  t060104spf.json
drwxr-xr-x 5 root root 4.0K Nov 18 22:08  ..
-rw-r--r-- 1 root root 8.8K Nov 24 00:00  t06NNNNXSPH.json
drwxr-xr-x 2 root root 4.0K Nov 24 00:00  .
pi@RPi:~/grott $ 
pi@RPi:~/grott $ sudo systemctl stop grott
pi@RPi:~/grott $ 
pi@RPi:~/grott $ 
pi@RPi:~/grott $ sudo cp 'examples/Record Layout'/t06NNNNXSPH.json t06NNNNXSPH.json
pi@RPi:~/grott $ 
pi@RPi:~/grott $ 
pi@RPi:~/grott $ ls -larth
total 196K
-rw-r--r-- 1 root root 7.5K Nov 18 22:05 README.md
drwxr-xr-x 2 root root 4.0K Nov 18 22:05 docker
-rw-r--r-- 1 root root 5.4K Nov 18 22:05 grottsniffer.py
-rw-r--r-- 1 root root 6.7K Nov 18 22:05 grottproxy.py
drwxr-xr-x 2 root root 4.0K Nov 18 22:05 wiki
-rw-r--r-- 1 root root  533 Nov 18 22:08 grott.service
drwxr-xr-x 5 root root 4.0K Nov 18 22:08 examples
drwxr-xr-x 8 pi   pi   4.0K Nov 23 23:51 ..
-rw-r--r-- 1 root root 1.5K Nov 24 00:00 grott.py
-rw-r--r-- 1 root root  79K Nov 24 00:00 grottconf.py
-rw-r--r-- 1 root root  28K Nov 24 00:00 grottdata.py
drwxr-xr-x 2 root root 4.0K Nov 24 00:01 __pycache__
-rw-r--r-- 1 root root 2.9K Nov 24 00:01 grott.ini
-rw-r--r-- 1 root root  12K Nov 24 00:01 Version_history.txt
drwxr-xr-x 8 root root 4.0K Nov 24 00:01 .git
-rw-r--r-- 1 root root 8.8K Nov 24 00:03 t06NNNNXSPH.json
drwxr-xr-x 7 root root 4.0K Nov 24 00:03 .
pi@RPi:~/grott $ 
pi@RPi:~/grott $ 
pi@RPi:~/grott $ sudo systemctl start grott
pi@RPi:~/grott $ 

```

From journalctl -u grott:

```
Nov 24 00:02:28 RPi systemd[1]: Stopping Grott Growatt monitor...
Nov 24 00:02:28 RPi systemd[1]: grott.service: Main process exited, code=killed, status=15/TERM
Nov 24 00:02:28 RPi systemd[1]: grott.service: Succeeded.
Nov 24 00:02:28 RPi systemd[1]: Stopped Grott Growatt monitor.
Nov 24 00:04:05 RPi systemd[1]: Started Grott Growatt monitor.
Nov 24 00:04:12 RPi grott[11854]: Grott Growatt logging monitor : 2.6.1f
Nov 24 00:04:12 RPi grott[11854]: Grott Command line parameters processed:
Nov 24 00:04:13 RPi grott[11854]:         verbose:              True
Nov 24 00:04:13 RPi grott[11854]:         config file:          grott.ini
Nov 24 00:04:13 RPi grott[11854]:         output file:          <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
Nov 24 00:04:13 RPi grott[11854]:         nomqtt:               False
Nov 24 00:04:13 RPi grott[11854]:         inverterid:           automatic
Nov 24 00:04:13 RPi grott[11854]:         pvoutput:             False
[... 1772 lines cut ...]
Nov 24 00:05:18 RPi grott[11854]:                  -  eharge1_tod          :  1.0
Nov 24 00:05:18 RPi grott[11854]:                  -  eharge1_tot          :  2742.7
Nov 24 00:05:18 RPi grott[11854]:                  -  elocalload_tod       :  1.9
Nov 24 00:05:18 RPi grott[11854]:                  -  elocalload_tot       :  429493232.6
Nov 24 00:05:18 RPi grott[11854]:          - MQTT jsonmsg:
Nov 24 00:05:18 RPi grott[11854]:                          {"device": "NTCIA17036", "time": "2021-11-24T00:05:17", "buffered": "no",
Nov 24 00:05:18 RPi grott[11854]:                          "values": {"pvstatus": 6, "pvpowerin": 0, "pv1voltage": 170, "pv1current":
Nov 24 00:05:18 RPi grott[11854]:                          0, "pv1watt": 1, "pv2voltage": 130, "pv2current": 0, "pv2watt": 1,
Nov 24 00:05:18 RPi grott[11854]:                          "pvpowerout": 2477, "pvfrequentie": 5005, "pvgridvoltage": 2377,
Nov 24 00:05:18 RPi grott[11854]:                          "pvgridcurrent": 14, "pvgridpower": 2562, "pvgridvoltage2": 0,
Nov 24 00:05:18 RPi grott[11854]:                          "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0,
Nov 24 00:05:18 RPi grott[11854]:                          "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 87445467, "eactoday":
Nov 24 00:05:18 RPi grott[11854]:                          160, "pvenergytoday": 160, "eactotal": 109416, "epvtotal": 85997,
Nov 24 00:05:18 RPi grott[11854]:                          "epv1today": 82, "epv1total": 66404, "epv2today": 63, "epv2total": 51130,
Nov 24 00:05:18 RPi grott[11854]:                          "pvtemperature": 281, "pvipmtemperature": 265, "pvboosttemp": 255,
Nov 24 00:05:18 RPi grott[11854]:                          "bat_dsp": 527, "pbusvolt": 4280, "eacharge_today": 19, "eacharge_total":
Nov 24 00:05:18 RPi grott[11854]:                          1816, "batterytype": 1, "uwsysworkmode": 6, "systemfaultword0": 0,
Nov 24 00:05:18 RPi grott[11854]:                          "systemfaultword1": 0, "systemfaultword2": 0, "systemfaultword3": 0,
Nov 24 00:05:18 RPi grott[11854]:                          "systemfaultword4": 32, "systemfaultword5": 0, "systemfaultword6": 0,
[... cut ...]
Nov 24 00:05:18 RPi grott[11854]:                          "SOC": 64, "pactouserr": 0, "pactousertot": 0, "pactogridr": 300,
Nov 24 00:05:18 RPi grott[11854]:                          "pactogridtot": 300, "plocaloadr": 3600, "plocaloadtot": 3600,
Nov 24 00:05:18 RPi grott[11854]:                          "spdspstatus": 6, "spbusvolt": 2955, "etouser_tod": 3, "etouser_tot": 2189,
Nov 24 00:05:18 RPi grott[11854]:                          "etogrid_tod": 171, "etogrid_tot": 127752, "edischarge1_tod": 51,
Nov 24 00:05:18

johanmeijer, 2021-11-23:
Oops the download button is gone indeed :(

The growatt server has an API capability but I have not found very good documentation of it yet. There is a python package available to use this API (https://pypi.org/project/growattServer/) and I managed to get some historical data from it (information on actual power delivered per time interval)  but certainly not the details we had with the CSV. 

I expected a T060120 record  for the smart meter (that is the case for a 5000TL-X, with smart meter) but maybe for a SPF type  inverter the the smart meter information is processed and included in de standard data record indeed. 
We might detect this when we get more fields from the datarecord. 

Everything looks good. You can change the t06NNNNXSPH.json and look for the additional information now. 

Be aware I do not say it is easy.  It can be a puzzle to find the right fields but it is doable. It is a pity we do not have the CSV anymore. 

I would be nice to have the (detailed) CSV information of one day/time and the grott trace record for the same time so we check if we find the right things. 

There are a lot of fields (e.g. fault/error codes) that normally contains no information. It will be hard to test if the right position and values for those fields are found. 

I am curious if you manage to find the additional fields. If I have to give more explanation please let me know. I also want to give it a try to found more fields but do not expect that to be done tomorrow.

cyberdiamond, 2021-11-25:
Luckily while testing I did did save some of the Grott output data and I do have those days with matching csv /xls files downloaded from Growatt server.
I'll attach them as they could be a good reference (including for anyone else to find in the future).

**Note:**
If looking at the times I am in Adelaide, Australia which in Standard Time is GMT+9.30 and these Grott logs are also in Daylight savings time (GMT+10.30).
For the csv files from Growatt server download I never updated the time to go forward 1 hour when we go into Daylight savings time so at this time of year they are 1 hour behind current times (so still GMT+9.30)

Also the first day 2021-11-19 I had not set the inverter type properly but that should not matter for the raw data.

[2021-11-19-grott-output-invtype-default.txt](https://github.com/johanmeijer/grott/files/7599754/2021-11-19-grott-output-invtype-default.txt)

[NTCIA17036 mix data - 2021-11-19_2021-11-19.xls](https://github.com/johanmeijer/grott/files/7599766/NTCIA17036.mix.data.-.2021-11-19_2021-11-19.xls)

[2021-11-20-grott-output-invtype-sph.txt](https://github.com/johanmeijer/grott/files/7599756/2021-11-20-grott-output-invtype-sph.txt)

[NTCIA17036 mix data - 2021-11-20_2021-11-20.xls](https://github.com/johanmeijer/grott/files/7599765/NTCIA17036.mix.data.-.2021-11-20_2021-11-20.xls)

[2021-11-21-grott-output-invtype-sph.txt](https://github.com/johanmeijer/grott/files/7599757/2021-11-21-grott-output-invtype-sph.txt)

[NTCIA17036 mix data - 2021-11-21_2021-11-21.xls](https://github.com/johanmeijer/grott/files/7599763/NTCIA17036.mix.data.-.2021-11-21_2021-11-21.xls)

johanmeijer, 2021-11-25:
Thanks. This should be enough to enhance the data processing and expose more fields for the SPH.

johanmeijer, 2022-01-02:
@cyberdiamond still try o find some time to do a totally mapping for the sph record. I almost there.

Other question you have the sdm 230 smart meter? How is it connected to the Shinelink? Do you use a raillog RF for this? 

This might be the reason why we do no see the smart meter info.

cyberdiamond, 2022-01-11:
Sorry I forgot to reply to this earlier.

Yes I have the Growatt SPM-E which is an Eastron SDM230-Modbus smart meter.
No I do not have a Raillog RF.
It is connected directly to the SPH5000 by cable using RS485 interface.
It is using Modbus RTU communication protocol.

Dorz90, 2022-02-17:
Hi @johanmeijer, @cyberdiamond,

I think we have very similar systems installed, possibly even by the same supplier ;) i am also in Adelaide, Australia.

My layout looks to be ever so slightly different to yours, but is not being picked up because it does not have a matching record layout. I have a Growatt SPH5000 inverter, 2x Growatt GBLI6531 6.5KWh batteries and a SDM230 meter. I am running grott in docker using environmental variables, but not decoding any data.

I have also noticed some similarities in terms of what is being reported by Grott/Shinelink.
Cyberdiamond's layouts are:
T065104XSPH (inverter, uses T06NNNNXSPH)
T065103XSPH (data record not defined)
T065129SPH (data record not defined)

My layouts are:
T055004XSPH (data record not defined) (assuming inverter)
T055003XSPH (data record not defined) (possibly Growatt GBLI6531 battery master connected to inverter via canbus. slave battery connected to master via RS485)
T055029SPH (data record not defined) (small amount of data, possibly export meter SDM230 with Growatt custom firmware)

comparing the layouts i think the original data appears to have some similarities, so the layout for the T06NNNNXSPH could also work for the T05NNNNXSPH inverter i have. is it possible to test this somehow?

Grott data below:

> - Growatt packet received:
> - Grott data ack record or data record   not defined no processing done
> 35303a3239f4b5
> e30001b000136001c000130001e0005474d542b38001f0013323032322d30322d31372031303a
> 000130000d0002313000140008475453573030303000150007312e302e312e350016000456312
> dad000400013500050001310006000233320008000a4d5943373931373031460009000130000a
> 00430005009950294e4143353931373031464d5943373931373031464e5443463934363032350
> - Growatt plain data:
> - Grott Growatt data decrypted
> - Growatt data decrypted V2
> - Record layout used :  T055029SPH
> - no matching record layout found, try   generic
> - layout   :    T055029SPH
> - Grott data record length 161
> - Grott automatic protocol detection
> \x51\x4e\x41\x77\x48\x5d\x4e\x95\xc1
> \x20\x6c\x4a\x6f\x68\x61\x67\x46\x77\x40\x5d\x5a\x51\x46\x59\x76\x45\x4f\x46
> \x43\x41\x47\x61\x6f\x74\x46\x44\x6f\x6b\x61\x75\x44\x47\x6c\x6f\x72\x26\x39
> \x47\x51\x44\x74\x52\x72\x68\x46\x4f\x44\x5a\x76\x5c\x5a\x77\x77\x74\x70\x11
> \x7e\x74\x46\x42\x6f\x7a\x61\x76\x45\x77\x72\x7b\x77\x69\x33\x20\x14\x25\x5f
> \x4f\x72\x65\x3a\x38\x37\x43\x7e\x43\x58\x47\x50\x32\x74\x4e\x72\x6e\x47\x61
> \x62\xda\x61\x70\x74\x46\x47\x6f\x72\x61\x75\x45\x47\x74\x6f\x75\x52\x46\x74
> \x38\x37\x43\x7e\x43\x58\x47\x50\x32\x3a\x13\x31\x29\x4e\x55\x42\x44\x75\x47
> \x00\x43\x00\x05\x00\x99\x50\x29\x09\x33\x2c\x42\x58\x45\x43\x77\x43\x29\x3a
> - Growatt original Data:
> <socket.socket fd=4, family=AddressFamily.AF_INET,   type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.8', 58266),   raddr=('47.91.67.66', 5279)>
> - Growatt packet received:
> - Grott data ack record or data record   not defined no processing done
> 0d05000055a3
> 000004150415000000d060d090d080d090d090d090d080d0a0d090d0a0d070d070d060d070d05
> 000000001c14dc0f28001832001b045f0e0048000601c80064163000000000000000480000000
> 85e00000000000000000000000000000000000000000000000000000000000003e80000016200
> 00002f410000000300015566000000300000d30e0000001a0000d60d0000003f00015b8800001
> 00000000000000000000019f30000000000000000000019f3000000fa00050c1e426800000010
> 00000000000000000051fa0216001c000000fb0000000000000000000000fb000000000000000
> 00000000000000000010000000000000000000003e80464000500000000000000000000000000
> 0000021b0f130f2d0000000000000000000000000000000000000000000000000000000000000
> 0000000000000000000000000000000000000000000000000000000000002dfac01ae01950194
> ed6ff100000018000173d80000001800016bd4000000000000000000000000000000000000000
> 97c001c000018bb00000000000000000000000000000000000000000000000000410002a96406
> 0000000000000000000000000000000000000000000000000000000000000000000018ba138d0
> c000500006d5d0c1801bc000036f40c4001c100003669000140a009ba07fc089f000000000000
> 00420005021950044e4143353931373031464e544346393436303235000000000000020000007
> - Growatt plain data:
> - Grott Growatt data decrypted
> - Growatt data decrypted V2
> - Record layout used :  none
> - no matching record layout found,   standard processing performed
> - no matching record layout found, try   generic
[... 74 lines cut ...]
> \x50\x65\xa1\x62\xdc\x07\xd2\x6b\x3b\x63\xc8\x70\x0b\x70\x37\x77\x61\x74\x74
> \x61\x74\x74\x47\x71\x87\x73\x05\x63\x04\x47\x72\x7d\xb7\x61\x46\x75\x91\x70
> \x54\x47\x8d\x21\x57\x61\x8b\x3a\x67\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
> \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x60\x58\x74\xb8\x3c\x4f\x77\x9e\x3a
> \xb3\x69\x62\x7c\xdf\x72\x7b\x77\x75\x7d\x1b\x4f\x8e\x6f\x77\x61\x74\x74\x47
> \x45\x45\x76\x40\x6f\x77\x60\x45\x74\x47\x72\x6f\x64\xc0\x74\x6c\x4d\x28\x66
> \x42\x72\x6f\x77\x61\x74\x74\x47\x72\x65\xfb\x61\x74\x06\x26\x13\x0e\x46\x53
> \x7d\xf9\x72\xe3\x74\x75\x72\x5d\x77\x53\x74\x71\x47\x40\x6f\x72\x61\x46\x74
> \x69\x7e\x2e\x55\x2e\x7b\x27\x61\x74\x74\x47\x72\x6f\x77\x61\x73\x8b\x4e\x90
> \x74\x56\x72\x65\x77\x53\x74\x53\x47\x76\x68\x7f\x6b\x5c\x66\x1b\x66\x3f\x70
> \x37\x01\x12\x13\x13\x0d\x67\x52\x4f\x77\x60\x79\xd9\x47\x72\x68\x91\x61\x76
> \x76\x61\x86\x17\x47\x73\x6f\x2d\x61\x74\x74\x47\x52\x4f\x57\x2f\x11\x03\x67
> \x35\x74\x06\x72\xa7\x77\xa9\x74\x74\x09\x26\x2c\x31\x58\x40\x42\x77\x40\x5a
> \x15\x33\x5e\x59\x51\x74\x74\x47\x72\x6f\x77\x61\x74\x75\x47\x72\x6a\xab\x61
[... cut ...]
> <socket.socket fd=4, family=AddressFamily.AF_INET,   type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.8', 58266),   raddr=('47.91.67.66', 5279)>
> - Growatt packet received:
> - Data less then minimum record   length, data not processed
> <socket.socket fd=4, family=AddressFamily.AF_INET,   type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.8', 58266),   raddr=('47.91.67.66', 5279)>
> - Growatt packet received:
> - Data less then minimum record   length, data not processed
> <socket.socket fd=5, family=AddressFamily.AF_INET,   type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.8', 5279),   raddr=('172.17.0.1', 48196)>

johanmeijer, 2022-02-17:
@Dorz90. Haven't defined a T05xxxxXSPH record yet [and somehow nobody missed it until now ;)]. 

It will indeed not differ much of the T06 record only everything is at another offset. I need to do some math but it is absolutely doable. Give me a couple of days. 

Do you use the plain python version or the docker image?

Dorz90, 2022-02-17:
@johanmeijer, thank you!

i pulled the docker image onto my Synology NAS.

johanmeijer, 2022-02-19:
@Dorz90 I added T05nnnnXSPH layout to 2.7.2. 

Docker images ledidobe/grott:beta and ledidobe/grott:2.7.2 are available with this change!

cyberdiamond, 2022-02-22:
> I think we have very similar systems installed, possibly even by the same supplier ;) i am also in Adelaide, Australia.

Most likely

My record might be slightly different as I had all the firmware upgraded about 8 months ago and it messed with some of the data. Eg. my export totals reset and other small things.

closed 2023-08-13 by johanmeijer