johanmeijer/grott #78: Trouble with re-routing

szupi-ipuzs, 2021-11-30:
Hi, i have tried following your guide for re-routing but something doesn't seem to work - after setting the IP via www on server.growatt.com (got the "successful" response), the status keeps showing original IP. Also, what's strange - the port doesn't seem to match the default (5279). After I modified the port to match the default (also got "successfull" response), the port changed to something completely different.
So far I was unable to receive even single stats record via grott. The growatt mobile application keeps working correctly though...
Could this be some anti-sniffing protection?
The only thing I haven't tried yet is setting the server by resetting the logger (will try that later). Any ideas what I might be doing wrong?

[image]

johanmeijer, 2021-12-02:
I am going to try it out this weekend if it still works for me (don't use it that often and it need to be done when the inverter is active). 

I suppose (advise) you want Grott in proxy mode so no additional iptables / network settings are needed at your server side. The only thing that is needed is changing the destination (growatt server) IP address. And normally it can be doen via the Growatt internet pages indeed. If that not works resetting and re-configuring the wifi stick is the most reliable option (I think). 

The IP address shown at the Growatt site your home IP address with a dynamic (arbitrary) port address.

szupi-ipuzs, 2021-12-03:
I've read your answer on another issue where you said about applying the setting on growatt server until it works - and after a few retries it worked! This time the "success" message appeared as a popup, not as a plain text message as before.
I am now running proxy mode and I can see the stats in HA.
Thanks!

closed 2021-12-03 by szupi-ipuzs