johanmeijer/grott #79: Unable to change IPadress via Growatt online

Tacoos33, 2021-11-30:
While atempting to change the IP adress to which the Growatt sends its data as described in:
https://github.com/johanmeijer/grott/wiki/Rerouting-Growatt-Wifi-TCPIP-data-via-your-Grott-Server#set-server-ip-via-dialogue
i get an error : set parameter failure
keeps recurring even after multiple tries
it is not the password, because then i get a different error code

gabrielpc1190, 2021-11-30:
I get this error for almost any parameter I try to change from the ShinePhone app on Android. If I do it from the server.growatt.com it says successful but doesn't do anything, so I keep trying until it does it.
I think it's not related to grott.
I wish we would have direct connection to the device instead of relying on servers on another country.

johanmeijer, 2021-12-02:
I will try it this weekend (need to do it at day time when the inverter is active) if it still works for me.

johanmeijer, 2021-12-05:
I have tried and it worked after a couple off retries. Seems to be normal that you have to do multiple times (see also: issue #78).

Hopefully it will work for you too.

Tacoos33, 2021-12-06:
Hi johann, i did multiple retries to no effect yet.
Roughly how many retries did you need? Does it state parameter set faultire when it dpoesn't succeed?
I was thinking that maybe it fails setting the new ip when it can't reach the saerver?|
Though as far as i know, grott is running and port 5279 is open and routed through docker to grott

johanmeijer, 2021-12-06:
It only takes a couple of retries. Not more then 5 or so (at my configuration). 

This is the screen. With successful change: 

[image]

I am not sure if the inverter fall back to the old IP setting if it can not reach the new one,  but sounds logical (if Growatt had build in some kin of resilience).

If it reaches Grott you should be able to see that in the log (docker logs ... command). Even if Grott is not able to send it to the Growatt server.    

I am not sure how to detect that the inverter can not connect to Grott. IP sniffer tooling is then necessary I think.

johanmeijer, 2021-12-06:
@Tacoos33I sent you a message via tweakers.

johanmeijer, 2021-12-14:
@Tacoos33 this problem is solved I think. 

I seems that for changing IP addresses (also the first time), and you have specified blockcmd = true (to block commands from outside), it is always necessary to specify: 

     noipf = True

in the [Generic] section of the .ini file

(or env variable : -e noipf="True" for the docker container )

Tacoos33, 2021-12-15:
Agreed, this solves the issue described above of not being able to change the IP in the Growatt interface.
Probably doesn't want to change if it can't find a suitable host over that IP.

closed 2021-12-15 by Tacoos33