johanmeijer/grott #80: Data less then minimum record length, data not processed

sjoerdtimmer, 2021-12-01:
My grott instance receives data from the ShineLanBox but all packets are less than minimum length. My Inverter was recently installed, could there have been a protocol update? Or is something else wrong? 

Context: 
I am running grott in a docker container (latest/db0c9d267f56)
mqtt is not on localhost but on another docker container which hostname i have configured through grott.ini. Everything else is default.  I did try the more recent 2.6.1f tag but the result is the same.

Here is a log for your convenience:
https://pastebin.com/L9G0RWWc

sjoerdtimmer, 2021-12-01:
[image]

johanmeijer, 2021-12-01:
Hi @sjoerdtimmer,

Did you defined your inverter at the Growatt website. At this moment Grott still needs that the communication between the inverter and the growatt server is established.  

If I look at the log , it seems that the inverter (or better grott proxy functionality) is trying to connect to the growatt server but get no response.

Grott skips the processing of records less then 100 bytes (or what is specified in minrecl in the .ini or gminrecl docker environmental variable) because these do not contain  iinteresting data. These records (e.g. to set up and keep alive the connections) are still being passed to the growatt server.

sjoerdtimmer, 2021-12-02:
My mistake. It was night and I has somehow assumed that it would at least send 0.0 values occasionally but this is apparently not the case :o

closed 2021-12-02 by sjoerdtimmer